crankyoldgit/IRremoteESP8266 #675: Feature request: Daikin AC comfort mode

NetNerd, 2019-04-19:
Some Daikin ACs using the original protocol (IRDaikinESP) have a "comfort" or "comfort airflow" mode, which is unsupported by this library. Unfortunately, if I'm reading the code correctly the bit that sets this is located in the first message sent, and that's thrown away by the library as if it's a useless header, so this may not be a trivial thing to add.

The bit that should be set has already been documented at https://github.com/blafois/Daikin-IR-Reverse.

　

I can test changes if needed.

crankyoldgit, 2019-04-19:
Yeah, that's going to require some very special B-grade skulduggery to hack into the code and still be backward compatible. I'll give it a shot, but not immediately as I'm working on some other major stuff with the library in prep for the next big release.

TL;DR: It should be do-able but it may take me a week or so before I get to it.

NetNerd, 2019-04-19:
Awesome! Hopefully it doesn't create a total mess.

closed 2019-04-28 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #678 (2019-04-28): DAIKIN/IRDaikinESP overhaul and add Comfort mode support.
* Enlarge the expected state size by 8 bytes.
  - Very first data section needed, because it has the "Comfort" setting in it.
  - Try to handle backward compatiblity with the old shorter size.
  - Add special case handling in IRMQTTServer to handle short size messages
    using the old size.
  - Update unit tests accordingly.
* Add set/getComfort() support.
  - Unit tests.
  - Fixes #675

* Refactor sendDaikin()/decodeDaikin() and IRDaikinESP class.
  - **[Bug]** Fix checksum calculation.
    - It was not including the last data byte of [... cut]
commit 69a912d0 (2019-04-28): DAIKIN/IRDaikinESP overhaul and add Comfort mode support. (#678)

* Enlarge the expected state size by 8 bytes.
  - Very first data section needed, because it has the "Comfort" setting in it.
  - Try to handle backward compatiblity with the old shorter size.
  - Add special case handling in IRMQTTServer to handle short size messages
    using the old size.
  - Update unit tests accordingly.
* Add set/getComfort() support.
  - Unit tests.
  - Fixes #675

* Refactor sendDaikin()/decodeDaikin() and IRDaikinESP class.
  - **[Bug]** Fix checksum calculation.
    - It was not including the last data [... cut]