crankyoldgit/IRremoteESP8266 #676: Will not compile on test mode

iwinyeah, 2019-04-20:
### Version/revison of the library used
v2.5.6

### Expected behavior
_What steps did you do and what should it have done?_
In my project,
when platformio.exe run, building is success
but platformio.exe test, buinding is not success
 
the compile error messags are below:
> Executing task: C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe test <

Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

=============================================== [test/*] Building... (1/3) ===============================================
Please wait...
In file included from .piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:3:0:
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:19:1: error: 'String' does not name a type
String uint64ToString(uint64_t input, uint8_t base = 10);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:20:1: error: 'String' does not name a type
String typeToString(const decode_type_t protocol,
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:23:1: error: 'String' does not name a type
String resultToSourceCode(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:24:1: error: 'String' does not name a type
String resultToTimingInfo(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:25:1: error: 'String' does not name a type
String resultToHumanReadableBasic(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:26:1: error: 'String' does not name a type
String resultToHexidecimal(const decode_results *result);
^
In file included from .piolibdeps\IRremoteESP8266_ID1089\src\IRrecv.cpp:17:0:
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:19:1: error: 'String' does not name a type
String uint64ToString(uint64_t input, uint8_t base = 10);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:20:1: error: 'String' does not name a type
String typeToString(const decode_type_t protocol,
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:23:1: error: 'String' does not name a type
String resultToSourceCode(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:24:1: error: 'String' does not name a type
String resultToTimingInfo(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:25:1: error: 'String' does not name a type
String resultToHumanReadableBasic(const decode_results *results);
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.h:26:1: error: 'String' does not name a type
String resultToHexidecimal(const decode_results *result);
^
*** [.pioenvs\nodemcu\lib942\IRremoteESP8266_ID1089\IRrecv.cpp.o] Error 1
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:47:1: error: 'String' does not name a type
String uint64ToString(uint64_t input, uint8_t base) {
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp: In function 'void serialPrintUint64(uint64_t, uint8_t)':
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:80:3: error: 'Serial' was not declared in this scope
Serial.print(uint64ToString(input, base));
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:80:42: error: 'uint64ToString' was not declared in this scope
Serial.print(uint64ToString(input, base));
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp: At global scope:
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:91:1: error: 'String' does not name a type
String typeToString(const decode_type_t protocol, const bool isRepeat) {
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:327:1: error: 'String' does not name a type
String resultToSourceCode(const decode_results *results) {
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:403:1: error: 'String' does not name a type
String resultToTimingInfo(const decode_results *results) {
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:432:1: error: 'String' does not name a type
String resultToHexidecimal(const decode_results *result) {
^
.piolibdeps\IRremoteESP8266_ID1089\src\IRutils.cpp:454:1: error: 'String' does not name a type
String resultToHumanReadableBasic(const decode_results *results) {
^
*** [.pioenvs\nodemcu\lib942\IRremoteESP8266_ID1089\IRutils.cpp.o] Error 1
=============================================== [ERROR] Took 2.24 seconds ===============================================

### Actual behavior
in file platformio.ini
with:

lib_deps =
	IRremoteESP8266@2.5.6

then platformio.exe test building is now success
remove this two line, building is success.

### Has this library/code previously worked as expected for you?
Yes, It work correctly, just not pass the test building only.

crankyoldgit, 2019-04-20:
Hi,
 I've never used the [PlatformIO test platform](https://docs.platformio.org/en/latest/plus/unit-testing.html#pio-unit-testing). Currently the library doesn't support it (after reading the docs).
Our current testing is done _outside_ of PlatformIO & using [Google's Testing framework](https://github.com/google/googletest) (see https://github.com/markszabo/IRremoteESP8266/wiki/Library-Maintainers-Guide#unit-tests for our library specific instructions)

This is the same environment the library uses in it's Travis CI system. (config here: https://github.com/markszabo/IRremoteESP8266/blob/master/.travis.yml#L54)

I tried running "Run other targets..." -> "PIO Test" etc, and it requires a PlatformIO account/login (which I haven't set up/don't really need) so I can't easily reproduce the problem or know if I've fixed it (without creating one)

I'm happy to add anything that will disable the PlatformIO test stuff for our library so your program can do it's thing etc. I'm just not sure what that is. Can you send a PR perhaps?

iwinyeah, 2019-04-20:
I found the building will success when I add the code below to IRrecv.h and IRsend.h:
---- IRrecv.h ----
#ifndef IRRECV_H_
#define IRRECV_H_
**// my code start
#if defined(ARDUINO) && defined(UNIT_TEST)
#include <WString.h>
#include <HardwareSerial.h>
#endif
// my code end**
#ifndef UNIT_TEST
#include <Arduino.h>
#endif

---- IRsend.h ----
#ifndef IRSEND_H_
#define IRSEND_H_

**// my code start
#if defined(ARDUINO) && defined(UNIT_TEST)
#include <WString.h>
#include <HardwareSerial.h>
#endif
// my code end**

#define __STDC_LIMIT_MACROS
#include <stdint.h>
#include "IRremoteESP8266.h"

crankyoldgit, 2019-04-22:
I set up a PlatformIO account, and I kind-of see what you mean.

I tried numerous things to try to get it behave with the current unit testing env (googletest) I use and the platformio one. I didn't have any success.

I tried a whole bunch of platformio options and tricks to get it to ignore/not perform any tests.
About the best/only thing that seemed to work was to remove the redundant "platformio.ini" from the root of the libraries files. It failed, but at least it didn't error.

Perhaps it's different when you are building/including this library from another project. Do you have a trivil example I can `git clone` to test with?

iwinyeah, 2019-04-27:
Did you try adding the below code
#if defined(ARDUINO) && defined(UNIT_TEST)
#include <WString.h>
#include <HardwareSerial.h>
#endif
to IRrecv.h and IRsend.h?
I have got the unit testing passed after add those code.
I guess the esp8266 redefine the class Serial and String not compatible with arduino.

crankyoldgit, 2019-05-02:
Yes. I tried that and a lot lot more.

Asking for this again as it may help:
> Perhaps it's different when you are building/including this library from another project. Do you have a trivil example I can git clone to test with?

crankyoldgit, 2019-05-18:
@iwinyeah Requesting again: "Perhaps it's different when you are building/including this library from another project. Do you have a trivial example I can git clone to test with?"

iwinyeah, 2019-05-22:
Just make a empty project, with the modified code,  The building will success.
I will create a project on github later.

iwinyeah, 2019-05-24:
Here is my unit test code: https://github.com/iwinyeah/IRRemoteTester

crankyoldgit, 2019-05-25:
Thanks. I'll try it out soon.

crankyoldgit, 2019-05-25:
Sadly, when I went to try it out, my 30 day free trial of "Professional Plan" with platformio has expired.
i.e.
```
Group PIO Plus Professional Trial
---------------------------------
Expire: 2019-05-22 09:44:09
```

So I can no longer test or try this. Sorry. :-(
Now, when I run the `test` from the menu, or `platformio test` I get:

```
PIO Plus (https://pioplus.com) v2.2.1
Error: You do not have permission for this operation. Please use `pio account show` command to check current per
missions.
Further details: https://pioplus.com/pricing.html or support@pioplus.com
```

I'm not going to pay $120/yr to add this feature so I can test the pio test feature for you. Sorry. I'm going to have to mark this `wontfix`. Again, sorry.

closed 2019-06-01 by crankyoldgit