crankyoldgit/IRremoteESP8266 #684: Gree library doesn’t work properly (AC White Westinghouse and Coventry)

enriqueolivieri, 2019-04-26:
### Version/revison of the library used
IRremoteESP8266 v2.5.6

### Expected behavior
Hi, I’ve done some tests in 2 air conditioners (a White Westinghouse and a Coventry). Both use Gree protocol and the library doesn’t work properly. 
1) I found that the method “IRsend::sendGree” has a mistake in the `// Footer #1`. `dutycycle` is `false` and it should  be `50` (50%). At the same time, `MSBfirst` is `true` but it is better to set `false` (even though it isn’t necessary because data is `0b010`).
2) Other change to make it work better is to set `kGreeHdrSpace = 4500`. I read this value of the signal with “IRrecvDumpV2".

crankyoldgit, 2019-04-27:
Thanks for the report! You are indeed correct on both points. Historic data does suggest that a hdrspace of 4500 would be a better fit. I'll create a PR for that later.

I've created a PR already for your first point. I guess it seemed it was working fine because most people use the `uint64_t` version rather than the `uint8_t *` version and my tests didn't catch it. :-/

closed 2019-04-27 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #686 (2019-04-27): Adjust kGreeHdrSpace to 4500
As reported in #684 and data reported in #386 a value of 4500 is a much better
choice over 4000.

Adjust value and unit tests to match.

Fixes #684
commit e1f499e0 (2019-04-27): Adjust kGreeHdrSpace to 4500 (#686)

As reported in #684 and data reported in #386 a value of 4500 is a much better
choice over 4000.

Adjust value and unit tests to match.

Fixes #684