crankyoldgit/IRremoteESP8266 #688: Fujitsu AR-JW2

kARTechnology, 2019-04-28:
Library   : v2.5.6

Not able to Control Fujitsu AirCon, branded as **Fujitsu General** in India & other Asian countries.

IRecvDumpv2 recognizes as Fujitsu AC but the commands are slightly different.
I tried to use the setRaw with the ir commands from above but the ac didn't respond, though I can see the ir led blink with a camera.

**CORRECT**
22:39:03.513 -> Encoding  : FUJITSU_AC
22:39:03.513 -> Code      : 1463001010FC08304100000000008F (120 bits)
22:39:03.513 -> Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 20C, Fan: 0 (AUTO), Swing: Off, 

**CORRECT**
22:39:07.511 -> Encoding  : FUJITSU_AC
22:39:07.545 -> Code      : 1463001010FC08304001000000008F (120 bits)
22:39:07.545 -> Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 0 (AUTO), Swing: Off, 

**CORRECT**
22:39:09.221 -> Encoding  : FUJITSU_AC
22:39:09.221 -> Code      : 1463001010FC08304002000000008E (120 bits)
22:39:09.221 -> Mesg Desc.: Power: On, Mode: 2 (DRY), Temp: 20C, Fan: 0 (AUTO), Swing: Off, 

**CORRECT**
22:39:10.897 -> Encoding  : FUJITSU_AC
22:39:10.897 -> Code      : 1463001010FC08304003000000008D (120 bits)
22:39:10.897 -> Mesg Desc.: Power: On, Mode: 3 (FAN), Temp: 20C, Fan: 0 (AUTO), Swing: Off, 

**CORRECT**
22:39:20.873 -> Encoding  : FUJITSU_AC
22:39:20.873 -> Code      : 1463001010FC08304001010000008E (120 bits)
22:39:20.908 -> Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 1 (HIGH), Swing: Off, 

**CORRECT**
22:39:23.069 -> Encoding  : FUJITSU_AC
22:39:23.069 -> Code      : 1463001010FC08304001020000008D (120 bits)
22:39:23.069 -> Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 2 (MED), Swing: Off, 

**CORRECT**
22:39:25.121 -> Encoding  : FUJITSU_AC
22:39:25.121 -> Code      : 1463001010FC08304001030000008C (120 bits)
22:39:25.121 -> Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 3 (LOW), Swing: Off, 

**CORRECT**
22:39:28.036 -> Encoding  : FUJITSU_AC
22:39:28.036 -> Code      : 1463001010FC08305001030000007C (120 bits)
22:39:28.036 -> Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 21C, Fan: 3 (LOW), Swing: Off, 

**Command: Step vane vertically**
22:39:32.598 -> Encoding  : FUJITSU_AC
22:39:32.598 -> Code      : 14630010106C (48 bits)
22:39:32.598 -> Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: Step vane vertically

**Command: Step vane vertically AUTO*
22:39:33.965 -> Encoding  : FUJITSU_AC
22:39:33.965 -> Code      : 14630010106D (48 bits)
22:39:33.965 -> Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, 

**Command: Step vane HORIZONTALLY**
22:39:35.063 -> Encoding  : FUJITSU_AC
22:39:35.097 -> Code      : 146300101079 (48 bits)
22:39:35.097 -> Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, 

**Command: Step vane HORIZONTALLY AUTO**
22:39:35.815 -> Timestamp : 000358.978
22:39:35.815 -> Encoding  : FUJITSU_AC
22:39:35.815 -> Code      : 14630010107A (48 bits)
22:39:35.815 -> Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off,

crankyoldgit, 2019-04-28:
1) Can you please provide short example code similar to the AC examples provided with the library, that illustrates how you are creating the class object and sending it?
2) Have you gotten transmitting to work at all for ANY device?
3) Have you tried using `sendRaw()` to send the raw message? This should confirm if your IR LED setup is working etc.

crankyoldgit, 2019-05-18:
@kARTechnology Friendly chase up. Can you answer the above questions?

kARTechnology, 2019-05-18:
@crankyoldgit 
1) Sorry for delay give me 2 days.
2) Tested on Sony LCD TV but didn't respond. Will try to clone remote and reply you.
3) I can see that the led is blinking slightly red(using IR LED only) when the commands are being sent.

crankyoldgit, 2019-05-23:
@kARTechnology Friendly ping

kARTechnology, 2019-05-23:
@crankyoldgit  
 Fujitsu AC sendRaw working from irRecvDumpv2.  but Library default one not working.
```

#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  
IRsend irsend(kIrLed); 
// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[243] = {3272, 1624,  400, 416,  398, 416,  398, 1232,  444, 372,  442, 1186,  400, 416,  442, 372,  398, 418,  398, 1232,  444, 1186,  444, 372,  398, 416,  398, 418,  396, 1232,  400, 1230,  446, 372,  396, 418,  398, 418,  396, 418,  398, 416,  398, 416,  398, 418,  396, 418,  398, 418,  396, 418,  398, 416,  398, 418,  396, 418,  398, 1232,  398, 418,  398, 418,  396, 418,  398, 416,  398, 418,  396, 418,  398, 418,  398, 1232,  398, 418,  398, 416,  398, 418,  398, 416,  398, 416,  398, 1232,  400, 1232,  398, 1232,  398, 1232,  400, 1230,  400, 1232,  398, 418,  396, 418,  398, 416,  398, 1232,  400, 416,  398, 418,  398, 416,  398, 420,  394, 418,  396, 418,  398, 416,  398, 418,  398, 1232,  398, 1232,  398, 418,  396, 418,  398, 1232,  398, 418,  396, 418,  398, 418,  398, 418,  396, 418,  396, 418,  398, 1232,  398, 1232,  398, 418,  398, 418,  396, 418,  398, 416,  398, 416,  398, 418,  398, 416,  398, 1232,  398, 418,  398, 418,  396, 418,  398, 416,  398, 418,  398, 418,  398, 416,  398, 418,  396, 418,  398, 418,  396, 418,  398, 416,  398, 418,  398, 418,  398, 416,  398, 418,  396, 418,  398, 418,  398, 416,  398, 418,  396, 418,  398, 416,  398, 416,  398, 418,  396, 418,  398, 418,  396, 418,  398, 418,  396, 418,  396, 418,  398, 418,  398, 1232,  400, 416,  398, 1230,  398, 1234,  398, 418,  396, 418,  398, 1232,  398, 418,  398};   

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
}

void loop() {  
  delay(2000);
  Serial.println("a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(rawData, 243, 38);  
}

```

kARTechnology, 2019-05-23:
the following **doesn't** work
`#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Fujitsu.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRFujitsuAC ac(kIrLed);

void printState() {
  // Display the settings.
  Serial.println("Fujitsu A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
  // Display the encoded IR sequence.
  unsigned char* ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  for (uint8_t i = 0; i < ac.getStateLength(); i++)
    Serial.printf("%02X", ir_code[i]);
  Serial.println();
}

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_Fujitsu.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting desired state for A/C.");
  ac.setCmd(kFujitsuAcCmdTurnOn);
  ac.setSwing(kFujitsuAcSwingBoth);
  ac.setMode(kFujitsuAcModeCool);
  ac.setFanSpeed(kFujitsuAcFanHigh);
  ac.setTemp(24);  // 24C
  //uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x71, 0x01, 0x01, 0x00, 0x00, 0x00, 0x5D};
  // ac.setRaw(state,15);
    ac.setTemp(24);  // 24C
}

void loop() {
  // Now send the IR signal.
  Serial.println("Sending IR command to A/C ...");
#if SEND_FUJITSU_AC
  ac.send();
#else  // SEND_FUJITSU_AC
  Serial.println("Can't send because SEND_FUJITSU_AC has been disabled.");
#endif  // SEND_FUJITSU_AC
  printState();
  delay(5000);
}`

**the following with the states added in setRaw works.**
`#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Fujitsu.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRFujitsuAC ac(kIrLed);

void printState() {
  // Display the settings.
  Serial.println("Fujitsu A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
  // Display the encoded IR sequence.
  unsigned char* ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  for (uint8_t i = 0; i < ac.getStateLength(); i++)
    Serial.printf("%02X", ir_code[i]);
  Serial.println();
}

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_Fujitsu.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting desired state for A/C.");
  ac.setCmd(kFujitsuAcCmdTurnOn);
  ac.setSwing(kFujitsuAcSwingBoth);
  ac.setMode(kFujitsuAcModeCool);
  ac.setFanSpeed(kFujitsuAcFanHigh);
  ac.setTemp(24);  // 24C
  **uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x71, 0x01, 0x01, 0x00, 0x00, 0x00, 0x5D};
   ac.setRaw(state,15);**
    ac.setTemp(24);  // 24C
}

void loop() {
  // Now send the IR signal.
  Serial.println("Sending IR command to A/C ...");
#if SEND_FUJITSU_AC
  ac.send();
#else  // SEND_FUJITSU_AC
  Serial.println("Can't send because SEND_FUJITSU_AC has been disabled.");
#endif  // SEND_FUJITSU_AC
  printState();
  delay(5000);
}`

kARTechnology, 2019-05-23:
**Not Working: (generated by default library**
  Power: On, Mode: 1 (COOL), Temp: 24C, Fan: 1 (HIGH), Swing: Off, Command: N/A
IR Code: **0x1463001010FE0930810101000000202D**

**Working: (generated by default library with the setRaw added**
  Power: On, Mode: 1 (COOL), Temp: 24C, Fan: 1 (HIGH), Swing: Off, Command: N/A
IR Code: **0x1463001010FC08308101010000004D**

so what is the difference that causes the default commands included in the library to fail?

kARTechnology, 2019-05-23:
figured out that
`**  ac.setModel(ARDB1);
**`

is working. what is the difference between these two remote models?

crankyoldgit, 2019-05-24:
I was going to suggest that.
The difference is they use a different state size, and it doesn't have any horizontal swing, and there are some other bits/bytes that are different in the protocols.

You could always look through the code in `ir_Fujitsu.*` to see the differences.

See #716 also.

kARTechnology, 2019-05-24:
But my ac does have horizontal and vertical swing.

kARTechnology, 2019-05-24:
Please edit
`void IRFujitsuAC::stepHoriz() {
  switch (_model) {
    case ARDB1:
      break;  // This remote doesn't have a horizontal option.
    default:
      _cmd = kFujitsuAcCmdStepHoriz;
  }
}`

to

`void IRFujitsuAC::stepHoriz() {

      _cmd = kFujitsuAcCmdStepHoriz;
  
}
`

and step vert and setp horiz working. tried editing for swing but swing not working.

kARTechnology, 2019-05-24:
do the raw commands from my remote for swing in vert and horiz help?

crankyoldgit, 2019-05-25:
Yes. More Data is always welcome. If you can, please include the _step_ commands. i.e. that move the vanes by one position. And any messages that the library decodes wrong too.

kARTechnology, 2019-05-28:
Here are the codes:
Step vane vertically
`uint8_t state[6] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x6C};`

Step vane horizontally
`uint8_t state[6] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x79};`

swing on/off vertically
`uint8_t state[6] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x6D};`

swing on/off horizontally
`uint8_t state[6] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x7A};`

kARTechnology, 2019-05-28:
Also for the step and swing, the ac.send() should not be used. because pressing the button on the remote just changes the position of the flaps / toggles on/off swing. the **library is becoming unusable** after performing stepvert or stephoriz

crankyoldgit, 2019-05-29:
@kARTechnology Can you please give me the sequence that caused it to not work as expected? i.e. before you added the `setCmd()` to fix it?

crankyoldgit, 2019-05-29:
@kARTechnology The latest update to PR #718 should give you what you want.
i.e. `toggleSwingVert()` & `toggleSwingHoriz()`

It may also fix your `setCmd()` problem, but with out more details, I can't be sure.

As usual, please test it as soon as you can. Thanks.

kARTechnology, 2019-05-29:
@crankyoldgit 

tested. **horiz swing & step horiz not working.**
also the output of printstate must not include the swing/step status for each command as swing/step are discrete, just toggle buttons

My code using pubsubclient, for controlling my ac
```
void callback(char* topic, byte* payload, unsigned int length) {

  Serial.print("Message arrived in topic: ");
  Serial.println(topic);
  Serial.print("Message:");
  for (int i = 0; i < length; i++)   Serial.print((char)payload[i]);

  Serial.println();

  if (strstr((char*)topic, "homeacbed/switch") > 0)
    if ((char)payload[0] == '1')
    {
      ac.setCmd(kFujitsuAcCmdTurnOn);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/switch", "1");
    }
    else if ((char)payload[0] == '0')
    {
      ac.setCmd(kFujitsuAcCmdTurnOff);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/switch", "0");
    }
  if (strstr((char*)topic, "homeacbed/mode") > 0)
    if (strncmp ((char*)payload, "cool", length) == 0)
    {
      ac.setMode(kFujitsuAcModeCool);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/mode", "cool");
    }
    else if (strncmp ((char*)payload, "dry", length) == 0)
    {
      ac.setMode(kFujitsuAcModeDry);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/mode", "dry");
    }
    else if (strncmp ((char*)payload, "fan", length) == 0)
    {
      ac.setMode(kFujitsuAcModeFan);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/mode", "fan");
    }
    else if (strncmp ((char*)payload, "auto", length) == 0)
    {
      ac.setMode(kFujitsuAcModeAuto);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/mode", "auto");
    }
  if (strstr((char*)topic, "homeacbed/fan") > 0)
    if (strncmp ((char*)payload, "high", length) == 0)
    { Serial.println("Set high");
      ac.setFanSpeed(kFujitsuAcFanHigh);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/fan", "high");
    }
    else if (strncmp ((char*)payload, "med", length) == 0)
    { Serial.println("Set med");
      ac.setFanSpeed(kFujitsuAcFanMed);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/fan", "med");
    }
    else if (strncmp ((char*)payload, "low", length) == 0)
    { Serial.println("Set low");
      ac.setFanSpeed(kFujitsuAcFanLow);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/fan", "low");
    }
    else if (strncmp ((char*)payload, "quiet", length) == 0)
    { Serial.println("Set quiet");
      ac.setFanSpeed(kFujitsuAcFanQuiet);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/fan", "quiet");
    }
    else if (strncmp ((char*)payload, "auto", length) == 0)
    { Serial.println("Set auto");
      ac.setFanSpeed(kFujitsuAcFanAuto);
      client.publish("/karthikx10@gmail.com/homeacbed/ping/fan", "auto");
    }
  if (strstr((char*)topic, "homeacbed/temp") > 0)
  {
    int temp = atoi( (char*)payload );
    ac.setTemp(temp);
    char snum[5];
    itoa(temp, snum, 10);
    client.publish("/karthikx10@gmail.com/homeacbed/ping/temp", snum);
  }
  **if (strstr((char*)topic, "homeacbed/swingvert") > 0)
    ac.toggleSwingVert();
  if (strstr((char*)topic, "homeacbed/swinghoriz") > 0)
    ac.toggleSwingHoriz();
  if (strstr((char*)topic, "homeacbed/stepvert") > 0)
    ac.stepVert();
  if (strstr((char*)topic, "homeacbed/stephoriz") > 0)
    ac.stepHoriz();**
  if (!(strstr((char*)topic, "ping") > 0))
  {
    ac.send();
    printState();
    ac.setCmd(kFujitsuAcCmdStayOn); 
  }
  Serial.println("-----------------------");
}
```

output of printstate:
```
Message arrived in topic: /karthikx10@gmail.com/homeacbed/**swingvert**
Message:1
Fujitsu A/C remote is in the following state:
  Model: 2 (ARDB1), Power: On, Mode: 1 (COOL), Temp: 19C, Fan: 1 (HIGH), Swing: `Vert`, Command: Toggle vertically swing
IR Code: 0x14630010106D
-----------------------
Message arrived in topic: /karthikx10@gmail.com/homeacbed/**swingvert**
Message:1
Fujitsu A/C remote is in the following state:
  Model: 2 (ARDB1), Power: On, Mode: 1 (COOL), Temp: 19C, Fan: 1 (HIGH), Swing: `Off`, Command: Toggle vertically swing
IR Code: 0x14630010106D
-----------------------
Message arrived in topic: /karthikx10@gmail.com/homeacbed/**swingvert**
Message:1
Fujitsu A/C remote is in the following state:
  Model: 2 (ARDB1), Power: On, Mode: 1 (COOL), Temp: 19C, Fan: 1 (HIGH), Swing: **Vert**, Command: Toggle vertically swing
IR Code: 0x14630010106D
-----------------------
Message arrived in topic: /karthikx10@gmail.com/homeacbed/**swingvert**
Message:1
Fujitsu A/C remote is in the following state:
  Model: 2 (ARDB1), Power: On, Mode: 1 (COOL), Temp: 19C, Fan: 1 (HIGH), Swing: **Off**, Command: Toggle vertically swing
IR Code: 0x14630010106D
-----------------------
```

instead of swing that info must be displayed in command.

crankyoldgit, 2019-05-29:
> instead of swing that info must be displayed in command.

For the other Fujitsu A/C's the **_long_** states include the `Swing:` bits in `state[10]`. Are you _sure_ subsequent messages don't have these bits in message when capturing the **real** remote?

I have an `AR-DB1` unit and it _does_ on mine. In fact, my unit doesn't have the special "toggleSwing" messages IIRC.

> `Fujitsu A/C remote is in the following state:`
`  Model: 2 (ARDB1), Power: On, Mode: 1 (COOL), Temp: 19C, Fan: 1 (HIGH), Swing: Vert, Command: N/A`

The output of `printState();` indicates you have the class object in `ARDB1` model mode.
You need to change it via `setModel(ARJW2);`. `ARDB1` has no Horizontal vane swing or horizontal modes. Your `ARJW2` does. Hence the class won't produce those messages while set to that model.
i.e. Not all models have the same features.

kARTechnology, 2019-05-29:
@crankyoldgit 
`setModel(ARJW2);` 
fixed the issue. thanks!

I am sure that subsequent messages don't have these bits in message when capturing the real remote.
its just toggle. also each step position has a different auto swing modes. will upload the user manual shorty with that swing part. request you to remove that "swing" and move it to command just like step vanes

this library only seems to lack timer, sleep functions but that is not necessary as the same can be done in arduino.

crankyoldgit, 2019-05-30:
> this library only seems to lack timer, sleep functions but that is not necessary as the same can be done in arduino.

Correct.

> I am sure that subsequent messages don't have these bits in message when capturing the real remote.

Can provide data to confirm that theory please?

In the meantime, I've added a boolean flag so you can get the functionality you desire.
e.g. `ac.toggleSwingHoriz(false)` should do what you want.

kARTechnology, 2019-05-30:
@crankyoldgit 
I turn on the AC.
press swing button, LED lights up on the indoor unit indicating swing is on.
now i close the IR transmitter of the remote and 
1) change fan speed to low
2) press the swing button
now uncover the ir transmitter and press the swing button. ( swing LED turn off without any change in fan speed)

this indicates the swing and step info do not have any relation to the other parameters and are just toggle switches.

I understand for other ac's the method you have deployed will apply, that is swing state is not toggle and each time the info is sent to the unit

crankyoldgit, 2019-05-30:
I understand that, but can you do the captures to confirm that. i.e. If I have the data I can better emulate your model.

kARTechnology, 2019-05-30:
@crankyoldgit what sequence of data captures required?

closed 2019-05-31 by crankyoldgit

crankyoldgit, 2019-05-31:
Using `IRrecvDumpV2` from the `master` branch:

Please capture **_all_** of the following:

Using the original remote, power off the A/C (i.e. start with it running in some mode other than auto), power on the A/C, set the mode to "Auto", make the unit swing horizontally, change the mode to Cool, make the unit step horizontally (should be swinging still in theory), then change the temp by one degree (up or down, don't care just tell me which), make the unit swing vertically (so both horiz and vert swing are happening), change the temp by one degree in the other direction (i.e. back to previous temp), stop the horizontal swing, step the vertical swing, change the mode to something other than cool, then stop the horizontal swing, and finally turn off the unit.

**_IF_** `IRrecvDumpV2` _successfully_ saw _every_ message as a **Fujitsu** one, then we don't need the raw or message timing info. i.e. you can remove that from the data to save some space. If it didn't, please include the raw/timing data for any messages it didn't identify as Fujitsu.

Basically, I'm mainly after the `state[]` and text descriptions of the messages (and the timestamps would be nice too) of **ALL** messages in the sequence in order. Please also annotate that data with what button you pressed to trigger that message.

With that I can try to ensure if the same class methods are called in the same sequence, we will get the exact same `state[]` information transmitted per each button press.

All in all, there should be approx 14+ `state[]` and description pairs.

Thanks in advance.

crankyoldgit, 2019-05-31:
If you get a chance, can you do that above test against the https://github.com/markszabo/IRremoteESP8266/tree/fujitsu_improvements branch.

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

commit f96590ec (2019-05-31): Add support for Fujitsu AR-REB1E & AR-JW2 remotes. (#718)

- Add support for Fujitsu ASYG7LMCA with remote AR-REB1E. Ref #716
- Extend Fujitsu A/C support.
- Handle Powerful/Turbo & Econo mode for AR-REB1E.
- Report model info as best we can.
  o AR-REB1E & AR-RAH2E appear to be the same except on special commands.
- Update Common A/C stuff to support new Turbo & Econo modes for Fujitsu A/C
- Update list of supported Fujitsu A/C systems.
- Add model support for Fujitsu AR-JW2 A/C remote
  o Seems this remote is the same protocol as AR-DB1, but with horizontal control.
- Add set/getQuietOutside [... cut]
PR #724 (2019-06-07): Update swing handling for Fujitsu A/Cs.
- Ditch using the internal swing state for some Fujitsu A/C models.
- Rework unit tests based on above.
- Refactor unit tests to use `EXPECT_STATE_EQ()`
- Update common A/C support to reflect model specific swing changes.

Fixes #688
commit 54481ef8 (2019-06-07): Update swing handling for Fujitsu A/Cs. (#724)

* Update swing handling for Fujitsu A/Cs.
- Ditch using the internal swing state for some Fujitsu A/C models.
- Rework unit tests based on above.
- Refactor unit tests to use EXPECT_STATE_EQ()
- Update common A/C support to reflect model specific swing changes.
- Unit tests to help debug Issue #726
  * The captured timings are very poor (too wide a spread of values). But the underlying data is sound.
- Improve Fujitsu A/C common A/C handling. #726 

Fixes #688