crankyoldgit/IRremoteESP8266 #689: Decode Daikin AC ARC433B69 Remote

samtd, 2019-04-28:
Library : v2.5.6
Board: NodeMCU v1.0

Timestamp : 000332.063
Encoding  : UNKNOWN
Code      : E0E32232 (220 bits)
Library   : v2.5.6
Remote Control: ARC433B69

I several test the library with other versions and i can’t receive ir commands from ARC433B69.
With another AC and remote control ARC433A it works perfect.
With ARC433B i can send commands with library IRremoteESP8266 and it works. 
For receiving IR codes from remote control I need to use the library 
danny source/Arduino_DY_IRDaikin for receiving. 

Timestamp : 000332.063
Encoding  : UNKNOWN
Code      : E0E32232 (220 bits)
Library   : v2.5.6

Raw Timing[439]:
   +  3402, -  1770,    +   382, -  1340,    +   382, -   480,    +   382, -   478, 
   +   382, -   480,    +   380, -  1342,    +   382, -   478,    +   356, -   504, 
   +   382, -   480,    +   380, -   478,    +   384, -  1342,    +   380, -   480, 
   +   380, -  1342,    +   382, -  1342,    +   382, -   478,    +   382, -  1340, 
   +   382, -  1340,    +   384, -  1340,    +   382, -  1342,    +   382, -  1340, 
   +   380, -   480,    +   382, -   480,    +   382, -  1296,    +   426, -   480, 
   +   380, -   480,    +   382, -   480,    +   380, -   480,    +   382, -   480, 
   +   382, -   478,    +   382, -  1342,    +   382, -  1342,    +   382, -  1340, 
   +   356, -  1368,    +   382, -   478,    +   382, -   480,    +   382, -   478, 
   +   380, -   480,    +   382, -   480,    +   382, -   480,    +   382, -   478, 
   +   382, -   480,    +   382, -   478,    +   358, -   504,    +   382, -   480, 
   +   380, -   480,    +   382, -   480,    +   382, -   480,    +   380, -   480, 
   +   382, -   478,    +   382, -   480,    +   382, -   478,    +   382, -   480, 
   +   354, -   506,    +   354, -   506,    +   380, -   480,    +   382, -   480, 
   +   382, -   480,    +   382, -   480,    +   380, -  1342,    +   382, -   480, 
   +   382, -   480,    +   382, -   478,    +   382, -   478,    +   382, -   478, 
   +   384, -   478,    +   382, - 29652,    +  3426, -  1772,    +   382, -  1340, 
   +   382, -   480,    +   380, -   478,    +   382, -   480,    +   382, -  1342, 
   +   382, -   480,    +   382, -   480,    +   382, -   478,    +   356, -   506, 
   +   382, -  1342,    +   380, -   480,    +   382, -  1340,    +   382, -  1340, 
   +   382, -   478,    +   356, -  1366,    +   382, -  1340,    +   384, -  1340, 
   +   382, -  1340,    +   382, -  1342,    +   382, -   478,    +   382, -   478, 
   +   382, -  1340,    +   382, -   478,    +   382, -   478,    +   382, -   478, 
   +   382, -   480,    +   382, -   480,    +   384, -   478,    +   358, -   504, 
   +   382, -   478,    +   382, -   480,    +   382, -   478,    +   382, -   480, 
   +   382, -   480,    +   382, -   478,    +   382, -   480,    +   382, -   478, 
   +   382, -   478,    +   382, -   478,    +   382, -   478,    +   384, -   478, 
   +   382, -   478,    +   360, -   500,    +   358, -   504,    +   382, -   478, 
   +   382, -   480,    +   382, -   480,    +   382, -   478,    +   382, -   478, 
   +   382, -  1340,    +   382, -  1342,    +   382, -   480,    +   380, -   480, 
   +   382, -  1342,    +   382, -   478,    +   382, -   480,    +   356, -   506, 
   +   382, -   478,    +   382, -   480,    +   382, -   480,    +   356, -   506, 
   +   382, -   478,    +   382, -   480,    +   382, -   478,    +   382, -   480, 
   +   382, -   478,    +   382, -   480,    +   380, -   480,    +   380, -   480, 
   +   382, -  1342,    +   382, -   478,    +   382, -  1342,    +   382, -   480, 
   +   382, -   480,    +   382, -   478,    +   382, -   478,    +   382, -   480, 
   +   382, -   478,    +   382, -   480,    +   356, -   504,    +   384, -   478, 
   +   382, -   480,    +   382, -   480,    +   380, -   480,    +   382, -   478, 
   +   382, -   480,    +   382, -   480,    +   382, -   478,    +   356, -   504, 
   +   384, -   478,    +   380, -   480,    +   382, -   480,    +   382, -   480, 
   +   382, -   478,    +   356, -   506,    +   382, -   478,    +   382, -   480, 
   +   380, -   480,    +   382, -   478,    +   382, -   480,    +   382, -   478, 
   +   382, -   480,    +   358, -   504,    +   382, -   478,    +   382, -   478, 
   +   356, -   504,    +   382, -   478,    +   382, -   480,    +   382, -   478, 
   +   382, -   478,    +   382, -   478,    +   382, -   480,    +   380, -   480, 
   +   382, -   480,    +   380, -   480,    +   356, -   506,    +   356, -   504, 
   +   382, -   480,    +   382, -   478,    +   382, -   478,    +   382, -   478, 
   +   382, -   478,    +   382, -   480,    +   382, -   478,    +   382, -   480, 
   +   382, -   480,    +   382, -  1340,    +   382, -  1342,    +   382, -   478, 
   +   384, -   478,    +   382, -   478,    +   382, -   480,    +   380, -   480, 
   +   382, -   478,    +   382, -   480,    +   356, -   506,    +   382, -   478, 
   +   382, -   480,    +   382, -   478,    +   356, -   506,    +   380, -   480, 
   +   382, -   478,    +   382, -   478,    +   382, -   478,    +   382, -   480, 
   +   382, -   480,    +   380, -   480,    +   382, -  1342,    +   382, -  1340, 
   +   382, -   480,    +   356, -   504,    +   382, -  1342,    +   382

uint16_t rawData[439] = {3402, 1770,  382, 1340,  382, 480,  382, 478,  382, 480,  380, 1342,  382, 478,  356, 504,  382, 480,  380, 478,  384, 1342,  380, 480,  380, 1342,  382, 1342,  382, 478,  382, 1340,  382, 1340,  384, 1340,  382, 1342,  382, 1340,  380, 480,  382, 480,  382, 1296,  426, 480,  380, 480,  382, 480,  380, 480,  382, 480,  382, 478,  382, 1342,  382, 1342,  382, 1340,  356, 1368,  382, 478,  382, 480,  382, 478,  380, 480,  382, 480,  382, 480,  382, 478,  382, 480,  382, 478,  358, 504,  382, 480,  380, 480,  382, 480,  382, 480,  380, 480,  382, 478,  382, 480,  382, 478,  382, 480,  354, 506,  354, 506,  380, 480,  382, 480,  382, 480,  382, 480,  380, 1342,  382, 480,  382, 480,  382, 478,  382, 478,  382, 478,  384, 478,  382, 29652,  3426, 1772,  382, 1340,  382, 480,  380, 478,  382, 480,  382, 1342,  382, 480,  382, 480,  382, 478,  356, 506,  382, 1342,  380, 480,  382, 1340,  382, 1340,  382, 478,  356, 1366,  382, 1340,  384, 1340,  382, 1340,  382, 1342,  382, 478,  382, 478,  382, 1340,  382, 478,  382, 478,  382, 478,  382, 480,  382, 480,  384, 478,  358, 504,  382, 478,  382, 480,  382, 478,  382, 480,  382, 480,  382, 478,  382, 480,  382, 478,  382, 478,  382, 478,  382, 478,  384, 478,  382, 478,  360, 500,  358, 504,  382, 478,  382, 480,  382, 480,  382, 478,  382, 478,  382, 1340,  382, 1342,  382, 480,  380, 480,  382, 1342,  382, 478,  382, 480,  356, 506,  382, 478,  382, 480,  382, 480,  356, 506,  382, 478,  382, 480,  382, 478,  382, 480,  382, 478,  382, 480,  380, 480,  380, 480,  382, 1342,  382, 478,  382, 1342,  382, 480,  382, 480,  382, 478,  382, 478,  382, 480,  382, 478,  382, 480,  356, 504,  384, 478,  382, 480,  382, 480,  380, 480,  382, 478,  382, 480,  382, 480,  382, 478,  356, 504,  384, 478,  380, 480,  382, 480,  382, 480,  382, 478,  356, 506,  382, 478,  382, 480,  380, 480,  382, 478,  382, 480,  382, 478,  382, 480,  358, 504,  382, 478,  382, 478,  356, 504,  382, 478,  382, 480,  382, 478,  382, 478,  382, 478,  382, 480,  380, 480,  382, 480,  380, 480,  356, 506,  356, 504,  382, 480,  382, 478,  382, 478,  382, 478,  382, 478,  382, 480,  382, 478,  382, 480,  382, 480,  382, 1340,  382, 1342,  382, 478,  384, 478,  382, 478,  382, 480,  380, 480,  382, 478,  382, 480,  356, 506,  382, 478,  382, 480,  382, 478,  356, 506,  380, 480,  382, 478,  382, 478,  382, 478,  382, 480,  382, 480,  380, 480,  382, 1342,  382, 1340,  382, 480,  356, 504,  382, 1342,  382};  // UNKNOWN E0E32232

crankyoldgit, 2019-04-29:
The code/protocol you've got there is currently not supported by this library.

It consists of two data sections: 8 bytes + 19 bytes. (ARC433B69)

The current Daikin protocols we support are:
5 bits + 8 bytes + 8 bytes + 19 bytes (`DAIKIN`)
20 bytes + 19 bytes (`DAIKIN2`)

I'll look at that other code and see what I can do to support it in this library. Don't hold your breath.

crankyoldgit, 2019-04-30:
FYI, the work thus far has been included in the newly released v2.6.0 of the library.

samtd, 2019-04-30:
I'm testing IRsend with two esp8266 and I have 44% of success decoding DAIKIN216.
With remote control I have full success.

I changed kTimeout and kMinUnknownSize but I didn’t have success.
[Irsend_test.txt](https://github.com/markszabo/IRremoteESP8266/files/3133212/Irsend_test.txt)
[original.txt](https://github.com/markszabo/IRremoteESP8266/files/3133213/original.txt)

crankyoldgit, 2019-05-01:
@samtd I've tweaked the protocol timings, so hopefully it will produce a more faithful message. Can you please test it out. Note: Some of this dependent on your hardware setup too. I can't control everything in software so there may not be much more I can do on that front.

I've added the Powerful/Turbo setting.

I'd *STRONGLY* suggest you convert your spreadsheet to a Google Sheets make it viewable by anyone. That way, it will be updated automatically when you change it, and no-one needs Excel ;-)

Per the FAQ, I am not going to try to break down the timers/clock etc settings to work out what each changed bit means and does. You'll need to work that out yourself and report back.
I've decoded far more a/c's than I care to admit and I don't have the time to do this for everyone. It's quite time consuming and challenging.
TL;DR: If you want a feature/setting to work, you are going to have to make it very plain/clear what bits need to change and where. I'll code it up, but I won't do the analysis.

crankyoldgit, 2019-05-18:
@samtd did you try any of the suggestions? Any more feedback?

crankyoldgit, 2019-05-24:
@samtd friendly ping. If I don't hear back soon, I'm going to assume all is now well and close the issue.

crankyoldgit, 2019-05-27:
Closing this due to inactivity.

closed 2019-05-27 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.