crankyoldgit/IRremoteESP8266 #697: Enhance support for ZH/JT-03 remote controler - AC Goodweather

ribeirodanielf, 2019-05-02:
Hi Guys,

I've being working on a Goodweather AC integration, using ZH/JT-03 remote controller and would like to share the results of this reverse engineering.
I'm attaching the code (zip) and also the Structure definition describing the protocol. It will be great if you can add support on the project.

Regards,

Daniel
[ZH_JT-03.txt](https://github.com/markszabo/IRremoteESP8266/files/3139514/ZH_JT-03.txt)
[ZHJT03.zip](https://github.com/markszabo/IRremoteESP8266/files/3139500/ZHJT03.zip)

crankyoldgit, 2019-05-03:
Sure, I can try to port that over so it's in the style of everything else.

Can you please include so rawData captures too, so I can use them for the unit tests?

ribeirodanielf, 2019-05-04:
Yes, give few more days and I will collect them.
Regards,

Daniel

crankyoldgit, 2019-05-05:
No rush. I'm on vacation this week and I can't get to the git client I've checked in the work I've done on this so far. So I won't be able to do anything for 7+ days.
I made some comments on your repo/pr.

ribeirodanielf, 2019-05-07:
Hi, Please find attached the file with some raw data.
Regards,
Daniel
[Raw IR ZHJT-03.txt](https://github.com/markszabo/IRremoteESP8266/files/3154100/Raw.IR.ZHJT-03.txt)

crankyoldgit, 2019-05-09:
Thanks. I'll take a look at it again next week when I'm back from vacation.

crankyoldgit, 2019-05-13:
@ribeirodanielf I just looked at the txt file you supplied. Can you please document what each of those messages actually is with respect to modes the A/C remote thinks the A/C unit should be in? i.e. So I can confirm the decoding of a message works?

ribeirodanielf, 2019-05-15:
Hi,

Basically first and last are On and Off. The middle are raise temperature
and down temperature.
Regards,

crankyoldgit, 2019-05-17:
@ribeirodanielf The first one does indeed seem to be a power on, and some are temp up/downs. But the last one is not a power off, that I can work out.

Maybe I've got something screwed up.

Can you please try the code in the `goodweather` branch here: https://github.com/markszabo/IRremoteESP8266/tree/goodweather

Can you let me know what works/doesn't work etc etc.?

I really need a few well documented example messages that states everything about the settings on the remote for the given messages. e.g. So I can be sure of got all the bits correct for each setting.

ribeirodanielf, 2019-05-22:
Thanks! Will check the branch and meanwhile collect more reliable
information.
Bit busy those days.
Regards,

Daniel

crankyoldgit, 2019-05-22:
I had a problem with my "power" bit detection/setting, so now I at least can see a difference. Still want you to test it out please. ;-)

closed 2019-05-27 by crankyoldgit

ribeirodanielf, 2019-05-27:
> Closed #697 <https://github.com/markszabo/IRremoteESP8266/issues/697> via

Hi David,

Just tested that and it works good for Goodweather commands detection.
Great job !!
Will find some time to go deep into the Timer function which is already
documented but not implemented.
Thanks for you support.
Regards,

crankyoldgit, 2019-05-27:
> Just tested that and it works good for Goodweather commands detection. Great job !! Thanks for you support.

Excellent.

Looking at that `Goodweather_test.txt` file, I noticed _Turbo_ doesn't seem to be a "Toggle" as such. It appears to be just a on/off setting. i.e. Pressing the _Turbo_ button doesn't generate a message that has the _Turbo_ bit set each time it is pressed. It seems to turn it on or off in the remote, not send a toggle message for each press.

Not sure if I have explained that clearly. So, another way. Can you please check the "extra" setting buttons of Turbo, Light, & Sleep generate a toggle message to the A/C **or** does it really change the setting and keep the setting in the message.

You code seemed to indicate it toggled, so I went with that. But I suspect it isn't a toggle.

You could test this by pressing "Turbo", so that it's on (the message will be something like `Turbo: Toggle` for on, and then press "Light" to turn it on too. (i.e. `Light: Toggle` too). If both are still set to `Turbo: Toggle, Light: Toggle` when you then change temp etc, then they probably are **NOT** toggles. i.e. We need to change the text message.

@ribeirodanielf Can you please confirm?

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

PR #715 (2019-05-27): Support for Goodweather A/Cs.
- Basic and detailed support for Goodweather A/C
- Unit tests
- Integration into Common A/C framework
- Added to various example code.

Fixes #697
commit 18344e4e (2019-05-27): Support for Goodweather A/Cs. (#715)

- Basic and detailed support for Goodweather A/C
- Unit tests
- Integration into Common A/C framework
- Added to various example code.

Fixes #697