crankyoldgit/IRremoteESP8266 #700: Hardware question: IR transmitter PCB

2016for, 2019-05-05:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
v2.6.0

### Expected behavior
_What steps did you do and what should it have done?_

IR receiving works but when I try to send codes nothing happens. I am using a IR transmitter PCB from Ali-express :
[image]

https://www.aliexpress.com/item/Free-Shipping-electronic-building-blocks-infrared-transmitter-module-IR-Transmitter-for-arduino/1972257325.html?spm=2114.10010108.1000013.1.78ea538evZM4Y7&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.90158.0&scm_id=1007.13339.90158.0&scm-url=1007.13339.90158.0&pvid=f2664f86-f072-4791-b4ef-259f19d770a5

Also, I don't know if I am supposed to use a 2n222 with this PCB or not?
(I am currently not)

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

Unable to send codes, it seems like the PCB red status LED does light up, but don't think anything is happening.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2019-05-05:
Looking at that module, you may not need the transistor. not sure.
I'd hook up VCC to the 5V VCC pin on your nodemcu/d1 mini board .. and data to the GPIO for what ever you you've chosen etc.

I'd recommend you go through the steps we recommend here: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#basic

Especially point 4.

2016for, 2019-05-05:
Thanks for the prompt reply mate.

I tried using an iPhone but I am not able to see the IR pulse.
(But I didn't see anything with the actual remote as well so I don't know if I can use an iPhone to check)

I'll have to probably try again with this board by placing it close to the AC.

crankyoldgit, 2019-05-07:
FYI. It can be hard to see. Yes, it's quite possible iPhones have an IR filter on their camera, so it might not show anything. I'd suggest some other digital camera/web cam if you have one.

If you have a spare normal LED, you might be able to connect it's pins temporarily in parrallel with the IR's ones on that board, it should then be visible with the naked eye.

crankyoldgit, 2019-05-07:
Reading the comments here (https://arduinomodules.info/ky-005-infrared-transmitter-sensor-module/) it looks like the 5V(VCC) connector isn't used on this module, so yes, my guess is you'll probably need to use a transistor as a switch to control 5V to the data pin on the module otherwise you going to be current limited by how much current an ESP8266 can provide via the GPIO. i.e. not enough for practical use purposes.

2016for, 2019-05-12:
Closing this issue. It seems like it works directly as well. But the problem is this is a cheap IR transmitter that works just within 10-20cm of the device.

closed 2019-05-12 by 2016for