crankyoldgit/IRremoteESP8266 #701: Hardware question/Feature request: Using pre-made devices?

2016for, 2019-05-06:
Tasmota Wiki has a great article on how to use those generic IR universal remotes from Aliexpress like this one https://github.com/arendst/Sonoff-Tasmota/wiki/YTF-IR-Bridge

Is there a way we can flash this library example (IRMQTTServer) onto pre-made devices?

https://user-images.githubusercontent.com/1296162/50830739-8b9f2780-12fd-11e9-83b3-bdaea3f50bb2.JPG

https://user-images.githubusercontent.com/1296162/50830740-8b9f2780-12fd-11e9-9a11-9572d9990e44.JPG

Looking forward to hear from your guys.

crankyoldgit, 2019-05-06:
If you can flash Tasmota on to it, you should be able to flash any ESP8266 binary the same way.
If you want to use the example code on this device, you would have to compile it with the correct GPIOs etc in advance.

ThaBoom, 2019-05-11:
I have flashed two of these already, the gpios are IR_LED 14 and IR_RX 5.  They have a button and an extra led.  I think you can flash them with a level converter on the two "data" pins of the usb cable they seem to go right to the rx/tx on the esp.

2016for, 2019-05-12:
@ThaBoom Thanks for the reply mate.
Looking forward to flash it. :)

Also, are you using a custom sketch on it? And are you connecting it to some home automation software like Home Assistant?

If yes. I am trying to achieve something like this via HA (https://github.com/markszabo/IRremoteESP8266/issues/702) any thoughts?

ThaBoom, 2019-05-13:
I am running IRMQTTServer and working on putting a dallas temp sensor on for my Home Assistant setup.   I have 9 rooms with mini splits in them the plan is to not have to run around the house to adjust them from heat to cool and vice versa.  I too would like that option but would like an option to only include the ac codes for what I need to save program space and keep OTA ability with the  onewire code.

crankyoldgit, 2019-05-13:
@ThaBoom You can disable the protocols you don't need/want and it should save you programming space.
See https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L55

crankyoldgit, 2019-05-13:
@2016for As someone (@ThaBoom) has confirmed it does indeed work/flash on to said device(s), confirming what I suggested, I'm going to mark this issue closed as there really isn't anything for the library to do etc to make it work.

closed 2019-05-13 by crankyoldgit

2016for, 2019-05-14:
Absolutely @crankyoldgit , Sorry I forgot to close this. :)