crankyoldgit/IRremoteESP8266 #702: Is there a way to set state of AC unit (for mqtt state topic for HA) using received IR signal?

2016for, 2019-05-06:
Seems like this Library is the thing I have been looking for so long! :)
Will help me integrate my Fujitsu AC to HA.

Goal is to be able to use both manual remote and wifi remote and keep things in sync with HA.

wifi IR receiver would be placed close to AC unit so as to "copy" any codes sent via manual remote to the AC unit.

Now, I want to be able to use the received codes to set the mqtt state topics to the received AC values so that my HA knows what the manual remote changed the AC to.

I know there would be some IR code misses here and there (AC might pick it up but wifi receiver might not, and vice versa) but I am ok with that.

Thanks in anticipation.
Looking forward to make this work.

crankyoldgit, 2019-05-06:
Currently, no. I'll look at it as a feature request. You'd also have the problem that a) the real remote will support more features than this library/example code does, b) the example code will support more features than HA's climate feature does, so you'd lose some info/functionality on top of possible missed messages etc. But if all you care about is basic functionality, then yes, could do what you want.

2016for, 2019-05-06:
Thanks a lot for the reply mate.

To be honest the Fujitsu codes in the library are all I need for my project.
Could you point me how I can achieve a very primitive sketch having only
FUJITSU HVAC codes, HA integration (using mqtt) and the ability to receive
IR codes and set state of HVAC back in Home assistant (using mqtt)
Although I won’t be sending any manual commands from the remote that the
library doesn’t support but even if in some rare case I do I don’t mind
them getting ignored.

Thanks again for the help.

crankyoldgit, 2019-05-06:
> Currently, no. I'll look at it as a feature request.

You may have missed this in the answer. Currently, there isn't a nice way to convert an inbound IR message into a _friendly_ MQTT message that HA will understand. I'll have to write that.

2016for, 2019-05-06:
Oh ok. Thanks for the help.
Appreciate your hard work. Great library man!

I did see something about Ir_server/received in the mqttserver example.
Maybe We can just modify that code bit to send Aircon/HVAC details as well
and for the time being We can use HA to adjust to the received settings via
mqtt (using Json template)

Otherwise the better solution would be to route Fujitsu Ir codes to HVAC
mqtt state topic within the sketch.

(Basically create another example for people wanting to control HVAC units
via mqtt)

Would be more than happy to help in any way possible.

Incoming IR messages (from an IR remote control) will be transmitted to
* the MQTT topic 'ir_server/received'. The MQTT message will be formatted
* similar to what is required to for the 'sent' topic.
* e.g. "3,C1A2F00F,32" (Protocol,Value,Bits) for simple codes
* or "18,110B805000000060110B807000001070" (Protocol,Value) for complex
codes
* Note: If the protocol is listed as -1, then that is an UNKNOWN IR
protocol.
* You can't use that to recreate/resend an IR message. It's only for
* matching purposes and shouldn't be trusted.

2016for, 2019-05-07:
http://osoyoo.com/2016/12/nodemcu-mqtt-ir-remoter/

This seems to be a simple sketch that’s sending received IR codes to mqtt,
can this help me create a sketch that at least sends the IR code received
on esp8266 to HA and leave the brains of decoding the message to HA for now.

(But it would be great if I can send a json because extracting info from
Json in Ha is fairly straightforward.)

Thanks again for all the help.

crankyoldgit, 2019-05-07:
If you just want the state code for the AC, IRMQTTServer will do that
already. See
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#128
(line 128)

It currently doesn't break it down to the individual settings for the AC,
that's what I'll have to add when I'm back from vacation.

2016for, 2019-05-07:
Ah. Sorry to bother you on your vacation. Enjoy mate. :)

Happy to wait for you to come back.

I’ll try to tinker around a little but yes the final goal would be to have
all the different settings sent to the mqtt such that all we need to do is
just add the sample config for HA (climate.mqtt) and the sketch handles the
incoming IR messages as well.

Have a great time buddy!

crankyoldgit, 2019-05-14:
I've added preliminary support for what I think you want to the branch `ac_to_common` branch: https://github.com/markszabo/IRremoteESP8266/tree/ac_to_common

Currently it's only for Kelvinator A/Cs (I'll add the others to that branch before I create a PR for it all)
Does this look like what you want? (e.g. https://github.com/markszabo/IRremoteESP8266/commit/cb1bd71b5eeeabe815f25d58ad08e1e63a5e5d40)
i.e. It takes a real IR A/C message, updates it's internal state of the A/C, & converts it to the various MQTT state topics so that Home Assistant etal can be kept in sync etc, well, as best it can.

2016for, 2019-05-14:
@crankyoldgit Oh my god man, you are a freaking genius.
This is exactly what I wanted to achieve. Plus the amazing "attention-to-detail" on your part (allowing/not allowing any remote to change settings, repeating IR signal from esp8266) are amazing!

Just a small question, I hope the IR receiver won't start decoding the signals transmitted from within the ESP8266 as well? I mean it might go into a loop or something.

Again, Thanks a lot for this. I have a FUJITSU AC and would definitely report back how it goes once the PR is complete. (I would love to help in any way possible other than coding. :P) (I have absolute zero experience with C++ or whatever language you're using) (although I kind of can try to make sense of what the code might be doing)

crankyoldgit, 2019-05-15:
> Just a small question, I hope the IR receiver won't start decoding the signals transmitted from within the ESP8266 as well? I mean it might go into a loop or something.

That's another reason to not have the `REPLAY_DECODED_AC_MESSAGE` set to `false`.
But yes, you really should keep the IR demodulator optically isolated from your transmitter IR LEDs.

> I have a FUJITSU AC and would definitely report back how it goes once the PR is complete. (I would love to help in any way possible other than coding. :P)

I've just added Fujitsu A/Cs to that branch, so feel free to download it and do some real-world testing.
Fujitsu is a bit odd-ball as it has a special "off" message which will probably reek havoc with the HA state. i.e. Just pressing "Off" on the remote transmits a different message than what the remote's state _normally_ is.

Let me know how you go.

2016for, 2019-05-15:
@crankyoldgit Awesome!
Will give it a shot and get back to you (over the weekend probably)

In terms of the "loop" possibility or the case where IR receiver picks up IR signals transmitted from ESP8266 itself, can we do something software-wise to prevent that?
(Maybe just turn off the receiver for a few seconds [probably "milliseconds" would also work considering how quick IR messages are] before/at time of transmitting signal from ESP8266?)

Again, Thanks a lot for the amazing work man!
Pretty insane skills you got there man!

crankyoldgit, 2019-05-16:
> I am planning to use those premade devices and I don't think they're optically isolated, so really hoping maybe a simple software solution can take care of this?

Let's cross that bridge when we get to it.

2016for, 2019-05-16:
> Let's cross that bridge when we get to it.

Absolutely buddy. Makes sense. :)

I know you have been hard at work, creating a masterpiece, can I ask you a newbie question?

Could you help me a little on how to configure mqtt settings and how do I maybe create a "Fujitsu_AC only" sketch in order to save some memory and hopefully improve performance?
(I tried the IRMQTTServer example in the master build a week or two back, the web UI was taking a lot of time to load, which I understand must be due to low memory on these chips)
(I also don't mind losing the web interface if that helps with performance over mqtt or something?)

So sorry for the dumb questions, I am still trying to learn arduino and ESP8266s. :)

Thanks again for the amazing work.
Your coding spree is pretty fun to read and helping me learn a lot. :)

crankyoldgit, 2019-05-17:
> Could you help me a little on how to configure mqtt settings and how do I maybe create a "Fujitsu_AC only" sketch in order to save some memory and hopefully improve performance?

First off, see: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

Secondly, https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L55 will help you disable protocols you don't want or need to save space etc.

> I tried the IRMQTTServer example in the master build a week or two back, the web UI was taking a lot of time to load, which I understand must be due to low memory on these chips

I doubt anything about the code is what is causing the performance problems you experienced.
What is more likely is: a) You didn't configure MQTT correctly (or disable it if you didn't have it). The code will be spending a lot of time trying to connect to the MQTT server rather than doing HTML things if you got that wrong. Check the serial output for debug/logging info on that. OR b) You've got WIFI issues or a dodgy chip.

If everything is set up correctly, everything should be quite fast, even with EVERYTHING turned on.

2016for, 2019-05-17:
@crankyoldgit Hey sorry for the newbie questions mate.

Will fix these things and see how it goes.

Just to be sure, mqtt settings go here right?

```
char MqttServer[kHostnameLength + 1] = "10.0.0.4";
char MqttPort[kPortLength + 1] = "1883";
char MqttUsername[kUsernameLength + 1] = "";
char MqttPassword[kPasswordLength + 1] = "";
char MqttPrefix[kHostnameLength + 1] = "";
```

crankyoldgit, 2019-05-17:
No. As of v1.0.0 of IRMQTTServer (2.6.0) of the library, MQTT settings are done in the initial setup menu on first flash/after a wipe. etc.
See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L42

2016for, 2019-05-17:
Awesome. Thanks a lot.

I tried compiling the sketch using platformio after changing the IRRemoteESP8266.h file changing everything to false except the 2 FUJITSU options.

I am getting this warning in b/w:
```
.piolibdeps/PubSubClient_ID89/src/PubSubClient.cpp: In member function 'boolean PubSubClient::beginPublish(const char*, unsigned int, boolean)':
.piolibdeps/PubSubClient_ID89/src/PubSubClient.cpp:456:18: warning: unused variable 'i' [-Wunused-variable]
uint16_t i;
^
```
However, the compile ultimately goes through. Anything I should be doing differently?

crankyoldgit, 2019-05-17:
That's an issue with the _PubSubClient_ library, not our library. It's also not a big issue. It's just a warning about a unused variable. You can safely ignore it.

> Also, even though I have set all others to false, it still compiles the cpp files for other protocols?

Yes, It will still compile everything. It just won't/shouldn't include a lot of the code in the resulting binary.

crankyoldgit, 2019-05-18:
@2016for If you can try out the code in PR #705 over the weekend, I'll make any changes needed before I merge it into the `master` branch. The PR has been code reviewed and approved, so I'm just waiting on feedback from you before I merge it.

2016for, 2019-05-18:
@crankyoldgit Absolutely buddy. Flashed it just now. Testing it.

How do I test whether the IR receiver is picking up IR transmitted from ESP8266 itself?

(I mean, when using manual remote in mqtt I am following topic "stat" to check, which I think is also the topic to which the inbuilt receiver sends the status after it transmits Aircon data)

2016for, 2019-05-18:
@crankyoldgit It seems to be working pretty well with a simple nodemcu.

My IR receiver does have a LED light to indicate code received. It does light up when I tell ESP to send code, however I am following mqtt topic: received
It seems to be only sending codes sent via manual remote not the remote itself. Do you think that's enough to say that the Receiver is not picking up codes from ESP itself?

2016for, 2019-05-18:
@crankyoldgit Also it seems like it does pick up codes, but it misses manual remote sometimes despite the LED turning on on the IR receiver + AC responds to those codes as well.

I think it happens more often when I just change temperature on the manual remote

crankyoldgit, 2019-05-19:
From what I can gather, the code in the PR seems to be working for you, hence I'll merge it in.

> (I mean, when using manual remote in mqtt I am following topic "stat" to check, which I think is also the topic to which the inbuilt receiver sends the status after it transmits Aircon data)

["received"](https://github.com/markszabo/IRremoteESP8266/blob/ac_to_common/examples/IRMQTTServer/IRMQTTServer.h#L68) is the only MQTT topic that will report what the IR demodulator receives.
The `stat` topics are just for indicating what the internal state of the ESP is in. i.e. if the **ESP** thinks the A/C should be in cool mode or not etc. This is mostly to confirm reception of the `cmnd` topics and to inform Home Assistant of any changes. You can't rely on the `stat` topics for proof the ESP has received a new IR code, as if the ESP detects there is no change of it's view of the state of the A/C it won't send an update via the `stat` topics.

> This would be problematic for me as I want it to replay the code it receives so that I am sure that the code received via manual remote updates AC as well as mqtt

Does that mean you've changed [REPLAY_DECODED_AC_MESSAGE](https://github.com/markszabo/IRremoteESP8266/blob/ac_to_common/examples/IRMQTTServer/IRMQTTServer.h#L115) to `true`?

There really is no sure-fire way that the code/module can be sure it has seen every IR command the A/C unit has seen. i.e. from the remote. e.g. The a/c could see the remote's signal and the ESP might not. Hence with IR, everything is an "optimistic mode" command. There is no confirmation back from the A/C via IR. This is one of the reasons A/C remotes typically send the entire state of the remote with each message; just in case a message was missed in a sequence.

Of course, The opposite is true, if the ESP sees the command from the remote, and the A/C unit doesn't.
Then yes, re-playing the signal is needed.

If both the A/C and the ESP see the message, it can be problematic if you use `REPLAY_DECODED_AC_MESSAGE`, as some A/C unit's use a toggle system for control of things like power. Imagine you sent a "power toggle" message from the IR remote. The A/C sees the message and goes from "Off" to "On". The ESP sees the message, and re-sends the "power toggle", which would then change it from the new "On" state to "Off".

I'll look at adding a configurable option to turn off IR capture while the ESP is sending an IR command in a later PR.

ThaBoom, 2019-05-19:
With this new commit it breaks the use_celsius function but the new functionality is working for me, should I make a new issue?  Also would you be interested in adding DHT Temp sensor to the code I have it added to my fork if you are.

2016for, 2019-05-19:
@crankyoldgit Awesome! That sounds great.

Luckily my AC ha a separate Off command. So replaying won't be a problem.

No I din't enable replay as of now, wanted to test default functionality first.

Also, it seems like you have created another branch and commit for the disable capture while transmitting, Thanks for that!

I think I should close this Issue as you have done a great job with this!

Also just one more thing it seems like the code sends only the changed parameters to mqtt "stat" topic.  I am worried about one thing, suppose I change AC mode via web server, and somehow the mqtt stat topic is missed by HA (I think mqtt messages sometimes get missed), later I change the temperature, the ESP will send only the temperature, won't it be better to send all current state related information at any change in ESP? (be it IR signal transmitted or received)

crankyoldgit, 2019-05-19:
> Also just one more thing it seems like the code sends only the changed parameters to mqtt "stat" topic. I am worried about one thing, suppose I change AC mode via web server, and somehow the mqtt [...]

The code regularly rebroadcasts all the `stat` topics (see https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.h#L74) every 10 minutes. So if HA somehow misses it, it will be get an update of what the ESP thinks the A/C's state is.

Plus, the MQTT messages for the `stat`s have the RETAIN flag set, so that if a) HA reboots etc, when it reconnects to the MQTT broker, it will get the last sent state. b) If the ESP reboots, it will fetch it's previous state from the MQTT broker via the same method, so it remembers etc.

I'll close this issue for now per request. Please create a new issue if there are any problems.

closed 2019-05-19 by crankyoldgit

crankyoldgit, 2019-05-19:
> With this new commit it breaks the use_celsius function but the new functionality is working for me, should I make a new issue?

@ThaBoom D'oh! Yes, please create a new issue for that. That's a bug.

2016for, 2019-05-19:
@crankyoldgit You're the man! Awesome.

Thanks a lot for the amazing work!

2016for, 2019-05-19:
@crankyoldgit Is there a way to enable sending all mqtt stat topics when things change on ESP?

I mean in my situation there will be a lot of power outages so can't retain that much on mqtt's way of handling things.

(If mqtt could just send everything that'll might have more chances of having everything in sync)

crankyoldgit, 2019-05-19:
> Is there a way to enable sending all mqtt stat topics when things change on ESP?

Currently no, you can't force it. But it will resend it's state via `stat` every 10 minutes.

Your MQTT broker(server) should remember the retain'ed `stat` messages through reboots of the MQTT broker.

In short, you should be just fine. It should "Just work!" (Famous last words)

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.