crankyoldgit/IRremoteESP8266 #703: Fujitsu AR-RAE1E

dneykov, 2019-05-10:
Hi all,

does anyone has experience with Fujitsu AR-RAE1E remote controller? Is it working with this library? 

Thanks

sheppy99, 2019-05-10:
Experience - yes
With this library - no
The 2 things I can contribute is that the off command is different from the rest, and that remote is subtly different from my other different numbered Fujitsu remote. Hardware wise it’s less finicky than my Daikin2 unit that replaced one of my Fujitsu’s. Swapping remotes half works before the unit sometimes goes into a sulk

ThaBoom, 2019-05-11:
I have a ar-reg1u remote and it would not work until I set the (H) swing off from the default (auto) as it was sending two extra bytes.  Oh unit number is 1 btw.  I have just flashed my 2nd tuya ir hub and have a few more to go.

crankyoldgit, 2019-05-13:
@ThaBoom Can you detail a bit more what you needed to do to get it to work etc and I might be able to add better support for that model?

ThaBoom, 2019-05-15:
@crankyoldgit I think its just selecting the right option for the correct output.  I was trial and error the diffrent protocols or unit numbers found that #1 would make it turn off and the output was correct with what my remote transmitted except for two extra bytes in there somewhere.  I turned both swings off and it started working figures since my units don't have the H swing.  I guess you could make the defaults off instead of auto and/or maybe document it in the wiki.

crankyoldgit, 2019-05-18:
@dneykov Did you try out the library to see if it works or not? i.e. Can we add it as a compatible model?

dneykov, 2019-05-18:
@crankyoldgit Yes, I did test it and it's work using the library. You can add it as compatible model.

Thanks

closed 2019-05-18 by dneykov

crankyoldgit, 2019-05-18:
@dneykov Excellent. Do you know what model type the library associates it with? e.g. `ARRAH2E` or `ARDB1` ?

dneykov, 2019-05-18:
@crankyoldgit In Aircon example I used model 1 from the dropdown

crankyoldgit, 2019-05-18:
Thanks. Noted in the code for future users.

commit 1534c47d (2019-05-18): Note AR-RAE1E is compatible with ARRAH2E

Fixes #703
commit f96590ec (2019-05-31): Add support for Fujitsu AR-REB1E & AR-JW2 remotes. (#718)

- Add support for Fujitsu ASYG7LMCA with remote AR-REB1E. Ref #716
- Extend Fujitsu A/C support.
- Handle Powerful/Turbo & Econo mode for AR-REB1E.
- Report model info as best we can.
  o AR-REB1E & AR-RAH2E appear to be the same except on special commands.
- Update Common A/C stuff to support new Turbo & Econo modes for Fujitsu A/C
- Update list of supported Fujitsu A/C systems.
- Add model support for Fujitsu AR-JW2 A/C remote
  o Seems this remote is the same protocol as AR-DB1, but with horizontal control.
- Add set/getQuietOutside [... cut]