crankyoldgit/IRremoteESP8266 #704: Daikin intelligent eye bit is wrong(?)

NetNerd, 2019-05-12:
Hello again. This is pretty minor, but it might be of interest. On my Daikin AC, the highest bit of Aux2 (byte 32) actually represents whether a weekly scheduled timer is set, not intelligent eye. (=0 if a weekly timer is set, otherwise 1)
I never noticed this before (because I just rely on a received IR signal to set this properly), but was thinking about it after seeing #699.

Sample data (all state bits):  
(no timers) `00010001 11011010 00100111 00000000 11000101 00000000 00000000 11010111 00010001 11011010 00100111 00000000 01000010 01111111 00001011 11011110 00010001 11011010 00100111 00000000 00000000 00111001 00110110 00000000 00110000 00000000 00000000 00000110 01100000 00100000 00000000 11000001 10000000 00000000 01111000`  
(weekly timer set) `00010001 11011010 00100111 00000000 11000101 00000000 00000000 11010111 00010001 11011010 00100111 00000000 01000010 01111111 00001011 11011110 00010001 11011010 00100111 00000000 00000000 00111001 00110110 00000000 00110000 00000000 00000000 00000110 01100000 00100000 00000000 11000001 00000000 00000000 11111000`

　

I'm not actually sure about this, but it's possible that intelligent eye never used this bit and it was a typo in docs.

* [The original source](https://github.com/mharizanov/Daikin-AC-remote-control-over-the-Internet/blob/master/Daikin-AC-remote.pde) definitely isn't error free. Note that powerful specifies the wrong bit if we are to assume zero-indexed numbering like other fields use.  
* The remote image on [the original project's page](https://harizanov.com/2012/02/control-daikin-air-conditioner-over-the-internet/) for this documentation seems to have a standard sensor button (it's possible the image is incorrect, but intelligent eye is a branding that Daikin has used for standard sensor mode)
* Bit 7 is exactly mirrored from bit 0 (maybe started counting the wrong way)
* There doesn't seem to be any code in the original source to verify behaviour of intelligent eye there

　

The other possibility is that Daikin reused bits for multiple functions, but it seems like there's no shortage of places to put new feature bits so this is probably unlikely.

Any thoughts?

　

Also, even though my AC uses 1 as default value, leaving this as 0 seems to have no effect unless the timers have actually been set. Other remote models seem to always send 0 from what I can find, so 1 probably shouldn't be used to initialise the bit.

crankyoldgit, 2019-05-13:
Can you please report exactly what model remote (and AC head unit model number too) you are using. I'd like to get some feedback from some other Daikin users too to see if it is a model specific thing or if it is a genuine error before I make changes. (@sillyfrog, @samtd, & @sheppy99 for feedback, as I can't verify this myself)

NetNerd, 2019-05-14:
My remote is an ARC452A4 and AC is a FTXS25LVMA.

Weekly scheduled timers are sent in a separate set of messages when initially set, but the LED seems to follow the status on the remote somehow. It's definitely something within the already used state data (the status is always correct after sending from ESP8266), but I admit I didn't verify this in code (just from comparing the data and finding the flipped bit here).

　

I tested behaviour to verify for certain that it is just an enable/disable bit that affects state here:

Test 1
1. set scheduled to turn off at 11:30 (let AC receive it)
2. disable timers using remote, but with IR led covered so AC doesn't know
3. adjust controls > timer LED on AC turns off
4. replace previously scheduled timer with a different one (off at 13:00), but cover IR led again so AC doesn't know
5. adjust controls > timer LED on AC turns on (is aware of a change)
6. AC turns off at 11:30 using last known timer (hasn't received the actual change, just an indicator)

Test 2
1. set scheduled timer to turn off at 13:00 (let AC receive it)
2. disable timers using remote, and this time let AC receive the data > timer LED on AC turns off 
3. set a scheduled timer (off at 13:00), but cover IR led again so AC doesn't know
4. timer LED on AC remains off when adjusting settings because there's no stored timers (even though they should be active as per the regular control message)

　

To verify that this bit is in captured data:

1. enable a timer
2. synchronise ESP8266 state to remote
3. sending data from ESP8266 leaves IR led on
4. disable timers, but with IR led covered
5. synchronise ESP8266 state to remote (ensure AC doesn't receive this signal)
6. sending data from ESP8266 turns IR led off

crankyoldgit, 2019-05-18:
Thanks for the update. I'll take a look soon.

crankyoldgit, 2019-05-18:
@NetNerd Yeah. Based on what you've described, it does indeed sounds like there is a separate/special message for transmitting the timer message. The enable/disable bit is probably sent in each/every message.

Rather than the binary you supplied above, can you please detail what you think the bits are via the `state[] ={...};` entries? It will make my job easier to update/reference the internal state of the protocol in the library.

Thanks in advance.

NetNerd, 2019-05-18:
I'm not quite sure what you mean by `state[] ={...};` entries. If you just mean what's changing in the standard control message, the only difference in the dumps above is the bit that is currently marked as being used for intelligent eye: `state[32] & 0b10000000` (`remote[kDaikinByteEye] & kDaikinBitEye`).

Do you want me to reverse the scheduling message too? (it looks pretty simple, just 3×26 bytes, but I'd need to collect data to figure out the time and temperature encoding for it and I'm just not sure what the benefit it would provide over using a remote)

crankyoldgit, 2019-05-18:
> I'm not quite sure what you mean by `state[] ={...};` entries

Each capture of a complex A/C via the `IRrecvDumpV2` example code should include a `uint8_ state[num] = {...};` line. That's what I'm referring to.

> Do you want me to reverse the scheduling message too?

Only if you want to. Collecting/post a sample of it may be useful if we can have the library ignore it etc if we don't want to support it, or if you put in the decoding analysis leg work we could support it.

TBH, I'm not sure why some people get hung up on having the Timer/Scheduling features work etc. I always think that sort of thing is better done by what ever home automation system/software people are using.
e.g. I'd rather have _Home Assistant_ or the ESP8266 turn on/off a device rather than rely on the A/C device itself as there are way way more possibilities there. But hey, people are different! :)

NetNerd, 2019-05-18:
Ah, I see. I'll try to reflash and dump in that format tomorrow then.

And yeah, I'm inclined to agree that configuring the AC schedule remotely is pretty pointless. I can see some benefit in having timers self-contained in the AC, but if someone's setting them manually anyway it really doesn't take much effort to grab the remote.

I think the library should already effectively ignore the schedule messages (unless listening for unknown protocols, but getting unknown data is expected then). It certainly doesn't get recognised as the Daikin decode type.  
And because they aren't sent in general use of the AC, it shouldn't impact functionality ever to just not handle these.

I'll probably collect some more data while I'm capturing the state[]... output, but probably won't do enough analysis for an implementation. Whether you create something to ignore it is your choice.

NetNerd, 2019-05-19:
Okay, here's the output from IRrecvDumpV2 for the standard control message with and without a weekly timer set.

without weekly timer:
`uint8_t state[35] = {0x11, 0xDA, 0x27, 0x00, 0xC5, 0x00, 0x00, 0xD7, 0x11, 0xDA, 0x27, 0x00, 0x42, 0xE3, 0x0B, 0x42, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x68, 0x32, 0x00, 0x30, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x80, 0x00, 0x83};`

with weekly timer:
`uint8_t state[35] = {0x11, 0xDA, 0x27, 0x00, 0xC5, 0x00, 0x00, 0xD7, 0x11, 0xDA, 0x27, 0x00, 0x42, 0xE3, 0x0B, 0x42, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x68, 0x32, 0x00, 0x30, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x00, 0x00, 0x03};`

NetNerd, 2019-05-19:
And here's what I know about the weekly timer setting messages right now: https://pastebin.com/t4GnEQHd

crankyoldgit, 2019-05-19:
> Okay, here's the output from IRrecvDumpV2 for the standard control message with and without a weekly timer set.

Can you also provide a similar set with intelligent eye on & off please?

NetNerd, 2019-05-19:
My "intelligent eye" feature (as written in the AC manual) is labelled as "sensor" on the remote and just toggles the standard sensor bit, so I can't help in locating the correct location (should one exist).

crankyoldgit, 2019-05-19:
Ahh. Now the conversation in this issue all makes more sense to me. Gotcha. I'll remove `setEye()` and `getEye()`.

closed 2019-05-26 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

commit 0da4a58e (2019-05-20): Remove set/getEye() for DaikinESP.
Turns out set/getEye() really was just set/getSensor().

Fixes #704
commit a8d0e2ba (2019-05-26): [BUG] set/getEye() uses wrong bit. Add set/getWeeklyTimerEnable() (#711)

* [BUG] set/getEye() uses wrong bit.
According to @NetNerd in #704 we are using the wrong bit for the
Intellegent Eye function of the A/C. 
It turns out that bit us used instead for the Weekly Timer enabled flag.
Add & update suitable routines and unit tests accordingly.
* Remove set/getEye() for DaikinESP.
Turns out set/getEye() really was just set/getSensor().

Fixes #704