crankyoldgit/IRremoteESP8266 #709: Not really an issue. Question about mqtt retain settings

2016for, 2019-05-19:
I have had troubles with "ghost switching" using Tasmota devices where the cause was retained mqtt messages.

Am I right in being paranoid about false triggers with retained mqtt messages?
If so, Is there a way to disable retain messages?

crankyoldgit, 2019-05-19:
The IRMQTTServer code uses the retained MQTT topics/messages to allow HA to keep in sync if it reboots/restarts, and for the IRMQTTServer to recover it's previous A/C climate state after a reset/reboot. Hence there currently is no setting to turn off "retain" for the climate stuff.

If you want to delete topics/messages that are no longer used, try the steps listed here: https://community.home-assistant.io/t/mqtt-how-to-remove-retained-messages/79029

closed 2019-05-19 by crankyoldgit

2016for, 2019-05-19:
@crankyoldgit You're the man! Awesome.

Thanks a lot for the amazing work!

2016for, 2019-05-19:
Does HA also send retain messages when we change stuff on HA?

Also, in case of reboot, whose settings are used? HA or ESP?

crankyoldgit, 2019-05-19:
No. The HA config I suggested in the code does **not** use retain. It doesn't need to.
HA sends _**all**_ commands via the `cmnd` topics, they have **no** retain flag on them from either HA or the ESP.
The `stat` topics are only used for status updates to HA (for non-optimistic mode) .. and for a couple of minutes they masquerade as `cmnd` topics for the ESP after it's _first_ connect to the MQTT broker. i.e. Recovers it's previous state from it on boot. The ESP should be the only one sending to the `stat` topics, and it does so with the RETAIN flag set.

The retained `stat` topics are stored not on the ESP or with HA, but by the MQTT broker. It (the broker) should remember them across a reboot. 

TL;DR: Only the `stat` (status) messages have retain on them. _Everything_ else is retain-less. In short, it's the best config I can think of to handle any/all outages and missed messages on the HA, ESP, & MQTT Broker sides.

2016for, 2019-05-19:
@crankyoldgit Love how you have thought about all the possibilities. Great attention to detail and a great mind!

Great work there mate!