crankyoldgit/IRremoteESP8266 #714: Cleaning the raw codesets

arihantdaga, 2019-05-20:
### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v 2.6.0

### Expected behavior

e.g.
I used IrRecv Dump V2 example code. And noticed at the raw Code timings. Even when i am pressing the same button, there is slight difference in the timings printed for the button. 

### Actual behavior
I wanted to know if we can clean the codesets, so as to get the same timing data for same buttons ? If its not implemented in the library yet, can someone point me in some direction ? i want to implement and contribute. 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Timestamp : 000679.554
Encoding  : NEC
Code      : 80BF49B6 (32 bits)
Library   : v2.6.0

Raw Timing[71]:
   +  9062, -  4446,    +   628, -  1642,    +   634, -   506,    +   628, -   506, 
   +   632, -   506,    +   628, -   506,    +   628, -   506,    +   632, -   506, 
   +   628, -   506,    +   628, -  1642,    +   632, -   506,    +   628, -  1642, 
   +   632, -  1642,    +   628, -  1646,    +   628, -  1642,    +   632, -  1642, 
   +   628, -  1642,    +   632, -   506,    +   628, -  1642,    +   632, -   506, 
   +   628, -   506,    +   634, -  1642,    +   628, -   506,    +   632, -   508, 
   +   628, -  1642,    +   632, -  1642,    +   628, -   506,    +   634, -  1640, 
   +   628, -  1646,    +   628, -   506,    +   632, -  1642,    +   628, -  1642, 
   +   632, -   506,    +   628, - 39686,    +  9062, -  2196,    +   632

uint16_t rawData[71] = {9062, 4446,  628, 1642,  634, 506,  628, 506,  632, 506,  628, 506,  628, 506,  632, 506,  628, 506,  628, 1642,  632, 506,  628, 1642,  632, 1642,  628, 1646,  628, 1642,  632, 1642,  628, 1642,  632, 506,  628, 1642,  632, 506,  628, 506,  634, 1642,  628, 506,  632, 508,  628, 1642,  632, 1642,  628, 506,  634, 1640,  628, 1646,  628, 506,  632, 1642,  628, 1642,  632, 506,  628, 39686,  9062, 2196,  632};  // NEC 80BF49B6
uint32_t address = 0xFD01;
uint32_t command = 0x92;
uint64_t data = 0x80BF49B6;

Timestamp : 000683.146
Encoding  : NEC
Code      : 80BF49B6 (32 bits)
Library   : v2.6.0

Raw Timing[71]:
   +  9060, -  4450,    +   602, -  1668,    +   628, -   512,    +   602, -   532, 
   +   602, -   536,    +   598, -   536,    +   602, -   532,    +   602, -   538, 
   +   598, -   538,    +   602, -  1666,    +   602, -   538,    +   602, -  1668, 
   +   602, -  1672,    +   628, -  1646,    +   604, -  1666,    +   606, -  1668, 
   +   628, -  1646,    +   628, -   506,    +   628, -  1646,    +   628, -   508, 
   +   628, -   512,    +   602, -  1668,    +   628, -   510,    +   604, -   532, 
   +   602, -  1668,    +   606, -  1668,    +   628, -   512,    +   628, -  1642, 
   +   628, -  1646,    +   628, -   510,    +   628, -  1642,    +   628, -  1646, 
   +   628, -   510,    +   628, - 39682,    +  9070, -  2196,    +   628

uint16_t rawData[71] = {9060, 4450,  602, 1668,  628, 512,  602, 532,  602, 536,  598, 536,  602, 532,  602, 538,  598, 538,  602, 1666,  602, 538,  602, 1668,  602, 1672,  628, 1646,  604, 1666,  606, 1668,  628, 1646,  628, 506,  628, 1646,  628, 508,  628, 512,  602, 1668,  628, 510,  604, 532,  602, 1668,  606, 1668,  628, 512,  628, 1642,  628, 1646,  628, 510,  628, 1642,  628, 1646,  628, 510,  628, 39682,  9070, 2196,  628};  // NEC 80BF49B6
uint32_t address = 0xFD01;
uint32_t command = 0x92;
uint64_t data = 0x80BF49B6;

```

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_
Upload the irrecvDumpV2 sketch. 

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

### Has this library/code previously worked as expected for you?
_No.

crankyoldgit, 2019-05-20:
> I used IrRecv Dump V2 example code. And noticed at the raw Code timings. Even when i am pressing the same button, there is slight difference in the timings printed for the button.

You will always get slightly different timings in the raw part of the message decode. It is dealing with the analogue world with the times between pulses etc.

What matters is the library detects it as the same _code_ each time, which it is. i.e.  `0x80BF49B6` both times.

All you need to do to send it is:
`irsend.sendNEC(0x80BF49B6);`

Thus, there is nothing really to 'clean' as such.

arihantdaga, 2019-05-20:
@crankyoldgit  Thank you very much for helping. If the protocol is decoded then we dont have any issue. However, i am dealing with lot of remotes(Specially ACs) where protocol is not detected. And i can only use raw codes. In that case, will it still be always ok for the code to have slight different timings ? I mean, i have tested, and it works sometimes but will it work always ? 

Softwares like AnalysIR has a feature to clean the IR Codeset, which clean the code and timings are consistent. I was wondering if there is any standard technique/algorithm to predict more precise timing...

crankyoldgit, 2019-05-20:
> In that case, will it still be always ok for the code to have slight different timings ? I mean, i have tested, and it works sometimes but will it work always ?

Not always, but pretty much always.

> I was wondering if there is any standard technique/algorithm to predict more precise timing...

https://github.com/markszabo/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py does a pretty good job of averaging the various timings for a given message. i.e. It's what I often use when adding support for a new protocol to the library.

arihantdaga, 2019-05-20:
Thank you @crankyoldgit . This will be a great help.

closed 2019-05-20 by arihantdaga