crankyoldgit/IRremoteESP8266 #716: TurnOnFujitsuAC incompatibility with uniaxial swing

u4mzu4, 2019-05-22:
### Version/revison of the library used
v2.6.0

### Expected behavior
Fujitsu air conditioner switch ON if use TurnOnFujitsuAC example

### Actual behavior
Fujitsu air conditioner doesn't switch ON if use TurnOnFujitsuAC example

### Steps to reproduce the behavior
Just compile, flash and run the example TurnOnFujitsuAC

#### Example code used
https://github.com/markszabo/IRremoteESP8266/tree/master/examples/TurnOnFujitsuAC

#### Circuit diagram and hardware used (if applicable)
https://github.com/chaeplin/rm-mini3-to-esp8266/blob/master/readme.md

### I have followed the steps in the [Troubleshooting Guide]
Found the issue #275, changed the line
from
ac.setSwing(kFujitsuAcSwingBoth);
to
ac.setSwing(kFujitsuAcSwingVert);

And after the modification the AC is working! :-) 

### Has this library/code previously worked as expected for you?
No.

### Other useful information
It would be great, if you can change the example line
ac.setSwing(kFujitsuAcSwingBoth);
to
ac.setSwing(kFujitsuAcSwingOff);
for wider compatibility of Fujitsu types.
Thanks in advaced

crankyoldgit, 2019-05-22:
What is the model number of the aircon & remote? I think you have a different model number than that is supported.
I may be able to add some detection/changes to support it.

I'll also look at making the change to the example suggested.

u4mzu4, 2019-05-22:
Hello, thanks for your quick response!
AC type: ASYG7LMCA
Remote: It's not available for me at the moment, but looks like this (based on the picture):
https://kzklima.hu/wp-content/uploads/fujitsu-oldalfali-split-klima-taviranyito-2.jpg

Anyway, thank you for your nice library and the hard work, I'm really-really happy with this. Smart AC was never so easy. :-)

crankyoldgit, 2019-05-22:
Thanks. The model number of the remote is quite important, so I'll wait till I have that before I try to add better support for it.

u4mzu4, 2019-05-22:
It could be possible, that the IR remote type is **UTY-LNTY**?
Based on Fujitsu website (https://www.fujitsu-general.com/eu/products/split/optionalparts.html).

crankyoldgit, 2019-05-22:
@u4mzu4 Could be, but we can wait till you check for the model number on the unit you have.
Our current supported models (used by the `setModel()` function) are listed here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Fujitsu.h#L72
i.e. `ARRAH2E` _(default)_ and `ARDB1`

u4mzu4, 2019-05-22:
The exact IR remote type is **AR-REB1E**.

crankyoldgit, 2019-05-23:
@u4mzu4 Thanks. I've created a branch to test the functionality of setting the correct model via `setModel()` for your remote/aircon etc. Can you please try it out?
Also, can you please provide me with some captures via `IRrecvDumpV2` for your remote, documenting what modes/settings they are. This may help me detect the difference between an AR-REB1E remote and a AR-RAH2E, which so far, only seem to differ between having horizontal swing options.

u4mzu4, 2019-05-23:
@crankyoldgit Thanks! The branch looks perfect, I'll test it as soon as I have some spare time. The biggest challenge is the AR-REB1E further capturing, because I've disassembled the receiver cables since the sending is working properly. (The big boss -my wife- wouldn't be happy if I made it inaccessible again.) There are some extra feature on the controller (set spoiler postition, economy mode, outdoor uint low noise option, 10°C antifreeze heating, powerful mode and timer), I think nobody is using them, but who knows.
Anyway, I have an ESP32 evalboard with builtin IR option (Olimex ESP32-EVB). Is there any painless possibility to port the code from ESP8266 to ESP32, or do I have to find another solution?

crankyoldgit, 2019-05-23:
> Is there any painless possibility to port the code from ESP8266 to ESP32, or do I have to find another solution?

Unfortunately, no. This library isn't supported on the ESP32. The original library this one is based on does support the ESP32 (https://github.com/z3t0/Arduino-IRremote) but it doesn't have the same protocol support.

u4mzu4, 2019-05-24:
Hello!

Here are some extra commands from AR-REB1E. I hope it will be helpful, and if you need any additional readout, do not hesitate to ask me. :-)

**Powerful mode:**
Timestamp : 000320.478
Encoding  : FUJITSU_AC
Code      : 146300101039C6 (56 bits)
Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.6.0
uint16_t rawData[115] = {3320, 1610,  432, 406,  432, 406,  432, 1220,  432, 406,  432, 1192,  458, 406,  432, 406,  432, 406,  432, 1218,  432, 1220,  432, 406,  432, 406,  432, 406,  432, 1192,  458, 1192,  460, 406,  432, 406,  432, 406,  432, 406,  432, 406,  432, 406,  432, 406,  432, 408,  432, 406,  432, 406,  430, 406,  432, 406,  432, 406,  432, 1190,  460, 406,  432, 408,  430, 406,  432, 406,  432, 406,  432, 406,  432, 406,  434, 1192,  458, 406,  432, 406,  432, 406,  432, 1194,  458, 406,  432, 406,  432, 1194,  456, 1196,  454, 1220,  432, 406,  432, 406,  432, 408,  430, 1194,  458, 1194,  456, 406,  432, 406,  430, 406,  432, 1194,  458, 1194,  458};  // FUJITSU_AC
uint8_t state[7] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x39, 0xC6};

**Economy mode**
Timestamp : 000449.237
Encoding  : FUJITSU_AC
Code      : 146300101009F6 (56 bits)
Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.6.0
uint16_t rawData[115] = {3350, 1552,  490, 350,  464, 374,  464, 1186,  488, 350,  464, 1186,  488, 348,  466, 374,  464, 374,  464, 1186,  488, 1162,  490, 350,  488, 350,  464, 374,  464, 1186,  488, 1162,  488, 350,  488, 350,  488, 350,  466, 372,  488, 350,  464, 374,  488, 348,  474, 366,  488, 350,  488, 348,  490, 350,  488, 350,  466, 372,  488, 1162,  490, 350,  488, 350,  488, 348,  490, 350,  488, 350,  488, 350,  490, 348,  490, 1162,  490, 350,  488, 350,  490, 348,  490, 1162,  490, 348,  490, 350,  488, 1162,  490, 348,  490, 348,  488, 350,  490, 348,  490, 350,  488, 1162,  490, 1160,  490, 348,  490, 1162,  490, 1162,  490, 1160,  490, 1162,  488};  // FUJITSU_AC
uint8_t state[7] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x09, 0xF6};

**Outdoor uint low noise**
Low noise:
Timestamp : 000487.258
Encoding  : FUJITSU_AC
Code      : 1463001010FE0930800110000000A09F (128 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 24C, Fan: 0 (AUTO), Swing: Vert, Command: N/A
Library   : v2.6.0

uint16_t rawData[259] = {3324, 1580,  464, 374,  464, 374,  464, 1186,  464, 374,  464, 1188,  464, 374,  464, 376,  462, 374,  464, 1188,  462, 1188,  464, 376,  464, 374,  462, 376,  464, 1186,  464, 1186,  464, 374,  462, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  462, 376,  464, 1188,  464, 374,  464, 374,  464, 376,  462, 374,  464, 374,  462, 376,  464, 374,  464, 1186,  464, 376,  462, 374,  464, 374,  462, 376,  464, 1188,  464, 1186,  464, 1188,  462, 1188,  464, 1186,  464, 1186,  466, 1186,  464, 1186,  464, 374,  466, 374,  464, 1186,  464, 374,  464, 374,  464, 374,  464, 374,  466, 372,  466, 372,  466, 374,  464, 374,  464, 1186,  490, 1162,  490, 348,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 1186,  490, 1162,  490, 348,  464, 374,  464, 374,  466, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  462, 376,  464, 1186,  488, 350,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  464, 374,  466, 372,  488, 350,  464, 374,  464, 376,  464, 374,  484, 352,  488, 350,  488, 348,  478, 362,  488, 350,  488, 350,  488, 350,  490, 348,  488, 350,  488, 350,  488, 350,  488, 348,  490, 348,  490, 348,  490, 348,  490, 348,  490, 348,  488, 350,  490, 1162,  490, 348,  488, 1162,  490, 1162,  490, 1162,  490, 1162,  490, 1160,  490, 1162,  490, 348,  490, 350,  488, 1162,  490};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x80, 0x01, 0x10, 0x00, 0x00, 0x00, 0xA0, 0x9F};

**Set vane vertically**
Timestamp : 000563.824
Encoding  : FUJITSU_AC
Code      : 14630010106C93 (56 bits)
Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: Step vane vertically
Library   : v2.6.0

uint16_t rawData[115] = {3352, 1556,  488, 350,  488, 376,  462, 1160,  490, 350,  486, 1164,  488, 374,  462, 376,  462, 376,  464, 1160,  488, 1164,  488, 374,  462, 376,  462, 376,  462, 1162,  490, 1162,  488, 376,  462, 376,  462, 350,  488, 376,  462, 378,  462, 376,  462, 376,  462, 376,  462, 352,  486, 376,  462, 376,  462, 376,  462, 376,  462, 1160,  490, 376,  462, 376,  462, 376,  462, 376,  462, 350,  488, 376,  462, 350,  488, 1162,  490, 376,  462, 376,  462, 376,  462, 376,  462, 376,  462, 1162,  490, 1160,  492, 374,  462, 1162,  490, 1162,  488, 376,  464, 1160,  490, 1162,  490, 374,  464, 352,  486, 1162,  490, 374,  462, 350,  488, 1162,  490};  // FUJITSU_AC
uint8_t state[7] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x6C, 0x93};

**10°C antifreezing heat**
Timestamp : 000210.287
Encoding  : FUJITSU_AC
Code      : 1463001010FE0930810B000000002024 (128 bits)
Mesg Desc.: Power: On, Mode: 3 (FAN), Temp: 24C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.6.0

Raw Timing[259]:
uint16_t rawData[259] = {3288, 1614,  428, 412,  426, 412,  428, 1222,  428, 412,  426, 1224,  426, 412,  426, 412,  426, 412,  424, 1226,  428, 1224,  424, 414,  426, 410,  426, 414,  426, 1224,  428, 1222,  428, 412,  426, 412,  426, 412,  426, 414,  426, 412,  428, 410,  426, 412,  428, 410,  426, 412,   [... cut ...]  410,  428, 412,  426, 412,  426, 412,  426, 414,  426, 410,  428, 410,  426, 412,  426, 412,  426, 412,  428, 410,  426, 410,  428, 410,  428, 410,  428, 410,  428, 410,  428, 410,  428, 410,  428, 410,  428, 410,  430, 408,  428, 410,  430, 1220,  460, 378,  460, 378,  458, 380,  456, 382,  458, 1192,  458, 380,  458, 380,  458, 1192,  458, 380,  460, 378,  460};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x81, 0x0B, 0x00, 0x00, 0x00, 0x00, 0x20, 0x24};

crankyoldgit, 2019-05-24:
@u4mzu4 If you want those modes added, you'll need to give me the `uint8_t state[16] = {...};` lines for each one with them ON and OFF. i.e. so only that function/setting has changed. I need that so I can be sure which bit(s) in the `state` have changed.

The `state[7]` ones I think are just fully self contained ... I think.

u4mzu4, 2019-05-25:
Hello again!
I appreciate your work so far with a great respect! :-)

What I've found in the meantime:
**Outdoor unit low noise [ON]**
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x80, 0x01, 0x00, 0x00, 0x00, 0x00, 0xA0, 0xAF};
**Outdoor unit low noise [OFF]**
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x80, 0x01, 0x00, 0x00, 0x00, 0x00, 0x20, 0x2F};

Regarding the **10°C antifreezing heat** it looks like you can only switch ON and there is no other option to switch OFF than power OFF, so I would love to forget it.

crankyoldgit, 2019-05-25:
I've added support for "Outside Quiet" to PR #718 for your model. Please try it out if you can.

u4mzu4, 2019-05-26:
I've tested the branch 
https://github.com/markszabo/IRremoteESP8266/tree/ARREB1E
and for me working as expected :-). Thanks again!

crankyoldgit, 2019-05-26:
Excellent. It should get merged with in 4 days or so.

closed 2019-05-31 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

u4mzu4, 2019-06-11:
Awesome, thanks again!

crankyoldgit, 2019-06-19:
@u4mzu4 As of [v2.6.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.6.2) this library now supports the ESP32.

commit f96590ec (2019-05-31): Add support for Fujitsu AR-REB1E & AR-JW2 remotes. (#718)

- Add support for Fujitsu ASYG7LMCA with remote AR-REB1E. Ref #716
- Extend Fujitsu A/C support.
- Handle Powerful/Turbo & Econo mode for AR-REB1E.
- Report model info as best we can.
  o AR-REB1E & AR-RAH2E appear to be the same except on special commands.
- Update Common A/C stuff to support new Turbo & Econo modes for Fujitsu A/C
- Update list of supported Fujitsu A/C systems.
- Add model support for Fujitsu AR-JW2 A/C remote
  o Seems this remote is the same protocol as AR-DB1, but with horizontal control.
- Add set/getQuietOutside [... cut]