crankyoldgit/IRremoteESP8266 #720: Store raw data to array to resend it later

inecoiec, 2019-05-26:
### Version/revison of the library used
IRremoteESP8266 v2.5.6
### template used 
IRecevDumpV2

Hi:
I want to manage an air conditioner ussing a ESP8266. I have a Nodemcu 1.0. 
The program must store 3 cool and 3 heat temperatures, the turn on and turn off codes (eight codes). The codes must be sent to AC ussing a web page.
    
To do this, I want to store data (in raw format, I do not want detect the code brand ) to a files stored in SPIFFS. This is not a problem .....for now.

I can detect the remote control and show the rawData to terminal. OK

My problem is that I can not find how to get the data recieved from the decode_results instance and store into array to store it to a file, or send it ussing IRSend.SendRaw propierties.

Thanks in advance

crankyoldgit, 2019-05-27:
`decode_results` needs to be converted to a `sendRaw()` compatible array first. You can't directly use `decode_results` as an argument for `sendRaw()`.

If you are only interested in 8 messages, I suggest you capture them with the _IRrecvDumpV2_ example code which will produce the C++ code for to use in `sendRaw()`.
For an example of how to use `sendRaw()` see https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino#L41 & https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino#L67

The routine `resultToSourceCode()` located here https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRutils.cpp#L558 will help you convert a `decode_results` to something that `sendRaw()` can use. You'll have to tweak it yourself, but it is what _IRrecvDumpV2_ uses.

Alternatively, if you want to just send _Raw_ messages via http, you can use the _IRMQTTServer_ example code as-is (or disable MQTT in it if you don't want that functionality) using a url like one of the canned examples at https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L773

inecoiec, 2019-05-27:
Thank you David for your fast answer !!
I will try to work with the resultToSurceCode() to get the code sampled with decode_result. It is some hard for me, but i will try. 

Thanks a lot again !!!
Omar

closed 2019-06-04 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

PR #721 (2019-06-04): Add function to convert decode_results to sendRaw() array.
`resultToRawArray()` will convert a `decode_results` to an array
suitable for calling `sendRaw()` with.

Unit tests added to cover the function.

Fixes #720
commit 493bd703 (2019-06-04): Add function to convert decode_results to sendRaw() array. (#721)

`resultToRawArray()` will convert a `decode_results` to an array
suitable for calling `sendRaw()` with.

Fixes #720