crankyoldgit/IRremoteESP8266 #722: Midea error code 62 on every commandop['9lol9uuooooooooooooooooooooooooooooooooooooooo

mariusmotea, 2019-05-28:
### Version/revison of the library used
v2.6.0

### Expected behavior
Basic control of the AC unit (On/Off, temp, fan). Only Off command is working as expected

### Actual behavior
Show `62` on the two digit display (probably error code 62?), no controls seams to work, still the ac ventilation is working.

### Steps to reproduce the behavior
```
Play with the library functions
```
#### Example code used
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Midea.h>
.......
const uint16_t kIrLed = D2;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRMideaAC ac(kIrLed);  // Set the GPIO to be used to sending the message
.............
ac.begin();

based on some conditions i can execute.
          ac.on();
          ac.off();
         ac.setFan(1);
         ac.setTemp(18);
then
        ac.send();

#### Circuit diagram and hardware used (if applicable)
Lolin D1 mini with Lolin IR controller.

### I have followed the steps in the [Troubleshooting Guide]
Yes
### Has this library/code previously worked as expected for you?
No

### Other useful information
I try to control the AC unit Midea from my home that is build from 3 internal units and one external compressor. On one of the internal units is state the model MSABAU-07HRFN1-QRD0GW (it was purchased in 2016). I believe the library is not generic for all Midea units and some IR code updates are required.

crankyoldgit, 2019-05-29:
It may be the case that your model isn't supported.
Have you tried capturing/decoding any messages from the actual remote with the [IRrecvDumpV2.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) example code at all? If so, what are they?
(e.g. Step 2, of the Advanced Troubleshooting steps/suggestions: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#advanced)

crankyoldgit, 2019-06-01:
Friendly reminder. Without some captures of the remote, we probably can't help you.

mariusmotea, 2019-06-01:
Thanks. I will provide such capture today.

Marius.

mariusmotea, 2019-06-01:
Please let me know if following capture is good.

https://pastebin.com/ewburFnX

Thanks again for your time.

crankyoldgit, 2019-06-01:
Thanks for the data you supplied. It was helpful.

I'm amazed your supplied suggested test code actually compiled without issues, as during testing I found the `on()` & `off()` methods in the `IRCoolixAC` class were actually missing.

The data you supplied showed that the library is totally capable of creating those messages. My only guess is you called the `IRCoolixAC` class methods with the wrong values for certain modes.
e.g. The wrong fan 'auto' speed for the given mode.

I was able to simulate every message you supplied when I used the correct values/parameters.

Your data showed me the only possible corruption issue could be the different fan "auto" settings. i.e. `kCoolixFanAuto` and `kCoolixFanAuto0`. The former is for most modes, the later is for AUTO and DRY modes. I've created PR #725 that forces the correct speed for the corresponding modes, to eliminate that possibility in future.

You can either try out the new code in this branch: https://github.com/markszabo/IRremoteESP8266/tree/issue722 It should fix all of the issues I can possibly see based on that data as to why you _might_ be having issues. (please test it and let me know)

or use the current library as-is, and use `setPower(true)` to turn the unit on, and `setPower(off)` to turn it off, and use the correct values for the auto fan speeds (see your pastebin text for those values in certain modes). As I said, the library is producing valid messages compared to your results and if used sanely, it should produce valid codes for your A/C unit(s) based on the data you supplied.

TL;DR: If you put the correct values into the `IRCoolixAC` class, as far as I can tell, it should be working exactly as the captured data suggests. See: https://github.com/markszabo/IRremoteESP8266/blob/issue722/test/ir_Coolix_test.cpp#L649

mariusmotea, 2019-06-01:
It works!

Thanks for your time. Did not know i need to use coolix library for a Midea brand AC unit. I will credit you in diyhue organization as this library will be used by users to control the AC unit from Hue applications using an emulated light.

closed 2019-06-01 by mariusmotea

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

PR #725 (2019-06-08): [BUG] Add missing on/off() to IRCoolixAC class.
Seems we were missing the fairly standard `on()`/`off()` class methods.

Enforce the correct "auto" fan speed when in the corresponding op mode.
Previously this _might_ have created invalid Coolix Codes.

Unit tests to confirm everything constructs correctly compared to supplied
real-world data and message construction matches real-world data.

Fixes #722
commit 11cef7d8 (2019-06-08): [BUG] Add missing on/off() to IRCoolixAC class. (#725)

* [BUG] Add missing on/off() to IRCoolixAC class.
Seems we were missing the fairly standard `on()`/`off()` class methods.

Enforce the correct "auto" fan speed when in the corresponding op mode.
Previously this _might_ have created invalid Coolix Codes.

Unit tests to confirm everything constructs correctly compared to supplied
real-world data and message construction matches real-world data.

Update supported model info for Coolix.

Fixes #722