crankyoldgit/IRremoteESP8266 #726: fujistu AR-RCE1C

huexpub, 2019-06-02:
thanks, last year, i use IRremoteESP8266 2.5.2 with tasmota ir-hvac works good , AR-RCE1C   with defaut Fujistu model ,but now latest code and use  IRMQTTSERVER can't work anymore, I see that the code changes a lot. How do I start to fix it?

crankyoldgit, 2019-06-02:
Do you have a raw capture from [IRrecvDumpV2](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) we can look at per chance?

Have you tried any of the different models (https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Fujitsu.h#L76) via `setModel()` yet?

huexpub, 2019-06-02:
i found the code defaut mode ARRAH2E in 2.5.2 and i see code  same most, i try other model  not action.
ok, i get some dump data,,,

        pnz.huex.pub

1443
\
d6593e89-0e8a-4e02-2bc6-f824463a2d03

64

fan:auto mode:auto temp:24  power：on

Timestamp : 000039.941
Encoding  : UNKNOWN
Code      : BA7CD518 (130 bits)
Library   : v2.6.0

Raw Timing[259]:
   +  3320, -  1644,    +   410, -   446,    +   382, -   450,    +   382, -  1258, 
   +   462, -   372,    +   382, -  1258,    +   382, -   450,    +   382, -   450, 
   +   384, -   458,    +   384, -  1254,    +   462, -  1178,    +   464, -   370, 
   +   382, -   450,    +   462, -   370,    +   384, -  1258,    +   382, -  1258, 
   +   382, -   458,    +   462, -   368,    +   384, -   526,    +   382, -   450, 
   +   384, -   448,    +   462, -   370,    +   384, -   448,    +   384, -   450, 
   +   384, -   460,    +   382, -   446,    +   382, -   450,    +   382, -   450, 
   +   382, -   450,    +   384, -  1230,    +   410, -   448,    +   384, -   448, 
   +   462, -   380,    +   382, -   446,    +   382, -   448,    +   384, -   450, 
   +   462, -   370,    +   384, -  1258,    +   382, -   448,    +   384, -   450, 
   +   384, -   458,    +   462, -   366,    +   382, -  1284,    +   384, -  1258, 
   +   382, -  1258,    +   460, -  1178,    +   384, -  1258,    +   460, -  1180, 
   +   462, -  1190,    +   384, -  1252,    +   384, -   448,    +   384, -   448, 
   +   384, -  1232,    +   488, -   370,    +   462, -   370,    +   384, -   424, 
   +   408, -   432,    +   410, -   422,    +   408, -   424,    +   486, -   346, 
   +   408, -   424,    +   410, -  1256,    +   384, -  1256,    +   384, -   448, 
   +   384, -   458,    +   384, -  1228,    +   486, -   370,    +   384, -   422, 
   +   410, -   422,    +   442, -   390,    +   410, -   424,    +   408, -   424, 
   +   488, -  1162,    +   410, -   418,    +   488, -   370,    +   460, -   344, 
   +   410, -   424,    +   408, -   422,    +   410, -   422,    +   410, -   450, 
   +   384, -   432,    +   488, -   342,    +   486, -   344,    +   410, -   422, 
   +   410, -   424,    +   486, -   344,    +   410, -   424,    +   410, -   448, 
   +   386, -   432,    +   486, -   342,    +   488, -   346,    +   486, -   344, 
   +   486, -   346,    +   410, -   422,    +   410, -   424,    +   408, -   450, 
   +   460, -   356,    +   410, -   446,    +   462, -   346,    +   408, -   424, 
   +   408, -   450,    +   384, -   422,    +   410, -   448,    +   384, -   424, 
   +   408, -   458,    +   384, -   446,    +   460, -   370,    +   384, -   448, 
   +   384, -   424,    +   408, -   424,    +   486, -   370,    +   384, -   586, 
   +   464, -   380,    +   460, -   368,    +   462, -   344,    +   408, -   448, 
   +   384, -   450,    +   384, -   448,    +   384, -  1256,    +   464, -   370, 
   +   462, -   380,    +   384, -  1254,    +   384, -  1232,    +   408, -  1256, 
   +   462, -  1180,    +   384, -   448,    +   382, -  1258,    +   384, -   450, 
   +   462, -   378,    +   384

uint16_t rawData[259] = {3320, 1644,  410, 446,  382, 450,  382, 1258,  462, 372,  382, 1258,  382, 450,  382, 450,  384, 458,  384, 1254,  462, 1178,  464, 370,  382, 450,  462, 370,  384, 1258,  382, 1258,  382, 458,  462, 368,  384, 526,  382, 450,  384, 448,  462, 370,  384, 448,  384, 450,  384, 460,  382, 446,  382, 450,  382, 450,  382, 450,  384, 1230,  410, 448,  384, 448,  462, 380,  382, 446,  382, 448,  384, 450,  462, 370,  384, 1258,  382, 448,  384, 450,  384, 458,  462, 366,  382, 1284,  384, 1258,  382, 1258,  460, 1178,  384, 1258,  460, 1180,  462, 1190,  384, 1252,  384, 448,  384, 448,  384, 1232,  488, 370,  462, 370,  384, 424,  408, 432,  410, 422,  408, 424,  486, 346,  408, 424,  410, 1256,  384, 1256,  384, 448,  384, 458,  384, 1228,  486, 370,  384, 422,  410, 422,  442, 390,  410, 424,  408, 424,  488, 1162,  410, 418,  488, 370,  460, 344,  410, 424,  408, 422,  410, 422,  410, 450,  384, 432,  488, 342,  486, 344,  410, 422,  410, 424,  486, 344,  410, 424,  410, 448,  386, 432,  486, 342,  488, 346,  486, 344,  486, 346,  410, 422,  410, 424,  408, 450,  460, 356,  410, 446,  462, 346,  408, 424,  408, 450,  384, 422,  410, 448,  384, 424,  408, 458,  384, 446,  460, 370,  384, 448,  384, 424,  408, 424,  486, 370,  384, 586,  464, 380,  460, 368,  462, 344,  408, 448,  384, 450,  384, 448,  384, 1256,  464, 370,  462, 380,  384, 1254,  384, 1232,  408, 1256,  462, 1180,  384, 448,  382, 1258,  384, 450,  462, 378,  384};  // UNKNOWN BA7CD518

fan:auto mode:auto temp:23  power：on

Timestamp : 000755.269
Encoding  : FUJITSU_AC
Code      : 1463001010FE09307000000000002040 (128 bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: On, Mode: 0 (AUTO), Temp: 23C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.6.0

Raw Timing[259]:
   +  3342, -  1610,    +   438, -   390,    +   438, -   394,    +   438, -  1204, 
   +   438, -   394,    +   438, -  1202,    +   440, -   392,    +   440, -   394, 
   +   438, -   404,    +   438, -  1198,    +   438, -  1202,    +   440, -   392, 
   +   440, -   394,    +   438, -   392,    +   440, -  1202,    +   438, -  1202, 
   +   440, -   402,    +   440, -   390,    +   438, -   394,    +   438, -   396, 
   +   438, -   394,    +   438, -   394,    +   438, -   394,    +   438, -   394, 
[... 1107 lines cut ...]
[... cut ...]  452,  380, 452,  380, 462,  380, 448,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 462,  380, 448,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 452,  380, 462,  380, 448,  380, 452,  380, 452,  380, 454,  380, 450,  380, 452,  380, 452,  380, 462,  380, 448,  382, 450,  382, 452,  382, 450,  382, 450,  382, 1258,  382, 452,  380, 462,  380, 1256,  380, 1260,  382, 450,  382, 1260,  382, 450,  382, 1258,  382, 452,  382, 458,  382};  // UNKNOWN BA4457CC

crankyoldgit, 2019-06-02:
I've had a look at the data, and yes. It appears so far, to be compatible with `ARRAH2E`.
i.e. `ac.setModel(ARRAH2E);` (1, in the Model field on IRMQTTServer)

Assuming you are capturing an actual remote signal, it seems very inconsistent. That is, the timings are all over the place. There could be a number of factors that are causing this.

I suggest you try the code in this branch: https://github.com/markszabo/IRremoteESP8266/tree/fujitsu_improvements

**or** make sure you set `ac.setSwing(kFujitsuAcSwingOff);` after you create the `IRFujitsuAC` object.
I think the default was to have both Vert & Horiz swing on by default, and the `ARRAH2E` doesn't have Horiz swing etc.

Given how out-of-whack the timings are on your signals (so bad that the library couldn't parse them regularly) you may need to tweak the valuses to suit your own device.
e.g. Try using values like:
```
kHdrMark   = 3320
kHdrSpace  = 1644
kBitMark   = 414
kOneSpace  = 1236
kZeroSpace = 417
```
here: https://github.com/markszabo/IRremoteESP8266/blob/fujitsu_improvements/src/ir_Fujitsu.cpp#L23

Example code to play with: https://github.com/markszabo/IRremoteESP8266/blob/fujitsu_improvements/examples/TurnOnFujitsuAC/TurnOnFujitsuAC.ino

2016for, 2019-06-02:
> and the `ARRAH2E` doesn't have Horiz swing etc.

I have an "AR-RHA2E" remote. It has both horizontal and vertical swing options.

I tried out the code before the very recent fujitsu changes (I think for a new model) (basically the one where IR capture to internal climate component was added) and things worked, will check latest code over next weekend and update you.

huexpub, 2019-06-02:
yes. i use TurnOnFujitsuAC.ino default can trun on  it ac.setSwing(kFujitsuAcSwingOff);, so maybe setSwing cause this problem、、、
so now IRMQTTSERVER.ino how to set?? i use web set swing (v&h) off  the iR not Transmitting signal ，must Contain swing V or H..

crankyoldgit, 2019-06-02:
I'm not completely sure what you are saying. So, let me start by stating what I think you are saying, and some more questions for you to answer.

1) I think you are saying that "Yes, it now works as expected if you use `ac.setSwing(kFujitsuAcSwingOff)`" ... correct?

2) In theory, if you set the Protocol to Fujitsu, the model to "1" and both the swing (v) and swing (h) to "off" it should work for you. If not, what are **all** the settings you have set on the AirCon page?

3) Which version/branch are you using when doing the above?

> and dump data yes all data was 130 bits,,,,,only off 58bits

If it is not being recognized as FUJITSU (33) (e.g. UNKNOWN is bad etc) and it should be 128 & 56 bits, not 130 & 58, then there still is another issue. However, if you can get the example code working and the A/C responds, then it's nothing to really worry about at the moment.

crankyoldgit, 2019-06-02:
> I have an "AR-RHA2E" remote. It has both horizontal and vertical swing options.

Yes @2016for , I was wrong. You are correct. My bad.

huexpub, 2019-06-02:
good new  i use 
const uint16_t kFujitsuAcHdrMark = 3210;
const uint16_t kFujitsuAcHdrSpace = 1680;
const uint16_t kFujitsuAcBitMark = 410;
const uint16_t kFujitsuAcOneSpace = 1230;
const uint16_t kFujitsuAcZeroSpace = 440;
const uint16_t kFujitsuAcMinGap = 8100;

all data can get  detail  , Wait a few minutes ， i put it markdown

huexpub, 2019-06-02:
Timestamp : 000145.638
Encoding  : FUJITSU_AC
Code      : 1463001010FE0930D1010400000020DA (128 bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: On, Mode: 1 (COOL), Temp: 29C, Fan: 4 (QUIET), Swing: Off, Command: N/A
Library   : v2.6.0

```

Raw Timing[259]:
   +  3282, -  1668,    +   384, -   444,    +   384, -   448,    +   384, -  1256, 
   +   384, -   448,    +   384, -  1254,    +   384, -   448,    +   384, -   448, 
   +   384, -   458,    +   384, -  1252,    +   386, -  1256,    +   384, -   448, 
   +   384, -   448,    +   384, -   448,    +   384, -  1256,    +   386, -  1254, 
   +   386, -   458,    +   384, -   442,    +   518, -   436,    +   386, -   446, 
   +   386, -   446,    +   388, -   442,    +   412, -   420,    +   412, -   420, 
   +   416, -   426,    +   414, -   414,    +   414, -   420,    +   414, -   418, 
   +   414, -   418,    +   416, -  1224,    +   416, -   416,    +   416, -   416, 
   +   416, -   426,    +   416, -   412,    +   416, -   414,    +   416, -   416, 
   +   440, -   394,    +   418, -  1220,    +   440, -   392,    +   416, -   414, 
   +   418, -   424,    +   440, -   388,    +   416, -  1226,    +   416, -  1224, 
   +   426, -  1216,    +   438, -  1200,    +   416, -  1224,    +   440, -  1202, 
   +   420, -  1230,    +   440, -  1196,    +   440, -   392,    +   414, -   418, 
   +   440, -  1200,    +   440, -   392,    +   416, -   416,    +   416, -   418, 
   +   438, -   402,    +   416, -   414,    +   418, -   414,    +   416, -   416, 
   +   434, -   398,    +   440, -  1200,    +   440, -  1200,    +   440, -   392, 
   +   440, -   404,    +   414, -  1222,    +   414, -   418,    +   426, -   406, 
   +   414, -   418,    +   414, -  1224,    +   416, -   418,    +   414, -  1226, 
   +   440, -  1210,    +   414, -  1224,    +   438, -   392,    +   418, -   414, 
   +   424, -   410,    +   414, -   418,    +   418, -   412,    +   416, -   416, 
   +   414, -   428,    +   438, -   392,    +   412, -   418,    +   440, -  1202, 
   +   414, -   418,    +   414, -   416,    +   414, -   418,    +   414, -   418, 
   +   422, -   420,    +   414, -   414,    +   438, -   392,    +   432, -   400, 
   +   438, -   394,    +   416, -   416,    +   414, -   418,    +   416, -   418, 
   +   438, -   404,    +   418, -   412,    +   438, -   392,    +   438, -   394, 
   +   438, -   394,    +   414, -   416,    +   414, -   418,    +   424, -   410, 
   +   414, -   428,    +   414, -   414,    +   440, -   392,    +   416, -   418, 
   +   438, -   394,    +   414, -   416,    +   420, -   412,    +   414, -   418, 
   +   414, -   428,    +   416, -   414,    +   440, -   392,    +   414, -   416, 
   +   440, -   394,    +   414, -   416,    +   416, -  1226,    +   438, -   394, 
   +   414, -   426,    +   440, -   388,    +   440, -  1200,    +   440, -   394, 
   +   438, -  1202,    +   416, -  1224,    +   420, -   410,    +   414, -  1228, 
   +   414, -  1232,    +   440

uint16_t rawData[259] = {3282, 1668,  384, 444,  384, 448,  384, 1256,  384, 448,  384, 1254,  384, 448,  384, 448,  384, 458,  384, 1252,  386, 1256,  384, 448,  384, 448,  384, 448,  384, 1256,  386, 1254,  386, 458,  384, 442,  518, 436,  386, 446,  386, 446,  388, 442,  412, 420,  412, 420,  416, 426,  414, 414,  414, 420,  414, 418,  414, 418,  416, 1224,  416, 416,  416, 416,  416, 426,  416, 412,  416, 414,  416, 416,  440, 394,  418, 1220,  440, 392,  416, 414,  418, 424,  440, 388,  416, 1226,  416, 1224,  426, 1216,  438, 1200,  416, 1224,  440, 1202,  420, 1230,  440, 1196,  440, 392,  414, 418,  440, 1200,  440, 392,  416, 416,  416, 418,  438, 402,  416, 414,  418, 414,  416, 416,  434, 398,  440, 1200,  440, 1200,  440, 392,  440, 404,  414, 1222,  414, 418,  426, 406,  414, 418,  414, 1224,  416, 418,  414, 1226,  440, 1210,  414, 1224,  438, 392,  418, 414,  424, 410,  414, 418,  418, 412,  416, 416,  414, 428,  438, 392,  412, 418,  440, 1202,  414, 418,  414, 416,  414, 418,  414, 418,  422, 420,  414, 414,  438, 392,  432, 400,  438, 394,  416, 416,  414, 418,  416, 418,  438, 404,  418, 412,  438, 392,  438, 394,  438, 394,  414, 416,  414, 418,  424, 410,  414, 428,  414, 414,  440, 392,  416, 418,  438, 394,  414, 416,  420, 412,  414, 418,  414, 428,  416, 414,  440, 392,  414, 416,  440, 394,  414, 416,  416, 1226,  438, 394,  414, 426,  440, 388,  440, 1200,  440, 394,  438, 1202,  416, 1224,  420, 410,  414, 1228,  414, 1232,  440};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0xD1, 0x01, 0x04, 0x00, 0x00, 0x00, 0x20, 0xDA};

```

Timestamp : 000146.459
Encoding  : FUJITSU_AC
Code      : 146300101002FD (56 bits)
Mesg Desc.: Model: 1 (ARRAH2E), Power: Off, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: N/A
Library   : v2.6.0

```

Raw Timing[115]:
   +  3312, -  1638,    +   414, -   416,    +   414, -   418,    +   414, -  1224, 
   +   416, -   418,    +   416, -  1224,    +   416, -   416,    +   414, -   418, 
   +   414, -   428,    +   414, -  1222,    +   416, -  1224,    +   414, -   416, 
   +   416, -   416,    +   416, -   418,    +   414, -  1226,    +   414, -  1226, 
   +   416, -   426,    +   416, -   412,    +   416, -   416,    +   546, -   414, 
   +   414, -   418,    +   414, -   416,    +   414, -   418,    +   416, -   416, 
   +   416, -   428,    +   414, -   414,    +   414, -   418,    +   414, -   418, 
[... cut ...] 4, 1256,  410, 448,  384, 450,  408, 424,  408, 422,  384, 450,  408, 434,  384, 1254,  382, 448,  382, 1256,  384, 1256,  408, 1232,  384, 1256,  410, 1230,  384, 1264,  384};  // FUJITSU_AC
uint8_t state[7] = {0x14, 0x63, 0x00, 0x10, 0x10, 0x02, 0xFD};

```

some data decode unknow but when you remote again next decode is right ,  i don't  know this value right?

i test mode 1 not work anymore , also with this  values ,, now web cant decode remote data ,so maybe Some files need to be changed ？

crankyoldgit, 2019-06-02:
> yeah ,not trun on AC in irmqtt web 

What values are you supplying on the "AirCon" form/page?
Is it able to control other devices with it? e.g. Are you using the correct GPIOs? Is the IR send circuit known to be working?

huexpub, 2019-06-02:
yes, i use aircon web , anthoner panasonic work well ,   i use cam to detected IR data ,yes both swing v & h NOT DATA output,

[image]

also test all model

crankyoldgit, 2019-06-02:
Based on the new data, I'd say you have a different model:
`Mesg Desc.: Model: 3 (ARREB1E), Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Command: Powerful, Outside Quiet: Off`

so use model 3 on the web form, and don't use powerful (turbo) or Economy via IRMQTTServer.
Use `ac.setModel(ARREB1E);` or `ac.setModel(3)` for other code etc.

In short, I would start with everything "off" on the AirCon page, except proto (Fuji) model (3) and power, mode, fan speed & temp should be fine to modify too. All the others, off.

crankyoldgit, 2019-06-02:
Especially the "swing" stuff, that should be off for now.

huexpub, 2019-06-02:
yeah ,you are right  ,model 3 with this value code work well,,,,,,swing Yes or no all ok ,thanks

crankyoldgit, 2019-06-02:
Unless you are using that `fujitsu_improvements` branch, I wouldn't use the swing options just now.

crankyoldgit, 2019-06-07:
Recent commits to the `master` branch should hopefully address the swing issue some what.

Marking this closed

closed 2019-06-07 by crankyoldgit