crankyoldgit/IRremoteESP8266 #727: IRMQTTServer fails to compile

davidw3591, 2019-06-02:
### Version/revison of the library used
V2.6.0

Hi, I'm trying to compile the example IRMQTTServer code but it fails with error: "'state_t' in namespace 'stdAc' does not name a type". I followed the instructions in IRMQTTServer.ino but can't get it to compile. I'm using Arduino software. What am I doing wrong?

Full error:
`Arduino: 1.8.5 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from D:\IRMQTTServer\IRMQTTServer.ino:278:0:

IRMQTTServer.h:214: error: 'state_t' in namespace 'stdAc' does not name a type

                  const stdAc::state_t state, const bool retain,

                        ^

IRMQTTServer.h:214: error: ISO C++ forbids declaration of 'state' with no type [-fpermissive]

                  const stdAc::state_t state, const bool retain,

                                       ^

IRMQTTServer.h:278: error: 'state_t' in namespace 'stdAc' does not name a type

 stdAc::state_t updateClimate(stdAc::state_t current, const String str,

 ^

IRMQTTServer.h:280: error: 'state_t' in namespace 'stdAc' does not name a type

 bool cmpClimate(const stdAc::state_t a, const stdAc::state_t b);

                       ^

IRMQTTServer.h:280: error: ISO C++ forbids declaration of 'a' with no type [-fpermissive]

 bool cmpClimate(const stdAc::state_t a, const stdAc::state_t b);

                                      ^

IRMQTTServer.h:280: error: 'state_t' in namespace 'stdAc' does not name a type

 bool cmpClimate(const stdAc::state_t a, const stdAc::state_t b);

                                               ^

IRMQTTServer.h:280: error: ISO C++ forbids declaration of 'b' with no type [-fpermissive]

 bool cmpClimate(const stdAc::state_t a, const stdAc::state_t b);

                                                              ^

IRMQTTServer.h:281: error: 'state_t' in namespace 'stdAc' does not name a type

 bool sendClimate(const stdAc::state_t prev, const stdAc::state_t next,

                        ^

IRMQTTServer.h:281: error: ISO C++ forbids declaration of 'prev' with no type [-fpermissive]

 bool sendClimate(const stdAc::state_t prev, const stdAc::state_t next,

                                       ^

IRMQTTServer.h:281: error: 'state_t' in namespace 'stdAc' does not name a type

 bool sendClimate(const stdAc::state_t prev, const stdAc::state_t next,

                                                   ^

[... 1346 lines cut ...]
IRMQTTServer:3317: error: 'celsiusToFahrenheit' was not declared in this scope

                                   : celsiusToFahrenheit(state.degrees);

                                                                      ^

IRMQTTServer:3321: error: 'climate' was not declared in this scope

   sendClimate(climate, state, MqttClimateStat, true, false, false,

               ^

IRMQTTServer:3321: error: 'state' was not declared in this scope

   sendClimate(climate, state, MqttClimateStat, true, false, false,

                        ^

exit status 1
'state_t' in namespace 'stdAc' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

crankyoldgit, 2019-06-03:
`state_t` is defined here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.h#L75

My first guess is you've got multiple versions of the library install, one of which is old and doesn't have this defined. Next, would be some how you've corrupted the library on your PC.
The solution for either would be hunt down and remove all copies of the library. Start with 'Library Manager' in the Arduino IDE and remove any that you can find. Then search your computer for files called "IRremoteESP8266.h" which may indicate where you've got another copy of the library.
Then, when they are all removed, re-add it via 'Library Manager' in the IDE. P.s. 1.8.5 of the IDE is quite old. I suggest you also look at updating it.
If that doesn't work, ensure you are compiling for the correct architecture. e.g. The build target is an ESP8266 board etc.

I can assure you the problem is 99.9% likely to be your local environment. Every change to the library has  complex tests run on it and is fully compiled using [Travis](https://travis-ci.org/markszabo/IRremoteESP8266) for the last couple of years.

crankyoldgit, 2019-06-03:
FYI, I just verified it builds fine under Arduino IDE 1.8.8 and Windows 10, using 2.6.0 of the library and ArduinoJson v5.13.5

davidw3591, 2019-06-03:
Thanks for the information.

Just tried to compile it on another PC using IDE 1.8.9.

In board manager, I added esp8266 v2.5.2.
[image]

Installed these libraries:
- ArduinoJson v5.13.5
- PubSubClient v2.7.0
- WiFiManager v0.14.0
- IRremoteESP8266 2.6.0

I still have the same issue. What am I overseeing?

crankyoldgit, 2019-06-03:
Honestly, I'm not sure. I just changed my libraries and esp8266 core to the same versions as you suggested. I even selected the same board type and SPIFFS size as you. (Though, you may need to set it to 1M)
I'm still on Arduino IDE 1.8.8 though, but it worked for me. I doubt the difference is in the IDE version.

Did you try a new copy of `IRMQTTServer` from the examples page?

If it is still the exact same messages as before, I'm still pretty sure it's something in your environment.

crankyoldgit, 2019-06-03:
[image]

crankyoldgit, 2019-06-03:
@davidw3591 FYI, I think I just accidentally reproduced your error messages.
I accidentally typed something at the start of the first line of `IRMQTTServer.ino`.

When I removed the junk characters, it went back to compiling like normal.

So check the very top of `IRMQTTServer.ino` or better still. Create a new copy via the "Examples" sub-menu.

davidw3591, 2019-06-03:
@crankyoldgit I got it to compile eventually. I removed the IRremoteESP8266 library again and reinstalled it with the library manager. Then created a new copy of IRMQTTServer via the "Examples" sub-menu and it compiled without errors.

Before I was using with the IRMQTTServer.ino and IRMQTTServer.h files which I extracted from a downloaded GIT zip file. Must have been somethng wrong with these files I guess?

Great piece of software by the way, it worked right out the box with my AirCon together with HomeAssistant. 😄 

Thank you very much for helping me out.

closed 2019-06-03 by davidw3591