crankyoldgit/IRremoteESP8266 #728: Irremote does not work at 160MHz

sblantipodi, 2019-06-04:
Hi,
using irremote latest on a D1 Mini and a or shield from lolin

Irremote works perfectly when D1 Mini runs at 80mhz but signal appears to be really "weak" when D1 Mini runs at 160MHz.

Please fix and congrats for the good work guys

crankyoldgit, 2019-06-05:
Have you tried using the [calibrate()](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L221) function?
e.g. Before you start transmitting for the first time after instantiating the `IRsend` object add:
```
irsend.calibrate();
```

I have not experience running the ESP8266 at 160Mhz. Care to share some minimal example code so I can reproduce your problem?

crankyoldgit, 2019-06-05:
Alternatively, can you please try the code in this branch? https://github.com/markszabo/IRremoteESP8266/tree/periodoffset_at_cpufreq

I ran a ESP8266 at 160Mhz and calculated the offset required at that freq.

I also tested a D1 Mini running at @ 160Mhz using PR #729 and without, but using the `calibrate()` function. It correctly controlled real IR devices using signals it generated.

Thus this issue is effectively closed but I'll leave it open until the PR is merged.

sblantipodi, 2019-06-06:
I will check as soon as I can if calibrate() will solve my 160MHz issues.
I will update the issue. thanks

crankyoldgit, 2019-06-07:
Based on your other Issue, I assume it went okay?

sblantipodi, 2019-06-07:
@crankyoldgit I'm using 80MHz now. Too many problems  to solve first. Can I use calibrate() even when not using rawdata? For example with samsung.h?

crankyoldgit, 2019-06-07:
`calibrate()` tunes any `IRsend` object/instance to have the internal timing offsets for the current given hardware. So, yes. Just call it once per instance, and after that, every thing else should use the correct timings for your hardware from then on. e.g. Put it in your `setup()` routine.

crankyoldgit, 2019-06-07:
After PR #729 is merged, you shouldn't need it, as the default should change based on the clock frequency defined in `F_CPU`

sblantipodi, 2019-06-07:
> After PR #729 is merged, you shouldn't need it, as the default should change based on the clock frequency defined in `F_CPU`

when will we see the update in the public repository?
I use platformio to download the libraries

crankyoldgit, 2019-06-07:
Probably in less than a week.

You can add a custom environment in your project's `platformio.ini` file, or modify your existing one by adding something like the follow to use a specific branch:

```
[env:whatever_youwant]
platform=espressif8266
framework=arduino
board=d1_mini
lib_deps =
  https://github.com/markszabo/IRremoteESP8266.git#periodoffset_at_cpufreq
```

closed 2019-06-08 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

commit f752f8de (2019-06-05): Remove ESP8266 check for kPeriodOffset.

Fixes #728
PR #729 (2019-06-08): Add kPeriodOffset for CPU Freq of 160MHz.
Add a default value that should be correct @ 160MHz cpu freq.

Fixes #728