crankyoldgit/IRremoteESP8266 #731: Need help on Daikin 160bit ARC423A5

pasna, 2019-06-06:
Hi,

I had download IRremoteESP8266 v.2.6.0 and try to add coding for my Daikin ARC423A5 which is 160 bit ( 7 bytes + 13 bytes).

Currently I cannot work around FAN (byte 17) and Temperature (byte 16) 

Turn OFF / COOL / 24 C / fan HIGH / no Swing
Row: 1
Hex: 11 DA 27 F0 0D 00 0F 11 DA 27  00  D3  30  11  00  00  1C  07  08  51
         /0 /1  /2  /3  /4  /5  /6   /7 /8 /9   /10  /11  /12  /13 /14 /15 /16 /17 /18 /19
Binary:
00010001 11011010 00100111 11110000
00001101 00000000 00001111 00010001
11011010 00100111 00000000 11010011
00110000 00010001 00000000 00000000
00011100 00000111 00001000 01010001

Turn ON / COOL / 24 C / fan HIGH / no Swing
Row: 2
Hex: 11 DA 27 F0 0D 00 0F 11 DA 27 00 D3 31 11 00 00 1C 07 08 52
Binary:
00010001 11011010 00100111 11110000
00001101 00000000 00001111 00010001
11011010 00100111 00000000 11010011
00110001 00010001 00000000 00000000
00011100 00000111 00001000 01010010

Turn ON / COOL / 25 C / fan AUTO / no Swing
Row: 1
Hex: 11 DA 27 F0 0D 00 0F 11 DA 27 00 D3 31 11 00 00 1E 0A 08 57
Binary:
00010001 11011010 00100111 11110000
00001101 00000000 00001111 00010001
11011010 00100111 00000000 11010011
00110001 00010001 00000000 00000000
00011110 00001010 00001000 01010111

Turn ON / COOL / 25 C / fan HIGH /  Swing ON
Row: 1
Hex: 11 DA 27 F0 0D 00 0F 11 DA 27 00 D3 31 F1 00 00 1E 07 08 34
Binary:
00010001 11011010 00100111 11110000
00001101 00000000 00001111 00010001
11011010 00100111 00000000 11010011
00110001 11110001 00000000 00000000
00011110 00000111 00001000 00110100

my decode from IRDump always show byte 17 with "70" instead of "07". I had try changeing bit mask one way or the other but unsuccessful, while byte 16 for temperature is totally unmanageable. Please help.
Great appreciation.

crankyoldgit, 2019-06-06:
Got a link to your code?

pasna, 2019-06-06:
currently I did with offline on my Mac. How to send my code? Should I attach folder of all codes on this thread?

crankyoldgit, 2019-06-06:
Best would be to push your changes to your own github fork of the library.
Pastebin or such will work too.
Or just upload/attach it here.
If it is only one or two functions, just include it in the issue message directly (Use the code markdown block etc so its formatted nicely)

crankyoldgit, 2019-06-06:
Have you mapped out what every value of the Temperature byte holds for every degree between the Min temp and the max temp on the remote? If so, what are they?

Ditto for the fan setting.

pasna, 2019-06-06:
[DAIKIN160BIT.txt](https://github.com/markszabo/IRremoteESP8266/files/3262079/DAIKIN160BIT.txt)
Am I sending the right file?

pasna, 2019-06-06:
https://github.com/pasna/IRremoteESP8266/compare/master...pasna-patch-1
these are files that I made change.

pasna, 2019-06-06:
sorry if I messed up.

crankyoldgit, 2019-06-07:
I got `DAIKIN160BIT.txt` fine, however, to help you work out the temp encoding, we are going to need the full range of temperature values, with at the very least the lowest temp and the highest temp. They are most important.

I'll try to look at you code shortly. But I think you've uploaded them with CR+LF for line endings, thus every line is flagged as changed.

crankyoldgit, 2019-06-07:
Your code:
```
uint8_t IRDaikin160::getFan() {
  uint8_t fan = remote_state[kDaikin160ByteFan] >> 1; // >> 4
  //if (fan != kDaikinFanQuiet && fan != kDaikinFanAuto) fan -= 2;
  if (fan == 0x03) {
    fan = 0;
  }else if (fan == 0x04) {
    fan = 1;
  }else if (fan == 0x05) {
    fan = 2;
  }else if (fan == 0x06) {
    fan = 3;
  }else if (fan == 0x07) {
    fan = 4;
  }else if (fan == 0x0a) {
    fan = 5;
  }else if (fan == 0x0b) {
    fan = 6;
  }
  return fan;
}
```

In short, you are doing a bit shift you don't need to do.

i.e.  `uint8_t fan = remote_state[kDaikin160ByteFan] >> 1;` should be: ` uint8_t fan = remote_state[kDaikin160ByteFan] & kDaikin160MaskFan;`

Or if you want it to behave like the other routines in the other classes, try:

```
// Set the speed of the fan, 1-5 or kDaikinFanAuto or kDaikinFanQuiet
void IRDaikin160::setFan(const uint8_t fan) {
  uint8_t fanset;
  if (fan == kDaikinFanQuiet || fan == kDaikinFanAuto)
    fanset = fan;
  else if (fan < kDaikinFanMin || fan > kDaikinFanMax)
    fanset = kDaikinFanAuto;
  else
    fanset = 2 + fan;
  // Set the fan speed bits, leave *upper* 4 bits alone
  remote_state[kDaikin160ByteFan] &= ~kDaikin160MaskFan;
  remote_state[kDaikin160ByteFan] |= fanset;
}

uint8_t IRDaikin160::getFan() {
  uint8_t fan = remote_state[kDaikin160ByteFan] & kDaikin160MaskFan;
  if (fan != kDaikinFanQuiet && fan != kDaikinFanAuto) fan -= 2;
  return fan;
}
```

Try that.

crankyoldgit, 2019-06-07:
Also try this (like above, untested but should give you the idea):
```
// Set the temp in deg C
void IRDaikin160::setTemp(const uint8_t temp) {
  uint8_t degrees = std::max(temp, kDaikinMinTemp);
  degrees = std::min(degrees, kDaikinMaxTemp) - kDaikinMinTemp;
  remote_state[kDaikin160ByteTemp] &= ~kDaikin160MaskTemp;
  remote_state[kDaikin160ByteTemp] |= (degrees << 1);
}

uint8_t IRDaikin160::getTemp(void) {
  return ((remote_state[kDaikin160ByteTemp] & kDaikin160MaskTemp) >> 1) + kDaikinMinTemp;
}
```

**_Note_** that remote_state[kDaikin160ByteTemp] (aka. byte 16), has it's highest bit set (`0b10000000`) when in **DRY** mode, and uses a temp setting of 10 degrees, according to your data in the text file.

crankyoldgit, 2019-06-07:
P.S. I look forward to the future Pull Request adding support for the new Daikin variant to the library. ;-)

Nice work so far!

pasna, 2019-06-07:
Good morning, 
Great appreciation on your fast reponse. 

I attach new file for full range temperature from 18ºC to 32ºC
[Daikin160Bits-Temperature-Full_range.txt](https://github.com/markszabo/IRremoteESP8266/files/3264766/Daikin160Bits-Temperature-Full_range.txt)

crankyoldgit, 2019-06-07:
FYI, your recent data contradicts the data you originally supplied in the first message of this issue.
i.e. the value of byte 16 was increasing with the increase temperature, now it is decreasing with the increase in temp.

So try the following:
```
// Set the temp in deg C
void IRDaikin160::setTemp(const uint8_t temp) {
  uint8_t degrees = std::max(temp, kDaikinMinTemp);
  degrees = kDaikinMaxTemp + 8 - std::min(degrees, kDaikinMaxTemp);
  remote_state[kDaikin160ByteTemp] &= ~kDaikin160MaskTemp;
  remote_state[kDaikin160ByteTemp] |= (degrees * 2);
}

uint8_t IRDaikin160::getTemp(void) {
  return kDaikinMaxTemp - (((remote_state[kDaikin160ByteTemp] & kDaikin160MaskTemp) / 2) - 8);
}
```

i.e. Byte 16 (`remote_state[kDaikin160ByteTemp]`) of `0x10` = **16** (dec). Which following the above `getTemp()` should be equal to 32 - ((**16** / 2) - 8) = 32 - (8 - 8) = 32 - 0 = _32C_
and Byte 16 of `0x2C` = **44** (dec) .. which should be 32 - (**44** / 2) - 8) = 32 - (22 - 8) = 32 - 14 = _18C_

and the other direction via `setTemp()` 
_32C_ = (32 + 8 - **32**) * 2 = (40 - **32**) * 2 = 8 * 2 = 16 = `0x10`
_18C_ = (32 + 8 - **18**) * 2 = (40 - **18**) * 2 = 22 * 2 = 44 = `0x2C`

pasna, 2019-06-07:
I think, I made mistake on my last file. It is another way round on the temperature value. I will work it out. 
[Daikin160Bits-Temperature-Full_range-2.txt](https://github.com/markszabo/IRremoteESP8266/files/3266104/Daikin160Bits-Temperature-Full_range-2.txt)

crankyoldgit, 2019-06-07:
> I will work it out.

Good luck.

pasna, 2019-06-10:
I worked out formula for temperature
```
byte16 
minimum temperature 18C = 0x10 = 16 (dec) 
maximum temoeratuire 32C = 0x2C = 44 (dec)
minnimum temp : (18*2) = 36 - 20 = 16(dec) = 0x10
maximum temp : (32*2) = 64 - 20 = 44(dec) = 0x2C 

// Set the temp in deg C
void IRDaikin160::setTemp(const uint8_t temp) {
  uint8_t degrees = std::max(temp, kDaikinMinTemp);
  degrees = (std::min(degrees, kDaikinMaxTemp) * 2) - 20;
  remote_state[kDaikin160ByteTemp] &= ~kDaikin160MaskTemp;
  remote_state[kDaikin160ByteTemp] |= degrees;
}

uint8_t IRDaikin160::getTemp(void) {
  return (remote_state[kDaikin160ByteTemp] & kDaikin160MaskTemp) +kDaikinMinTemp;
}

```
Please correct me if I am wrong.
Best regards.

crankyoldgit, 2019-06-10:
Looks right to me, but roof is in your own testing.

crankyoldgit, 2019-06-12:
@pasna Is your fork/code in a state that I can merge it back in to the main library?

pasna, 2019-06-13:
@crankyoldgit I am working on the decode, still need improvement. Currently I can control my AC but when I try to dump from AC remote on serial monitor still shows "UNKNOWN" and secondly my IR distant only 30 cm even though I pass signal through a BC338 transistor.

crankyoldgit, 2019-06-13:
I can handle the base decoder i.e. converting the signal into a `uint8_t state[]` array. If you've got the stuff that can read/write to char array, we could be done. If "decode", means _"You still need to work out what all the bits in the array do"_, then yeah, we've got a lot more to do, that you'll probably need to work out.

What I'll need to do the _base_ decoder will be a good raw capture of a single message from the `IRrecvDumpV2` example code.

pasna, 2019-06-13:
I mean, when I capture signal with IRrecvDumpV2, it isn't converting rawData into hexadecimal array. I don't really know where was my mistake. I have all the rawData signal in files which got from IRrecvDumpV2 and IRLearn (AnalysIR), if these will do and if you please.

crankyoldgit, 2019-06-13:
Yes, the ones from `IRrecvDumpV2` would be great.

pasna, 2019-06-13:
Here is the file of rawdata from IRrecvDumpV2.
Thanks

[DAIKIN160BIT-rawdata-irrecvdumpv2.txt](https://github.com/markszabo/IRremoteESP8266/files/3286261/DAIKIN160BIT-rawdata-irrecvdumpv2.txt)

crankyoldgit, 2019-06-13:
Thanks. I'll try to take a look at it tomorrow.

crankyoldgit, 2019-06-14:
@pasna If you want to try out the `daikin_ARC423A5` branch at: https://github.com/markszabo/IRremoteESP8266/tree/daikin_ARC423A5
It should decode the messages into `uint8_t state[]`s for you via the `IRrecvDumpV2` in that branch.

Hopefully it will make your life easier. It's based off **only** off the first sample in the `txt` file you supplied.

Happy analyzing/decoding. Let me know if there is a problem.

crankyoldgit, 2019-06-16:
FYI, that PR is now merged in to the `master` branch, please use that instead now.

pasna, 2019-06-20:
@crankyoldgit I had made further coding on IRMQTTServer to work for Daikin160bit, how to send these changes to get approved. Should I do pull request, please advise.

crankyoldgit, 2019-06-20:
A PR is preferred. It won't be the first A/C we support that doesn't have vent positions worked out. They can always be added later.

When you submit a PR, I can then edit that branch directly and I can do cleanup and add testing in your branch.

pasna, 2019-06-25:
Done testing IRrecvDumpV2, vertical swing does not respond correctly - show always "ON". My remote ARC423A5 and my aircon mode FTE12HV2S has vertical vent position. I will modify and send new PR.
Have a great day.

crankyoldgit, 2019-06-25:
Looking forward to it. I just used your code. All the bugs are yours. ;-)

crankyoldgit, 2019-06-27:
FYI @pasna  I've updated #777 to include most of your changes in your recent PR.
Please test out that branch.

I noticed you made changes to the other Daikin protocols. I didn't include those changes. Do you have those units too? Are the changes you made based on real data/testing with real devices?

Also, for your future PRs. Could you possible send them from a normal clone of the library? Manually diff'ing the files is time consuming to port it over etc using the way you are sending them now.

pasna, 2019-06-28:
@crankyoldgit I had test Daikin160 latest branch and it works perfectly, on IRMQTTServer and IRrecvDumpV2. All readings reflect correctly except the interpretation of Vertical Vent Position. Currently Daikin160SwingVHighest reflect highest angle of 140º and Daikin160SwingVLowest reflect lowest angle of 90º . This depend on how the construction was initially design.

On Daikin216bit protocol, I do have remote unit and A/C, my A/C model is FTE18GV2S. The reason for minor adjustment was that IRrecvDumpV2 from remote decode correctly while IRMQTTServer send incorrect FAN speed when read on IRrecvDumpV2.

I did not realized that my current way of PR would create difficulties. I would be glad to follow whichever better method to reduce burden, I am non programmer and have little knowledge on github, but I will be much happy to learn. Please advise any channel that I can study the method?
Have a great weekend.
Best regards.

crankyoldgit, 2019-06-28:
> All readings reflect correctly except the interpretation of Vertical Vent Position. Currently Daikin160SwingVHighest reflect highest angle of 140º and Daikin160SwingVLowest reflect lowest angle of 90º . This depend on how the construction was initially design.

I think we are having a "Lost in Translation" experience.
I'm not sure what you mean.
Is `Daikin160SwingVLowest` the lowest position, or not? If not, what should the value of the lowest setting be?

Is `Daikin160SwingVHighest` the Highest position, or not? If not, what should the value of the highest setting be?

>

crankyoldgit, 2019-06-28:
> I did not realized that my current way of PR would create difficulties. I would be glad to follow whichever better method to reduce burden, I am non programmer and have little knowledge on github, but I will be much happy to learn. Please advise any channel that I can study the method?

Try reading:
https://help.github.com/en/articles/creating-a-pull-request & https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

I'd start from a fresh/clean client. e.g.
`git clone https://github.com/markszabo/IRremoteESP8266.git`
then
`cd IRremoteESP8266`
then
`git checkout Daikin160` to start working on the `Daikin160` branch.
Edit the files you need to in their existing positions.
When you are done, create a commit from all of your changes.
e.g.
`git commit -a`
When you are ready to push & make a PR.
`git push` (You may need to follow some instructions on telling it to add it to **your fork** on github.)

If you then go to your fork of the library on github. e.g. https://github.com/pasna/IRremoteESP8266
or https://github.com/pasna/IRremoteESP8266/tree/Daikin160
It will probably have a message saying here a new branch called Daikin160 etc. and to 'Compare' etc or create a PR to `markszabo/master` and then follow it's instructions.

pasna, 2019-06-28:
[image]
Hi again,
I attached a picture file, hopefully will explain what I mean. 
Picture 1 on the left is when command send "Lowest position" (0x11) the vent open up the least,  and Picture 2 on the right is when command send "Highest position" (0x51 ) the vent open up the widest.

crankyoldgit, 2019-06-28:
Thanks. That explains a lot. I've reversed the order. Can you please try the latest commit to the Daikin160 branch.

closed 2019-07-01 by crankyoldgit

pasna, 2019-07-01:
Good Morning @crankyoldgit ,
Tested the latest modification, IRrecvDumpV2 and IRMQTTServer work great. All orders are correct now.
Thank you and best regards.

crankyoldgit, 2019-07-01:
Excellent. Thanks for all your hard work, and thanks for confirming it's working.

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

breezer-11, 2025-08-25:
Good day @crankyoldgit!

I am working on Daikin's ARC480A21 remote for my project, I would like to know if its compatible with other Daikin's IR protocol here

crankyoldgit, 2025-08-26:
Only one way to find out for sure, is for _you_ to test it. Unless it is listed in:  https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md

PR #777 (2019-07-01): Daikin160: Add detailed & common a/c support.
* Now supports:
  - Power
  - Mode
  - Temperature
  - Fan Speed
  - Vertical Swing
  - toString
  - common a/c api.

* Unit test coverage for new features.
* Update example code to reflect changes.

This commit is based on @pasna's work. I'm only taking credit for bug fixes &
unit tests.

Obsoletes PR #771 & #772 (too difficult to merge cleanly)
Fixes #731
commit f620ea67 (2019-07-01): Daikin160: Add detailed & common a/c support. (#777)

* Daikin160: Add detailed & common a/c support.
* Now supports:
  - Power
  - Mode
  - Temperature
  - Fan Speed
  - Vertical & Horizontal Swing
  - toString
  - common a/c api.

* Unit test coverage for new features.
* Update example code to reflect changes.

This is based on @pasna's work. I'm only taking credit for bug fixes & unit tests.

Obsoletes PR #771, #772, & #786 (too difficult to merge cleanly)
Fixes #731