crankyoldgit/IRremoteESP8266 #733: Need help for IRCoolixAC

buzzik, 2019-06-06:
Tried to make coolix IR remote.
All is work if I use irsend.sendCOOLIX(****); with codes that I get from Dump v2 sketch.

But I've tried to make sketch using ir_Coolix.h   (as referense I took samsung controll example)
Here is the sketch:
`
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>

#include <ESP8266WiFi.h>
#include <MQTTClient.h>
#include <WiFiUdp.h>

WiFiUDP ntpUDP;
WiFiClient WiFiclient;
MQTTClient client;

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRCoolixAC ac(kIrLed);     // Set the GPIO used for sending messages.

const char* ssid     = "net";
const char* password = "7426170027";

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.
uint64_t StrToHex(const char* str)
{
  return (uint64_t) strtoull(str, 0, 16);
}
void printState() {
  // Display the settings.
  Serial.println("Samsung A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
}
// Example of data captured by IRrecvDumpV2.ino
void messageReceived(String &topic, String &payload) {
  Serial.println("incoming mqtt: " + topic + " - " + payload);
  irsend.sendCOOLIX(StrToHex(payload.c_str()));
  if (topic == "/livingroom/conditioner/state") { ac.setPower(payload == "ON"); }
  if (topic == "/livingroom/conditioner/temp") { ac.setTemp(payload.toInt()); }
  if (topic == "/livingroom/conditioner/raw") { ac.setRaw(StrToHex(payload.c_str())); }
  if (topic == "/livingroom/conditioner/send") { ac.send(StrToHex(payload.c_str())); }
  if (topic == "/livingroom/conditioner/fan") {
    uint8_t fanMode = kCoolixFanMin;
    switch (payload.toInt()) {
      case 0:
      fanMode = kCoolixFanMin;
      break;
      case 1:
      fanMode = kCoolixFanMed;
      break;
      case 2:
      fanMode = kCoolixFanMax;
      break;
      case 3:
      fanMode = kCoolixFanAuto;
      break;
    }
    ac.setFan(fanMode);
  }
  if (topic == "/livingroom/conditioner/mode") {
    // const uint8_t kCoolixCool = 0b000;
    // const uint8_t kCoolixDry = 0b001;
    // const uint8_t kCoolixAuto = 0b010;
    // const uint8_t kCoolixHeat = 0b011;
    // const uint8_t kCoolixFan = 0b100;
    uint8_t acMode = kCoolixCool;
[... 105 lines cut ...]
   +   606, -  1604,    +   646, -   516,    +   606, -   524,    +   604, -  1606,
   +   642, -  1604,    +   638, -  1608,    +   644, -   518,    +   606, -   522,
   +   606, -   520,    +   606, -   520,    +   606, -   522,    +   606, -   522,
   +   604, -   522,    +   604, -   522,    +   606, -  1604,    +   644, -  1604,
   +   644, -  1602,    +   644, -  1604,    +   642, -  1604,    +   642

uint16_t rawData[399] = {4532, 4440,  644, 520,  606, 520,  608, 520,  598, 526,  606, 520,  606, 520,  604, 522,  598, 530,  606, 1604,  644, 1604,  642, 1604,  644, 1
602,  646, 1602,  644, 1602,  644, 1602,  644, 1604,  642, 518,  604, 524,  602, 524,  606, 520,  604, 524,  602, 524,  602, 524,  600, 526,  600, 1610,  644, 1604,  64
0, 1608,  640, 1604,  640, 1606,  642, 1604,  638, 1608,  636, 1610,  638, 524,  570, 556,  572, 554,  570, 556,  572, 554,  572, 554,  570, 556,  570, 558,  570, 1640,
  604, 1642,  606, 1640,  636, 1610,  640, 1608,  640, 1606,  638, 1608,  642, 1604,  642, 4962,  4554, 4414,  644, 518,  606, 522,  608, 518,  606, 520,  606, 520,  60
6, 520,  606, 520,  606, 522,  606, 1606,  644, 1602,  644, 1600,  646, 1602,  642, 1604,  644, 1602,  644, 1602,  644, 1604,  642, 520,  604, 520,  606, 520,  606, 520
,  606, 520,  606, 522,  606, 520,  606, 522,  606, 1606,  644, 1602,  644, 1604,  644, 1602,  642, 1604,  644, 1602,  644, 1602,  644, 1604,  644, 518,  606, 520,  606
, 520,  606, 520,  606, 520,  606, 520,  606, 522,  604, 522,  606, 1604,  644, 1604,  644, 1604,  642, 1602,  646, 1602,  644, 1604,  642, 1604,  644, 1604,  644, 5294
,  4552, 4414,  644, 1602,  644, 516,  604, 1608,  642, 1604,  642, 520,  600, 526,  602, 1608,  642, 520,  600, 526,  608, 1604,  640, 520,  600, 526,  600, 1612,  638
, 1606,  642, 520,  598, 1612,  642, 520,  600, 1610,  644, 1602,  644, 1604,  636, 1608,  642, 520,  566, 1642,  646, 1604,  644, 1602,  644, 518,  604, 520,  604, 522
,  606, 520,  606, 1604,  644, 516,  606, 522,  606, 1606,  644, 1602,  642, 1604,  642, 518,  606, 522,  600, 526,  606, 520,  606, 522,  606, 520,  604, 522,  606, 52
0,  608, 1604,  644, 1602,  644, 1602,  642, 1604,  644, 1604,  642, 4962,  4552, 4414,  642, 1604,  646, 516,  606, 1604,  644, 1602,  638, 522,  606, 520,  606, 1604,
  642, 522,  606, 520,  604, 1606,  638, 522,  606, 520,  606, 1606,  642, 1604,  642, 520,  604, 1608,  642, 520,  606, 1606,  642, 1604,  644, 1602,  644, 1602,  644,
 518,  606, 1604,  644, 1604,  644, 1602,  642, 518,  606, 522,  604, 522,  606, 520,  606, 1604,  646, 516,  606, 524,  604, 1606,  642, 1604,  638, 1608,  644, 518,
606, 522,  606, 520,  606, 520,  606, 522,  606, 522,  604, 522,  604, 522,  606, 1604,  644, 1604,  644, 1602,  644, 1604,  642, 1604,  642};  // COOLIX 0
uint64_t data = 0x0;

`

What im doing wrong?

crankyoldgit, 2019-06-07:
Code nitpick (But probably not the issue):
```
return (uint64_t) strtoull(str, NULL, 16);
```

It probably doesn't matter, but you don't have a `ac.begin();` call.

But the issue is probably here:

```
void messageReceived(String &topic, String &payload) {
  Serial.println("incoming mqtt: " + topic + " - " + payload);
  irsend.sendCOOLIX(StrToHex(payload.c_str()));   // *** This will be sent every time, even if the payload is "ON" which will convert to zero.
  if (topic == "/livingroom/conditioner/state") { ac.setPower(payload == "ON"); }
```

buzzik, 2019-06-07:
> Code nitpick (But probably not the issue):
>
> ```
> return (uint64_t) strtoull(str, NULL, 16);
> ```
>
> It probably doesn't matter, but you don't have a `ac.begin();` call.
>
> But the issue is probably here: [...]

Oh, Probably its bug of copy-paste. This part 

> irsend.sendCOOLIX(StrToHex(payload.c_str()));

 is have escape charecter in the begining.

Here is the clear code
`#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>

#include <ESP8266WiFi.h>
#include <MQTTClient.h>
#include <WiFiUdp.h>

WiFiUDP ntpUDP;
WiFiClient WiFiclient;
MQTTClient client;

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRCoolixAC ac(kIrLed);     // Set the GPIO used for sending messages.

const char* ssid     = "net";
const char* password = "7426170027";

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.
uint64_t StrToHex(const char* str)
{
  return (uint64_t) strtoull(str, 0, 16);
}
void printState() {  // Display the settings.

  Serial.println("Samsung A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
}
void messageReceived(String &topic, String &payload) {
  Serial.println("incoming mqtt: " + topic + " - " + payload);
  if (topic == "/livingroom/conditioner/state") { ac.setPower(payload == "ON"); }
  if (topic == "/livingroom/conditioner/temp") { ac.setTemp(payload.toInt()); }
  if (topic == "/livingroom/conditioner/raw") { ac.setRaw(StrToHex(payload.c_str())); }
  if (topic == "/livingroom/conditioner/fan") {
    uint8_t fanMode = kCoolixFanMin;
    switch (payload.toInt()) {
      case 0:
      fanMode = kCoolixFanMin;
      break;
      case 1:
      fanMode = kCoolixFanMed;
      break;
      case 2:
      fanMode = kCoolixFanMax;
      break;
      case 3:
      fanMode = kCoolixFanAuto;
      break;
    }
    ac.setFan(fanMode);
  }
  if (topic == "/livingroom/conditioner/mode") {
    uint8_t acMode = kCoolixCool;
    switch (payload.toInt()) {
      case 0:
      acMode = kCoolixCool;
      break;
      case 1:
      acMode = kCoolixDry;
      break;
      case 2:
      acMode = kCoolixAuto;
      break;
      case 3:
      acMode = kCoolixHeat;
      break;
      case 4:
      acMode = kCoolixFan;
      break;
    }
    ac.setMode(acMode);
  }
  ac.send();
}

void connect() {
  client.connect("livingRoomConditioner", "mr4k", "Yz10tyevm");
  Serial.println("\nconnected!");
  client.subscribe("/livingroom/conditioner/state");
  client.subscribe("/livingroom/conditioner/temp");
  client.subscribe("/livingroom/conditioner/fan");
  client.subscribe("/livingroom/conditioner/mode");
  client.subscribe("/livingroom/conditioner/raw");

}
void setup() {
  irsend.begin();
  Serial.begin(115200);
  delay(200);
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  Serial.println("begin NTP");
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  delay(2000);
  Serial.print("connecting to MQTT broker...");
  client.begin("192.168.31.200", WiFiclient);
  client.onMessage(messageReceived);
  // Set up what we want to send. See ir_Samsung.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting initial state for A/C.");
  ac.setPower(false);
  ac.setFan(kCoolixFanMed);
  ac.setMode(kCoolixCool);
  ac.setTemp(22);
  printState();
  connect();
}
void loop() {
  client.loop();
  if(!client.connected()) {
    connect();
  }
}
`

and where should I put "ac.begin();" ? Didn't see it in examples.
Thank you for your answers

buzzik, 2019-06-07:
My bad, will fix and try.

> Code nitpick (But probably not the issue):
> 
> ```
> return (uint64_t) strtoull(str, NULL, 16);
> ```

Sry, what its mean and how to fix? My english not so good, dont understand "nitpick " word :)
function 
`uint64_t StrToHex(const char* str)
{
  return (uint64_t) strtoull(str, 0, 16);
}`

need to convert String to uint32_t
to send it via void setRaw(const uint32_t new_code);
Is there better way to convert it ?

crankyoldgit, 2019-06-07:
It means its a very minor/trivial issue.
The fix is, you should use `NULL` instead of `0`. but as I said, it probably isn't the issue.

The real issue is this line:
```
  irsend.sendCOOLIX(StrToHex(payload.c_str()));
```

It is going to be sent every time you receive a MQTT message.
**IF** payload is not a number, it will send a zero.

e.g. When your program gets a MQTT message with a topic of "/livingroom/conditioner/state" and a payload of "ON", for example. It will send zero due to that line as "ON" is not a hexadecimal number. That is, `strtoul()` returns `0` if it encounters an error. i.e. It starts to read a string that doesn't start with `-`, `+`, or a digit.

crankyoldgit, 2019-06-07:
Basically, you probably want to delete that line.

buzzik, 2019-06-07:
> Basically, you probably want to delete that line.

Yes, I understand, that was my mistake. I thought I cleaned up everything:) Will try your solution, thanks!

buzzik, 2019-06-07:
Everithing works great, thanks!

closed 2019-06-07 by buzzik

Lordes21, 2019-06-25:
Hello. I m trying to decode my samsung ac remote to use it in MIT App Inventor 2 to control my ac from smartphone. Can anybody please help me with arduino code? Im using a IR receiver TSOP1136.

crankyoldgit, 2019-06-26:
Please see the FAQ item for help with writing your own code:
https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

Some models of Samsung A/C are already supported, you might find the following helpful.
If `IRrecvDumpV2.ino` detects it correctly then it should work.
https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Samsung.cpp

The following example code isn't for Samsung, but can be easily adapted:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/TurnOnToshibaAC/TurnOnToshibaAC.ino