crankyoldgit/IRremoteESP8266 #734: Samsung AR12KSFPEWQNET works but not good

sblantipodi, 2019-06-06:
Hi,
I'm using this example for my Samsung AR12KSFPEWQNET

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/ControlSamsungAC/ControlSamsungAC.ino

It works but not good. 
Wings can be set on and off when fan is in
ac.setFan(kSamsungAcFanHigh);
Every other states has no effect on wings.

Power mode and quiet mode does not work at all.
I tried ac.setPower(true) and ac.setQuiet(true) with no luck.

Quiet mode is pretty important to me, any idea on how to fix those problems?

Thanks

crankyoldgit, 2019-06-07:
The on/off issue is/was a know bug. It has been fixed **after** the 2.6.0 release. See https://github.com/markszabo/IRremoteESP8266/pull/695 , so either grab the `master` branch, or wait for the next release.

The swing issue is an unresolved issue analysis issue. See https://github.com/markszabo/IRremoteESP8266/issues/505#issuecomment-428023616
e.g. I pointed out there is more to the swing bits than @Hollako said there was.

More data and analysis is needed. If you want to do that, that'd be great. I'm happy to fix/improve the code but I don't have a Samsung A/C and remote so I can't do it myself.

If you can capture some data (with `IRrecvDumpV2`) with Quiet on and off I can try to fix what ever problem that may be.

sblantipodi, 2019-06-07:
Here some rawreading I used that works ok while sending raw.

This is my working 
QUIET MODE ON:
624,17360,3076,8902,520,476,520,1472,520,476,520,474,520,476,520,476,520,474,522,476,520,478,518,1476,516,500,494,502,548,448,546,450,544,452,522,1468,520,1474,520,1472,520,1472,520,1472,520,476,520,476,518,478,516,480,516,500,496,500,494,502,550,446,546,450,544,452,524,472,522,474,518,476,520,476,520,474,522,474,520,474,520,476,520,474,520,476,518,478,518,480,516,480,516,502,494,502,548,1444,524,472,522,472,520,474,518,478,518,476,520,476,520,1472,520,1470,520,1472,520,1474,516,2980,2998,8980,498,1498,548,448,526,470,544,452,524,472,520,474,520,476,520,476,520,476,520,1472,520,474,520,476,520,1474,518,1476,516,1496,496,1498,548,446,546,1446,524,1468,518,1474,520,1472,520,1472,520,1472,520,1474,518,1476,518,480,516,500,496,528,520,1446,544,1446,524,1470,518,476,520,476,520,474,520,476,520,474,520,476,520,474,520,476,520,476,518,1476,516,482,514,502,548,448,548,1442,544,452,522,474,518,476,518,476,520,476,520,474,520,476,520,1472,520,1470,522,1474,518,1476,536

QUIET MODE ON, SWING OFF
624,17808,2998,8962,518,478,518,1474,518,500,496,500,494,502,548,448,546,450,524,472,522,474,520,1472,520,476,522,474,520,476,520,474,522,474,520,1472,520,1474,518,1478,514,1498,494,1498,546,450,544,452,522,474,520,476,520,476,520,476,520,476,520,476,520,476,520,474,522,474,520,476,520,478,518,480,516,482,514,502,494,502,548,448,546,450,544,452,522,474,520,476,518,476,520,476,520,476,520,1472,520,476,520,474,522,476,520,478,518,478,518,480,514,1498,494,1498,546,1446,524,1468,522,2976,3058,8906,520,1472,522,476,520,474,520,476,522,474,520,478,520,478,518,480,516,500,494,1498,496,502,548,448,544,1448,522,1472,518,1474,520,1472,520,476,520,1472,522,1472,520,1476,518,1478,514,1498,550,1444,546,1448,522,1470,518,476,520,476,520,476,520,1472,520,1472,520,1472,520,478,518,478,518,480,516,502,494,502,494,502,548,448,544,452,522,472,520,1472,518,478,520,476,520,476,520,1472,520,476,520,476,520,476,520,478,518,480,516,480,514,502,494,1498,548,1444,524,1468,520,1472,522

QUIET MODE ON, SWING ON
624,17788,3078,8908,516,476,518,1474,520,476,520,476,522,474,520,476,520,476,520,476,518,478,518,1478,514,502,494,502,550,446,546,450,544,452,524,1468,520,1472,520,1472,522,1472,520,1472,520,476,520,476,520,478,516,480,518,500,496,500,494,502,550,448,546,450,544,452,522,474,518,476,518,478,520,476,520,474,522,474,520,476,520,474,522,476,520,478,518,478,518,480,516,500,496,502,546,450,548,1444,530,466,522,474,520,476,518,476,520,476,520,476,520,1472,520,1472,520,1474,520,1474,516,2984,2996,8984,498,1498,550,446,546,450,544,452,522,474,518,478,518,478,520,476,520,476,520,1470,522,476,520,474,522,1474,518,478,516,480,516,500,496,1498,550,1444,544,1448,522,1470,518,478,520,1472,522,476,520,1472,520,1472,520,476,520,478,518,480,516,1498,550,1444,546,1448,544,452,522,472,520,476,518,476,520,476,520,474,520,476,522,474,520,476,520,1472,520,478,518,480,516,480,516,1498,550,446,548,448,546,450,524,472,520,476,518,476,520,476,522,1472,520,1472,520,1470,522,1472,520

POWER MODE:
624,17376,3058,8904,538,454,520,1472,520,476,520,474,520,476,520,476,520,476,520,478,518,478,518,1496,496,502,550,446,548,1442,540,456,524,470,520,1472,518,1474,520,1472,520,1472,520,1472,520,476,520,476,518,480,516,480,514,502,550,448,548,448,544,450,544,450,522,474,520,474,520,476,520,476,522,474,520,476,520,476,520,476,520,476,520,476,520,478,518,478,516,480,516,502,514,480,550,446,548,448,544,452,524,472,520,478,516,478,518,474,522,476,520,1470,522,1472,520,1472,520,1474,516,2984,3046,8928,498,1498,550,446,546,450,524,470,522,474,518,476,518,478,520,476,520,476,520,1472,520,476,520,476,520,476,518,1474,518,482,514,1496,496,502,548,1444,524,1468,522,1468,520,1472,520,1472,520,1472,520,1470,520,1474,518,1474,516,1496,550,446,548,1444,544,1446,524,1470,518,476,520,476,520,474,520,474,520,476,520,474,520,474,520,476,520,478,518,1494,498,1498,550,1442,548,1446,522,1470,520,474,520,478,520,476,520,474,520,476,520,476,520,474,520,1472,520,1476,516,1494,498,1498,518

POWER MODE ON, SWING OFF
624,17786,3078,8904,520,476,518,1474,520,476,520,476,520,476,520,476,520,476,520,476,520,478,516,1478,516,502,494,502,548,1444,544,452,524,472,520,1470,520,1474,520,1472,520,1470,522,1472,520,476,520,478,518,480,516,480,516,502,494,502,548,448,546,450,544,452,522,474,520,476,518,476,520,476,520,476,520,476,520,476,520,474,520,476,520,476,520,478,518,480,516,500,496,502,494,502,548,448,546,450,526,470,524,472,520,476,518,476,520,476,520,476,520,1472,520,1472,520,1472,520,1474,516,2984,2996,8984,498,1498,550,446,546,450,544,450,524,472,518,478,518,478,520,476,520,476,520,1472,520,476,520,476,520,478,518,1496,498,480,516,1496,552,446,550,1444,544,1448,522,1472,518,1474,522,1472,520,1472,522,1470,522,1474,518,1476,516,1498,494,502,550,1444,544,1450,520,1402,588,478,520,476,520,476,520,476,520,476,520,476,520,474,520,476,520,478,516,1478,514,1498,494,1498,548,1446,524,1470,518,478,518,476,520,476,520,476,520,476,520,476,520,476,520,1474,518,1476,516,1496,496,1500,518

POWER MODE ON, SWING ON
624,17802,2996,8962,518,476,520,1476,516,482,514,502,494,502,548,448,546,450,534,462,522,474,518,1474,518,478,520,474,520,1472,522,474,520,476,520,1474,520,1476,516,1496,496,1498,550,1444,540,456,524,472,520,476,518,478,520,476,520,476,520,474,520,476,520,474,520,476,520,478,518,478,518,478,516,480,516,502,494,502,548,448,546,450,544,452,524,472,520,474,518,478,520,476,520,476,520,476,520,476,520,476,520,476,520, [... cut ...] 2,494,502,520,1472,546,1446,524,1468,520,1474,520,476,520,1472,522,1470,522,1472,520,1474,518,1476,516,1496,496,1498,550,1444,524,472,522,474,520,476,518,1474,520,1472,522,1470,522,474,520,476,520,476,518,478,518,478,516,500,496,502,494,502,550,446,544,1446,524,472,520,476,518,476,520,1472,522,474,520,476,520,474,520,476,520,476,520,476,518,480,516,1496,496,1498,548,1444,546,1446,520

Hope it can help. please let me know if you find out how to solve the Quier, Power and swing mode.
@crankyoldgit

crankyoldgit, 2019-06-07:
Can you give me the equivalent of "QUIET MODE ON" with just the QUIET mode off?
i.e. Where only that setting has changed so I can compare the two.

As stated before, the power on/off should be solved if you use the `master` branch. It was fixed in PR #695

sblantipodi, 2019-06-07:
> Can you give me the equivalent of "QUIET MODE ON" with just the QUIET mode off?
> i.e. Where only that setting has changed so I can compare the two.
> 
> As stated before, the power on/off should be solved if you use the `master` branch. It was fixed in PR #695

I'm at work now, I will give you that in 8 hours more or less.
Do you need something else even for swing and power mode?

Thanks

crankyoldgit, 2019-06-07:
To be clear, I need two captures from you to fix Quiet:
1) The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet **Off**
2) The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet **On**
That is so we can have the situation that we can see what bits change for Quiet mode.
i.e. Exactly the same except for the quiet setting being changed.

Power on/off is already fixed. So no data as yet is needed for the moment.

If you want Swing to work, you are going to have to do the analysis yourself to workout and explain what bits are involved. i.e. you have to explain/work out which bits change when the swing setting is changed for ALL modes etc of the Samsung remote. Seriously, read through #505.

I suggest we tackle Quiet mode first. That should be easy.

crankyoldgit, 2019-06-07:
PS. I don't need the **raw** data. Just the `uint8_t state[14] ={ ... };` lines with a description of what you set the remote too should be fine.

sblantipodi, 2019-06-07:
ok this evening I will give you the infos. thanks

sblantipodi, 2019-06-07:
I don't understood why the same button creates different raw data every time. 
every raw data works as expected even if it is different.

crankyoldgit, 2019-06-07:
Read this: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#the-libraryexample-code-doesnt-understand-my-air-conditioner-remote-

> Simple IR device protocols use a 12-48 bit value mapping a single button press. A/C remotes however tend to send the entire state of the remote to the device with each button press. e.g. Desired Temp., Operating Mode, Fan Speed, Fan Direction(s), Economy Mode(s), Sleep Modes, Time & Timer Modes, etc, plus a checksum. In short, imagine it sending every button on the remote for each button press. This can produce a message of around 128 bits of data, ...

sblantipodi, 2019-06-07:
it does not answer my question because same button, in the same state procudes different raw data on my remote.
in the same state 
remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet Off
I can have different raw readings. 

I have tried different readings for the same state and it works ok but can't understand why.
probably there is somethings related to current time?

crankyoldgit, 2019-06-07:
Raw messages are a representation of an analogue message. i.e. If I taped someone Italian saying "Buon giorno", and played it back to you. You'd think I spoke Italian. But the same person who I recorded could say it again, and it would be slightly different. That's what `sendRaw()` does, it's like a `.wav` recording.

The next level the library has, if the base protocol is supported is to effectively convert the sounds to text. i.e. `.wav` to `.txt`. A much smaller. Similarly, the library essentially converts the "raw" format into a short "state[]" format. It can then also covert the text to speech, if we follow the same analogie. However, it's now digital, so it will always produce the same output to the speaker.
It doesn't understand the message, it just know how to covert sounds to letters, and vice versa.

The final level, if supported, is to understand what the `text` means. eg. grammar, and how to construct a sentence, to be fluent. That is, to covert the `state[]` into various AirCon settings, and vice versa.

Does that make sense?

sblantipodi, 2019-06-07:
> Raw messages are a representation of an analogue message. i.e. If I taped someone Italian saying "Buon giorno", and played it back to you. You'd think I spoke Italian. But the same person who I recorded could say it again, and it would be slightly different. That's what `sendRaw()` does, it's like a `.wav` recording.
> 
> The next level the library has, if the base protocol is supported is to effectively convert the sounds to text. i.e. `.wav` to `.txt`. Much smaller. Similarly, the library essentially converts the "raw" format into a short "state[]" format. It can then also covert the text to speech, if we follow the same analogie. However, it's now digital, so it will always produce the same output to the speaker.
> It doesn't understand the message, it just know how to covert sounds to letters, and vice versa.
> 
> The final level, if supported, is to understand what the `text` means. eg. grammar, and how to construct a sentence, to be fluent. That is, to covert the `state[]` into various AirCon settings, and vice versa.
> 
> Does that make sense?

wow. I really appreciate this explanations. you made my day. never understood before.
thanks! see you later for the ir codes :)

crankyoldgit, 2019-06-07:
Congrats. You have just inspired an FAQ entry. https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#why-is-the-raw-data-for-a-button-or-ac-state-always-different-for-each-capture

sblantipodi, 2019-06-07:
@crankyoldgit  the output I will attach is:
The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet On
The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet Off

repeated 2 times.
hope that helped

Timestamp : 000079.217
Encoding  : SAMSUNG_AC
Code      : 02820F000020F001F2FE710011F0 (112 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
Library   : v2.6.0

Raw Timing[233]:
   +   632, - 17340,    +  3076, -  8906,    +   518, -   478,    +   516, -  1476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   518, -   478,    +   518, -  1476, 
   +   516, -   502,    +   494, -   502,    +   494, -   502,    +   550, -   446, 
   +   546, -   450,    +   546, -  1446,    +   520, -  1474,    +   518, -  1474, 
   +   520, -  1472,    +   520, -  1474,    +   518, -   476,    +   520, -   476, 
   +   520, -   478,    +   518, -   480,    +   516, -   482,    +   514, -   502, 
   +   494, -   502,    +   548, -   446,    +   546, -   448,    +   546, -   450, 
   +   544, -   452,    +   520, -   476,    +   518, -   478,    +   520, -   476, 
   +   520, -   476,    +   520, -   474,    +   522, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   518, -   478,    +   518, -   480, 
   +   516, -   480,    +   516, -   502,    +   494, -   502,    +   550, -  1444, 
   +   546, -   450,    +   534, -   462,    +   520, -   476,    +   518, -   478, 
   +   518, -   476,    +   520, -   476,    +   520, -  1472,    +   522, -  1472, 
   +   520, -  1472,    +   522, -  1474,    +   518, -  2984,    +  2996, -  8984, 
   +   498, -  1498,    +   550, -   446,    +   546, -   450,    +   546, -   450, 
   +   528, -   470,    +   518, -   476,    +   518, -   478,    +   518, -   476, 
   +   520, -   476,    +   520, -  1472,    +   520, -   476,    +   520, -   476, 
   +   520, -  1474,    +   518, -  1476,    +   518, -  1498,    +   494, -  1500, 
   +   550, -   446,    +   546, -  1444,    +   524, -  1470,    +   518, -  1474, 
   +   520, -  1472,    +   520, -  1472,    +   522, -  1472,    +   520, -  1474, 
   +   518, -  1476,    +   516, -   482,    +   516, -   500,    +   494, -   502, 
   +   550, -  1442,    +   548, -  1444,    +   544, -  1448,    +   520, -   476, 
   +   518, -   478,    +   520, -   476,    +   520, -   474,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   478, 
   +   518, -  1476,    +   516, -   482,    +   514, -   502,    +   550, -   446, 
   +   550, -  1442,    +   546, -   450,    +   524, -   472,    +   522, -   474, 
   +   518, -   478,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -  1472,    +   520, -  1474,    +   520, -  1476,    +   516, -  1478, 
   +   536

uint16_t rawData[233] = {632, 17340,  3076, 8906,  518, 478,  516, 1476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  518, 478,  518, 1476,  516, 502,  494, 502,  494, 502,  550, 446,  546, 450,  546, 1446,  520, 1474,  518, 1474,  520, 1472,  520, 1474,  518, 476,  520, 476,  520, 478,  518, 480,  516, 482,  514, 502,  494, 502,  548, 446,  546, 448,  546, 450,  544, 452,  520, 476,  518, 478,  520, 476,  520, 476,  520, 474,  522, 476,  520, 476,  520, 476,  520, 476,  518, 478,  518, 480,  516, 480,  516, 502,  494, 502,  550, 1444,  546, 450,  534, 462,  520, 476,  518, 478,  518, 476,  520, 476,  520, 1472,  522, 1472,  520, 1472,  522, 1474,  518, 2984,  2996, 8984,  498, 1498,  550, 446,  546, 450,  546, 450,  528, 470,  518, 476,  518, 478,  518, 476,  520, 476,  520, 1472,  520, 476,  520, 476,  520, 1474,  518, 1476,  518, 1498,  494, 1500,  550, 446,  546, 1444,  524, 1470,  518, 1474,  520, 1472,  520, 1472,  522, 1472,  520, 1474,  518, 1476,  516, 482,  516, 500,  494, 502,  550, 1442,  548, 1444,  544, 1448,  520, 476,  518, 478,  520, 476,  520, 474,  520, 476,  520, 476,  520, 476,  520, 476,  520, 478,  518, 1476,  516, 482,  514, 502,  550, 446,  550, 1442,  546, 450,  524, 472,  522, 474,  518, 478,  520, 476,  520, 476,  520, 476,  520, 1472,  520, 1474,  520, 1476,  516, 1478,  536};  // SAMSUNG_AC
uint8_t state[14] = {0x02, 0x82, 0x0F, 0x00, 0x00, 0x20, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};

Timestamp : 000107.067
Encoding  : SAMSUNG_AC
Code      : 02920F000000F001F2FE710011F0 (112 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
Library   : v2.6.0

Raw Timing[233]:
   +   608, - 17368,    +  3078, -  8904,    +   520, -   478,    +   516, -  1476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   478,    +   518, -   478,    +   518, -  1476, 
   +   516, -   500,    +   494, -   502,    +   550, -  1442,    +   546, -   452, 
   +   544, -   450,    +   522, -  1470,    +   520, -  1472,    +   522, -  1472, 
   +   520, -  1474,    +   520, -  1472,    +   520, -   476,    +   518, -   478, 
   +   518, -   478,    +   516, -   480,    +   516, -   482,    +   514, -   502, 
   +   548, -   448,    +   548, -   448,    +   544, -   452,    +   536, -   460, 
   +   520, -   474,    +   520, -   478,    +   518, -   476,    +   522, -   476, 
[... cut ...] 520, 1472,  520, 1472,  522, 474,  520, 478,  518, 478,  518, 1478,  514, 1498,  550, 1444,  546, 450,  544, 450,  522, 474,  520, 476,  518, 478,  520, 476,  520, 474,  522, 474,  520, 476,  520, 1472,  522, 476,  520, 478,  518, 480,  516, 1498,  494, 502,  550, 446,  548, 448,  546, 450,  546, 450,  520, 476,  518, 476,  520, 1474,  520, 1472,  520, 1472,  520, 1474,  518};  // SAMSUNG_AC
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};

crankyoldgit, 2019-06-07:
Quiet on: `uint8_t state[14] = {0x02, 0x82, 0x0F, 0x00, 0x00, 0x20, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};`
Quiet off: `uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};`

The differences are in `state[1]` & `state[5]`

sblantipodi, 2019-06-07:
> Quiet on: `uint8_t state[14] = {0x02, 0x82, 0x0F, 0x00, 0x00, 0x20, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};`
> Quiet off: `uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};`
> 
> The differences are in `state[1]` & `state[5]`

so how can we fix the problem?

crankyoldgit, 2019-06-08:
See PR #736 for the changes.

You can use the code in the `Issue734` branch here: https://github.com/markszabo/IRremoteESP8266/tree/Issue734

Please test it out and let me know how it goes. It should work based on the data you've supplied.

sblantipodi, 2019-06-08:
@crankyoldgit 
included the lib this way:
lib_deps =   https://github.com/markszabo/IRremoteESP8266.git#Issue734

          acir.setQuiet(true);
          acir.send();

it doesn't work. quiet is not enabled, AC is noisy

sblantipodi, 2019-06-08:
any help appreciated

crankyoldgit, 2019-06-08:
If you take a look at: https://github.com/markszabo/IRremoteESP8266/commit/bfcc92ffb0a506039d0bc122d55060901cae31cd

You will see I can/have (re)constructed the original state/message you reported as working when using the latest fix (the one you should be/are trying to use).

My guess is either a) you are not using the class/library correctly, or b) it isn't using the updated code/branch. i.e. It's probably still using the plain 2.6.0 release., or c) there is something else we don't understand about the Quiet setting.

If c), What is the state you have the `acir` class object in? To get a partial idea, what does the following output (untested & uncompiled example psuedo code):
```
acir.setQuiet(true);
Serial.print("Samsung AC is in the following state: ");
Serial.println(acir.toString);
acir.send();
```

or better still
```
acir.setQuiet(true);
Serial.print("Samsung AC is in the following state: ");
Serial.println(acir.toString);
uint8_t state[kSamsungAcStateLength] = acir.getRaw();
for (int i = 0; i < kSamsungAcStateLength; i++) {
  Serial.printf("%d: %02x\n", i, state[i]);
}
acir.send();
```

sblantipodi, 2019-06-08:
@crankyoldgit 
ok it works, I was wrong because I forgot to set the 
acir.setQuiet(false);
when I want to exit the quiet state and switching from quiet to low produced no audible effect :)

now the only things that remains uncovered is the powerful state.

do you think that it will be easy to fix like the quiet mode?

crankyoldgit, 2019-06-08:
If you want to do the same for the powerful/turbo state, sure. i.e. capture a similar pair of messages that should help isolate the bits.

sblantipodi, 2019-06-08:
@crankyoldgit 
> If you want to do the same for the powerful/turbo state, sure. i.e. capture a similar pair of messages that should help isolate the bits.

The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet Off POWER ON
The remote set to: On, Cool, 16C, Fan Auto, Swing Off, Beep Off Clean Off, & Quiet Off POWER OFF

repeated 2 times.

Timestamp : 000012.630
Encoding  : SAMSUNG_AC
Code      : 02920F000000F001A2FE77001FF0 (112 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 16C, Fan: 7 (TURBO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
Library   : v2.6.0

Raw Timing[233]:
   +   624, - 17366,    +  3074, -  8912,    +   522, -   474,    +   518, -  1472, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   518, -  1478, 
   +   516, -   500,    +   496, -   502,    +   494, -  1500,    +   546, -   450, 
   +   546, -   450,    +   544, -  1448,    +   520, -  1474,    +   518, -  1474, 
   +   518, -  1474,    +   520, -  1472,    +   520, -   476,    +   520, -   476, 
   +   518, -   478,    +   518, -   480,    +   516, -   480,    +   516, -   502, 
   +   494, -   502,    +   550, -   446,    +   548, -   448,    +   544, -   452, 
   +   522, -   474,    +   520, -   476,    +   518, -   478,    +   518, -   478, 
   +   520, -   476,    +   520, -   474,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   518, -   478,    +   518, -   480, 
   +   516, -   480,    +   516, -   502,    +   494, -   502,    +   550, -   446, 
   +   546, -   450,    +   546, -   450,    +   544, -   452,    +   520, -   478, 
   +   518, -   478,    +   520, -   476,    +   520, -  1472,    +   520, -  1472, 
   +   520, -  1472,    +   522, -  1474,    +   518, -  2982,    +  2998, -  8986, 
   +   496, -  1498,    +   550, -   446,    +   550, -   446,    +   546, -   450, 
   +   546, -   450,    +   522, -   474,    +   518, -   478,    +   518, -   478, 
   +   520, -   476,    +   520, -  1474,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -  1474,    +   518, -   480,    +   516, -  1498, 
   +   496, -   502,    +   494, -  1500,    +   548, -  1444,    +   546, -  1446, 
   +   518, -  1474,    +   520, -  1474,    +   520, -  1472,    +   520, -  1472, 
   +   520, -  1474,    +   518, -  1476,    +   518, -  1496,    +   496, -   502, 
   +   494, -  1498,    +   550, -  1442,    +   546, -  1448,    +   520, -   476, 
   +   518, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   518, -   478, 
   +   520, -  1476,    +   516, -  1498,    +   496, -  1500,    +   548, -  1444, 
   +   546, -  1448,    +   522, -   472,    +   520, -   476,    +   518, -   478, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -  1474,    +   518, -  1476,    +   518, -  1476,    +   516, -  1496, 
   +   542

uint16_t rawData[233] = {624, 17366,  3074, 8912,  522, 474,  518, 1472,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  518, 1478,  516, 500,  496, 502,  494, 1500,  546, 450,  546, 450,  544, 1448,  520, 1474,  518, 1474,  518, 1474,  520, 1472,  520, 476,  520, 476,  518, 478,  518, 480,  516, 480,  516, 502,  494, 502,  550, 446,  548, 448,  544, 452,  522, 474,  520, 476,  518, 478,  518, 478,  520, 476,  520, 474,  520, 476,  520, 476,  520, 476,  520, 476,  518, 478,  518, 480,  516, 480,  516, 502,  494, 502,  550, 446,  546, 450,  546, 450,  544, 452,  520, 478,  518, 478,  520, 476,  520, 1472,  520, 1472,  520, 1472,  522, 1474,  518, 2982,  2998, 8986,  496, 1498,  550, 446,  550, 446,  546, 450,  546, 450,  522, 474,  518, 478,  518, 478,  520, 476,  520, 1474,  520, 476,  520, 476,  520, 476,  520, 1474,  518, 480,  516, 1498,  496, 502,  494, 1500,  548, 1444,  546, 1446,  518, 1474,  520, 1474,  520, 1472,  520, 1472,  520, 1474,  518, 1476,  518, 1496,  496, 502,  494, 1498,  550, 1442,  546, 1448,  520, 476,  518, 476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  520, 476,  518, 478,  520, 1476,  516, 1498,  496, 1500,  548, 1444,  546, 1448,  522, 472,  520, 476,  518, 478,  520, 476,  520, 476,  520, 476,  520, 476,  520, 1474,  518, 1476,  518, 1476,  516, 1496,  542};  // SAMSUNG_AC
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0xFE, 0x77, 0x00, 0x1F, 0xF0};

Timestamp : 000018.647
Encoding  : UNKNOWN
Code      : 7B551B62 (117 bits)
Library   : v2.6.0

Raw Timing[233]:
   +   652, - 17336,    +  3078, -  8910,    +   562, -   456,    +   546, -  1448, 
   +   550, -   446,    +   552, -   444,    +   552, -   444,    +   550, -   446, 
   +   550, -   446,    +   552, -   446,    +   550, -   446,    +   552, -  1440, 
   +   550, -   446,    +   550, -   446,    +   550, -  1470,    +   478, -   518, 
   +   502, -   492,    +   536, -  1458,    +   542, -  1450,    +   552, -  1440, 
   +   552, -  1442,    +   552, -  1442,    +   550, -   446,    +   550, -   446, 
   +   550, -   446,    +   552, -   444,    +   550, -   446,    +   550, -   446, 
   +   550, -   472,    +   524, -   472,    +   480, -   516,    +   510, -   488, 
   +   538, -   458,    +   542, -   452,    +   548, -   448,    +   550, -   446, 
[... cut ...] 52, 1638,  356, 470,  526, 1466,  510, 1484,  542, 1450,  550, 1442,  552, 1442,  552, 1440,  552, 1440,  552, 1440,  552, 444,  550, 444,  552, 444,  196, 356,  1466, 526,  1466, 510,  1484, 544,  452, 548,  448, 552,  444, 552,  444, 552,  442, 554,  442, 552,  444, 554,  442, 554,  444, 552,  1440, 552,  444, 552,  444, 552,  444, 552,  1466, 482,  514, 508,  488, 536,  460, 548,  448, 550,  446, 552,  444, 552,  444, 552,  1440, 552,  1440, 552,  1440, 552,  1440, 532 };  // UNKNOWN 3F824381

crankyoldgit, 2019-06-08:
For what ever reason, the two "Powerful Off" messages have poor quality data in them that stops them from being detected as valid Samsung AC messages.
I hacked the bad value to make it parse, and it results in:
```
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xF2, 0xFE, 0x71, 0x00, 0x11, 0xF0};
```

The "Powerful On" state is:
```
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 16C, Fan: 7 (TURBO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0xFE, 0x77, 0x00, 0x1F, 0xF0};
```
So it looks like "powerful" is controlled by `state[8]`, `state[10]`, & `state[12]`. As they are the only ones that differ.
`state[10]` is also used by Clean mode.
`state[12]` is used by Mode and Fan speed. Bits 0b01110000 for mode, & 0b00001110 for the Fan

The bits that change in `state[12]` are just the fan bits (seems likely)

So that leaves [8] & [9].

I'll code something up shortly.

crankyoldgit, 2019-06-08:
Question: Can the powerful and quiet settings be on at the same time? Or does one turn off the other?

crankyoldgit, 2019-06-08:
@sblantipodi FYI the `Issue734` branch has been updated to include "Powerful" support for Samsung.

Please test and let me know.

sblantipodi, 2019-06-08:
@crankyoldgit 
even If I write acir.setPower(true) before send it print
  Power: On, Mode: 1 (COOL), Temp: 25C, Fan: 2 (LOW), Swing: Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off

why? it does not work for now... am I wrong in something?

sblantipodi, 2019-06-08:
PS: tested ESP at 160MHz it works as before now, thanks!

crankyoldgit, 2019-06-08:
`setPower()` controls if the A/C is operating or not. `true` is A/C On, `false` is A/C Off.
`setPowerful()` controls if the A/C is in powerful (aka Turbo) mode. `true` is Turbo on, `false` is Turbo Off.

It depends what you are trying to do. Can you try explaining it a different way?

crankyoldgit, 2019-06-08:
I'll also ask this again:

Question: Can the powerful and quiet settings be on at the same time? Or does one turn off the other?

sblantipodi, 2019-06-08:
@crankyoldgit 
ok with setPowerful() it works like a charm.

there is no reason to turn both on, power and quet, I haven't got any automation to try this, I'm sorry.
can't wait to see this branch merged on the master branch... 
is publishing to the official arduino repo automatic when you push on the master branch?

how do you push src to the official repo?

thanks.

if you need some other try I'm here.
thanks for the help and the excellent work man

crankyoldgit, 2019-06-09:
> ok with setPowerful() it works like a charm.

Excellent.

> there is no reason to turn both on, power and quet, I haven't got any automation to try this, I'm sorry.

Yes, you do. The remote should show you. i.e. What happens when you hit the "quiet" button on the remote when the remote is in "powerful" mode? and vice versa.

Do they turn the other mode/setting off?

> can't wait to see this branch merged on the master branch...
is publishing to the official arduino repo automatic when you push on the master branch?

No. It only happens when we make a [release](https://github.com/markszabo/IRremoteESP8266/releases). I tend to make [releases](https://github.com/markszabo/IRremoteESP8266/releases) from the `master` branch every month or two.

However, you now know how to track the `master` branch in platformio. So you can keep up with the latest fixes and improvements if you want. The changes in PR #736 should get merged into `master` within 6 or so days, assuming no problems.

> if you need some other try I'm here.

I believe the only thing not working as expected is `setSwing()` and `getSwing()`, as I noted in #505 there are some odd bits that change that still need explaining. If you can try and work out what those bits do, that would be a great help.

sblantipodi, 2019-06-09:
Unfortunately I don't know how to work with that bits 
I can do some tests for you but really no idea on how to solve the swing problem 

At the moment I workarounded the problem by setting fan to high while moving swings and then back to the previous setting. In this way I can move the swing without any problem.

crankyoldgit, 2019-06-09:
Then capturing the `state[]` lines for what the remote 's message is for the _correct_ setting, when using the equivalent `setBlah()` steps don't work as expected, and reporting them here is a good place to start. i.e. Not using an work-a-rounds by setting the fan to high etc.

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

sblantipodi, 2019-06-09:
If I hit the quiet button on the remote when powerful is on it power off the powerful mode and starts the quiet one. And viceversa. 

The hard things of the wings is that it remembers the position of the wings for every mode, cool, hot, auto, quiet, powerful.

Can't wait to upgrade to 2.6.1 doing it as soon as I can :)

Good work man

crankyoldgit, 2019-06-09:
> If I hit the quiet button on the remote when powerful is on it power off the powerful mode and starts the quiet one. And viceversa.

I figured it would. I'll add code to emulate that shortly. It's not a vital bit of emulation at this point. ;-)

> The hard things of the wings is that it remembers the position of the wings for every mode, cool, hot, auto, quiet, powerful.

Ah. That's a problem. Not sure we can easily emulate that given the common use patterns.

> Can't wait to upgrade to 2.6.1 doing it as soon as I can :)

The Arduino Library Manager should pick up the new version with in 24-ish hours.

sblantipodi, 2019-06-09:
hi @crankyoldgit in the changelog of the platformio library manager I can see this
Correct set/getQuiet for Samsung A/C (#736)
but there is no mention for the setPowerful...

does 2.6.1 contains even the setPowerful fix?

crankyoldgit, 2019-06-09:
yes.

First line of [Features]:
`Add set/getPowerful for Samsung A/C (#736)`

crankyoldgit, 2019-06-10:
If everything is now working well enough for you, I'd like to close out this issue. If there is something broken or wrong with the newly changed code, we can create a new issue for it.

sblantipodi, 2019-06-10:
Sure thanks. It works well. Tested it in all the new features and it works well.

closed 2019-06-10 by crankyoldgit