crankyoldgit/IRremoteESP8266 #743: Missing examples/documentation for working with Gree AC

C0rn3j, 2019-06-09:
So my AC vendor - Gree - is supported by this library - https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Gree.h
https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Gree.cpp

But I can't find documentation on how to work with it properly. The best I got is rip the [Mitsubishi AC example](https://github.com/markszabo/IRremoteESP8266/tree/master/examples) and try/gaze from the source code as to how to work with it.

There's lots of resources in the FAQ/troubleshooting for when the AC is NOT supported, but not as much for when it is.

I got my AC working via modifying the Mitsubishi code slightly, but I'd appreciate a proper example. Maybe I'm just overlooking something?

```
IRGreeAC ac(SEND_PIN);
...
Serial.println("Setting desired state for A/C.");
ac.on();
ac.setFan(1);
ac.setMode(kGreeCool);
ac.setTemp(24);
```

crankyoldgit, 2019-06-09:
There are approx 20 Aircon protocols supported out of a total of 60+ IR protocols included in this library. We don't want to create and maintain example code for each and every protocol out there.
Example code is just that, an example. They are not instruction manuals or documentation.
A lot of the AirCon code is very similar between protocols. Hence as you discovered, only a few examples are actually needed. You being able to work out how to make a slightly different example/program do what you want is exactly how it is meant to be.

I am sure the current documentation isn't very newbie friendly and is lacking. However, there are plenty of examples that use almost every function of this library under either the [examples](https://github.com/markszabo/IRremoteESP8266/tree/master/examples) or [test](https://github.com/markszabo/IRremoteESP8266/tree/master/test) directories.

Users of this library vary from _"please give me a firmware so I can just load up"_, to _"I'm trying to do something really bizarre and odd-ball with your library that is totally out of spec"_, to people who's first post is them adding a complete IR A/C protocol including Unit Tests for it. It's hard to aim the documentation at a level that suits everyone. This problem is not a unique to this library.

If you feel there is missing documentation, feel free to contribute some. This is an open-source project after all. Pull Requests (PRs) and contributions are welcome. The more people who help and contribute, the better it can become.

C0rn3j, 2019-06-09:
I get not wanting to have XX examples of essentially the same thing.
Some usage description on top of the source code for the given device would be sufficient, I'll see about adding that for Gree sometime in the future.

I think having a 'Supported Protocols' entry on the top of the FAQ such as this would've saved me rest of the hassle:

```
We support XYZ...
We also support several Air Conditioners -  A B C, ....
Check out the [Example codes] to get you started, if your AC does not have a specific example, adapt an existing example for another AC.
```
The 'example code' is referred to throughout the entire FAQ but isn't really linked from it, so that'd be solved too.

What do you think? I can't really add this myself other than making this boilerplate, as I am unfamiliar with the project.

crankyoldgit, 2019-06-10:
Links to example code I can do. I'll look at something that can automate creating a list of protocols that are supported. A manually maintained list will quickly get out of date.

crankyoldgit, 2019-06-10:
Links added in the wiki pages to example code and circuits. Please take a look and point out anywhere you think I've missed.

closed 2019-06-15 by crankyoldgit

C0rn3j, 2019-06-15:
Thanks guys, this is perfect!

commit 5d8147ce (2019-06-14): Generate supported brand/model info from src.

FYI @sillyfrog

Fixes #743
commit c0581d16 (2019-06-14): Generate a list of supported protocols.
* Tool to build the document.
* List send only, send & decode, and supported detailed A/C protocols.
* Add reference to new doc.

Fixes #743
PR #755 (2019-06-15): Scrape Supported Protocols and generate SupportedProtocols.md
* Tool to build the document.
* Add reference to new doc.

* Other
  - Move existing python unit tests to Python3
  - Use Python3's pylint
  - Fix pylint issues found after move.
  - Convince CI/Travis to install/use convince Python3 stuff.

Fixes #743
commit 711901c5 (2019-06-15): Scrape Supported Protocols and generate SupportedProtocols.md (#755)

* Tool to build the document.
* Add reference to new doc.
* Other
  - Move existing python unit tests to Python3
  - Use Python3's pylint
  - Fix pylint issues found after move.
  - Convince CI/Travis to install/use convince Python3 stuff.

Fixes #743