crankyoldgit/IRremoteESP8266 #744: New Lib

buzzik, 2019-06-09:
I have a conditioner that not recognizable by dumb. It says unknown. 
Can we somhow get a lib for it? I can get a dump of commands and live it here

crankyoldgit, 2019-06-10:
FYI, there is a [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#the-libraryexample-code-doesnt-understand-my-air-conditioner-remote-) for this type of question.

Can you please supply the Make (brand) and Model number of the remote, and the unit that is controlled by it? Are the captures you supplied from the actual remote or the app?

I'll look to see what can be done to add bare-minimum support for it when we have more data.

buzzik, 2019-06-10:
Here is a codes that I captured
https://docs.google.com/spreadsheets/d/15L6SpD9bAHEyOcly9gkm9eEAbjaEanxQhgI9DY4sMr4/edit?usp=sharing

Some of them recognizing like TCL112AC but some like UNKNOWN

LBS-TOR07
remote - LEBERG (nothing on other side)

buzzik, 2019-06-10:
P.S.
I've tried TCL112AC  lib and its works! Even those command thar recognizes as UNKNOWN from remote. I think we can close issuse. Thank you so much for help. You are Cool!

closed 2019-06-14 by buzzik

crankyoldgit, 2019-06-16:
FYI, the mentioned code changes have been included in the new v2.6.2 release.

buzzik, 2019-06-17:
Thank you ! Will check

commit 1a6cdb28 (2019-06-10): Increase tolerance value for TCL112AC protocol.
Increase the tolerance by 5% to help match poor TCL112AC signals.

Fixes #744
PR #745 (2019-06-16): Increase tolerance value for TCL112AC protocol.
Increase the tolerance by 5% to help match poor TCL112AC signals.

Fixes #744
commit 283e8b21 (2019-06-16): Increase tolerance value for TCL112AC protocol. (#745)

Increase the tolerance by 5% to help match poor TCL112AC signals.

Fixes #744