crankyoldgit/IRremoteESP8266 #750: Thank you for sharing!!

1043717432, 2019-06-12:
Thank you for sharing!!

How to add temperature sensor in air conditioning

crankyoldgit, 2019-06-12:
This is a problem you'll need to work out with [Home Assistant](https://www.home-assistant.io/), not IRremoteESP8266.

closed 2019-06-12 by crankyoldgit

1043717432, 2019-06-12:
Thank you for your reply. How to configure the temperature sensor in Home Assistant and in the air conditioner?

1043717432, 2019-06-12:
> 这是一个问题，您需要使用[Home Assistant](https://www.home-assistant.io/)，而不是IRremoteESP8266。

Thank you for your reply. How to configure the temperature sensor in Home Assistant and in the air conditioner?

crankyoldgit, 2019-06-12:
Again, you need to ask for help on the Home Assistant site. **Not here**. You want & need to make changes in the Home Assistant configuration. We are not equipped to help you here with Home Assistant configuration issues.

1043717432, 2019-06-12:
Thank you, clear!