crankyoldgit/IRremoteESP8266 #762: Cant receive data while using TFT_eSPI libary

Shockwave2k, 2019-06-19:
### Version/revision of the library used
2.6.2

### Expected behavior
IR-Receiver should receive IR-Control Input

### Actual behavior
1. load IRremoteESP8266.
2. load TFT_eSPI.
3. Sending data via IR-Controller
4. No Data received

### Steps to reproduce the behavior
As soon as i load the TFT_eSPI lib it seem to collide and stops working

#### Example code used
The IRrecvDemo just works fine but my own code seems to break as soon as i load the TFT panel libary.
https://gist.github.com/Shockwave2k/03c5f54ab573a87ab5ab8f7443c738c4

#### Circuit diagram and hardware used (if applicable)
[image]
- NodeMCU v3 (esp8266)
- [ST7789](https://de.aliexpress.com/item/1-3-zoll-IPS-HD-TFT-ST7789-Stick-IC-240-240-SPI-Kommunikation-3-3-V/32880846744.html?spm=a2g0s.9042311.0.0.37ea4c4dPrd9Mc)
- [AM2302 DHT22](https://de.aliexpress.com/item/High-Precision-AM2302-DHT22-Digital-Temperature-Humidity-Sensor-Module-For-arduino-Uno-R3/32759158558.html?spm=a2g0s.9042311.0.0.27424c4dL0UgNK)

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
Yes without the TFTPanel libary

### Other useful information
As already said it seems to collide with the TFT Libary which you can find [here](https://github.com/Bodmer/TFT_eSPI?utm_source=platformio&utm_medium=piohome).

crankyoldgit, 2019-06-19:
My immediate guess is they are both using the (same) timer/interrupt.
I suggest you check if the TFT Library uses a timer interrupt. If it does, that's the likely culprit.

crankyoldgit, 2019-06-19:
Alternatively, try using D0/GPIO 16 for the IR receiver data pin.
Other than that, I can't see any obvious mistakes in your example code.

crankyoldgit, 2019-06-19:
Or remove the DHT and use D2 (instead of D6) and see how that goes.

Shockwave2k, 2019-06-19:
No luck on switching to D0/D2. Moving the DHT to D6 also works fine. So i guess its your mentioned timer/interrupt problem? Could use some help/hints on that as i'm pretty new to the c++ language.

crankyoldgit, 2019-06-19:
I had a quick look over the TFT_eSPI & SPI libraries. I couldn't see anything obvious standing out.
So I'm mostly out of ideas other than my first thought.

Have you tried commenting out everything from your example code you pointed to leaving just the IR stuff, and see if the IR part still works? If it doesn't then, well, you've got some basic problem which I can probably help with. If it does, then slowly add the DHT and/or TFT parts in to see which thing breaks it.

Shockwave2k, 2019-06-19:
Yes thats what i tried already.
https://gist.github.com/Shockwave2k/876280ec041b09559d449874309c2041

This gist just works fine. It includes the DHT sensor but not the Display.
If you have any tips how i could maybe debug it in Visual Studio Code / IntelliJ / Arduino IDE i would give it a try.

Of course any other tip is also welcome :smile:

crankyoldgit, 2019-06-20:
I suggest you take it up/report an issue with the TFT_eSPI library. They might know if it uses some timer or plays with or disables the interrupts on other pins etc.

Here are the major references in our code to the ESP8266 timer I can think of:

https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L47
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L84
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L117
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L213
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.cpp#L62

I could probably come up with some HORRIBLE kludge to make our stuff work without a timer, but it would require your code to essentially pretend to operate like a timer/interrupt etc.
e.g. every time you go through your main loop, check if it has started capturing something, then wait for N milliseconds (or do something) to see if more has been added, if not, tell it finish/decode, otherwise keep waiting until nothing gets added. It'd be completely unsupported & experimental etc, but might get you out of your bind.

crankyoldgit, 2019-06-20:
Dumb thing to try: If you remove the TFT display physically from the circuit, but leave the code in, Does it still play up? Just wondering if it is drawing too much current from the 3.3v onboard voltage regulator.
It's unlikely .. but worth a shot.
e.g. the IR demodulator might not be getting enough juice to work with everything hooked up.

Shockwave2k, 2019-06-20:
Tried the TFT removal but sadly it not about the power.
I will give it a shot in the TFT_eSPI libary. Maybe the have an idea.
Will keep you updated if i get a response.

crankyoldgit, 2019-06-20:
Good luck. Fingers crossed.

Shockwave2k, 2019-06-24:
I dont know what i have changed... but somehow it now works...
will close it for now and post an update if i find out what caused the issue.

closed 2019-06-24 by Shockwave2k