crankyoldgit/IRremoteESP8266 #764: Include AC Infrared to MQTT adapter ZH/JT-03

AndreyShpilevoy, 2019-06-19:
Hi, I found, that is some potential possibility to include already prepared code to this repo ([like here - Mitsubishi Heavy Industry Air Conditioner Code from ToniA add please](https://github.com/markszabo/IRremoteESP8266/issues/660)).

I also found a lib with my AC - [AC Infrared to MQTT adapter ZH/JT-03](https://github.com/arthurkrupa/ac-ir-mqtt-zhjt03)
Can it be included as well?

This lib work with Remote controls that compatible with such AC: 

- Chigo
- Komeco
- Lloyd
- Blaupunkt
- EPiK
- Neoclima

Thank you in advance!

AndreyShpilevoy, 2019-06-19:
Here are some examples of Raw signals from `IRrecvDumpV2`

POWER ON
MODE COOL
Temperature: 26
Air Speed slow
Swing vertical: off
Swing horizontal: on

`Timestamp : 002496.734
Encoding  : UNKNOWN
Code      : EE182D95 (99 bits)
Library   : v2.6.2

Raw Timing[197]:
   +  6112, -  7392,    +   540, -   602,    +   516, -   578,    +   522, -   604, 
   +   540, -   554,    +   540, -   554,    +   540, -   576,    +   518, -   576, 
   +   516, -   554,    +   540, -   608,    +   542, -   554,    +   540, -   554, 
   +   540, -   576,    +   518, -   604,    +   516, -   556,    +   540, -   576, 
   +   546, -   580,    +   542, -   578,    +   542, -   602,    +   518, -   554, 
   +   542, -   554,    +   568, -   582,    +   540, -   554,    +   540, -   582, 
   +   540, -   578,    +   518, -   582,    +   542, -   576,    +   544, -   530, 
   +   566, -   534,    +   562, -   534,    +   562, -   552,    +   542, -   582, 
   +   540, -   604,    +   518, -   608,    +   542, -   554,    +   540, -   582, 
   +   540, -   604,    +   518, -   580,    +   540, -   606,    +   544, -   554, 
   +   542, -   554,    +   542, -   580,    +   542, -   576,    +   520, -   554, 
   +   540, -   578,    +   518, -   578,    +   518, -   582,    +   544, -   552, 
   +   570, -   580,    +   544, -   580,    +   542, -   554,    +   542, -   604, 
   +   520, -   576,    +   520, -   580,    +   540, -   556,    +   540, -   556, 
   +   542, -   584,    +   566, -   580,    +   542, -  1622,    +   542, -   554, 
   +   542, -  1620,    +   544, -   604,    +   520, -  1642,    +   518, -  1674, 
   +   548, -   560,    +   564, -   580,    +   544, -   554,    +   544, -   552, 
   +   544, -   554,    +   542, -   556,    +   542, -   576,    +   522, -   554, 
   +   542, -   556,    +   542, -   580,    +   542, -  1670,    +   520, -   578, 
   +   520, -  1622,    +   542, -   580,    +   518, -  1646,    +   520, -   558, 
   +   568, -   552,    +   546, -  1628,    +   566, -   580,    +   544, -  1668, 
   +   522, -   576,    +   520, -   578,    +   520, -  1670,    +   522, -   576, 
   +   522, -  1670,    +   496, -  1676,    +   570, -   560,    +   566, -   532, 
   +   564, -  1648,    +   544, -  1670,    +   522, -  1650,    +   544, -   552, 
   +   544, -   576,    +   520, -  7390,    +   544

uint16_t rawData[197] = {6112, 7392,  540, 602,  516, 578,  522, 604,  540, 554,  540, 554,  540, 576,  518, 576,  516, 554,  540, 608,  542, 554,  540, 554,  540, 576,  518, 604,  516, 556,  540, 576,  546, 580,  542, 578,  542, 602,  518, 554,  542, 554,  568, 582,  540, 554,  540, 582,  540, 578,  518, 582,  542, 576,  544, 530,  566, 534,  562, 534,  562, 552,  542, 582,  540, 604,  518, 608,  542, 554,  540, 582,  540, 604,  518, 580,  540, 606,  544, 554,  542, 554,  542, 580,  542, 576,  520, 554,  540, 578,  518, 578,  518, 582,  544, 552,  570, 580,  544, 580,  542, 554,  542, 604,  520, 576,  520, 580,  540, 556,  540, 556,  542, 584,  566, 580,  542, 1622,  542, 554,  542, 1620,  544, 604,  520, 1642,  518, 1674,  548, 560,  564, 580,  544, 554,  544, 552,  544, 554,  542, 556,  542, 576,  522, 554,  542, 556,  542, 580,  542, 1670,  520, 578,  520, 1622,  542, 580,  518, 1646,  520, 558,  568, 552,  546, 1628,  566, 580,  544, 1668,  522, 576,  520, 578,  520, 1670,  522, 576,  522, 1670,  496, 1676,  570, 560,  566, 532,  564, 1648,  544, 1670,  522, 1650,  544, 552,  544, 576,  520, 7390,  544};  // UNKNOWN EE182D95`

MODE COOL
Temperature: 26
Air Speed high
Swing vertical: off
Swing horizontal: off

`Timestamp : 002701.579
Encoding  : UNKNOWN
Code      : 3E16682B (99 bits)
Library   : v2.6.2

Raw Timing[197]:
   +  6148, -  7352,    +   582, -   540,    +   576, -   516,    +   496, -   632, 
   +   598, -   516,    +   578, -   514,    +   582, -   512,    +   580, -   540, 
   +   496, -   602,    +   542, -   602,    +   580, -   514,    +   582, -   538, 
   +   554, -   542,    +   520, -   576,    +   516, -   582,    +   600, -   542, 
   +   580, -   514,    +   582, -   538,    +   524, -   570,    +   496, -   602, 
   +   570, -   528,    +   600, -   544,    +   580, -   514,    +   582, -   514, 
   +   580, -   514,    +   552, -   570,    +   518, -   580,    +   572, -   526, 
   +   600, -   542,    +   582, -   512,    +   582, -   514,    +   580, -   540, 
   +   496, -   630,    +   574, -   572,    +   580, -   514,    +   580, -   514, 
   +   582, -   514,    +   582, -   540,    +   498, -   630,    +   574, -   572, 
   +   580, -   540,    +   582, -  1606,    +   522, -   572,    +   496, -  1670, 
   +   542, -   556,    +   596, -   502,    +   598, -   542,    +   584, -   540, 
   +   580, -   540,    +   526, -   602,    +   574, -   552,    +   598, -   542, 
   +   582, -   540,    +   580, -   516,    +   548, -   546,    +   524, -   572, 
   +   508, -   620,    +   570, -   576,    +   580, -  1608,    +   584, -   514, 
[... 68 lines cut ...]
   +   542, -   558,    +   566, -   578,    +   540, -   552,    +   542, -   552, 
   +   544, -   552,    +   540, -   554,    +   540, -   554,    +   544, -   554, 
   +   568, -   552,    +   542, -   580,    +   542, -   574,    +   522, -   552, 
   +   544, -   576,    +   542, -   576,    +   546, -   556,    +   566, -   552, 
   +   542, -   552,    +   544, -   600,    +   520, -   552,    +   540, -   580, 
[... cut ...] 4, 574,  522, 602,  518, 554,  532, 1654,  522, 554,  542, 578,  546, 558,  562, 554,  544, 552,  544, 602,  520, 554,  544, 602,  520, 554,  542, 584,  568, 552,  546, 550,  546, 530,  566, 1648,  542, 1644,  546, 552,  544, 552,  542, 602,  518, 1624,  568, 550,  544, 532,  568, 1646,  544, 554,  542, 1642,  522, 552,  546, 552,  544, 1646,  542, 578,  518, 1652,  568, 560,  564, 602,  520, 1642,  522, 1646,  542, 554,  542, 556,  542, 1644,  522, 1652,  568, 7378,  548};  // UNKNOWN 7BB3E22A`

AndreyShpilevoy, 2019-06-19:
Here are some examples of Raw signals from `IRrecvDumpV2`

POWER ON
MODE COOL
Temperature: 26
Air Speed slow
Swing vertical: off
Swing horizontal: on

```
Timestamp : 002496.734
Encoding  : UNKNOWN
Code      : EE182D95 (99 bits)
Library   : v2.6.2

Raw Timing[197]:
   +  6112, -  7392,    +   540, -   602,    +   516, -   578,    +   522, -   604, 
   +   540, -   554,    +   540, -   554,    +   540, -   576,    +   518, -   576, 
   +   516, -   554,    +   540, -   608,    +   542, -   554,    +   540, -   554, 
   +   540, -   576,    +   518, -   604,    +   516, -   556,    +   540, -   576, 
   +   546, -   580,    +   542, -   578,    +   542, -   602,    +   518, -   554, 
   +   542, -   554,    +   568, -   582,    +   540, -   554,    +   540, -   582, 
   +   540, -   578,    +   518, -   582,    +   542, -   576,    +   544, -   530, 
   +   566, -   534,    +   562, -   534,    +   562, -   552,    +   542, -   582, 
   +   540, -   604,    +   518, -   608,    +   542, -   554,    +   540, -   582, 
   +   540, -   604,    +   518, -   580,    +   540, -   606,    +   544, -   554, 
   +   542, -   554,    +   542, -   580,    +   542, -   576,    +   520, -   554, 
   +   540, -   578,    +   518, -   578,    +   518, -   582,    +   544, -   552, 
   +   570, -   580,    +   544, -   580,    +   542, -   554,    +   542, -   604, 
   +   520, -   576,    +   520, -   580,    +   540, -   556,    +   540, -   556, 
   +   542, -   584,    +   566, -   580,    +   542, -  1622,    +   542, -   554, 
   +   542, -  1620,    +   544, -   604,    +   520, -  1642,    +   518, -  1674, 
   +   548, -   560,    +   564, -   580,    +   544, -   554,    +   544, -   552, 
   +   544, -   554,    +   542, -   556,    +   542, -   576,    +   522, -   554, 
   +   542, -   556,    +   542, -   580,    +   542, -  1670,    +   520, -   578, 
   +   520, -  1622,    +   542, -   580,    +   518, -  1646,    +   520, -   558, 
   +   568, -   552,    +   546, -  1628,    +   566, -   580,    +   544, -  1668, 
   +   522, -   576,    +   520, -   578,    +   520, -  1670,    +   522, -   576, 
   +   522, -  1670,    +   496, -  1676,    +   570, -   560,    +   566, -   532, 
   +   564, -  1648,    +   544, -  1670,    +   522, -  1650,    +   544, -   552, 
   +   544, -   576,    +   520, -  7390,    +   544

uint16_t rawData[197] = {6112, 7392,  540, 602,  516, 578,  522, 604,  540, 554,  540, 554,  540, 576,  518, 576,  516, 554,  540, 608,  542, 554,  540, 554,  540, 576,  518, 604,  516, 556,  540, 576,  546, 580,  542, 578,  542, 602,  518, 554,  542, 554,  568, 582,  540, 554,  540, 582,  540, 578,  518, 582,  542, 576,  544, 530,  566, 534,  562, 534,  562, 552,  542, 582,  540, 604,  518, 608,  542, 554,  540, 582,  540, 604,  518, 580,  540, 606,  544, 554,  542, 554,  542, 580,  542, 576,  520, 554,  540, 578,  518, 578,  518, 582,  544, 552,  570, 580,  544, 580,  542, 554,  542, 604,  520, 576,  520, 580,  540, 556,  540, 556,  542, 584,  566, 580,  542, 1622,  542, 554,  542, 1620,  544, 604,  520, 1642,  518, 1674,  548, 560,  564, 580,  544, 554,  544, 552,  544, 554,  542, 556,  542, 576,  522, 554,  542, 556,  542, 580,  542, 1670,  520, 578,  520, 1622,  542, 580,  518, 1646,  520, 558,  568, 552,  546, 1628,  566, 580,  544, 1668,  522, 576,  520, 578,  520, 1670,  522, 576,  522, 1670,  496, 1676,  570, 560,  566, 532,  564, 1648,  544, 1670,  522, 1650,  544, 552,  544, 576,  520, 7390,  544};  // UNKNOWN EE182D95
```

MODE COOL
Temperature: 26
Air Speed high
Swing vertical: off
Swing horizontal: off

```
Timestamp : 002701.579
Encoding  : UNKNOWN
Code      : 3E16682B (99 bits)
Library   : v2.6.2

Raw Timing[197]:
   +  6148, -  7352,    +   582, -   540,    +   576, -   516,    +   496, -   632, 
   +   598, -   516,    +   578, -   514,    +   582, -   512,    +   580, -   540, 
   +   496, -   602,    +   542, -   602,    +   580, -   514,    +   582, -   538, 
   +   554, -   542,    +   520, -   576,    +   516, -   582,    +   600, -   542, 
   +   580, -   514,    +   582, -   538,    +   524, -   570,    +   496, -   602, 
   +   570, -   528,    +   600, -   544,    +   580, -   514,    +   582, -   514, 
   +   580, -   514,    +   552, -   570,    +   518, -   580,    +   572, -   526, 
   +   600, -   542,    +   582, -   512,    +   582, -   514,    +   580, -   540, 
   +   496, -   630,    +   574, -   572,    +   580, -   514,    +   580, -   514, 
   +   582, -   514,    +   582, -   540,    +   498, -   630,    +   574, -   572, 
   +   580, -   540,    +   582, -  1606,    +   522, -   572,    +   496, -  1670, 
   +   542, -   556,    +   596, -   502,    +   598, -   542,    +   584, -   540, 
[... 76 lines cut ...]
   +   544, -   552,    +   540, -   554,    +   540, -   554,    +   544, -   554, 
   +   568, -   552,    +   542, -   580,    +   542, -   574,    +   522, -   552, 
   +   544, -   576,    +   542, -   576,    +   546, -   556,    +   566, -   552, 
   +   542, -   552,    +   544, -   600,    +   520, -   552,    +   540, -   580, 
   +   570, -   530,    +   568, -   552,    +   542, -   554,    +   542, -   554, 
   +   542, -   602,    +   522, -   578,    +   542, -   576,    +   518, -   584, 
   +   568, -   578,    +   542, -   552,    +   544, -   574,    +   520, -   576, 
   +   520, -   580,    +   540, -   584,    +   566, -   528,    +   566, -   534, 
[... cut ...] 574,  522, 602,  518, 554,  532, 1654,  522, 554,  542, 578,  546, 558,  562, 554,  544, 552,  544, 602,  520, 554,  544, 602,  520, 554,  542, 584,  568, 552,  546, 550,  546, 530,  566, 1648,  542, 1644,  546, 552,  544, 552,  542, 602,  518, 1624,  568, 550,  544, 532,  568, 1646,  544, 554,  542, 1642,  522, 552,  546, 552,  544, 1646,  542, 578,  518, 1652,  568, 560,  564, 602,  520, 1642,  522, 1646,  542, 554,  542, 556,  542, 1644,  522, 1652,  568, 7378,  548};  // UNKNOWN 7BB3E22A
```

crankyoldgit, 2019-06-20:
What is the Make and Model numbers of your Remote and A/C Units involved?

AndreyShpilevoy, 2019-06-20:
AC: Neoclima NS-09AHTI
Remote: ZH/TY-01

I just realized, that my remote and remote by my link are different...
ZH/TY-01 vs ZH/JT-01
But can it be, that they use the same protocol?

crankyoldgit, 2019-06-20:
Thanks for the info. I'll try to get it soon.

Why don't you try their project and see if it supports your device first? i.e. can send/decode your remote etc.

AndreyShpilevoy, 2019-06-20:
I found this library in the middle of the night and had no time to check it. 
Most probably I will check it today and will share results of it.

AndreyShpilevoy, 2019-06-20:
Unforchenaelly I can't check this lib by myself for now.
I don't know why, but this lib call `wdt reset` in runtime right after connecting to MQTT server.

AndreyShpilevoy, 2019-06-21:
@crankyoldgit IT's alive! ALIVE!!!
I just checked the version from "neoclima" branch.
I have send `irsend.sendNeoclima(state)` where took `state` from one of your tests.
As far as I can understand - it was code from my first code example (according to Raw).

My AC immediately tuned on with such params:
```
POWER ON
MODE COOL
Temperature: 26
Air Speed slow
Swing vertical: off
Swing horizontal: on
```

crankyoldgit, 2019-06-22:
> My AC immediately tuned on with such params:

Excellent! I'll marking it as working then.

Yes, it was from the first example you reported.

I had a look at the code you referenced. And frankly it's very hard to understand.

Can you report what the `uint8_t state[12] = {...};` lines are for everything but power off, fan and mode of auto, but vary the temp starting at the lowest, and add one degree/increment till it is at it's highest temp.

That will help make sure we've got the bit ordering right, and work out what bits control the temp setting etc.

AndreyShpilevoy, 2019-06-22:
Yep, here it is: [Link](https://drive.google.com/file/d/1DKMiIuQPQIYjoViTSLoYyMaAD36iXszr/view?usp=sharing)
```
MODE COOL
Temperature: <from 16 to 32 and vice versa>
Air Speed slow
Swing vertical: off
Swing horizontal: on

crankyoldgit, 2019-06-23:
Thanks. That helps a lot. That has shown we got the bit ordering wrong.

I'll update the PR soon, with the changes etc. Note: All the `state[]`s for it will be changed because of it. So, you can discard anything collect thus far.

AndreyShpilevoy, 2019-06-23:
Ok, thank you!
Do you need any other information?

crankyoldgit, 2019-06-23:
Yep, you can re-collect the same data as before, but with the updated code in that branch.

As far as I can easily tell, `state[9]` contains the temperature. `state[5]` contains a code for the button pressed. e.g. I'd guess based on your spreadsheet data that the different values are the "temp up" and "temp down" buttons. It seems to correlate well with your data.

and finally `state[11]` is the checksum. Once we have a pool of the corrected data, we can try to figure out what sort of calculation it is.

AndreyShpilevoy, 2019-06-23:
Ok, here it is: [Link](https://drive.google.com/file/d/1yzRtMukUCokKtxDFRosbtSdMWaWs5qV3/view?usp=sharing)

crankyoldgit, 2019-06-23:
Thanks. That helped work out the checksum, which I've now added to the code.
With the latest commit I just added to the branch, it should show the Temperature when it detect/decodes it too.

Now, you have the fun part. Working out what all the bits & bytes mean for all the other settings.
e.g.
`state[5]`: 0x02 means "Temp Up" button was pressed , 0x03 means "Temp Down" button was pressed.

Once you've nailed down what bits do what, I'll help code it up and add it to the library.

AndreyShpilevoy, 2019-06-23:
Oh, here a [Link](https://drive.google.com/file/d/1kjYk4zS9NQcMQhFkak-L4mp4UuaAIesW/view?usp=sharing) to file with all functions (except Timer).

BTW. I not sure should be something changed with IRrecvDumpV2, but for now, my output looks like this:
```
Timestamp : 005275.045
Encoding  : NEOCLIMA
Code      : 000000000002006A002AA53B (96 bits)
Library   : v2.6.2

Raw Timing[197]:
   +  6150, -  7354,    +   582, -   566,    +   494, -   626,    +   542, -   552, 
   +   578, -   516,    +   578, -   542,    +   552, -   542,    +   548, -   544, 
   +   494, -   600,    +   488, -   632,    +   554, -   542,    +   578, -   568, 
   +   552, -   542,    +   520, -   574,    +   466, -   628,    +   540, -   554, 
   +   578, -   570,    +   552, -   568,    +   522, -   600,    +   514, -   580, 
   +   572, -   522,    +   552, -   542,    +   578, -   544,    +   552, -   544, 
   +   552, -   542,    +   552, -   570,    +   466, -   630,    +   514, -   580, 
   +   576, -   518,    +   578, -   518,    +   578, -   572,    +   552, -   542, 
   +   550, -   546,    +   494, -   628,    +   540, -   554,    +   576, -   518, 
   +   576, -   518,    +   578, -   570,    +   554, -   542,    +   548, -   574, 
   +   466, -   630,    +   514, -   608,    +   576, -  1610,    +   582, -   542, 
   +   556, -   540,    +   524, -   572,    +   492, -   604,    +   490, -   632, 
   +   576, -   518,    +   576, -   572,    +   554, -   542,    +   554, -   568, 
   +   492, -   608,    +   568, -   550,    +   548, -   546,    +   578, -   518, 
   +   580, -   544,    +   554, -   546,    +   576, -  1610,    +   492, -   608, 
   +   536, -  1654,    +   598, -   572,    +   554, -  1608,    +   550, -  1612, 
   +   540, -   610,    +   576, -   550,    +   576, -   516,    +   580, -   568, 
   +   524, -   572,    +   492, -   606,    +   538, -   556,    +   566, -   554, 
   +   576, -   520,    +   576, -   572,    +   554, -  1608,    +   580, -   544, 
   +   550, -  1584,    +   522, -   580,    +   488, -  1702,    +   598, -   518, 
   +   578, -   516,    +   580, -  1610,    +   578, -   572,    +   466, -  1702, 
   +   596, -   520,    +   578, -   518,    +   580, -  1582,    +   582, -   568, 
   +   552, -  1610,    +   494, -  1704,    +   596, -  1586,    +   580, -   544, 
   +   552, -  1610,    +   580, -  1608,    +   496, -  1700,    +   596, -   522, 
   +   576, -   518,    +   578, -  7342,    +   552

uint16_t rawData[197] = {6150, 7354,  582, 566,  494, 626,  542, 552,  578, 516,  578, 542,  552, 542,  548, 544,  494, 600,  488, 632,  554, 542,  578, 568,  552, 542,  520, 574,  466, 628,  540, 554,  578, 570,  552, 568,  522, 600,  514, 580,  572, 522,  552, 542,  578, 544,  552, 544,  552, 542,  552, 570,  466, 630,  514, 580,  576, 518,  578, 518,  578, 572,  552, 542,  550, 546,  494, 628,  540, 554,  576, 518,  576, 518,  578, 570,  554, 542,  548, 574,  466, 630,  514, 608,  576, 1610,  582, 542,  556, 540,  524, 572,  492, 604,  490, 632,  576, 518,  576, 572,  554, 542,  554, 568,  492, 608,  568, 550,  548, 546,  578, 518,  580, 544,  554, 546,  576, 1610,  492, 608,  536, 1654,  598, 572,  554, 1608,  550, 1612,  540, 610,  576, 550,  576, 516,  580, 568,  524, 572,  492, 606,  538, 556,  566, 554,  576, 520,  576, 572,  554, 1608,  580, 544,  550, 1584,  522, 580,  488, 1702,  598, 518,  578, 516,  580, 1610,  578, 572,  466, 1702,  596, 520,  578, 518,  580, 1582,  582, 568,  552, 1610,  494, 1704,  596, 1586,  580, 544,  552, 1610,  580, 1608,  496, 1700,  596, 522,  576, 518,  578, 7342,  552};  // NEOCLIMA
uint8_t state[12] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00, 0x6A, 0x00, 0x2A, 0xA5, 0x3B};
```

crankyoldgit, 2019-06-23:
Yes, the output should have subtly changed **if** you updated to the latest in that branch and are using the new version or `IRrecvDumpV2` in that branch. e.g. check if yours has these lines in it:

https://github.com/markszabo/IRremoteESP8266/blob/neoclima/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L187
```
#if DECODE_NEOCLIMA
  if (results->decode_type == NEOCLIMA) {
    IRNeoclimaAc ac(0);
    ac.setRaw(results->state);
    description = ac.toString();
  }
#endif  // DECODE_NEOCLIMA
```

It should start outputting details of the settings as they get added.

e.g.
`Temp: 26C` etc.

I'll take a look at what you linked shortly.

crankyoldgit, 2019-06-23:
Earlier on you said:

> POWER ON
MODE COOL
Temperature: 26
Air Speed slow
Swing vertical: off
Swing horizontal: on

I can only see one `swing` setting in your spreadsheet. Is the `swing` listed there the vertical or horizontal? and where is the setting for the other one?

Note: I've added another commit to that branch that covers power, mode, and fan speed.

AndreyShpilevoy, 2019-06-23:
Yep, updated lib, but forgot to update `IRrecvDumpV2`.
Now everything is OK.

```
IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Timestamp : 000245.806
Encoding  : NEOCLIMA
Code      : 000000000003006A0028A53A (96 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 24C, Fan: 1 (Low)
Library   : v2.6.2
```

`Swing vertical` == `Swing` (on remote)
`Swing horizontal` == `Air Flow` (on remote)

I missed only the `Timer` button.
Timer button a bit tricky to check.

crankyoldgit, 2019-06-24:
Major update to the branch. So please update and test when you can. It should do pretty much everything now.

Note: Fresh isn't added/supported as I think you need to experiment/work out why it has odd `state[5]` behaviour. To be honest, so does Turbo to a point. I think part of `state[5]` may include the control bits for Turbo and Fresh.

So I suggest you try other buttons and modes with Turbo & Fresh on/off etc to see if you can isolate what's going on there.

Other than that, I think it's pretty much done.
You don't need to add Timer support etc if you don't want/need it. It's typically easier to control via some other Home Automation system anyway.

AndreyShpilevoy, 2019-06-24:
I tested it and found some minor issues: [Link](https://drive.google.com/file/d/1tA09Gu_ZqDcHucscnqzv0V3cIUWOE0d1/view?usp=sharing)

1. Fan mode is shown wrongly. Looks like copypaste error. For `Auto` and `Hight` shown `Auto`.
For `Middle` and `Low` - `Low`.
2. Turbo buttons - shown as UNKNOWN 
3. Follow me - always off

After it will be fixed - I think it's ready to be merged to master branch.

crankyoldgit, 2019-06-25:
Re: 3. Follow.

When you say Follow is on `state[8]` has had two different values in the data in your various spreadsheets. i.e. `0x5D` & `0x5F`. You will need to see if there are other cases and what might be causing changes in that byte. Until you've worked out what bits need to change and predictably we should probably disable it.

Re: 2. Turbo.
Ahhh. You incorrectly [reported](https://drive.google.com/file/d/1kjYk4zS9NQcMQhFkak-L4mp4UuaAIesW/view) the value of `state[5]` for Turbo. I see your mistake now, and I'll fix my mistake based on it.

Re: 1. Fan.
Yep, I think I found what I did wrong. Fixed (I hope)

Re: Fresh.
I think I've figured out Fresh. Added.

AndreyShpilevoy, 2019-06-25:
I checked it again - `IRrecvDumpV2` work like a charm.
Thank you!

crankyoldgit, 2019-06-26:
But what about the odd behaviour of **Follow**?

AndreyShpilevoy, 2019-06-26:
I checked it with different Mod, fun, turbo and some other settings - seems like everything was recognized correctly.

crankyoldgit, 2019-06-26:
Okay, if you say so, but some how you generated it and recorded it on the spreadsheet.

AndreyShpilevoy, 2019-06-26:
Can it be a defect because of not clear signal transmission from remote to IR module?

crankyoldgit, 2019-06-26:
VERY unlikely. The last byte of the `state[]` is a checksum, and that is checked when it decodes the message. It won't say it's a NEOCLIMA message with it being wrong.

AndreyShpilevoy, 2019-06-26:
I will check it again and will produce results in Excel for Follow function, to be sure.
Can it be helpful for us?

AndreyShpilevoy, 2019-06-26:
If I will make one more spreadsheet for Follow function - can it help us to be sure, that everything is ok?

crankyoldgit, 2019-06-26:
According to your spreadsheet when Follow was **ON** it recorded the state as:
`uint8_t state[12] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x13, 0x00, 0x6A, 0x5F, 0x29, 0xA5, 0xAA};`

which currently decodes to:
`Power: On, Mode: 1 (COOL), Temp: 25C, Fan: 3 (Low), Swing(V): Off, Swing(H): On, Sleep: Off, Turbo: Off, Hold: Off, Ion: Off, Eye: Off, Light: Off, Follow: Off, 8C Heat: Off, Fresh: Off, Button: 19 (Follow)`

Note: Follow is detected as **OFF**.

We currently detect Follow as **ON** when `state[8]` is `0x5D` and **OFF** when it is `0x00`. Here in this case it was `0x5F` which really indicates it was probably on.

My guess is you had some other feature we haven't decoded thus far on at the same time or `Follow` is not just a simple on/off switch.

crankyoldgit, 2019-06-26:
> If I will make one more spreadsheet for Follow function - can it help us to be sure, that everything is ok?

You could just set the remote to "Follow" being ON. .. and mash buttons (other than Follow) one at a time till DumpV2 says follow is "Off" and just remember what the heck you pressed to trigger that. ;)

AndreyShpilevoy, 2019-06-26:
Ok, here it is: [Link](https://drive.google.com/open?id=1UBDTmXfe5yv82ESBzj6WB-Z48tXSrkZN)

Btw, I think we should just remove this function :)
I found what does it mean.

After activation, it starts to track the actual temperature and send it to AC.
If we will not implement Timer - I have no idea why we should implement this function.

crankyoldgit, 2019-06-28:
Disabled `setFollow()` and enhanced `getFollow()` to handle 0x5F and 0x5D.

So, once merged, I think we can close this issue out.

closed 2019-06-28 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

commit 1deb3637 (2019-06-24): Neoclima: Add remaining settings and common A/C support.
* Added set/get for:
  - SwingV
  - SwingH
  - Sleep
  - Turbo
  - Button,
  - Hold
  - Ion
  - Light
  - 8CHeat
  - Eye
  - Follow
* Add Common A/C support for Neoclima A/C
* Update IRMQTTServer.
* Unit tests to cover all of the above.

Fixes #764
PR #767 (2019-06-28): Experimental detailed support for Neoclima protocol.
* `sendNeoclima()` & `decodeNeoclima()` added.
* `IRNeoclimaAc` class added with full support.
  - Self decoding works.
  - Real data works.
* Common A/C support.
* Example code updated for Neoclima support.
* Minor other code cleanup.

Fixes #764
commit 6f416901 (2019-06-28): Experimental detailed support for Neoclima protocol. (#767)

* `sendNeoclima()` & `decodeNeoclima()` added.
* `IRNeoclimaAc` class added with full support.
  - Self decoding works.
  - Real data works.
* Common A/C support.
* Example code updated for Neoclima support.
* Minor other code cleanup.

Fixes #764