crankyoldgit/IRremoteESP8266 #768: IRrecvDumpV2 doesn't work on Olimex ESP32-EVB board

u4mzu4, 2019-06-22:
### Version/revision of the library used
v2.6.2

### Expected behavior
Some output on Serial Monitor

### Actual behavior
"IRrecvDumpV2 is now running and waiting for IR input on Pin 39"

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Not applicable.

### Steps to reproduce the behavior
1.) open IRrecvDumpV2.ino
2.) modifie the line
const uint16_t kRecvPin = 14;
to
const uint16_t kRecvPin = 39;
3.) compile, download and run

#### Example code used
IRrecvDumpV2.ino

#### Circuit diagram and hardware used (if applicable)
https://github.com/OLIMEX/ESP32-EVB/raw/master/HARDWARE/REV-D/ESP32-EVB_Rev_D.pdf

### I have followed the steps in the [Troubleshooting Guide]
Not yet.

### Has this library/code previously worked as expected for you?
Unfortunately not.

### Other useful information
IR receiver was never tested on this board before. There is an internal 100k pullup resistor in the RPM7236-H8 receiver chip (36 Khz).

crankyoldgit, 2019-06-22:
According to the pdf link you supplied, it uses a RPM7236 demodulator.
A google search (and the item number) for it indicates it's a 36kHz (as you later pointed out) demodulator.
(See https://au.mouser.com/datasheet/2/348/rohm%20semiconductor_rpm7236-1201702.pdf)

See the Troubleshooting guide in the wiki (https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#basic-1)

It *might* work for you some of the time, but not 100% reliably for receiving non-36kHz protocols. Most protocols are 38kHz. It's by far the most common and industry default.

Try testing it out with devices that use 36kHz protocols such as RC5/RC6/RCMM (i.e. some old TVs, Xbox 360), LaserTag, Trotec, Panasonic, Denon & Magiquest

crankyoldgit, 2019-06-22:
Unfortunately, my only ESP32 board doesn't have GPI 39 exposed, so I can't test on that particular pin.
I tried GPI34 & 35. I got no result. I tested GPIO33 and also got no result. GPIO14 works perfectly though on my ESP32 board. Weird. All of those >33 GPIOs should work as far as I can tell.

u4mzu4, 2019-06-22:
GPIO32 also? 
GPIOs 32-39 are connected to ADC1 channels, and it could cause interference with the SW interrupt (somehow).

crankyoldgit, 2019-06-22:
I think I've found the problem. I've now got it reliably working on every GPIO/GPI I've tried. inc. 35, which is an INPUT ONLY GPI. Thus I think it should work on 39.

Can you please try/confirm branch [Issue768](https://github.com/markszabo/IRremoteESP8266/tree/Issue768) / PR #774 fixes it?

u4mzu4, 2019-06-22:
Tested, finally works on GPIO39 as well. :-)

crankyoldgit, 2019-06-23:
Excellent. Thanks for the confirmation.
Re-opening the issue until the code is merged into `master`.

closed 2019-06-25 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

commit c99e5064 (2019-06-23): [BUG] Ensure `IRrecv`'s GPIO is set to input mode.
* Testing on different ESP32 boards highlighted that unlike ESP8266's
  the GPIO needs to be explicitly set to INPUT (or INPUT_PULLUP) mode for
  interrupts to work correctly.

* Adjust list of available GPIO/GPI's on ESP32 in IRMQTTServer.
* Add option to use `INPUT_PULLUP` via `IRrecv.enableIRIn(true)`.
  - Default is false. i.e. `INPUT`

Tested on GOOUUU-ESP32 & a Olimex ESP32-EVB board. Confirmed working.

Fixes #768
PR #774 (2019-06-25): [BUG] Ensure `IRrecv`'s GPIO is set to input mode.
* Testing on different ESP32 boards highlighted that unlike ESP8266's
  the GPIO needs to be explicitly set to INPUT (or INPUT_PULLUP) mode for
  interrupts to work correctly.

* Adjust list of available GPIO/GPI's on ESP32 in IRMQTTServer.
* Add option to use `INPUT_PULLUP` via `IRrecv.enableIRIn(true)`.
  - Default is false. i.e. `INPUT`

Fixes #768
commit c33d975d (2019-06-25): [BUG] Ensure `IRrecv`'s GPIO is set to input mode. (#774)

* Testing on different ESP32 boards highlighted that unlike ESP8266's
  the GPIO needs to be explicitly set to INPUT (or INPUT_PULLUP) mode for
  interrupts to work correctly.

* Adjust list of available GPIO/GPI's on ESP32 in IRMQTTServer.
* Add option to use `INPUT_PULLUP` via `IRrecv.enableIRIn(true)`.
  - Default is false. i.e. `INPUT`

Tested on GOOUUU-ESP32 & a Olimex ESP32-EVB board. Confirmed working.

Fixes #768