crankyoldgit/IRremoteESP8266 #769: Can I reflash bin file using web interface? Basically have a premade device, accessing the gpio pins of which is difficult

2016for, 2019-06-22:
Currently have Tasmota on it. Tasmota has web firmware upgrade, was planning to install .bin file using that, however can I do it other way around as well, so that if something doesn't work I can just re-flash it using a web interface or something?

crankyoldgit, 2019-06-22:
Probably. I haven't tested it myself, but it should work in principle.
There is nothing in `IRMQTTServer`'s OTA upgrade stuff that checks what `.bin` file you are upload, only if it fits. Just watch out if you are using a device with only a small amount of flash. e.g. A Sonoff device etc.

Try it on another more easily accessible device first if you want to be sure.

Let us know how you go.

2016for, 2019-06-23:
> Probably. I haven't tested it myself, but it should work in principle.
> There is nothing in `IRMQTTServer`'s OTA upgrade stuff that checks what `.bin` file you are upload, only if it fits. Just watch [...]

Awesome. Thanks for that. How would I go about flashing .bin file once I am on your IRMQTTServer firmware?

crankyoldgit, 2019-06-23:
From the Menu, click the [Admin] button. "Update firmware" should be at the bottom of the page assuming you have set a password. If you haven't set a password, you will need to "Wipe Settings" and re-setup the device & the password in the initial config session.

e.g. 
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L271

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L45

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L286

2016for, 2019-06-23:
I am trying to flash over Tasmota with sonoff-minimal.bin (342kb) a d1_mini .bin file I compiled using gitpod with IRMQTTServer platform.ino (497kb) but while trying to flash it says 

> Program flash size is larger than real flash size
 
Any tips?

crankyoldgit, 2019-06-23:
Have you checked what Flash Size you have set in the Build Target? And how much flash does the device have?
e.g. Did you build an image with a 4Meg target and are trying to fit it onto a device with only 1M of flash?

2016for, 2019-06-24:
> Have you checked what Flash Size you have set in the Build Target? And how much flash does the device have?
> e.g. Did you build an image with a 4Meg target and are trying to fit it onto a device with only 1M of flash?

Oh that was the case. I was using env: d1_mini

I created a new env in platformio.ini with board=esp01_01 (I have a generic Tuya device which probably uses 1MB esp8266 chips)

That should do it right?

crankyoldgit, 2019-06-24:
I think you're about to find out. Let us know.

2016for, 2019-06-24:
Worked! :)

Added the Tuya device on HA, IR-receiving is working, trying to figure out problem with sending.

Also there's a status LED available on the device, where do I set the gpio for that?
(in Tasmota config the GPIO for LED is:
`GPIO_LED1_INV,    // GPIO04 Blue Led - Link status`

crankyoldgit, 2019-06-24:
No idea. IRMQTTServer doesn't have any settings for that. You'll have add/work that out for yourself.

crankyoldgit, 2019-06-24:
Seems this issue is now complete. Closing.

closed 2019-06-24 by crankyoldgit