crankyoldgit/IRremoteESP8266 #775: Cant learn Panasonic AC IR code correctly

jamietang68, 2019-06-22:
Version 2.5.5 & 2.6.1

The hardware is that I'm using is Tuya universal IR controller with customised software. The sending function is working well, just that the learning function cant get Panasonic AC code right.

Initially i tried with v2.5.5 few months back, and recently i updated to 2.6.1 but both are having same issue.

I'll paste both the code that learned from the software and the correct code that is working below.

1. Version 2.5.5 Raw data & Hex code that is not working, AC On signal
Raw data 
WARNING: IR code is too big for buffer (>= 300). This result shouldn't be trusted until this is resolved. Edit & increase kCaptureBufferSize.
Encoding  : UNKNOWN
Code      : B79C9C5 (150 bits)

Raw Timing[299]:
   +  3514, -  1682,    +   468, -   402,    +   470, -  1270,    +   494, -   376, 
   +   468, -   402,    +   492, -   378,    +   492, -   378,    +   492, -   378, 
   +   468, -   402,    +   492, -   376,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   468, -   402,    +   468, -  1272,    +   468, -   402, 
   +   494, -   376,    +   492, -   376,    +   494, -   376,    +   468, -   402, 
   +   468, -   402,    +   468, -   402,    +   466, -  1272,    +   492, -  1248, 
   +   470, -  1272,    +   466, -   426,    +   444, -   424,    +   446, -  1272, 
   +   492, -   376,    +   468, -   402,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   468, -   402,    +   466, -   402,    +   470, -   400, 
   +   492, -   378,    +   468, -   402,    +   492, -   378,    +   492, -   376, 
   +   468, -   402,    +   468, -   402,    +   492, -   398,    +   448, -   402, 
   +   468, -   402,    +   468, -   402,    +   468, -   424,    +   446, -   402, 
   +   468, -   402,    +   468, -   402,    +   470, -   400,    +   492, -   376, 
   +   468, -   402,    +   468, -   402,    +   468, -   402,    +   466, -   402, 
   +   470, -   422,    +   444, -   402,    +   468, -  1272,    +   468, -  1272, 
   +   468, -   402,    +   468, -   402,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   492, - 76220,    +  3542, -  1682,    +   494, -   376, 
   +   468, -  1270,    +   494, -   376,    +   494, -   376,    +   494, -   376, 
   +   468, -   400,    +   494, -   376,    +   494, -   376,    +   492, -   378, 
   +   492, -   378,    +   494, -   376,    +   468, -   402,    +   468, -   402, 
   +   494, -  1244,    +   494, -   376,    +   478, -   392,    +   494, -   374, 
   +   468, -   404,    +   468, -   402,    +   468, -   400,    +   468, -   402, 
   +   494, -  1246,    +   468, -  1272,    +   468, -  1272,    +   468, -   426, 
   +   444, -   400,    +   468, -  1272,    +   468, -   402,    +   496, -   374, 
   +   468, -   402,    +   468, -   402,    +   470, -   400,    +   492, -   376, 
   +   492, -   378,    +   468, -   402,    +   468, -   400,    +   468, -   402, 
   +   466, -   402,    +   468, -   402,    +   466, -   402,    +   468, -   402, 
   +   494, -   374,    +   470, -   402,    +   466, -  1274,    +   492, -  1248, 
   +   468, -  1272,    +   468, -   402,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   468, -  1272,    +   494, -  1246,    +   470, -   402, 
   +   468, -  1272,    +   468, -   402,    +   492, -   378,    +   494, -   376, 
   +   492, -   398,    +   472, -   378,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   468, -   400,    +   494, -  1246,    +   468, -  1272, 
   +   468, -   402,    +   468, -  1272,    +   468, -   424,    +   444, -  1272, 
   +   468, -  1272,    +   468, -  1272,    +   470, -   400,    +   468, -   402, 
   +   468, -   400,    +   470, -   400,    +   468, -   402,    +   468, -   402, 
   +   468, -   402,    +   468, -   402,    +   492, -   376,    +   494, -   376, 
   +   494, -   376,    +   470

Hex code (not working)
0000 006B 0004 0092 0086 0040 0011 000F 0012 0030 0012 000E 0011 000F 0012 000E 0012 000E 0012 000E 0011 000F 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0011 0030 0011 000F 0012 000E 0012 000E 0012 000E 0011 000F 0011 000F 0011 000F 0011 0030 0012 002F 0012 0030 0011 0010 0011 0010 0011 0030 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0012 000F 0012 000E 0011 000F 0012 000E 0012 000E 0011 000F 0011 000F 0012 000F 0011 000F 0011 000F 0011 000F 0011 0010 0011 000F 0011 000F 0011 000F 0012 000F 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0012 0010 0011 000F 0011 0030 0011 0030 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0012 019A 0088 0040 0012 000E 0011 0030 0012 000E 0012 000E 0012 000E 0011 000F 0012 000E 0012 000E 0012 000E 0012 000E 0012 000E 0011 000F 0011 000F 0012 002F 0012 000E 0012 000F 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0012 002F 0011 0030 0011 0030 0011 0010 0011 000F 0011 0030 0011 000F 0013 000E 0011 000F 0011 000F 0012 000F 0012 000E 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0011 000F 0012 000E 0012 000F 0011 0030 0012 002F 0011 0030 0011 000F 0011 000F 0011 000F 0011 000F 0011 0030 0012 002F 0012 000F 0011 0030 0011 000F 0012 000E 0012 000E 0012 000F 0012 000E 0011 000F 0011 000F 0011 000F 0011 000F 0012 002F 0011 0030 0011 000F 0011 0030 0011 0010 0011 0030 0011 0030 0011 0030 0012 000F 0011 000F 0011 000F 0012 000F 0011 000F 0011 000F 0011 000F 0011 000F 0012 000E 0012 000E 0012 000E 0012 0E6C

____________________________________________________________________________________________________

2. Version 2.6.1 (increased buffer size to 301), no more error, but still not working
Raw Data
T: Timestamp : 361105T: Encoding  : UNKNOWN
Code      : 9DCF5C22 (66 bits)
N: Raw Timing[131]:
   +  3514, -  1680,    +   474, -   398,    +   492, -  1248,    +   468, -   422, 
   +   474, -   374,    +   496, -   394,    +   474, -   376,    +   468, -   400, 
   +   494, -   376,    +   494, -   376,    +   494, -   376,    +   494, -   376, 
   +   494, -   374,    +   496, -   374,    +   494, -  1244,    +   496, -   374, 
   +   496, -   374,    +   494, -   376,    +   492, -   376,    +   496, -   374, 
   +   468, -   402,    +   494, -   376,    +   470, -  1270,    +   494, -  1246, 
   +   496, -  1244,    +   494, -   376,    +   496, -   374,    +   494, -  1246, 
   +   496, -   374,    +   496, -   374,    +   496, -   374,    +   494, -   374, 
   +   494, -   376,    +   494, -   376,    +   494, -   396,    +   474, -   376, 
   +   494, -   374,    +   496, -   374,    +   496, -   374,    +   496, -   374, 
   +   494, -   374,    +   494, -   394,    +   474, -   376,    +   494, -   396, 
   +   474, -   374,    +   496, -   374,    +   494, -   376,    +   494, -   376, 
   +   494, -   374,    +   470, -   400,    +   494, -   374,    +   496, -   376, 
   +   494, -   376,    +   496, -   372,    +   494, -   376,    +   494, -   376, 
   +   494, -   376,    +   468, -   402,    +   468, -  1272,    +   468, -  1272, 
   +   494, -   374,    +   468, -   402,    +   494, -   376,    +   494, -   374, 
   +   494, -   376,    +   494
N: << 

Hex code
0000 006B 0000 0042 0086 0040 0012 000F 0012 002F 0011 0010 0012 000E 0013 000F 0012 000E 0011 000F 0012 000E 0012 000E 0012 000E 0012 000E 0012 000E 0013 000E 0012 002F 0013 000E 0013 000E 0012 000E 0012 000E 0013 000E 0011 000F 0012 000E 0012 0030 0012 002F 0013 002F 0012 000E 0013 000E 0012 002F 0013 000E 0013 000E 0013 000E 0012 000E 0012 000E 0012 000E 0012 000F 0012 000E 0012 000E 0013 000E 0013 000E 0013 000E 0012 000E 0012 000F 0012 000E 0012 000F 0012 000E 0013 000E 0012 000E 0012 000E 0012 000E 0012 000F 0012 000E 0013 000E 0012 000E 0013 000E 0012 000E 0012 000E 0012 000E 0011 000F 0011 0030 0011 0030 0012 000E 0011 000F 0012 000E 0012 000E 0012 000E 0012 0ED8

_____________________________________________________________________________________________________

IR Code that is working (copied online)
Hex code
0000 0070 0000 00DC 0081 0040 0011 000F 0011 002F 0010 0010 0010 0010 0010 0010 0010 000F 0012 000F 0010 000F 0011 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0030 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0011 000F 0010 0030 0010 0030 0010 0030 0011 000F 0010 0010 0010 0030 0010 0010 0010 0010 0011 000F 0011 000F 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0011 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 000F 0011 0010 0010 0010 0010 0010 00 [... cut ...] 11 0010 0010 0010 0011 0010 0011 0010 0010 0010 0011 0010 0011 0010 0010 0010 0010 0010 0011 0010 0010 0010 0011 0010 0011 0010 0010 0030 0011 000F 0011 0010 0011 0010 0011 0010 0010 0010 0010 0010 0011 0030 0011 0010 0011 0010 0011 0010 0011 0010 0010 0010 0010 0010 0010 0010 0011 0010 0011 0010 0011 0010 0011 0010 0011 0010 0010 0010 0011 0010 0010 0010 0010 0010 0010 0030 0011 0010 0011 0030 0010 0030 0010 0010 0011 0010 0011 0030 0011 0030 0010 0E74

Any idea how can I fix it? 

Thanks alot!

crankyoldgit, 2019-06-23:
> The hardware is that I'm using is Tuya universal IR controller with customised software.

What software are you using? If you are using someone else's code, please take it up with them.
i.e. we can't support users of someone else's project.

As mentioned in: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with
> Fixing code in some other maintained project that includes our library, unless you're one of the maintainers of that project. You should probably report an issue with them instead of us.

If you want me to help you, I'm going to need some _raw_ captures using [IRrecvDumpV2](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) of messages from your remote.

From the info you have so far reported, a) I'm not sure what you are trying to do, so can you please explain it better or a different way so I can understand. b) I _think_ the HEX codes you are reporting are `PRONTO` codes. How are you converting the signal to Pronto?  c) Your "no error" (2nd) dump seems to be a partial capture given the size of the first one. So, try again, or try increasing the _timeout_ value in the code.

crankyoldgit, 2019-06-28:
Friendly chase up.

crankyoldgit, 2019-07-05:
Another chase up. If we don't get a response in a few days I'll mark this closed due to inactivity.

crankyoldgit, 2019-07-09:
Closing this after two+ weeks of no response. If you reply, I'll reopen it.

closed 2019-07-09 by crankyoldgit