crankyoldgit/IRremoteESP8266 #778: Can you add AUX AC

gujiys, 2019-06-24:
POWER ON
MODE COOL
Temperature: 32
Air Speed auto
Swing vertical: off
Timestamp : 002055.546
                      Encoding  : ELECTRA_AC
                                            Code      : C3C7E000A0002000002000408A (104 bits)
                                                                                             Library   : v2.6.0

Raw Timing[211]:
                   +  8990, -  4548,    +   552, -  1712,    +   558, -  1712,    +   560, -   618,
                                                                                                       +   522, -   584,    +   558, -   620,    +   520, -   622,    +   520, -  1712,
                                                                      +   560, -  1712,    +   560, -  1712,    +   560, -  1712,    +   560, -  1712,
                                     +   558, -   584,    +   558, -   620,    +   520, -   622,    +   520, -  1712,
    +   562, -  1712,    +   560, -   582,    +   554, -   584,    +   556, -   582,
                                                                                        +   558, -   584,    +   558, -   620,    +   520, -  1712,    +   560, -  1712,
                                                       +   562, -  1712,    +   560, -   582,    +   556, -   582,    +   556, -   584,
                      +   558, -   584,    +   560, -   620,    +   522, -   582,    +   558, -   584,
                                                                                                          +   558, -   594,    +   544, -   584,    +   554, -   582,    +   556, -   620,
                                                                         +   522, -   584,    +   558, -   620,    +   522, -  1712,    +   560, -   584,
                                        +   560, -  1712,    +   560, -   582,    +   556, -   582,    +   556, -   584,
       +   556, -   584,    +   558, -   582,    +   560, -   584,    +   558, -   584,
                                                                                           +   560, -   618,    +   520, -   582,    +   556, -   582,    +   556, -   620,
                                                          +   520, -   584,    +   558, -   584,    +   558, -  1714,    +   560, -   582,
                         +   560, -   618,    +   520, -   582,    +   556, -   582,    +   554, -   620,
                                                                                                             +   520, -   584,    +   558, -   620,    +   520, -   622,    +   520, -   582,
                                                                            +   558, -   620,    +   520, -   582,    +   556, -   582,    +   556, -   584,
                                           +   556, -   584,    +   558, -   620,    +   520, -   622,    +   520, -   584,
          +   558, -   618,    +   520, -   582,    +   556, -   582,    +   556, -   620,
                                                                                              +   520, -   584,    +   558, -   584,    +   558, -  1712,    +   562, -   582,
                                                             +   560, -   618,    +   520, -   584,    +   554, -   584,    +   554, -   620,
                            +   520, -   584,    +   558, -   582,    +   558, -   584,    +   558, -   584,
                                                                                                                +   560, -   618,    +   520, -   582,    +   556, -   582,    +   556, -   584,
                                                                               +   558, -   584,    +   556, -   586,    +   556, -   584,    +   558, -  1714,
                                              +   562, -   618,    +   520, -   584,    +   554, -  1712,    +   558, -   620,
             +   520, -  1712,    +   560, -   582,    +   558, -   584,    +   558, -   584,
                                                                                                 +   560, -  1710,    +   556

uint16_t rawData[211] = {8990, 4548,  552, 1712,  558, 1712,  560, 618,  522, 584,  558, 620,  520, 622,  520, 1712,  560, 1712,  560, 1712,  560, 1712,  560, 1712,  558, 584,  558, 620,  520, 622,  520, 1712,  562, 1712,  560, 582,  554, 584,  556, 582,  558, 584,  558, 620,  520, 1712,  560, 1712,  562, 1712,  560, 582,  556, 582,  556, 584,  558, 584,  560, 620,  522, 582,  558, 584,  558, 594,  544, 584,  554, 582,  556, 620,  522, 584,  558, 620,  522, 1712,  560, 584,  560, 1712,  560, 582,  556, 582,  556, 584,  556, 584,  558, 582,  560, 584,  558, 584,  560, 618,  520, 582,  556, 582,  556, 620,  520, 584,  558, 584,  558, 1714,  560, 582,  560, 618,  520, 582,  556, 582,  554, 620,  520, 584,  558, 620,  520, 622,  520, 582,  558, 620,  520, 582,  556, 582,  556, 584,  556, 584,  558, 620,  520, 622,  520, 584,  558, 618,  520, 582,  556, 582,  556, 620,  520, 584,  558, 584,  558, 1712,  562, 582,  560, 618,  520, 584,  554, 584,  554, 620,  520, 584,  558, 582,  558, 584,  558, 584,  560, 618,  520, 582,  556, 582,  556, 584,  558, 584,  556, 586,  556, 584,  558, 1714,  562, 618,  520, 584,  554, 1712,  558, 620,  520, 1712,  560, 582,  558, 584,  558, 584,  560, 1710,  556};  // ELECTRA_AC
                                                    uint8_t state[13] = {0xC3, 0xC7, 0xE0, 0x00, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x40, 0x8A};
``
POWER ON
MODE COOL
Temperature: 16
Air Speed auto
Swing vertical: off
`Timestamp : 002461.564
                      Encoding  : ELECTRA_AC
                                            Code      : C347E000A0002000002000410B (104 bits)
                                                                                             Library   : v2.6.0

Raw Timing[211]:
                   +  8986, -  4552,    +   548, -  1716,    +   556, -  1716,    +   556, -   622,
                                                                                                       +   518, -   624,    +   516, -   586,    +   558, -   586,    +   556, -  1716,
                                                                      +   560, -  1714,    +   556, -  1716,    +   554, -  1716,    +   556, -  1716,
                                     +   556, -   624,    +   518, -   624,    +   518, -   586,    +   556, -  1716,
    +   558, -   620,    +   518, -   586,    +   552, -   590,    +   548, -   624,
                                                                                        +   518, -   624,    +   518, -   586,    +   554, -  1716,    +   558, -  1716,
                                                       +   558, -  1714,    +   558, -   586,    +   552, -   590,    +   548, -   624,
                      +   518, -   586,    +   556, -   586,    +   556, -   624,    +   518, -   586,
                                                                                                          +   556, -   592,    +   548, -   586,    +   552, -   586,    +   554, -   624,
                                                                         +   516, -   624,    +   518, -   622,    +   518, -  1716,    +   558, -   626,
                                        +   518, -  1716,    +   556, -   586,    +   552, -   586,    +   552, -   624,
       +   518, -   624,    +   518, -   624,    +   518, -   624,    +   518, -   626,
                                                                                           +   518, -   594,    +   544, -   586,    +   554, -   586,    +   556, -   624,
                                                          +   518, -   624,    +   518, -   624,    +   518, -  1714,    +   558, -   588,
                         +   556, -   592,    +   546, -   586,    +   552, -   586,    +   554, -   624,
                                                                                                             +   518, -   624,    +   518, -   622,    +   518, -   624,    +   516, -   586,
                                                                            +   556, -   594,    +   544, -   586,    +   552, -   586,    +   552, -   624,
[... 33 lines cut ...]
                      +   518, -   624,    +   518, -   586,    +   556, -   624,    +   518, -   586,
                                                                                                          +   556, -   622,    +   518, -   586,    +   552, -   590,    +   548, -   624,
[... cut ...]  624,  516, 624,  518, 1716,  558, 586,  556, 622,  518, 586,  552, 594,  544, 624,  518, 586,  556, 624,  516, 624,  518, 626,  516, 622,  516, 1732,  540, 592,  546, 624,  518, 624,  518, 586,  556, 624,  518, 1716,  558, 622,  516, 1716,  554, 1716,  556, 624,  518, 1716,  556, 586,  556, 624,  518, 1714,  558, 1716,  552};  // ELECTRA_AC
                                                      uint8_t state[13] = {0xC3, 0x47, 0xE0, 0x00, 0x60, 0x00, 0x20, 0x00, 0x00, 0x20, 0x00, 0x41, 0xCB};

`

crankyoldgit, 2019-06-25:
Please read the FAQ on this: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_

Please see #764 as a recent example of the work **you** will need to do in order for someone to be able to add the details you want.

huexpub, 2019-06-28:
> Please read the FAQ on this: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_
>
> Please see #764 as a [...]

https://github.com/ToniA/arduino-heatpumpir/blob/master/AUXHeatpumpIR.h   maybe useful!

crankyoldgit, 2019-06-28:
What are the Brand and Model numbers of BOTH the remote and the A/C head unit please?

gujiys, 2019-06-28:
brand: AUX
Main engine model: KFR-35GW/BpNFW=3
Model of remote control:  YKR-T/011

crankyoldgit, 2019-06-28:
Thanks for the info.
I've added the protocol as best I can based on the link you mentioned.

Please try out the code & examples in the branch https://github.com/markszabo/IRremoteESP8266/tree/electra_detailed

Please pay extra attention to the Swing settings. I'm not 100% sure about them.

crankyoldgit, 2019-07-02:
Have you tried out that branch yet?

closed 2019-07-04 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

commit 34414acd (2019-07-04): Experimental detailed support for Electra/AUX protocol
Based on: 
https://github.com/ToniA/arduino-heatpumpir/blob/master/AUXHeatpumpIR.cpp
* Add common a/c support
* Add unit tests to cover new settings
* Example code updated.

Fixes #778
PR #788 (2019-07-04): Experimental detailed support for Electra/AUX protocol
Based on: 
https://github.com/ToniA/arduino-heatpumpir/blob/master/AUXHeatpumpIR.cpp
* Add common a/c support
* Add unit tests to cover new settings
* Example code updated.

Fixes #778
commit 08be4cb1 (2019-07-04): Experimental detailed support for Electra/AUX protocol (#788)

Based on: 
https://github.com/ToniA/arduino-heatpumpir/blob/master/AUXHeatpumpIR.cpp
* Add common a/c support
* Add unit tests to cover new settings
* Example code updated.

Fixes #778