crankyoldgit/IRremoteESP8266 #780: Few issues with the IRMQTTServer

2016for, 2019-06-25:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

2.6.2 (AirCon shows 1.12 beta)

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

1. Repeater is not working. I have enabled repeater using #define REPLAY_DECODED_AC_MESSAGE true
2. Also IRMQTTServer doesn't resend a code if you don't change any settings. I mean because of the nature of IR sometimes the code isn't effectively picked up by the AC and we need to send the same code again, but in order to do so right now we have to change something then change back to what we want and then update and send.
3. Currently, the IRMQTTServer sends the current state at boot, which I am pretty sure will have bad consequences in my setup (frequent powercuts). How do I disable that?

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

1. Well it is not repeating the code received, I want it to repeat it. :)
2. As far as 2nd one is concerned it would be great if we send IR command every time even if it's the same code (this won't have any negatives, and the positives are peace of mind)

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2019-06-25:
1. May be a bug. Can you please turn on DEBUG (in the .h file) and give me the serial output, and a copy of the Info page?

2 & 3 are really feature requests. Please create a separate issue for them.

crankyoldgit, 2019-06-25:
Also, See https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue

Especially points 1, 8, & 11. ;-P

closed 2019-07-02 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

PR #784 (2019-07-02): [BUG] IRMQTTServer: REPLAY_DECODED_AC_MESSAGE not working.
* Bugs:
  - Fix REPLAY_DECODED_AC_MESSAGE behaviour.

* Feature requests:
  - Add MQTT_CLIMATE_IR_SEND_ON_RESTART compile-time flag. (defaults to off)
  - Add the ability resend existing climate state via MQTT & HTTP.
    - e.g. Topic: `ir_server/ac/cmnd/resend` with a payload: `resend`

Fixes #780
commit 9a8e8856 (2019-07-02): [BUG] IRMQTTServer: REPLAY_DECODED_AC_MESSAGE not working. (#784)

* Bugs:
  - Fix REPLAY_DECODED_AC_MESSAGE behaviour.

* Feature requests:
  - Add MQTT_CLIMATE_IR_SEND_ON_RESTART compile-time flag. (defaults to off)
  - Add the ability resend existing climate state via MQTT & HTTP.
    - e.g. Topic: `ir_server/ac/cmnd/resend` with a payload: `resend`

Fixes #780
PR #797 (2019-07-04): Additional fix for `REPLAY_DECODED_AC_MESSAGE`
Fixes #780
commit fd04e4fb (2019-07-04): Additional fix for `REPLAY_DECODED_AC_MESSAGE` (#797)

Fixes #780