crankyoldgit/IRremoteESP8266 #781: FUJITSU AC - AR-RHA2E - Swing modes not working

2016for, 2019-06-25:
### Version/revision of the library used

2.6.2 master. remote model: AR-RHA2E

### Expected behavior
_What steps did you do and what should it have done?_

IR receiver is able to pickup swingv and swingh as auto (sent using manual remote) in order to test if swing works on AC.

However when trying to send via AirCon Control/HA, swing doesn't work on AC.

crankyoldgit, 2019-06-25:
More information please.
Can you capture the message from the remote as to what the ESP should be sending instead of what it does?

AND

What commands, state, & sequence etc are you sending to the ESP in order to get it to produce the desired swing?

crankyoldgit, 2019-06-25:
e.g. https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue

Points 1, 8, 10, & 11-13. :-P

2016for, 2019-06-26:
Sorry for that. :)
I actually don't have access to serial monitor for the device. Can I get those logs/commands from the web GUI?

crankyoldgit, 2019-06-26:
Not really.

I suggest you monitor the MQTT broker for messages to the relevant topics (e.g. `ir_server/ac/cmnd/+` &   `ir_server/ac/stat/+`) and capture the IR messages with a different ESP running IRrecvDumpV2.

Without some data or a more detailed description of what is going on I'm afraid I can't help you.

crankyoldgit, 2019-06-28:
Just chasing this up ...

2016for, 2019-06-29:
[image]

code sent by ESP: 33,1463001010FE0930610030000000201F
code sent by remote: 33,1463001010FE09306000300000002020

2016for, 2019-06-29:
sending swingv/h: auto from one ESP to other reflects swingv/h: off

2016for, 2019-06-29:
[image]

code sent by ESP: 33,1463001010FE0930610130000000201E
code sent remote: 33,1463001010FE09305000300000002030

crankyoldgit, 2019-06-29:
> (same settings but both swing off= 33,1463001010FE09305000000000002060)
(same settings but swingv: on {shows as auto on webUI}= 33,1463001010FE09305000000000002050)
(same settings but swingh: on {shows as auto on webUI}= 33,1463001010FE09305000000000002040)

Where are these settings coming from? They don't make any sense. The last byte of a Fujitsu A/C state that long is the checksum. And it's changing for all three values where the rest of the state is exactly the same, thus the checksum should be the same in all three.

Can use `IRrecvDumpV2` (like I suggested) to capture the actual remote settings? I don't think we should trust the values IRMQTTServer is reporting for captures if the above values were reported by it.

crankyoldgit, 2019-06-29:
Oh, and I mean the full capture from IRrecvDumpV2, not just the `state[]` lines, as something fishy is going on.

crankyoldgit, 2019-06-29:
Also from your swing off message:
> code sent remote: 33,1463001010FE09305000300000002030

As far as I can tell (using the library, and manual verification, that this is actually with both swings set to on.

crankyoldgit, 2019-06-29:
@2016for As far as I can tell the library is doing the right thing (See https://github.com/markszabo/IRremoteESP8266/commit/0355e29b9eb9f7db0c1f8009c647d8531256e678)
 but it's hard to tell because your supplied data doesn't match what you say it is in some cases (or at least what I think it is)

crankyoldgit, 2019-07-03:
Friendly chase up for more data.

2016for, 2019-07-05:
So sorry, will get back to you soon over the weekend.

crankyoldgit, 2019-07-09:
The weekend has passed. Any luck?

crankyoldgit, 2019-07-16:
Another week, another reminder.

crankyoldgit, 2019-07-18:
Yet another chase up.

crankyoldgit, 2019-07-21:
Last chase up. If I don't hear from you by Monday, I'm going to mark this issue closed due to inactivity.

crankyoldgit, 2019-07-23:
Marking this close due to inactivity.

closed 2019-07-23 by crankyoldgit