crankyoldgit/IRremoteESP8266 #783: IRMQTTServer turns off AC when changing temperature

silvanverschuur, 2019-06-25:
I'm using the IRMQTTServer to control my Whirlpool AC unit. I can send a `on` message to the `ir_server/ac/cmnd/power` topic. The AC unit turns on, this works fine.

When I change the temperature afterwards (`ir_server/ac/cmnd/temp` topic) the unit turns off. 
This also happens when changing the mode from fan -> cool for example. So all parameters are sent to the unit when sending a power on, changing values turn the unit off.

Whirlpool PACW12HP
Remote: DG11J2-31

I've tried IRrecvDumpv2 to see if the remote is detected correctly. All buttons are recognized, unit is detected as Model: 1 (DG11J13A).

# Version/revision of the library used
Version 2.6.2

crankyoldgit, 2019-06-26:
This sounds like a bug.
Can you please capture and record the sequence you want to have happen via `IRrecvDumpV2` and put it in here?
e.g. Record what you want the device to do, but using your remote to create the signals, rather than the ESP generating them.

Whirlpool has only recently been added & is experimental, and we may not have it's behaviour sorted out perfectly. Thus we need to compare it to what it really should be doing.

silvanverschuur, 2019-06-26:
I've captured the power on (temperature on the remote is 23 degrees) and an increase temperature message (temp from 23 -> 24 degrees). It seems to me that the code contains every parameter (mode, temp, fan speed, etc). So when I increase the temperature all the other parameters are also transmitted.

**Power**
```
IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Timestamp : 000004.052
Encoding  : WHIRLPOOL_AC
Code      : 83060452000080000000000000D600010000000001 (168 bits)
Mesg Desc.: Model: 1 (DG11J13A), Power toggle: On, Mode: 2 (COOL), Temp: 23C, Fan: 0 (AUTO), Swing: Off, Light: On, Clock: 00:00, On Timer: Off, Off Timer: Off, Sleep: Off, Super: Off, Command: 1 (POWER)
Library   : v2.6.2

Raw Timing[343]:
   +  9044, -  4454,    +   648, -  1600,    +   650, -  1604,    +   644, -   490, 
   +   644, -   488,    +   650, -   492,    +   644, -   498,    +   644, -   502, 
   +   644, -  1614,    +   644, -   480,    +   650, -  1600,    +   650, -  1610, 
   +   644, -   494,    +   644, -   492,    +   644, -   498,    +   644, -   502, 
   +   644, -   488,    +   650, -   474,    +   650, -   484,    +   644, -  1604, 
   +   648, -   488,    +   650, -   492,    +   644, -   498,    +   646, -   502, 
   +   644, -   488,    +   648, -   480,    +   644, -  1606,    +   648, -   484, 
   +   648, -   490,    +   648, -  1614,    +   648, -   494,    +   650, -  1622, 
   +   644, -   488,    +   648, -   480,    +   644, -   484,    +   648, -   484, 
   +   644, -   494,    +   644, -   494,    +   648, -   494,    +   648, -   498, 
   +   648, -   488,    +   650, -   480,    +   644, -   484,    +   644, -   488, 
   +   644, -   492,    +   644, -   494,    +   648, -   494,    +   648, -   498, 
   +   650, -   470,    +   650, -  7908,    +   644, -   480,    +   650, -   484, 
   +   644, -   488,    +   644, -   488,    +   648, -   494,    +   648, -   494, 
   +   648, -   502,    +   644, -  1614,    +   648, -   476,    +   648, -   484, 
   +   648, -   488,    +   648, -   494,    +   644, -   498,    +   648, -   498, 
   +   648, -   502,    +   644, -   488,    +   648, -   480,    +   644, -   484, 
   +   644, -   488,    +   644, -   488,    +   648, -   494,    +   644, -   498, 
   +   644, -   502,    +   644, -   494,    +   644, -   480,    +   648, -   480, 
   +   648, -   484,    +   648, -   490,    +   644, -   494,    +   648, -   492, 
   +   650, -   498,    +   650, -   488,    +   650, -   480,    +   644, -   484, 
   +   644, -   488,    +   644, -   494,    +   644, -   494,    +   648, -   498, 
   +   644, -   502,    +   644, -   488,    +   648, -   480,    +   644, -   484, 
   +   644, -   484,    +   648, -   488,    +   650, -   492,    +   644, -   498, 
   +   650, -   498,    +   648, -   488,    +   644, -   480,    +   648, -   480, 
   +   648, -   484,    +   648, -   488,    +   650, -   492,    +   644, -   498, 
   +   644, -   502,    +   644, -   488,    +   650, -   480,    +   648, -  1604, 
   +   644, -  1608,    +   648, -   488,    +   650, -  1614,    +   648, -   498, 
   +   648, -  1618,    +   648, -  1596,    +   648, -  7904,    +   644, -   480, 
   +   648, -   480,    +   650, -   484,    +   648, -   490,    +   648, -   488, 
   +   648, -   498,    +   644, -   502,    +   644, -   488,    +   650, -  1600, 
   +   648, -   486,    +   648, -   484,    +   648, -   488,    +   648, -   494, 
   +   644, -   498,    +   648, -   502,    +   648, -   488,    +   650, -   480, 
   +   652, -   480,    +   648, -   488,    +   648, -   494,    +   648, -   494, 
   +   648, -   498,    +   648, -   498,    +   650, -   488,    +   650, -   474, 
   +   648, -   488,    +   640, -   484,    +   650, -   488,    +   648, -   490, 
   +   648, -   492,    +   650, -   498,    +   648, -   488,    +   648, -   476, 
   +   648, -   484,    +   644, -   488,    +   644, -   490,    +   648, -   494, 
   +   644, -   498,    +   644, -   502,    +   644, -   488,    +   648, -   476, 
   +   648, -   484,    +   644, -   484,    +   648, -   488,    +   650, -   488, 
   +   650, -   498,    +   644, -   502,    +   648, -   490,    +   648, -  1610, 
   +   648, -   484,    +   648, -   488,    +   650, -   488,    +   648, -   498, 
   +   644, -   502,    +   644, -   502,    +   648, -   476,    +   644

uint16_t rawData[343] = {9044, 4454,  648, 1600,  650, 1604,  644, 490,  644, 488,  650, 492,  644, 498,  644, 502,  644, 1614,  644, 480,  650, 1600,  650, 1610,  644, 494,  644, 492,  644, 498,  644, 502,  644, 488,  650, 474,  650, 484,  644, 1604,  648, 488,  650, 492,  644, 498,  646, 502,  644, 488,  648, 480,  644, 1606,  648, 484,  648, 490,  648, 1614,  648, 494,  650, 1622,  644, 488,  648, 480,  644, 484,  648, 484,  644, 494,  644, 494,  648, 494,  648, 498,  648, 488,  650, 480,  644, 484,  644, 488,  644, 492,  644, 494,  648, 494,  648, 498,  650, 470,  650, 7908,  644, 480,  650, 484,  644, 488,  644, 488,  648, 494,  648, 494,  648, 502,  644, 1614,  648, 476,  648, 484,  648, 488,  648, 494,  644, 498,  648, 498,  648, 502,  644, 488,  648, 480,  644, 484,  644, 488,  644, 488,  648, 494,  644, 498,  644, 502,  644, 494,  644, 480,  648, 480,  648, 484,  648, 490,  644, 494,  648, 492,  650, 498,  650, 488,  650, 480,  644, 484,  644, 488,  644, 494,  644, 494,  648, 498,  644, 502,  644, 488,  648, 480,  644, 484,  644, 484,  648, 488,  650, 492,  644, 498,  650, 498,  648, 488,  644,  [... cut ...] 484,  648, 484,  648, 494,  644, 498,  646, 498,  648, 488,  644, 480,  646, 484,  644, 484,  644, 488,  644, 494,  644, 498,  644, 498,  648, 484,  648, 474,  650, 480,  644, 484,  650, 488,  644, 494,  644, 494,  650, 496,  650, 488,  644, 480,  644, 1606,  644, 484,  650, 488,  644, 492,  644, 498,  644, 502,  644, 470,  644};  // WHIRLPOOL_AC
uint8_t state[21] = {0x83, 0x06, 0x00, 0x62, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xE2, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x02};
```

crankyoldgit, 2019-06-27:
@silvanverschuur Can I please get you to try using the library and example code from the `master` branch. I think I may have already fixed this in #758 
Let me know if that solves your problem or not.

As for other captures, I'd really like to know if the settings are really "toggles" or not.
i.e. they only appear when you change the setting, not in every copy of the messages

silvanverschuur, 2019-06-27:
@crankyoldgit I can confirm that the `master` branch fixes my problem. Everything now works as expected. Thanks!

closed 2019-06-27 by silvanverschuur