crankyoldgit/IRremoteESP8266 #785: Firmware upload failing

2016for, 2019-06-26:
### Version/revision of the library used

2.6.2

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

Tried using same firmware.bin that is currently on ESP device (both of them 430kb)
Also tried using sonoff-minimal.bin (350kb)
Made sure OTA password is enabled
Wiped settings to make sure OTA password is set

crankyoldgit, 2019-06-26:
No idea. Without the serial/debug turned on, I can't tell what the issue might be.

The code that does it is here: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L2252

The only reason it should fail is if it can't fit or upload the file. e.g. Not enough free space.
It works fine on all the devices I've tested it on.

crankyoldgit, 2019-07-03:
FYI, I've tried to emulate what you did on a 1Meg ESP. You have approx ~250k of free sketch space to upload into.
So, you need a binary that is smaller than that to upload with. If you go with some really trivial OTA example code, you might get it to squeeze in to that space, and then be able to upload another image via OTA. That's all if you can't get to the serial interface to upload via serial.

crankyoldgit, 2019-07-03:
If you choose a build option of 64k SPIFFS for your 1Meg board in the Arduino IDE, you can upload an ~444k (vs ~250k) firmware size from a standard build of IRMQTTServer.

closed 2019-07-04 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

2016for, 2019-07-05:
> If you choose a build option of 64k SPIFFS for your 1Meg board in the Arduino IDE, you can upload an ~444k (vs ~250k) firmware size from a standard build of IRMQTTServer.

Hey thanks for the suggestions, I am actually compiling the bin files using platform-io on gitpod, any suggestions how I can do this?

Plus, any future things I should keep in mind because I would be re-uploading the firmware a few times via OTA to finally come up with the firmware of IRMQTTServer that suits my needs.

crankyoldgit, 2019-07-06:
> Hey thanks for the suggestions, I am actually compiling the bin files using platform-io on gitpod, any suggestions how I can do this?

See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/platformio.ini#L61

2016for, 2019-07-06:
> > Hey thanks for the suggestions, I am actually compiling the bin files using platform-io on gitpod, any suggestions how I can do this?
> 
> See: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/platformio.ini#L61

Oh sorry I meant how to go about the 64k SPIFFS thing?

crankyoldgit, 2019-07-06:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/platformio.ini#L68

2016for, 2019-07-09:
Thanks a lot. Can I upload using this build on a normal old 1mb firmware already on the device with probably 250kb space left like you mentioned.

crankyoldgit, 2019-07-09:
No. The IRMQTTServer code **will not** compile down to less than 250k. You need to find something smaller than that if you can.

I suggest you try with the smallest available example code e.g. Follow the instructions https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#web-browser
and make sure you are using a target of 1M with 64k SPIFFS. If you are lucky, that might produce a firmware.bin file that's <= 250k or what ever space your device has left.

Once you've got that, you'll probably be able to upload it and then you can recompile/upload a better version of IRMQTTServer or Tasmota etc that will allow enough space for OTA uploading.

crankyoldgit, 2019-07-09:
Tip: Try older versions of the esp8266 core as well e.g. 2.3.0 etc. It may help produce a small enough binary to get you out of your bind.

PR #795 (2019-07-04): Show available sketch space for OTA uploads.
* Update docs/comments for dealing with low flash memory boards.
* Add a Platformio build environment for ESP8266s with
  1M of flash and minimal SPIFFS space.

Fixes #785
commit 786b843c (2019-07-04): Show available sketch space for OTA uploads. (#795)

* Update docs/comments for dealing with low flash memory boards.
* Add a Platformio build environment for ESP8266s with
  1M of flash and minimal SPIFF space.

Fixes #785