crankyoldgit/IRremoteESP8266 #787: MQTT problem

hugodelaroza, 2019-06-27:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
2.5.6
### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

Send a command by MQTT

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

I try to send a 100bit IR command using MQTT but the command is not sent to de final device

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

i can not send a raw with 100 bits via mqtt

when I sent it via http it works perfect!

http://192.168.0.22/ir?type=30&code=38000,4480,4430,550,1634,524,528,524,1612,548,1636,524,504,550,528,526,1634,526,528,524,504,548,1636,524,530,524,504,550,1632,526,1634,526,528,524, 1608,550,504,548,1612,546,1636,522,1612,548,1612,546,504,550,1612,548,1636,526,1634,526,502,550,528,524,530,522,530,524,1608,552,504,548,528,524,1634,526,1610,550,1634,524,506,548,530,524,504,578,498,554,498,524,530,524,528,526,528,554,1604,526,1634,524,1610,550,1634,524,1636,524,5244,4450,4430,550,1634,526,526,526,1632,526,1608,550,528,522,530,522,1636,552,500,524,528,580,1578,526,506,546,530,524,1634,524,1612,546,504,550,1634,526,526,526,1634,526,1634,526,1634,526,1634,526,528,524,1634,526,1634,554,1606,526,528,524,504,550,526,556,472,578,1582,548,502,550,528,524,1636,524,1610,550,1612,548,528,584,470,524,528,524,530,524,504,550,526,556,498,524,502,550,1636,524,1610,550,1610,550,1634,526,1610,550

but it does not work when I send it via mqtt

command_topic: "IR02/send"
payload_on: "38000,4480,4430,550,1634,524,528,524,1612,548,1636,524,504,550,528,526,1634,526,528,524,504,548,1636,524,530,524,504,550,1632,526,1634,526,528,524,1608,550,504,548,1612,546,1636,522,1612,548,1612,546,504,550,1612,548,1636,526,1634,526,502,550,528,524,530,522,530,524,1608,552,504,548,528,524,1634,526,1610,550,1634,524,506,548,530,524,504,578,498,554,498,524,530,524,528,526,528,554,1604,526,1634,524,1610,550,1634,524,1636,524,5244,4450,4430,550,1634,526,526,526,1632,526,1608,550,528,522,530,522,1636,552,500,524,528,580,1578,526,506,546,530,524,1634,524,1612,546,504,550,1634,526,526,526,1634,526,1,634,526,1634,526,1634,526,528,524,1634,526,1634,554,1606,526,528,524,504,550,526,556,472,578,1582,548,502,550,528,524,1636,524,1610,550,1612,548,528,584,470,524,528,524,530,524,504,550,526,556,498,524,502,550,1636,524,1610,550,1610 , 550,1634,526,1610,550 "

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

I changed the pubsubclient.h to 2048
#define MQTT_MAX_PACKET_SIZE 2048

I'm using a wemos d1 mini

THANKS A LOT!

crankyoldgit, 2019-06-27:
I think you have two problems:
1) You are missing the leading "30," to specify raw mode. See https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L90
2) you have spaces in your raw string. Try it with only digits and ","s in the MQTT string.

hugodelaroza, 2019-06-28:
Hi!, It´s Working now!! the problem was the missin of 30 numer!!

Thanks a lot!!

closed 2019-06-28 by crankyoldgit