crankyoldgit/IRremoteESP8266 #791: Problem switching off SAMSUNG AC AR12HSSDBWKNEU

fabriors73, 2019-07-01:
Hello I'm completely new of github. I'm in these days trying to create a remote for my Samsung AC - AR12HSSDBWKNEU.
I have not tested all the option but, so far, everything seems to work fine except the switch off that doesn't work.

### Version/revision of the library used: 2.6.2

### Used CODE
I used the example test found ControlSamsungAC.ino

### Actual behavior
Does not switch off.

if it can be usefull I captured the code sent by the original IR pilot using IRrecvDumpV2 on arduino  UNO. this is the result.

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[115]: 
     + 450, - 550     + 450, -1500     + 500, - 500     + 450, - 550
     + 450, - 500     + 500, - 500     + 450, - 550     + 450, - 500
     + 500, - 500     + 500, -1450     + 500, - 500     + 500, - 500
     + 450, -1500     + 500, -1500     + 450, - 550     + 450, -1500
     + 450, -1500     + 500, -1500     + 450, -1500     + 500, -1450
     + 500, - 500     + 500, - 500     + 450, - 500     + 500, - 500
     + 500, - 500     + 450, - 500     + 500, - 500     + 500, - 500
     + 450, - 550     + 450, - 550     + 450, - 500     + 450, - 550
     + 450, - 550     + 450, - 500     + 500, - 500     + 450, - 550
     + 450, - 500     + 500, - 500     + 500, - 500     + 450, - 500
     + 500, - 500     + 500, - 500     + 450, - 500     + 500, - 500
     + 500, - 500     + 450, - 550     + 450, - 550     + 450, - 500
     + 450, - 550     + 450, - 550     + 450, - 500     + 500, - 500
     + 450, - 550     + 450, - 500     + 500, -1500     + 450, -1500
     + 500, -2950     +2950
unsigned int  rawData[115] = {450,550, 450,1500, 500,500, 450,550, 450,500, 500,500, 450,550, 450,500, 500,500, 500,1450, 500,500, 500,500, 450,1500, 500,1500, 450,550, 450,1500, 450,1500, 500,1500, 450,1500, 500,1450, 500,500, 500,500, 450,500, 500,500, 500,500, 450,500, 500,500, 500,500, 450,550, 450,550, 450,500, 450,550, 450,550, 450,500, 500,500, 450,550, 450,500, 500,500, 500,500, 450,500, 500,500, 500,500, 450,500, 500,500, 500,500, 450,550, 450,550, 450,500, 450,550, 450,550, 450,500, 500,500, 450,550, 450,500, 500,1500, 450,1500, 500,2950, 2950};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

I will be thankful if you can give me any help or suggestion.

crankyoldgit, 2019-07-01:
Hi,
 I'm sorry it didn't work for you. However, you have reminded me I needed to update that example code.

Can you please try this code instead, and let me know if it works okay for you?

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Samsung.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRSamsungAc ac(kIrLed);     // Set the GPIO used for sending messages.

void printState() {
  // Display the settings.
  Serial.println("Samsung A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
}

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_Samsung.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting initial state for A/C.");
  ac.off();
  ac.setFan(kSamsungAcFanLow);
  ac.setMode(kSamsungAcCool);
  ac.setTemp(25);
  ac.setSwing(false);
  printState();
}

void loop() {
  // Turn the A/C unit on
  Serial.println("Turn on the A/C ...");
  ac.on();
  ac.send();
  printState();
  delay(15000);  // wait 15 seconds
  // and set to cooling mode.
  Serial.println("Set the A/C mode to cooling ...");
  ac.setMode(kSamsungAcCool);
  ac.send();
  printState();
  delay(15000);  // wait 15 seconds

  // Increase the fan speed.
  Serial.println("Set the fan to high and the swing on ...");
  ac.setFan(kSamsungAcFanHigh);
  ac.setSwing(true);
  ac.send();
  printState();
  delay(15000);

  // Change to Fan mode, lower the speed, and stop the swing.
  Serial.println("Set the A/C to fan only with a low speed, & no swing ...");
  ac.setSwing(false);
  ac.setMode(kSamsungAcFan);
  ac.setFan(kSamsungAcFanLow);
  ac.send();
  printState();
  delay(15000);

  // Turn the A/C unit off.
  Serial.println("Turn off the A/C ...");
  ac.off();
  ac.send();
  printState();
  delay(15000);  // wait 15 seconds
}
```

crankyoldgit, 2019-07-01:
> if it can be usefull I captured the code sent by the original IR pilot using IRrecvDumpV2 on arduino UNO. this is the result.

Unfortunately, `IRrecvDumpV2 on arduino UNO`  means you are using the https://github.com/z3t0/Arduino-IRremote version of the library. That library has problems capturing large A/C messages. I urge you to use [this library](https://github.com/markszabo/IRremoteESP8266) instead using an ESP device to capture the signal as it is more functional and capable.

I suspect that what you have reported/recorded is part of a Samsung A/C that is too big to fit or parts were missing due to the Arduino. i.e. a poor capture. Try again with this library and an ESP device.

fabriors73, 2019-07-01:
> 
> 
> Hi,
> I'm sorry it didn't work for you. However, you have reminded me I needed to update that example code.
> 
> Can you please try this code instead, and let me know if it works okay for you?
> 
> ```

Hello, thank you for your quick reply.
 I did change the code sending only as.send(); anyway the result does not change.
I can hear 2 "beep". It seems that the AC is switched off and immediately on again.

Identical when powering on I can hear 2 "beep" instead of one.

fabriors73, 2019-07-01:
> Unfortunately, `IRrecvDumpV2 on arduino UNO` means you are using the https://github.com/z3t0/Arduino-IRremote version of the library. That library has problems capturing large A/C messages. I urge [...]

I again captured the code by using the NODEMCU board. 
In fact the code from arduino is partial the new one is this:

Timestamp : 000015.801
Encoding  : SAMSUNG_AC
Code      : 02B20F000000C001D20F0000000001F2FE719011C0 (168 bits)
Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 25C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off
Library   : v2.6.2

Raw Timing[349]:
   +   588, - 17834,    +  3004, -  8956,    +   574, -   440,    +   478, -  1510, 
   +   482, -   512,    +   482, -   512,    +   482, -   512,    +   482, -   512, 
   +   482, -   512,    +   482, -   510,    +   482, -   512,    +   482, -  1488, 
   +   500, -   514,    +   480, -   512,    +   480, -  1532,    +   456, -  1532, 
   +   546, -   448,    +   472, -  1514,    +   478, -  1510,    +   482, -  1506, 
   +   482, -  1504,    +   482, -  1506,    +   482, -   512,    +   480, -   514, 
   +   482, -   512,    +   478, -   514,    +   480, -   514,    +   478, -   514, 
   +   480, -   538,    +   454, -   540,    +   450, -   522,    +   470, -   544, 
   +   576, -   418,    +   478, -   516,    +   480, -   514,    +   480, -   514, 
   +   480, -   512,    +   482, -   512,    +   482, -   512,    +   482, -   490, 
   +   502, -   492,    +   502, -   492,    +   500, -   512,    +   482, -   512, 
   +   480, -   514,    +   480, -   512,    +   480, -   538,    +   454, -   540, 
   +   448, -   544,    +   520, -   474,    +   522, -   472,    +   554, -   440, 
   +   480, -   512,    +   482, -   512,    +   482, -   512,    +   482, -   512, 
   +   482, -  1504,    +   482, -  1508,    +   480, -  2988,    +  2978, -  8958, 
   +   500, -  1508,    +   480, -   512,    +   482, -   538,    +   456, -   538, 
   +   448, -   544,    +   548, -   446,    +   476, -   518,    +   478, -   516, 
   +   480, -   512,    +   482, -  1506,    +   482, -   510,    +   482, -   510, 
   +   482, -  1506,    +   482, -   512,    +   482, -  1508,    +   480, -  1490, 
   +   496, -  1508,    +   480, -  1534,    +   540, -  1448,    +   472, -  1516, 
   +   478, -   514,    +   480, -   514,    +   482, -   510,    +   482, -   512, 
   +   482, -   492,    +   502, -   490,    +   502, -   510,    +   482, -   512, 
   +   482, -   514,    +   480, -   514,    +   480, -   512,    +   480, -   514, 
   +   480, -   512,    +   480, -   540,    +   452, -   518,    +   472, -   522, 
   +   570, -   444,    +   476, -   518,    +   480, -   492,    +   502, -   512, 
   +   482, -   512,    +   484, -   510,    +   482, -   512,    +   482, -   514, 
   +   480, -   512,    +   480, -   512,    +   482, -   514,    +   480, -   514, 
   +   478, -   516,    +   478, -   518,    +   476, -   514,    +   478, -   540, 
   +   454, -   538,    +   450, -   544,    +   452, -   542,    +   472, -   520, 
   +   480, -  2986,    +  2980, -  8956,    +   500, -  1508,    +   482, -   492, 
   +   500, -   514,    +   480, -   512,    +   482, -   512,    +   480, -   514, 
   +   480, -   514,    +   480, -   514,    +   480, -   514,    +   480, -  1506, 
   +   480, -   540,    +   508, -   484,    +   452, -  1536,    +   472, -  1514, 
   +   478, -  1508,    +   482, -  1506,    +   480, -   512,    +   482, -  1506, 
   +   482, -  1506,    +   482, -  1506,    +   482, -  1506,    +   482, -  1530, 
   +   456, -  1532,    +   450, -  1538,    +   472, -  1514,    +   478, -   516, 
   +   480, -   514,    +   482, -   510,    +   482, -  1506,    +   482, -  1504, 
   +   482, -  1506,    +   482, -   512,    +   482, -   512,    +   480, -   512, 
   +   480, -   514,    +   480, -   508,    +   486, -  1532,    +   450, -   542, 
   +   452, -   542,    +   472, -  1514,    +   480, -  1508,    +   482, -   512, 
   +   482, -   510,    +   484, -   510,    +   482, -  1506,    +   482, -   510, 
   +   484, -   512,    +   480, -   514,    +   480, -   514,    +   480, -   512, 
   +   480, -   512,    +   480, -   538,    +   454, -   538,    +   452, -   542, 
   +   452, -  1536,    +   552, -  1434,    +   484

uint16_t rawData[349] = {588, 17834,  3004, 8956,  574, 440,  478, 1510,  482, 512,  482, 512,  482, 512,  482, 512,  482, 512,  482, 510,  482, 512,  482, 1488,  500, 514,  480, 512,  480, 1532,  456, 1532,  546, 448,  472, 1514,  478, 1510,  482, 1506,  482, 1504,  482, 1506,  482, 512,  480, 514,  482, 512,  478, 514,  480, 514,  478, 514,  480, 538,  454, 540,  450, 522,  470, 544,  576, 418,  478, 516,  480, 514,  480, 514,  480, 512,  482, 512,  482, 512,  482, 490,  502, 492,  502, 492,  500, 512,  482, 512,  480, 514,  480, 512,  480, 538,  454, 540,  448, 544,  520, 474,  522, 472,  554, 440,  480, 512,  482, 512,  482, 512,  482, 512,  482, 1504,  482, 1508,  480, 2988,  2978, 8958,  500, 1508,  480, 512,  482, 538,  456, 538,  448, 544,  548, 446,  476, 518,  478, 516,  480, 512,  482, 1506,  482, 510,  482, 510,  482, 1506,  482, 512,  482, 1508,  480, 1490,  496, 1508,  480, 1534,  540, 1448,  472, 1516,  478, 514,  480, 514,  482, 510,  482, 512,  482, 492,  502, 490,  502, 510,  482, 512,  482, 514,  480, 514,  480, 512,  480, 514,  480, 512,  480, 540,  452, 518,  472, 522,  570, 444,  476, 518,  480, 492,  502, 512,  482, 512,  484, 510,  482, 512,  482, 5 [... cut ...] 4,  482, 510,  482, 1506,  482, 1504,  482, 1506,  482, 512,  482, 512,  480, 512,  480, 514,  480, 508,  486, 1532,  450, 542,  452, 542,  472, 1514,  480, 1508,  482, 512,  482, 510,  484, 510,  482, 1506,  482, 510,  484, 512,  480, 514,  480, 514,  480, 512,  480, 512,  480, 538,  454, 538,  452, 542,  452, 1536,  552, 1434,  484};  // SAMSUNG_AC
uint8_t state[21] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0xF2, 0xFE, 0x71, 0x90, 0x11, 0xC0};

crankyoldgit, 2019-07-02:
> Hello, thank you for your quick reply.
> I did change the code sending only as.send(); anyway the result does not change.
> I can hear 2 "beep". It seems that the AC is switched off and immediately on [...]

Thanks for the feedback. I'll look into it. Yes. The two beeps is expected. Samsung uses a special "extended" length message for powering on and off the AC. When the library detects a user wants to change the power state, it tries to use a pre-canned on or off message depending, and then it sends what it expects the state to be anyway.

I'll look at your supplied data shortly and see if I can work out what is going on.

crankyoldgit, 2019-07-02:
> I again captured the code by using the NODEMCU board.
> In fact the code from arduino is partial the new one is this:
> 
> Timestamp : 000015.801
> Encoding : SAMSUNG_AC
> Code : 02B20F000000C001D20F0000000001F2FE719011C0 (168 bits)

Thanks. Assuming you captured that from the actual remote, then that confirms the library is sending the correct special _off_ message as it is exactly the same as stored in the [sendOff()](https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Samsung.cpp#L393) function.

The problem is probably that when the A/C sees a normal commands following an off command, it turns back on, which I thought it wouldn't if it's power bits were set to off too.

I've modified the library in the [Issue791 branch](https://github.com/markszabo/IRremoteESP8266/tree/Issue791) such that when it detects it's powering off the A/C, it _only_ sends the extended message (that you captured) and nothing else. Hopefully that will fix it.

Can you please try out the the [Issue791 branch](https://github.com/markszabo/IRremoteESP8266/tree/Issue791) and tell me if it fixes it for you?
https://github.com/markszabo/IRremoteESP8266/tree/Issue791

fabriors73, 2019-07-02:
Hi, I introduced the modification you suggested in the library and the off command is working correctly.
Many thanks for your help.

fabriors73, 2019-07-02:
By the way, just for info, following your idea, I removed the sendOn() command in case of power on and just left the sendSamsungAC() command. 
In my case this is enough to switch on the AC.

crankyoldgit, 2019-07-03:
Excellent. I'm glad it's working for you now. Thanks for the extra feedback on the power on situation.
I'll still leave it there just in case it is needed for other models.

BTW. Can you please let me know if "AR12HSSDBWKNEU" is the model number for the A/C unit or the remote. If it's the unit, can you also tell me the remote's model number if you can find it. So I can add it to the list of supported devices etc.

closed 2019-07-03 by crankyoldgit

crankyoldgit, 2019-07-04:
The changes mentioned thus far have been included in the new v2.6.3 release of the library.

fabriors73, 2019-07-05:
Hello,
sorry I've seen the mail just now.
AR12HSSDBWKNEU" is the model number for the A/C unit . I Have not found any
number on the remote.
BR
Fabrizio

PR #792 (2019-07-03): Change `ControlSamsungAC` example to not use `sendExtended()`
* Don't send normal state when powering off.

Fixes #791
commit e9385e11 (2019-07-03): Change `ControlSamsungAC` example to not use `sendExtended()` (#792)

* Change `ControlSamsungAC` example to not use `sendExtended()`
* Don't send normal state when powering off.
* Add AR12HSSDBWKNEU to supported models.

Fixes #791