crankyoldgit/IRremoteESP8266 #794: IRremoteESP8266

yunxiang521, 2019-07-02:
crankyoldgit 你好：
      前几天看到你的IRremoteESP8266库很感兴趣，可在操作开启家里的美的空调时无法使用。型号是KFR-26GW/WCBN8A3 也是产品太新的缘故吧，我使用朋友家里老旧的格力型号空调。很好的开启了。效果非常不错。我在使用IRrecvDump接收KFR-26GW/WCBN8A3遥控器的电平延时方法，可是接收到的数值都是高达5位6位的数据，延时应该没有这么高吧，不知道那里出了问题，特向你请教。

crankyoldgit, 2019-07-02:
Google Translate's conversion:

> Crankyoldgit Hello:
I was very interested in seeing your IRremoteESP8266 library a few days ago, but it can't be used when operating the beautiful air conditioner at home. The model is KFR-26GW/WCBN8A3. It is also a reason why the product is too new. I use the old Gree air conditioner from my friend's house. Very good to open. The effect is very good. I am using IRrecvDump to receive the level delay method of KFR-26GW/WCBN8A3 remote control, but the received values ​​are up to 5 digits and 6 digits of data. The delay should not be so high. I don't know if there is a problem. Ask your advice.

yunxiang521, 2019-07-02:
I also use Google English conversion, I am very sorry that my English ability is really limited, but I will try to understand. Air conditioning is Midea's

yunxiang521, 2019-07-02:
250000901 (35 bits)
Raw (74): {8970, 4404678, 1662624, 534650, 534648, 536652, 534648, 536648, 534652, 532650, 1662650, 512674, 510676, 1636650, 510672, 512676, 508670, 512672, 514672, 510672, 512674, 508670, 514670, 514674, 510668, 516670, 514672, 512670, 514672, 510670, 1620670, 512672, 1640646, 516668, 516670, 1642620, 542670};
DEPRECATED: Please use IRrecvDumpV2.ino instead!
This is the data I captured, and the values are very large. Use is invalid.

crankyoldgit, 2019-07-02:
@yunxiang521 
English:
Unfortunately, I don't understand Chinese. Sadly Google Translate isn't doing a great job of explaining the nuance of your question. Especially the last half of your message. It's not very understandable.

I think you are using the `IRrecvDump` program, I strongly suggest you use the [IRrecvDumpV2](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) program instead.

Chinese (via Google Translate):
不幸的是，我不懂中文。遗憾的是，谷歌翻译在解释你问题的细微差别方面做得并不好。特别是你邮件的后半部分。这不是很容易理解。

我认为您正在使用“IRrecvDump”程序，我强烈建议您使用“IRrecvDumpV2”程序。

crankyoldgit, 2019-07-02:
> DEPRECATED: Please use IRrecvDumpV2.ino instead!

弃用：请改用IRrecvDumpV2.ino！

crankyoldgit, 2019-07-02:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDump/IRrecvDump.ino#L5

"""
****已弃用 - 请勿使用*****

除非你知道自己在做什么，否则你应该使用IRrecvDumpV2.ino草图/示例代替捕获和解码IR消息。
几乎所有方面都更好，更快，更详细。

此代码仅用于遗留原因，并作为如何使用IRremoteESP8266库的另一个简单示例。

截至2017年11月，将不再更新或支持。
你被警告了。

****已弃用 - 请勿使用*****
"""

yunxiang521, 2019-07-02:
Looking at your first reply, I found that you don't know Chinese, which is really an embarrassing thing. According to your method, I used IRrecvDumpV2.ino to get a positive solution.

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed =D5;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[73] = {8968, 4432,  644, 1622,  666, 516,  670, 514,  672, 1616,  694, 516,  666, 518,  670, 514,  642, 542,  646, 1666,  644, 518,  670, 514,  666, 1646,  642, 518,  668, 518,  668, 514,  668, 516,  668, 516,  668, 516,  672, 512,  668, 516,  670, 514,  668, 516,  670, 514,  648, 534,  668, 516,  672, 512,  670, 514,  672, 512,  674, 1640,  644, 516,  672, 1640,  642, 518,  672, 512,  668, 1644,  648, 512,  668};
uint16_t rawDataa[73] = {8966, 4402,  676, 1640,  650, 534,  650, 532,  652, 532,  650, 532,  652, 534,  650, 534,  650, 532,  652, 1638,  676, 508,  676, 508,  676, 1614,  676, 506,  676, 506,  678, 506,  676, 508,  676, 508,  676, 508,  676, 506,  676, 508,  676, 506,  676, 508,  678, 506,  676, 506,  678, 506,  676, 508,  676, 508,  676, 508,  676, 1612,  678, 506,  676, 1614,  678, 506,  676, 508,  676, 1614,  676, 506,  676};
// Example Samsung A/C state captured from IRrecvDumpV2.ino
uint8_t samsungState[kSamsungAcStateLength] = {
    0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0,
    0x01, 0xE2, 0xFE, 0x71, 0x40, 0x11, 0xF0};

void setup() {
  irsend.begin();
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);

}

void loop() {
  irsend.sendRaw(rawData, 73, 38);  // Send a raw data capture at 38kHz.
  delay(3000);
    irsend.sendRaw(rawDataa, 73, 38);  // Send a raw data capture at 38kHz.
  delay(3000);
}

Because there is only Gree air conditioning around, this is a value of Gree. When I get home, I will use Midea's empty debugging test. Thank you very much for your reply.

crankyoldgit, 2019-07-02:
FYI

uint16_t rawData[73] = {8968, 4432, 644, 1622, 666, 516, 670, 514, 672, 1616, 694, 516, 666, 518, 670, 514, 642, 542, 646, 1666, 644, 518, 670, 514, 666, 1646, 642, 518, 668, 518, 668, 514, 668, 516, 668, 516, 668, 516, 672, 512, 668, 516, 670, 514, 668, 516, 670, 514, 648, 534, 668, 516, 672, 512, 670, 514, 672, 512, 674, 1640, 644, 516, 672, 1640, 642, 518, 672, 512, 668, 1644, 648, 512, 668};
is
`irsend.sendTeco(0x250000909);`

uint16_t rawDataa[73] = {8966, 4402, 676, 1640, 650, 534, 650, 532, 652, 532, 650, 532, 652, 534, 650, 534, 650, 532, 652, 1638, 676, 508, 676, 508, 676, 1614, 676, 506, 676, 506, 678, 506, 676, 508, 676, 508, 676, 508, 676, 506, 676, 508, 676, 506, 676, 508, 678, 506, 676, 506, 678, 506, 676, 508, 676, 508, 676, 508, 676, 1612, 678, 506, 676, 1614, 678, 506, 676, 508, 676, 1614, 676, 506, 676};
is
`irsend.sendTeco(0x250000901);`

yunxiang521, 2019-07-04:
Thank you very much for your data, which can be used in my air conditioning. The effect is very good. Thank you

crankyoldgit, 2019-07-05:
Good luck with your air conditioner.

closed 2019-07-05 by crankyoldgit