crankyoldgit/IRremoteESP8266 #796: Storing the content of rawData only

matbks, 2019-07-03:
Hi,

I want to read an ir signal and save the raw data in firebase.

I am only interested on this part of the result:
uint16_t rawData[41] = **{2694, 864,  498, 840,  502, 368...}**

How can i get only this part? not the whole results

if i use 
`void loop() {
  // Check if the IR code has been received.
    if (irrecv.decode(&results)) {
    // Output the results as source code
    Serial.println(resultToSourceCode(&results));
    Serial.println("");  // Blank line between entries
    yield();             // Feed the WDT (again)*/
  }
}`

i am gonna get

`uint16_t rawData[41] = {2722, 840,  498, 840,  472, 418,  498, 428,  442, 850,  920, 428,  472, 418,  472, 418,  502, 422,  442, 418,  472, 422,  498, 422,  442, 418,  502, 424,  468, 424,  468, 424,  472, 424,  886, 424,  498, 840,  472, 422,  472};  // RC6 C
uint32_t address = 0x0;
uint32_t command = 0xC;
uint64_t data = 0xC;`

and i just want to save the first part of it.

crankyoldgit, 2019-07-03:
See: 
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRutils.cpp#L717
https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRutils.cpp#L565

and finally see: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

crankyoldgit, 2019-07-03:
Xref #500

closed 2019-07-03 by crankyoldgit

matbks, 2019-07-03:
Thank you.
I edited the IRUtils.cpp , just toke off all the stuff i didn't want from the resultToSourceCode(&results) function.