crankyoldgit/IRremoteESP8266 #800: IRutils.cpp htmlEscape function

jimmys01, 2019-07-04:
Hello,
https://github.com/markszabo/IRremoteESP8266/blob/6313a7098443e1a7732d0b077165b0a6c91fa7ee/src/IRutils.cpp#L243
htmlEscape in IRutils.cpp is a bit to specific function for use in a library for IR don't you think? (only usefull in one example code) Perhaps you can move it there?
It interferes with https://github.com/letscontrolit/ESPEasy/blob/a8ab3bcd7c5ea24c85abd48a64a2d24fa8045f81/src/StringConverter.ino#L419
I know this is not your issue per se or whats is appropriate, the change to be made here or in ESPeasy code. I was hopping for your input. 
Thank you :-)

crankyoldgit, 2019-07-04:
Hey, I like your implementation of it better than mine. ;)
I'll move it to the `namespace IRutils` so it shouldn't conflict in future releases.
It had been in the example code previously, and I moved it out because I wanted to make the example code simpler to understand, plus I can't unit test the code in the example code. Hence why it was moved out of the examples. i.e. the more I can move out of the examples, the more I can have under tests.

crankyoldgit, 2019-07-04:
@jimmys01 If you care to try out #801 It should probably fix your problem. Please let me know.

FYI, I tried your project's routine, but you can't replace "&" and ";" correctly with your project's approach.
e.g. `"&" `-> `"&amp;" `-> `"&amp&semi;"` when it should go to just `"&amp;"`.
Yes, I know you don't neuter `';'`s in your routines, but to be XSS-safe I believe that's a char that needs to be escaped.
I got halfway through converting mine over when I remembered why I did it that way in the first place. :-/

jimmys01, 2019-07-04:
Ok I will try it out tommorow

closed 2019-07-05 by crankyoldgit

jimmys01, 2019-07-05:
Works like a charm

crankyoldgit, 2019-07-26:
FYI, the code changes thus far have been included in the v2.6.4 release of the library.

PR #801 (2019-07-05): Move `htmlEscape()` to the IRutils namespace
Fixes #800
commit 86f17342 (2019-07-05): Move `htmlEscape()` to the IRutils namespace (#801)

Fixes #800