crankyoldgit/IRremoteESP8266 #802: whirlpool ac library not compiling

jenish-jain, 2019-07-05:
when i tried using the Ir library for whirlpool ac after making the desired modification it gives me this error and is not compiling.

this is the error i am getting on compiling

Archiving built core (caching) in: C:\Users\Admin\AppData\Local\Temp\arduino_cache_61975\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_8b68dfc3b7466d55e02a0a5b9f4b639e.a
sketch\whirpool_Whirpool.ino.cpp.o: In function `printState()':

C:\Users\Admin\Desktop\whirpool_Whirpool/whirpool_Whirpool.ino:44: undefined reference to `IRWhirlpoolAc::on()'

sketch\whirpool_Whirpool.ino.cpp.o: In function `setup':

C:\Users\Admin\Desktop\whirpool_Whirpool/whirpool_Whirpool.ino:49: undefined reference to `IRWhirlpoolAc::on()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

crankyoldgit, 2019-07-05:
Can you point to or show us your code for `whirpool_Whirpool.ino`? With out it we probably can't help you.

jenish-jain, 2019-07-05:
this is my code

/* Copyright 2017, 2018 David Conran
*
* An IR LED circuit *MUST* be connected to the ESP8266 on a pin
* as specified by kIrLed below.
*
* TL;DR: The IR LED needs to be driven by a transistor for a good result.
*
* Suggested circuit:
*     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
*
* Common mistakes & tips:
*   * Don't just connect the IR LED directly to the pin, it won't
*     have enough current to drive the IR LED effectively.
*   * Make sure you have the IR LED polarity correct.
*     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
*   * Typical digital camera/phones can be used to see if the IR LED is flashed.
*     Replace the IR LED with a normal LED if you don't have a digital camera
*     when debugging.
*   * Avoid using the following pins unless you really know what you are doing:
*     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
*     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
*     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
*   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
*     for your first time. e.g. ESP-12 etc.
*/
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Whirlpool.h>
int temp;
const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRWhirlpoolAc ac(kIrLed);  // Set the GPIO used for sending messages.

void printState() {
  // Display the settings.
  Serial.println("Whirlpool A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
  // Display the encoded IR sequence.
  unsigned char* ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  for (uint8_t i = 0; i < kWhirlpoolAcStateLength; i++)
    Serial.printf("%02X", ir_code[i]);
  Serial.println();
}

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_Mitsubishi.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting desired state for A/C.");
//  ac.on();
  ac.setFan(1);
  ac.setMode(kWhirlpoolAcCool);
  ac.setTemp(26);
  ac.setFan(kWhirlpoolAcFanAuto);
}

void loop() {
  // Now send the IR signal.
#if SEND_WHIRLPOOL_AC
  Serial.println("Sending IR command to A/C ...");
  for(temp = 16; temp <= 23; temp ++)
  {
    ac.setTemp(temp);
    ac.send();
#endif  // SEND_MITSUBISHI_AC
  printState();
  
  delay(5000);
  }
}

crankyoldgit, 2019-07-05:
Ah. The `IRWhirlpoolAc` class/code does NOT have an `on()`/`off()`/`setPower()`. It's an odd protocol unfortunately.
You need to use `setPowerToggle()` See: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Whirlpool.cpp#L340

Whirlpool A/Cs don't seem to have a simple on/off function / bit in there message. They seem to use a system where if a certain bit is set in the message (i.e. via `ac.setPowerToggle(true);` it will cause the A/C unit to go from off->on or on->off. So you will need to manage that yourself by only setting `ac.setPowerToggle(true);` when you want the A/C to switch its power state, otherwise keep it `ac.setPowerToggle(false);`

jenish-jain, 2019-07-05:
ok, thanks.

jenish-jain, 2019-07-05:
thanks for the support, by the way, do you decode all the AC companies IR protocol and develop the algorithms for them or are some of them available on the companies website. (i mean the decoding algorithms)
jenish jain

crankyoldgit, 2019-07-05:
> or are some of them available on the companies website.

Hahahahaha .. I wish. [end sarcasm]

Sadly, no. They (aircon protocols) are almost all reverse engineered by some poor soul.

Some the more simple (think TV remotes etc) protocols do have published specification, but ~80% don't.
e.g. NEC, Lego, ...

jenish-jain, 2019-07-05:
Haha ok . I thought I was dumb enough to not find them any where in the manuals or documentations.
Thanks a lot for the good work😆 you have saved many hours of hardwork for everyone.

crankyoldgit, 2019-07-05:
Thanks, and good luck with your sketch/project.

closed 2019-07-05 by crankyoldgit