crankyoldgit/IRremoteESP8266 #803: [ESP32] Problems using IRremote and Hardware-based PWM at the same time

MorpheusDe97, 2019-07-06:
### Version/revision of the library used
v2.6.3

### Actual behavior
When I use the bare library, everything works as expected. But as soon, as I activate some hardware-based PWM on a completely other pin to dim a LED, this library does not work as expected anymore. The IR-LED just seems to light continuously. 

### Steps to reproduce the behavior
I have the following setup: 
ESP32 with 

- an IR-LED (driven by an NPN transistor) on PIN 32
- an RGB-LED on Pins 18, 19 and 21

As soon, as I activate the PWM on Pins 18, 19 and 21 using 
`	ledcSetup(PWM_CHANNEL_RED, PWM_FREQUENCY, PWM_RESOUTION);
	ledcSetup(PWM_CHANNEL_BLUE, PWM_FREQUENCY, PWM_RESOUTION);
	ledcSetup(PWM_CHANNEL_GREEN, PWM_FREQUENCY, PWM_RESOUTION);
	ledcAttachPin(LED_RED_PIN, PWM_CHANNEL_RED);
	ledcAttachPin(LED_BLUE_PIN, PWM_CHANNEL_BLUE);
	ledcAttachPin(LED_GREEN_PIN, PWM_CHANNEL_GREEN);
	ledcWrite(PWM_CHANNEL_RED, HighRedDutyCycle);
	ledcWrite(PWM_CHANNEL_GREEN, HighDutyCycle);
	ledcWrite(PWM_CHANNEL_BLUE, HighDutyCycle);
`
I can see the infrared LED sending continiously. 
I know that this library is timerInterrupt-based. But is there the possibility, or a known issue that there may appear unwanted interferences between using hardware-PWM and Infrared-sending?

crankyoldgit, 2019-07-07:
Actually, the library doesn't use any of the Interrupts or Timers on the _sending_ side. They are only utilised for _receiving_. Thus, I think your problem may be more fundamental.

There are no known issues with sending on the ESP32 platform, but ESP32 support is only new so there is a possibility you are the first to encounter it.

Re: https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal

1. How certain are you your hardware circuit is correct? The LED should most certainly be _off_ almost all of the time.
2. Have you used the simple sending [example code](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino)? Does it work as expected?
3. Are you invoking the library correctly? e.g. correct pin. Calling `irsend.begin()` before you send the first time etc.
4. What troubleshooting steps have you taken?
5. Can you simplify your code to the smallest possible example that demonstrates your problem and post it here?

crankyoldgit, 2019-07-09:
Chasing up on this. Any progress?

crankyoldgit, 2019-07-11:
Another friendly chase up.

MorpheusDe97, 2019-07-11:
Hello, sorry for the late reply! 
Using a completely new Arduino project running just your example-code and adding the pwm-functions to it, everything works like a charm. Probably, there was any kind of different problem as I directly tried to embed your library into a quite big project. I'll dig a bit deeper into the topic, but currently, it seems like the issue I came across is **not caused by your library**! 
I'll keep you updated as soon as I know something new.

crankyoldgit, 2019-07-11:
**Phew!** I'll mark this issue closed for now then.

closed 2019-07-11 by crankyoldgit