crankyoldgit/IRremoteESP8266 #804: Problem with Sending Daikin data

ilyasrois, 2019-07-07:
How to send data using irsend.sendDaikin. Because I want to send the data that I've recorded using example IRrecvDumpV2. In this case I want to control Daikin AC.

here my data : 
Timestamp : 003660.810
Library   : v2.6.3

Encoding  : DAIKIN160
Code      : 11DA27F00D000F11DA2700D331F100001C0A002D (160 bits)
Mesg Desc.: Power: On, Mode: 3 (COOL), Temp: 24C, Fan: 10 (AUTO), Vent Position (V): 15 (Auto)
Raw Timing[327]:
   +  4926, -  2092,    +   318, -  1784,    +   342, -   632,    +   362, -   634, 
   +   364, -   632,    +   366, -  1758,    +   342, -   634,    +   336, -   660, 
   +   364, -   634,    +   362, -   632,    +   364, -  1760,    +   344, -   632, 
   +   362, -  1758,    +   344, -  1758,    +   340, -   634,    +   364, -  1758, 
   +   342, -  1758,    +   342, -  1760,    +   342, -  1758,    +   342, -  1758, 
   +   344, -   634,    +   362, -   634,    +   362, -  1758,    +   342, -   634, 
   +   362, -   634,    +   364, -   634,    +   362, -   634,    +   362, -   634, 
   +   362, -   634,    +   364, -  1758,    +   342, -  1760,    +   344, -  1756, 
   +   342, -  1758,    +   344, -  1758,    +   344, -   630,    +   362, -  1760, 
   +   342, -  1760,    +   342, -   634,    +   362, -   634,    +   362, -   634, 
   +   362, -   634,    +   362, -   634,    +   364, -   634,    +   362, -   634, 
   +   362, -   634,    +   362, -   636,    +   362, -   634,    +   362, -   634, 
   +   362, -   632,    +   364, -  1760,    +   342, -  1758,    +   342, -  1760, 
   +   340, -  1760,    +   342, -   632,    +   362, -   634,    +   362, -   634, 
   +   362, -   634,    +   364, - 28570,    +  4952, -  2066,    +   346, -  1754, 
   +   342, -   632,    +   294, -   704,    +   366, -   630,    +   366, -  1756, 
   +   342, -   632,    +   364, -   632,    +   294, -   704,    +   336, -   660, 
   +   366, -  1756,    +   342, -   632,    +   362, -  1760,    +   316, -  1784, 
   +   348, -   628,    +   364, -  1756,    +   316, -  1786,    +   344, -  1756, 
   +   342, -  1758,    +   290, -  1812,    +   346, -   630,    +   366, -   628, 
   +   364, -  1758,    +   290, -   688,    +   366, -   630,    +   366, -   630, 
   +   366, -   630,    +   362, -   632,    +   364, -   634,    +   292, -   704, 
   +   292, -   704,    +   366, -   630,    +   368, -   628,    +   366, -  1754, 
   +   318, -  1784,    +   346, -   630,    +   366, -   630,    +   366, -  1754, 
   +   290, -   688,    +   292, -  1830,    +   346, -  1754,    +   342, -  1760, 
   +   314, -   660,    +   366, -   630,    +   368, -   628,    +   368, -  1754, 
   +   290, -  1812,    +   344, -   630,    +   366, -   630,    +   366, -  1754, 
   +   290, -   686,    +   292, -   704,    +   366, -   630,    +   368, -  1754, 
   +   342, -  1760,    +   288, -  1810,    +   346, -  1754,    +   344, -   632, 
   +   294, -   704,    +   292, -   706,    +   364, -   630,    +   368, -   628, 
   +   368, -   628,    +   364, -   632,    +   336, -   658,    +   294, -   704, 
   +   366, -   630,    +   364, -   630,    +   368, -   628,    +   364, -   632, 
   +   362, -   632,    +   294, -   704,    +   310, -   686,    +   364, -   632, 
   +   366, -   628,    +   368, -  1756,    +   314, -  1786,    +   344, -  1756, 
   +   344, -   632,    +   362, -   632,    +   294, -   704,    +   292, -   706, 
   +   366, -  1754,    +   346, -   628,    +   364, -  1758,    +   314, -   660, 
   +   366, -   630,    +   366, -   630,    +   368, -   628,    +   364, -   634, 
   +   336, -   658,    +   294, -   706,    +   364, -   632,    +   366, -   630, 
   +   366, -   630,    +   364, -   632,    +   364, -   634,    +   310, -  1812, 
   +   344, -   632,    +   366, -  1754,    +   342, -  1760,    +   318, -   658, 
   +   366, -  1756,    +   344, -   632,    +   364, -   632,    +   292

uint16_t rawData[327] = {4926, 2092,  318, 1784,  342, 632,  362, 634,  364, 632,  366, 1758,  342, 634,  336, 660,  364, 634,  362, 632,  364, 1760,  344, 632,  362, 1758,  344, 1758,  340, 634,  364, 1758,  342, 1758,  342, 1760,  342, 1758,  342, 1758,  344, 634,  362, 634,  362, 1758,  342, 634,  362, 634,  364, 634,  362, 634,  362, 634,  362, 634,  364, 1758,  342, 1760,  344, 1756,  342, 1758,  344, 1758,  344, 630,  362, 1760,  342, 1760,  342, 634,  362, 634,  362, 634,  362, 634,  362, 634,  364, 634,  362, 634,  362, 634,  362, 636,  362, 634,  362, 634,  362, 632,  364, 1760,  342, 1758,  342, 1760,  340, 1760,  342, 632,  362, 634,  362, 634,  362, 634,  364, 28570,  4952, 2066,  346, 1754,  342, 632,  294, 704,  366, 630,  366, 1756,  342, 632,  364, 632,  294, 704,  336, 660,  366, 1756,  342, 632,  362, 1760,  316, 1784,  348, 628,  364, 1756,  316, 1786,  344, 1756,  342, 1758,  290, 1812,  346, 630,  366, 628,  364, 1758,  290, 688,  366, 630,  366, 630,  366, 630,  362, 632,  364, 634,  292, 704,  292, 704,  366, 630,  368, 628,  366, 1754,  318, 1784,  346, 630,  366, 630,  366, 1754,  290, 688,  292, 1830,  346, 1754,  342, 1760,  314, 660,  366, 630,  368, 628,  368, 1754,  290, 1812,  344, 630,  366, 630,  366, 1754,  290, 686,  292, 704,  366, 630,  368, 1754,  342, 1760,  288, 1810,  346, 1754,  344, 632,  294, 704,  292, 706,  364, 630,  368, 628,  368, 628,  364, 632,  336, 658,  294, 704,  366, 630,  364, 630,  368, 628,  364, 632,  362, 632,  294, 704,  310, 686,  364, 632,  366, 628,  368, 1756,  314, 1786,  344, 1756,  344, 632,  362, 632,  294, 704,  292, 706,  366, 1754,  346, 628,  364, 1758,  314, 660,  366, 630,  366, 630,  368, 628,  364, 634,  336, 658,  294, 706,  364, 632,  366, 630,  366, 630,  364, 632,  364, 634,  310, 1812,  344, 632,  366, 1754,  342, 1760,  318, 658,  366, 1756,  344, 632,  364, 632,  292};  // DAIKIN160
uint8_t state[20] = {0x11, 0xDA, 0x27, 0xF0, 0x0D, 0x00, 0x0F, 0x11, 0xDA, 0x27, 0x00, 0xD3, 0x31, 0xF1, 0x00, 0x00, 0x1C, 0x0A, 0x00, 0x2D};

crankyoldgit, 2019-07-07:
```
IRsend irsend(IR_LED);
uint8_t state[20] = {0x11, 0xDA, 0x27, 0xF0, 0x0D, 0x00, 0x0F, 0x11, 0xDA, 0x27, 0x00, 0xD3, 0x31, 0xF1, 0x00, 0x00, 0x1C, 0x0A, 0x00, 0x2D};
irsend.begin();
irsend.sendDaikin160(state);
```

crankyoldgit, 2019-07-09:
How did you go?

ilyasrois, 2019-07-11:
> How did you go?

oh great it worked very well. Thanks a lot for helping me

closed 2019-07-11 by crankyoldgit