crankyoldgit/IRremoteESP8266 #805: Mitsubishi Heavy Ac on example not working

jenish-jain, 2019-07-07:
i tried using the Mitsubishi heavy on the example on my Mitsubishi Heavy duty AC but it isn't working

crankyoldgit, 2019-07-07:
FYI, there are two different types of Mitsubishi Heavy A/Cs supported. 152 & 88. The example code is setup for 152.
Have you determined what variant you have using `IRrecvDumpV2`?

Also, if you want help, you are going to need to provide a lot more information than you have thus far.

jenish-jain, 2019-07-07:
mine is 88 how to set it up for that.

crankyoldgit, 2019-07-07:
As it says in the example code:
```
  // Set up what we want to send. See ir_MitsubishiHeavy.(cpp|h) for all the
  // options.
```

Basically, Change all occurrences of 152 to 88 in the example code. (except in the baud rate of course)

jenish-jain, 2019-07-07:
ok thanks got it.

jenish-jain, 2019-07-07:
in creating libraries for all types of AC you must have created some sort of documentation or record of how many types of AC remotes are their per company or the features associated with each company remote. like some ac supports vertical a well as horizontal swing whereas some do not, can you share that with me? i need it for my project to make a remote interface for all different company remotes.
jenish jain

closed 2019-07-07 by jenish-jain

crankyoldgit, 2019-07-07:
> i need it for my project to make a remote interface for all different company remotes.

That already exists in a number of different projects. For instance, see https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino
Which mostly uses [sendAc()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.h#L42)

For your list of supported features, you'll have to browse the source code here: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp

The closest thing we have to a list of remote etc per company is: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md