crankyoldgit/IRremoteESP8266 #809: Whirpool AC Power Command On/Off instead of Toggle

Hollako, 2019-07-12:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
Used Library is: v2.6.2

### Expected behavior
To be able to send a specified Power command to Whirpool AC On or Off instead to send Power Toggle command. 

e.g.
IRWhirlpoolAc acir (IR_PIN)
and to send the following commands when needed:
- acir.on() 
- acir.off() 
instead of acir.setPowerToggle(1)

Note that when i try to compile the sketch with acir.on() or acir.off()  i get an error during the compiling. Only acir.setPowerToggle(1) is compiled.

### Other useful information
Note that all the commands are working and an Hisence AC and all is working propoerly. Only the Power On/Off commands are needed.

crankyoldgit, 2019-07-13:
Hey @Hollako , unfortunately Whirlpool A/Cs don't appear to have a true on/off setting so I can't simply add it to the library. If they did, I would have added it. They only appear to have a toggle power setting as far as I know (and that's backed up by other users reporting it.)
See https://github.com/crankyoldgit/IRremoteESP8266/issues/802#issuecomment-508718428 for some more detail/confirmation.

Hollako, 2019-07-13:
Well noted !! 

Thank you.

closed 2019-07-13 by Hollako