crankyoldgit/IRremoteESP8266 #810: IRMQTTServer doesn't seem to work reliably when sending full strings

sheppy99, 2019-07-13:
### Version/revision of the library used
v2.6.3

### Expected behavior
AC turns ON and OFF using the same setup I was using yesterday
### Actual behavior
Turned AC on from a Rule as I did before I upgraded by sending a MQTT Message to it
In this case for I sent
53,11DA270001E242F0280C8004B016240000A9C33511DA270000092600A0000006600000C19060F8
This appeared instantly in MQTT.fx on ir_server/send with nothing back on ir_server/sent
One Minute later the AC turned ON
Sometime later I then sent an OFF command 
53,11DA270001F5C2F0280C8004B016240000A9C3C811DA270000082500A0000006600000C19060F6
which didn't turn it OFF.

### Steps to reproduce the behavior
Compile IRMQTTServer, load it onto a Huzzah Board and send a Daikin2 hex string from OpenHAB as I've been doing for months
#### Example code used
see above for hex string

### Has this library/code previously worked as expected for you?
Yes using IRMQTTServer from 07/04/19
### Other useful information
I'm now running the 07/4/19 dated version of IRMQTTServer with the latest core and latest IRremoteESP8266 library and it works OK so this isn't an important fix. I updated to see if the ESP plays nicer with my UniFi AP-AC-Lite without a few dropouts per day. I could also see a few MQTT messages such as ir_server/ac/stat/econo off with the new version in case it proves anything other than I had a newer version.

crankyoldgit, 2019-07-13:
Hi @sheppy99 , Thanks for the report. I recently changed how it handles processing the "older" style a/c strings. So I probably broke something. I'll try to replicate the issue to see if I can work out what I did wrong.

crankyoldgit, 2019-07-13:
Hey Kevin,
 If you can try the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue810 it should fix it for you. A simple "D'oh!" programming bug. If you don't want to copy the entire branch down, it's a simple code change which you could probably hand edit in. See #811 

If you can confirm it's working for you I'll merge it straight away. My testing seems to indicate it's fine for me now.

sheppy99, 2019-07-13:
Hi David,
Thanks for the prompt response, I’ll take a look either tomorrow or Monday and let you know.
Cheers

closed 2019-07-14 by crankyoldgit

sheppy99, 2019-07-15:
Hi David,
I just loaded the code and it seems to work.
On a different note I had 5 disconnections over 2 days with the recompiled older version on the latest 2.5.2. core and latest IR library. I'm guessing this wouldn't work with the older 2.3.0 core?
Curiously enough I recently updated 10 x SOnOff PoW's to the latest 2.5.2 core V6.6.0 Tasmota and 3 of them were throwing regular LWT dropoff's in my logs, downgrading the dodgy 3 to V6.6.0 with the 2.3.0 core fixed the problem yet the other 7 sitting very close by are happy with V2.5.2 - weird

crankyoldgit, 2019-07-15:
Kevin,
 Thanks for the confirmation.

 Yeah, the UniFi APs can be hit and miss with 2.5.2 core. It was disastrous with the 2.4.x versions.
For example:
> Version: v1.3.0
Built: Jul 4 2019 17:59:43
Period Offset: -5us
IR Lib Version: 2.6.3
ESP8266 Core Version: 2_5_2
[...]
> Server: 10.0.0.4:1883 (Connected 10 days 18 hours 27 minutes 3 seconds ago)
> Disconnections: 0

Some times it's like the above, and it's perfect. Sometimes it's a PITA. Dunno.

To answer your question: Yes. IRMQTTServer should totally work with v2.3.0 core. The only downside I am aware of are the known security vulnerabilities in that version of the esp8266 core.

crankyoldgit, 2019-07-26:
FYI, the code changes thus far have been included in the v2.6.4 release of the library.

PR #811 (2019-07-14): [Bug] IRMQTTServer: Fix sending >64 bit codes.
* Off by one error in string pointer. Was pointing to the separator char, not the data part of the string.

Fixes #810
commit 5eb211c0 (2019-07-14): [Bug] IRMQTTServer: Fix sending >64 bit codes. (#811)

* Off by one error in string pointer.

Fixes #810