crankyoldgit/IRremoteESP8266 #814: Thank you for sharing this project. Thank you for your work.

1043717432, 2019-07-15:
Thank you for sharing this project. Thank you for your work.

When A/C uses TTL monitoring, it is found that the remote control a/c opens the switch on and the TTL status shows off.

Ask for help
Timestamp : 002188.163
                      Library   : v2.6.3

Encoding  : GREE
                Code      : 59072050012000C0 (64 bits)
                                                      Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 23C, Fan: 1, Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Auto, Swing Vertical Pos: 1 (Auto)
Raw Timing[139]:
                   +  8928, -  4450,    +   642, -  1642,    +   640, -   554,    +   638, -   554,
                                                                                                       +   640, -  1642,    +   666, -  1616,    +   666, -   530,    +   640, -  1640,
                                                                                                                                                                                           +   640, -   554,    +   640, -  1642,    +   640, -  1642,    +   640, -  1642,
                                                                +   668, -   528,    +   664, -   530,    +   662, -   530,    +   666, -   528,
                                                                                                                                                    +   640, -   554,    +   638, -   554,    +   664, -   528,    +   640, -   554,
                         +   666, -   528,    +   664, -   530,    +   664, -  1616,    +   640, -   554,
                                                                                                             +   638, -   556,    +   640, -   554,    +   666, -   528,    +   664, -   528,
                                                                                                                                                                                                 +   640, -   554,    +   664, -  1616,    +   666, -   528,    +   638, -  1642,
                                                                      +   668, -   528,    +   666, -   528,    +   640, -  1642,    +   666, -   528,
                                                                                                                                                          +   640, - 19850,    +   644, -  1640,    +   642, -   554,    +   638, -   554,
                               +   640, -   554,    +   640, -   554,    +   640, -   554,    +   640, -   554,
                                                                                                                   +   640, -   556,    +   638, -   554,    +   638, -   556,    +   638, -   554,
                                                                                                                                                                                                       +   640, -   554,    +   666, -   530,    +   640, -  1640,    +   640, -   554,
                                                                            +   666, -   528,    +   666, -   528,    +   640, -   554,    +   666, -   528,
                                                                                                                                                                +   638, -   554,    +   640, -   554,    +   664, -   530,    +   666, -   528,
                                     +   640, -   554,    +   640, -   554,    +   640, -   554,    +   640, -   554,
                                                                                                                         +   664, -   530,    +   640, -   554,    +   638, -   554,    +   640, -  1642,
                                                                                                                                                                                                             +   666, -  1616,    +   666

uint16_t rawData[139] = {8928, 4450,  642, 1642,  640, 554,  638, 554,  640, 1642,  666, 1616,  666, 530,  640, 1640,  640, 554,  640, 1642,  640, 1642,  640, 1642,  668, 528,  664, 530,  662, 530,  666, 528,  640, 554,  638, 554,  664, 528,  640, 554,  666, 528,  664, 530,  664, 1616,  640, 554,  638, 556,  640, 554,  666, 528,  664, 528,  640, 554,  664, 1616,  666, 528,  638, 1642,  668, 528,  666, 528,  640, 1642,  666, 528,  640, 19850,  644, 1640,  642, 554,  638, 554,  640, 554,  640, 554,  640, 554,  640, 554,  640, 556,  638, 554,  638, 556,  638, 554,  640, 554,  666, 530,  640, 1640,  640, 554,  666, 528,  666, 528,  640, 554,  666, 528,  638, 554,  640, 554,  664, 530,  666, 528,  640, 554,  640, 554,  640, 554,  640, 554,  664, 530,  640, 554,  638, 554,  640, 1642,  666, 1616,  666};  // GREE
                                                                                                                                                                                                    uint8_t state[8] = {0x59, 0x07, 0x20, 0x50, 0x01, 0x20, 0x00, 0xC0};

 Mesg Desc.: Power: Off, The correct display is ON

1043717432, 2019-07-15:
The state of my A/C remote control operation is ON, and the state of my A/C air conditioner is MODE is ON. In the source code, POWER refers to ON/OFF distinguishing code, not MODE.

crankyoldgit, 2019-07-15:
Hi, Thanks for the report.

I'm confused as to what you mean by:
> When A/C uses TTL monitoring, it is found that the remote control a/c opens the switch on and the TTL status shows off.

What is TTL in this context? https://en.wikipedia.org/wiki/TTL
Can you please elaborate more on what the problem is?

1043717432, 2019-07-15:
[image]
I wrote IRrecv DumpV2 in esp8266 to monitor and control remote operation.

1043717432, 2019-07-15:
The current problem is that the remote control operation of green air conditioner is completely correctly fed back. The only problem is that the power state feedback is off, not the correct on state.

1043717432, 2019-07-15:
=====================When I press the "power" button ON================
Timestamp: 005576.294
Library: v2.6.3
Encoding: GREE
Code: 59072050012000C0 (64 bits)
Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 23C, Fan: 1, Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Auto, Swing Vertical Pos: 1 (Auto)
uint16_t rawData[139] = {8920, 4460,  628, 1654,  630, 564,  630, 564,  630, 1650,  606, 1676,  632, 564,  632, 1648,  630, 564,  630, 1652,  632, 1650,  632, 1652,  632, 562,  634, 560,  630, 564,  628, 564,  628, 566,  630, 564,  604, 590,  630, 564,  630, 562,  632, 562,  628, 1652,  630, 566,  630, 564,  628, 564,  630, 564,  630, 564,  632, 562,  632, 1648,  632, 564,  628, 1652,  632, 562,  632, 564,  632, 1650,  630, 564,  628, 19860,  632, 1652,  632, 562,  630, 564,  630, 564,  630, 564,  630, 562,  630, 564,  628, 564,  630, 564,  630, 564,  630, 564,  628, 564,  630, 564,  630, 1650,  632, 564,  628, 566,  628, 566,  630, 564,  630, 564,  630, 564,  632, 562,  628, 564,  632, 562,  630, 562,  630, 564,  630, 564,  630, 564,  630, 564,  630, 564,  632, 562,  632, 1648,  630, 1650,  632};  // GREE
uint8_t state[8] = {0x59, 0x07, 0x20, 0x50, 0x01, 0x20, 0x00, 0xC0};

=======================When I press the "power" button OFF===================
Timestamp : 005648.523
Library   : v2.6.3
Encoding  : GREE
 Code      : 5107205001200040 (64 bits)
Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 23C, Fan: 1, Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Auto, Swing Vertical Pos: 1 (Auto)
uint16_t rawData[139] = {8928, 4450,  642, 1642,  640, 554,  664, 530,  638, 556,  638, 1642,  640, 556,  638, 1642,  640, 556,  638, 1642,  640, 1642,  640, 1642,  640, 554,  640, 554,  638, 556,  638, 556,  638, 556,  664, 530,  638, 554,  638, 556,  638, 554,  638, 554,  638, 1642,  664, 530,  666, 528,  638, 556,  638, 556,  638, 556,  638, 556,  664, 1616,  640, 556,  664, 1618,  640, 556,  638, 554,  664, 1616,  640, 556,  638, 19852,  640, 1642,  668, 528,  638, 556,  636, 556,  638, 556,  640, 554,  640, 554,  638, 554,  664, 530,  664, 530,  640, 554,  638, 554,  642, 552,  638, 1644,  640, 554,  638, 554,  638, 554,  638, 556,  638, 554,  638, 556,  638, 556,  638, 556,  664, 530,  638, 556,  638, 554,  638, 556,  636, 556,  664, 530,  638, 556,  638, 556,  638, 1642,  640, 554,  638};  // GREE
uint8_t state[8] = {0x51, 0x07, 0x20, 0x50, 0x01, 0x20, 0x00, 0x40};

1043717432, 2019-07-15:
Brand: green

Remote control model: YBOFB2

1043717432, 2019-07-15:
SmartIRRepeater

1043717432, 2019-07-15:
SmartIRRepeater
=======================When I press the "power" button OFF===================
000012.341: A 64-bit GREE message was successfully retransmitted.
=====================When I press the "power" button ON================
000027.031: A 64-bit GREE message was successfully retransmitted.

1043717432, 2019-07-15:
SmartIRRepeater
=======================When I press the "power" button OFF===================

1043717432, 2019-07-15:
SmartIRRepeater
=======================When I press the "power" button OFF===================
30 30 30 37 39 35 2E 36 30 33 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A 
=====================When I press the "power" button ON================
30 30 30 38 31 38 2E 31 39 39 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A

1043717432, 2019-07-15:
After remote control button ON air conditioner a/c state, the wind speed and temperature are adjusted by remote control. Power in Air Conditioner Control is always off state.
[image]

crankyoldgit, 2019-07-15:
HI @1043717432 , previous data indicated the 3rd byte (`state[2]`) of Gree A/C messages was involved with power on/off control.

I've created a branch that has the fix for your particular circumstances, while leaving things compatible with existing systems .. I hope.

Can you please test the branch of the code in https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814 ?

I'll need you to confirm the ESP can power on/off your GREE a/c using the class method (e.g. via the "Aircon" page in the IRMQTTServer example code.)

This all could just be your A/C's codes are subtly different from everyone elses, but fingers crossed this will keep everyone happy.

1043717432, 2019-07-15:
> HI @ 1043717432，先前的数据表明`state[2]`Gree A / C消息的第3个字节（）涉及电源开/关控制。
> 
> 我已经创建了一个分支，它可以针对您的特定情况进行修复，同时保持与现有系统的兼容性。我希望如此。
> 
> 您能否在[https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814中](https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814)测试代码的分支？
> 
> 我需要你确认ESP可以使用类方法打开/关闭你的GREE a / c（例如通过IRMQTTServer示例代码中的“Aircon”页面。）
> 
> 这一切都可能只是你的A / C的代码与每个人的细微差别，但是手指越过这将使每个人都开心。

Thank you. How should I test it?

1043717432, 2019-07-15:
> HI @ 1043717432，先前的数据表明`state[2]`Gree A / C消息的第3个字节（）涉及电源开/关控制。
>
> 我已经创建了一个分支，它可以针对您的特定情况进行修复，同时保持与现有系统的兼容性。我希望如此。 [...]

Thank you. I'll test it right away and give you feedback.

crankyoldgit, 2019-07-15:
If you clone/pull/download that [branch](https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814) and compile using it, IRrecvDumpV2 and IRMQTTServer should now detect the Power on/off correctly, and hopefully be able to control (construct and send new messages) the power functions of your A/C.

crankyoldgit, 2019-07-15:
> SmartIRRepeater
=======================When I press the "power" button OFF===================
30 30 30 37 39 35 2E 36 30 33 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A
=====================When I press the "power" button ON================
30 30 30 38 31 38 2E 31 39 39 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A

No idea where you are getting this output from. `SmartIRRepeater` shouldn't be producing that.

1043717432, 2019-07-15:
> > SmartIRRepeater 
> > =======================当我按下“电源”按钮时================= == 
> > 30 30 30 37 39 35 2E 36 30 33 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A 
> > =====================当我按下“电源”按钮时== ============== 
> > 30 30 30 38 31 38 2E 31 39 39 3A 20 41 20 36 34 2D 62 69 74 20 47 52 45 45 20 6D 65 73 73 61 67 65 20 77 61 73 20 73 75 63 63 65 73 73 66 75 6C 6C 79 20 72 65 74 72 61 6E 73 6D 69 74 74 65 64 2E 0A
> 
> 不知道从哪里得到这个输出。`SmartIRRepeater`不应该那样做。

putty  com

1043717432, 2019-07-15:
Thank you. The 64 different original codes I got by using various device learning buttons are all in ON state.
Thank you. The 64 different original codes I got by using various device learning buttons are all in ON state.

Using your test test test file for compilation, the problem still exists. The ON and OFF button states can not be distinguished clearly, and other functions can be recognized normally.
```
=====================RM3 PRO=====================
Code: 2600920000012794143616121611173515361636141216121513153714121536161215121612151216121612161116111711163516111712161214131511181016351611173516111711173615101700028e1612171016131413141216121513151216111711161117111611183316121512181017101810161116121611161
====================homebridgr rm3======================
JgBOAAABJpURNxETEhISNhMRFBAUNBQQFBATERISFBITERQQFBAUEBISERMRExETEhISNhQQFBAUEBQQFBATERI2ERMRNxETEhISNhMRFAANBQAAAAAAAAAAAAA=```

1043717432, 2019-07-15:
[image]
Code: 59072050012000C0 (64 bits)  
Send boot successfully

crankyoldgit, 2019-07-15:
> putty com

Putty is just a terminal program. I meant, what program is running on the ESP. It is NOT `SmartIRRepeater` as it doesn't produce that sort of output. None of the example code supplied by this library does.

crankyoldgit, 2019-07-15:
> @crankyoldgit Thank you. The 64 different original codes I got by using various device learning buttons are all in ON state.

I don't understand what you are saying above.

> 
> Using your test file for compilation, the problem still exists. The key states ON and OFF cannot be distinguished clearly.

Again, I'm not sure what you are saying here. The tests I added were based on the data you supplied, and due to the changes I made, it should now detect correctly. If it doesn't, then I think you may not be building the binary using the updated library.

> ```
> =====================RM3 PRO=====================
> Code: [...]

I have no idea what this is from. It certainly isn't from something in our library.

crankyoldgit, 2019-07-15:
> [image]
> Code: 59072050012000C0 (64 bits)
> Send boot successfully

No. I meant please try to send an "on" command via the "Aircon" menu/page. Not the home page.
e.g. The page/image you included earlier in: https://github.com/crankyoldgit/IRremoteESP8266/issues/814#issuecomment-511286479

1043717432, 2019-07-15:
> > [image]
> > 代码：59072050012000C0（64位）
> > 成功发送启动
> 
> 不，我的意思是请尝试通过“Aircon”菜单/页面发送“开启”命令。不是主页。
> 例如，您之前包含的页面/图像：[＃814（评论）](https://github.com/crankyoldgit/IRremoteESP8266/issues/814#issuecomment-511286479)

The test failed. Under the Aircon menu, the use of green to send ON failed, the air conditioner has a'beep'sound, but the boot failed.

1043717432, 2019-07-15:
> > [image]
> > 代码：59072050012000C0（64位）
> > 成功发送启动
> 
> 不，我的意思是请尝试通过“Aircon”菜单/页面发送“开启”命令。不是主页。
> 例如，您之前包含的页面/图像：[＃814（评论）](https://github.com/crankyoldgit/IRremoteESP8266/issues/814#issuecomment-511286479)

The test failed. Under the Aircon menu, Gree is used to send'on'. The air conditioner has a'beep', but on turns off. And there was no response to send'OFF'.

crankyoldgit, 2019-07-15:
I've updated https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814 to support different models of GREE a/c.
In `IRMQTTServer`, you will need to set it to **GREE** protocol, and model **2** (YBOFB2) for it to work with your A/C unit.

`IRrecvDumpV2` should correctly identify your Model **if** it is a message with the power set to "On". "Off" messages appear to be identical between the models so it can't accurately detect the models yet in those cases.

1043717432, 2019-07-16:
> I've updated https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814 to support different models of GREE a/c.
> In `IRMQTTServer`, you will need to set it to **GREE** protocol, and model **2** [...]

Thank you for your kindness. It's perfect.

Air conditioning brand: green

Remote control model: YBOFB/YAPOF3

crankyoldgit, 2019-07-16:
Thanks for confirming it is working as expected.

1043717432, 2019-07-18:
> Thanks for confirming it is working as expected.

Thank you. I'm disturbing you again.

At present, there is a problem to ask for help. It is normal to turn on the infrared signal transmitted by remote control of air conditioner, the panel lamp of air conditioner and the feedback lamp in firmware. But when firmware is used to adjust the air conditioning temperature, the air conditioning panel lamp is turned off. Send on the air conditioning panel lamp is invalid.

crankyoldgit, 2019-07-18:
I believe that is controlled by the `setLight(true);` (On) or `setLight(false);` (Off) functions of the `IRGreeAC` class.

That should be controllable via the "Light" option in the "Aircon" menu/page of `IRMQTTServer` program.

1043717432, 2019-07-18:
> I believe that is controlled by the `setLight(true);` (On) or `setLight(false);` (Off) functions of the `IRGreeAC` class.
> 
> That should be controllable via the "Light" option in the "Aircon" menu/page of `IRMQTTServer` program.

The Linght option in the IRMQTTSERVER program page is ON, but when the IRMQTTSERVER page sends a temperature adjustment request, the air conditioning panel lamp is off.

1043717432, 2019-07-18:
> I believe that is controlled by the `setLight(true);` (On) or `setLight(false);` (Off) functions of the `IRGreeAC` class.
> 
> That should be controllable via the "Light" option in the "Aircon" menu/page of `IRMQTTServer` program.

It seems that the transmitted status code is the opposite.

crankyoldgit, 2019-07-18:
Can you please capture the messages from your remote for the on & off for the "light" using `IRrecvDumpV2` and include them here? It may be the case that this is another difference for your model compared to all the other models.

1043717432, 2019-07-18:
Okay, I'll use dumpv2 to capture it right away. There are four different brands of air conditioners in my home. Now the problems of two remote control models of green air conditioner have been basically repaired by you. Thank you.

1043717432, 2019-07-19:
> Can you please capture the messages from your remote for the on & off for the "light" using `IRrecvDumpV2` and include them here? It may be the case that this is another difference for your model compared to all the other models.

···
Timestamp : 000074.638
                      Library   : v2.6.3

Encoding  : KELVINATOR
                      Code      : 790420501100008079042070000050C0 (128 bits)
                                                                             Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 5 (MAX), Turbo: Off, Quiet: Off, XFan: Off, IonFilter: Off, Light: On, Swing (Horizontal): On, Swing (Vertical): On
Raw Timing[279]:
                   +  9012, -  4468,    +   666, -  1642,    +   664, -   542,    +   662, -   544,
                                                                                                       +   662, -  1644,    +   664, -  1642,    +   664, -  1642,    +   666, -  1640,
                                                                                                                                                                                           +   666, -   542,    +   664, -   542,    +   664, -   540,    +   666, -  1640,
                                                                +   664, -   542,    +   666, -   542,    +   638, -   568,    +   664, -   540,
                                                                                                                                                    +   664, -   540,    +   662, -   544,    +   662, -   544,    +   664, -   542,
                         +   662, -   544,    +   664, -   540,    +   664, -  1642,    +   666, -   540,
                                                                                                             +   664, -   542,    +   664, -   542,    +   664, -   542,    +   666, -   540,
                                                                                                                                                                                                 +   664, -   542,    +   666, -  1642,    +   666, -   540,    +   662, -  1644,
                                                                      +   664, -   542,    +   662, -   544,    +   664, -  1642,    +   664, -   544,
                                                                                                                                                          +   664, - 19972,    +   664, -  1642,    +   640, -   568,    +   638, -   568,
                               +   662, -   544,    +   662, -  1644,    +   660, -   544,    +   664, -   542,
                                                                                                                   +   662, -   544,    +   664, -   542,    +   662, -   542,    +   662, -   542,
                                                                                                                                                                                                       +   636, -   568,    +   638, -   568,    +   662, -   544,    +   662, -   544,
                                                                            +   636, -   568,    +   660, -   544,    +   638, -   568,    +   664, -   542,
                                                                                                                                                                +   662, -   544,    +   636, -   568,    +   638, -   570,    +   660, -   544,
                                     +   636, -   570,    +   662, -   542,    +   662, -   542,    +   664, -   544,
                                                                                                                         +   660, -   544,    +   664, -   544,    +   636, -   570,    +   636, -   568,
                                                                                                                                                                                                             +   662, -  1642,    +   638, - 40012,    +  8982, -  4498,    +   664, -  1644,
                                                                                  +   662, -   544,    +   636, -   570,    +   636, -  1670,    +   638, -  1668,
                                                                                                                                                                      +   636, -  1670,    +   664, -  1644,    +   638, -   568,    +   660, -   544,
                                           +   662, -   544,    +   660, -  1644,    +   662, -   544,    +   660, -   544,
                                                                                                                               +   662, -   544,    +   662, -   542,    +   662, -   544,    +   636, -   570,
    +   636, -   570,    +   636, -   570,    +   636, -   570,    +   636, -   570,
                                                                                        +   636, -  1670,    +   638, -   568,    +   664, -   542,    +   660, -   546,
                                                                                                                                                                            +   636, -   568,    +   660, -   544,    +   662, -   544,    +   636, -  1668,
[... cut ...] 36, 568,  660, 546,  636, 570,  660, 546,  636, 570,  660, 546,  662, 1644,  662, 544,  662, 1644,  636, 570,  636, 570,  636, 570,  636, 568,  636, 570,  636, 570,  660, 544,  636, 1670,  662, 1646,  636};  // KELVINATOR
                                                                                                                                                                 uint8_t state[16] = {0x79, 0x04, 0x00, 0x50, 0x11, 0x00, 0x00, 0x80, 0x79, 0x04, 0x00, 0x70, 0x00, 0x00, 0x50, 0xC0};

1043717432, 2019-07-19:
> Can you please capture the messages from your remote for the on & off for the "light" using `IRrecvDumpV2` and include them here? It may be the case that this is another difference for your model compared to all the other models.
···Timestamp : 000074.638
Library : v2.6.3

Encoding : KELVINATOR
Code : 790420501100008079042070000050C0 (128 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 5 (MAX), Turbo: Off, Quiet: Off, XFan: Off, IonFilter: Off, Light: On, Swing (Horizontal): On, Swing (Vertical): On
Raw Timing[279]:
+ 9012, - 4468, + 666, - 1642, + 664, - 542, + 662, - 544,
+ 662, - 1644, + 664, - 1642, + 664, - 1642, + 666, - 1640,
+ 666, - 542, + 664, - 542, + 664, - 540, + 666, - 1640,
+ 664, - 542, + 666, - 542, + 638, - 568, + 664, - 540,
+ 664, - 540, + 662, - 544, + 662, - 544, + 664, - 542,
+ 662, - 544, + 664, - 540, + 664, - 1642, + 666, - 540,
+ 664, - 542, + 664, - 542, + 664, - 542, + 666, - 540,
+ 664, - 542, + 666, - 1642, + 666, - 540, + 662, - 1644,
+ 664, - 542, + 662, - 544, + 664, - 1642, + 664, - 544,
+ 664, - 19972, + 664, - 1642, + 640, - 568, + 638, - 568,
+ 662, - 544, + 662, - 1644, + 660, - 544, + 664, - 542,
+ 662, - 544, + 664, - 542, + 662, - 542, + 662, - 542,
+ 636, - 568, + 638, - 568, + 662, - 544, + 662, - 544,
+ 636, - 568, + 660, - 544, + 638, - 568, + 664, - 542,
+ 662, - 544, + 636, - 568, + 638, - 570, + 660, - 544,
+ 636, - 570, + 662, - 542, + 662, - 542, + 664, - 544,
+ 660, - 544, + 664, - 544, + 636, - 570, + 636, - 568,
+ 662, - 1642, + 638, - 40012, + 8982, - 4498, + 664, - 1644,
+ 662, - 544, + 636, - 570, + 636, - 1670, + 638, - 1668,
+ 636, - 1670, + 664, - 1644, + 638, - 568, + 660, - 544,
+ 662, - 544, + 660, - 1644, + 662, - 544, + 660, - 544,
+ 662, - 544, + 662, - 542, + 662, - 544, + 636, - 570,
+ 636, - 570, + 636, - 570, + 636, - 570, + 636, - 570,
+ 636, - 1670, + 638, - 568, + 664, - 542, + 660, - 546,
+ 636, - 568, + 660, - 544, + 662, - 544, + 636, - 1668,
+ 638, - 1668, + 638, - 1670, + 662, - 544, + 660, - 546,
+ 636, - 1670, + 662, - 544, + 660, - 19978, + 662, - 546,
+ 660, - 544, + 660, - 544, + 664, - 542, + 636, - 570,
+ 660, - 544, + 660, - 546, + 636, - 568, + 662, - 546,
+ 636, - 568, + 638, - 568, + 662, - 544, + 660, - 544,
+ 636, - 570, + 660, - 544, + 636, - 570, + 660, - 544,
+ 662, - 544, + 662, - 542, + 638, - 568, + 664, - 1642,
+ 662, - 544, + 636, - 1670, + 664, - 542, + 662, - 544,
+ 638, - 568, + 636, - 568, + 660, - 544, + 662, - 542,
+ 638, - 570, + 636, - 1668, + 638, - 1670, + 664

uint16_t rawData[279] = {9012, 4468, 666, 1642, 664, 542, 662, 544, 662, 1644, 664, 1642, 664, 1642, 666, 1640, 666, 542, 664, 542, 664, 540, 666, 1640, 664, 542, 666, 542, 638, 568, 664, 540, 664, 540, 662, 544, 662, 544, 664, 542, 662, 544, 664, 540, 664, 1642, 666, 540, 664, 542, 664, 542, 664, 542, 666, 540, 664, 542, 666, 1642, 666, 540, 662, 1644, 664, 542, 662, 544, 664, 1642, 664, 544, 664, 19972, 664, 1642, 640, 568, 638, 568, 662, 544, 662, 1644, 660, 544, 664, 542, 662, 544, 664, 542, 662, 542, 662, 542, 636, 568, 638, 568, 662, 544, 662, 544, 636, 568, 660, 544, 638, 568, 664, 542, 662, 544, 636, 568, 638, 570, 660, 544, 636, 570, 662, 542, 662, 542, 664, 544, 660, 544, 664, 544, 636, 570, 636, 568, 662, 1642, 638, 40012, 8982, 4498, 664, 1644, 662, 544, 636, 570, 636, 1670, 638, 1668, 636, 1670, 664, 1644, 638, 568, 660, 544, 662, 544, 660, 1644, 662, 544, 660, 544, 662, 544, 662, 542, 662, 544, 636, 570, 636, 570, 636, 570, 636, 570, 636, 570, 636, 1670, 638, 568, 664, 542, 660, 546, 636, 568, 660, 544, 662, 544, 636, 1668, 638, 1668, 638, 1670, 662, 544, 660, 546, 636, 1670, 662, 544, 660, 19978, 662, 546, 660, 544, 660, 544, 664, 542, 636, 570, 660, 544, 660, 546, 636, 568, 662, 546, 636, 568, 638, 568, 662, 544, 660, 544, 636, 570, 660, 544, 636, 570, 660, 544, 662, 544, 662, 542, 638, 568, 664, 1642, 662, 544, 636, 1670, 664, 542, 662, 544, 638, 568, 636, 568, 660, 544, 662, 542, 638, 570, 636, 1668, 638, 1670, 664}; // KELVINATOR
uint8_t state[16] = {0x79, 0x04, 0x20, 0x50, 0x11, 0x00, 0x00, 0x80, 0x79, 0x04, 0x20, 0x70, 0x00, 0x00, 0x50, 0xC0};

Timestamp : 000086.117
Library : v2.6.3

Encoding : KELVINATOR
Code : 790400501100008079040070000050C0 (128 bits)
Mesg Desc.: Power: On, Mode: 1 (COOL), Temp: 20C, Fan: 5 (MAX), Turbo: Off, Quiet: Off, XFan: Off, IonFilter: Off, Light: Off, Swing (Horizontal): On, Swing (Vertical): On
Raw Timing[279]:
+ 9008, - 4496, + 662, - 1644, + 664, - 544, + 662, - 544,
+ 662, - 1642, + 662, - 1644, + 664, - 1644, + 662, - 1644,
+ 638, - 568, + 638, - 568, + 664, - 542, + 662, - 1644,
+ 664, - 544, + 662, - 546, + 662, - 544, + 636, - 568,
+ 660, - 544, + 662, - 544, + 638, - 568, + 662, - 542,
+ 662, - 544, + 636, - 568, + 638, - 568, + 636, - 568,
+ 638, - 568, + 662, - 544, + 662, - 544, + 662, - 542,
+ 638, - 568, + 636, - 1670, + 636, - 570, + 636, - 1670,
+ 662, - 544, + 638, - 568, + 660, - 1644, + 638, - 570,
+ 662, - 19974, + 638, - 1670, + 638, - 568, + 636, - 568,
+ 636, - 570, + 662, - 1644, + 664, - 544, + 662, - 544,
+ 662, - 544, + 662, - 544, + 662, - 544, + 638, - 568,
+ 638, - 568, + 662, - 544, + 662, - 542, + 662, - 542,
+ 662, - 544, + 662, - 542, + 662, - 544, + 662, - 542,
+ 662, - 544, + 662, - 544, + 662, - 544, + 662, - 544,
+ 660, - 546, + 660, - 546, + 636, - 568, + 638, - 568,
+ 636, - 570, + 662, - 544, + 636, - 568, + 636, - 570,
+ 662, - 1642, + 638, - 40012, + 8982, - 4498, + 636, - 1670,
[... cut ...]  570, 634, 1670, 638, 570, 660, 19976, 664, 544, 636, 570, 636, 568, 636, 570, 636, 568, 636, 568, 636, 570, 636, 570, 636, 570, 636, 570, 636, 568, 636, 568, 660, 546, 636, 568, 636, 568, 660, 546, 636, 570, 660, 546, 636, 570, 660, 546, 662, 1644, 662, 544, 662, 1644, 636, 570, 636, 570, 636, 570, 636, 568, 636, 570, 636, 570, 660, 544, 636, 1670, 662, 1646, 636}; // KELVINATOR
uint8_t state[16] = {0x79, 0x04, 0x00, 0x50, 0x11, 0x00, 0x00, 0x80, 0x79, 0x04, 0x00, 0x70, 0x00, 0x00, 0x50, 0xC0};

crankyoldgit, 2019-07-19:
@1043717432 Have you tried using "KELVINATOR" as the protocol in IRMQTTServer? That's what the library thinks the A/C / remote's protocol is.

1043717432, 2019-07-19:
No, since the last time you updated the GREE protocol, Model 2, I set it up for my A/C unit.

1043717432, 2019-07-19:
Wrong. The remote control forgot to bring it. I'll try it again tomorrow.

1043717432, 2019-07-19:
[image]
The status panel lamp answers correctly.

But the A/C state OFF and ON are misidentified.

Right should be ON

crankyoldgit, 2019-07-19:
Again, can you please capture & document the same settings/sequences you want via your IR remote, and capture it via `IRrecvDumpV2`?

I'm pretty sure the Kelvinator protocol is correct (as it is an aircon I have) but your's may be subtly different in some way.

1043717432, 2019-07-20:
Thank you. The Kelvinator protocol is correct, and it's perfect for Gree A/C operations. Swing (V) and light are perfect. Remote control model: YAPOF3 perfect

But for: YBOF2 currently operates Swing (V), IRMQ TServer, web Aircon status Swing (V) is always off when remotely controlling any operation.

crankyoldgit, 2019-07-20:
> But for: YBOF2 currently operates Swing (V), IRMQ TServer, web Aircon status Swing (V) is always off when remotely controlling any operation.

Can you please be more verbose in your explanations of the issues you are having? I'm finding it very difficult to understand what your problems are. Google Translate is probably not helping matters, but the more descriptive & detailed you can be the more context I have to work on to understand you and thus, help you.

**_Again, can you please capture & document the same settings/sequences you want via your IR remote, and capture it via IRrecvDumpV2?_**

Without data, it is hard to see what is going wrong. The more information you give me, the faster & better the response from me will be.

1043717432, 2019-07-20:
I'm sorry, I'm Chinese, and my English expression is not good. I'll test it again and give you feedback. Thank you.

crankyoldgit, 2019-07-20:
I'm not faulting your English, as it is far far better than my Chinese. ;-) I'm just trying to save frustration and improve our communication level and speed.

crankyoldgit, 2019-07-22:
> I'm sorry, I'm Chinese, and my English expression is not good. I'll test it again and give you feedback. Thank you.

Chasing this up.

1043717432, 2019-07-22:
===========When the ON button is pressed,==============
···Timestamp : 001308.688
                      Library   : v2.6.3

Encoding  : GREE
                Code      : 190A0050002000F0 (64 bits)
                                                      Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 26C, Fan: 1, Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: Off, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 0 (Last Pos)
Raw Timing[139]:
                   +  8996, -  4476,    +   650, -  1648,    +   650, -   554,    +   648, -   552,
                                                                                                       +   648, -  1648,    +   650, -  1650,    +   648, -   554,    +   648, -   554,
                                                                                                                                                                                           +   648, -   556,    +   646, -   554,    +   648, -  1650,    +   648, -   554,
                                                                +   648, -  1650,    +   652, -   552,    +   648, -   554,    +   648, -   554,
                                                                                                                                                    +   648, -   554,    +   648, -   554,    +   648, -   554,    +   648, -   554,
                         +   648, -   554,    +   648, -   554,    +   648, -   556,    +   646, -   554,
                                                                                                             +   648, -   554,    +   648, -   554,    +   648, -   554,    +   648, -   554,
                                                                                                                                                                                                 +   648, -   554,    +   648, -  1650,    +   648, -   554,    +   648, -  1650,
                                                                      +   648, -   554,    +   648, -   552,    +   648, -  1648,    +   648, -   554,
                                                                                                                                                          +   648, - 19986,    +   650, -   556,    +   646, -   554,    +   648, -   554,
                               +   650, -   552,    +   648, -   554,    +   648, -   554,    +   648, -   554,
                                                                                                                   +   648, -   554,    +   648, -   554,    +   648, -   554,    +   648, -   554,
                                                                                                                                                                                                       +   650, -   554,    +   648, -   554,    +   648, -  1648,    +   650, -   554,
                                                                            +   646, -   554,    +   648, -   554,    +   650, -   554,    +   648, -   554,
                                                                                                                                                                +   648, -   554,    +   650, -   552,    +   648, -   554,    +   648, -   554,
                                     +   648, -   554,    +   650, -   552,    +   648, -   554,    +   648, -   554,
                                                                                                                         +   648, -   552,    +   650, -  1648,    +   650, -  1650,    +   650, -  1648,
                                                                                                                                                                                                             +   650, -  1648,    +   648

uint16_t rawData[139] = {8996, 4476,  650, 1648,  650, 554,  648, 552,  648, 1648,  650, 1650,  648, 554,  648, 554,  648, 556,  646, 554,  648, 1650,  648, 554,  648, 1650,  652, 552,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 556,  646, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 1650,  648, 554,  648, 1650,  648, 554,  648, 552,  648, 1648,  648, 554,  648, 19986,  650, 556,  646, 554,  648, 554,  650, 552,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  648, 554,  650, 554,  648, 554,  648, 1648,  650, 554,  646, 554,  648, 554,  650, 554,  648, 554,  648, 554,  650, 552,  648, 554,  648, 554,  648, 554,  650, 552,  648, 554,  648, 554,  648, 552,  650, 1648,  650, 1650,  650, 1648,  650, 1648,  648};  // GREE
                                                                                                                                                                                                  uint8_t state[8] = {0x19, 0x0A, 0x00, 0x50, 0x00, 0x20, 0x00, 0xF0};
================When the power button is ON, turn on the light ON.======
Timestamp : 001344.969
                      Library   : v2.6.3

Encoding  : GREE
                Code      : 190A2050002000F0 (64 bits)
                                                      Mesg Desc.: Power: Off, Mode: 1 (COOL), Temp: 26C, Fan: 1, Turbo: Off, IFeel: Off, WiFi: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 0 (Last Pos)
Raw Timing[139]:
[... cut ...]  622, 582,  620, 1676,  622, 582,  620, 582,  622, 580,  622, 582,  622, 580,  620, 582,  620, 582,  620, 580,  622, 582,  620, 580,  622, 580,  622, 580,  622, 580,  622, 582,  620, 1676,  622, 1676,  622, 1676,  622, 580,  622};  // GREE
                                                                                                                                                                                                uint8_t state[8] = {0x11, 0x0A, 0x00, 0x50, 0x00, 0x20, 0x00, 0x70};

crankyoldgit, 2019-07-22:
I think I found the bug/mistake. I had Light & Turbo swapped. See: https://github.com/crankyoldgit/IRremoteESP8266/commit/96c061185ad3957aa8e76b08ac2dad99da9e984e

I've updated https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814, so please download and recompile everything based on that, and let me know how it goes.

1043717432, 2019-07-22:
> I think I found the bug/mistake. I had Light & Turbo swapped. See: [96c0611](https://github.com/crankyoldgit/IRremoteESP8266/commit/96c061185ad3957aa8e76b08ac2dad99da9e984e)
> 
> I've updated https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue814, so please download and recompile everything based on that, and let me know how it goes.

ok,Thank you.!

1043717432, 2019-07-22:
> I think I found the bug/mistake. I had Light & Turbo swapped. See: [96c0611](https://github.com/crankyoldgit/IRremoteESP8266/commit/96c061185ad3957aa8e76b08ac2dad99da9e984e)
>
> I've updated [...]

··· \Arduino\libraries\IRremoteESP8266-Issue814\src\IRac.cpp:32:25: fatal error: gtest/gtest.h: No such file or directory

 #include "gtest/gtest.h"

                         ^

compilation terminated.

exit status 1
Hello! Am I compiling directly?

Or compile IRremoteESP 8266-Issue814 test?

crankyoldgit, 2019-07-22:
I have no idea what you have done or are doing. Clearly what ever it is is not right.

You should just be downloading it the same as you were doing previously/earlier in this issue.

1043717432, 2019-07-22:
Thank you. You can compile it.

But the sweeper still can't get the right feedback!

crankyoldgit, 2019-07-22:
What do you mean by "Sweeper"? Do you mean the "Swing". As in vertical control of where the air flows?

Or do you mean something else?

Did the Light/LED thing work at all?

crankyoldgit, 2019-07-22:
Or did you mean "Sleep"?

crankyoldgit, 2019-07-23:
Just chasing this up.

crankyoldgit, 2019-07-25:
> But the sweeper still can't get the right feedback!

Another chase up. I still don't know what you mean by "sweeper".

crankyoldgit, 2019-07-26:
FYI, the code changes thus far have been included in the v2.6.4 release of the library.

crankyoldgit, 2019-07-28:
Another ping to chase answers for my questions.

crankyoldgit, 2019-07-30:
Closing this due to lack of response and that we seem to have resolved everything else.

commit 5a562098 (2019-07-15): Gree: Add differing model support.
* Add set/getModel(). e.g. YAW1F (1) [default] & YBOFB2 (2)
* Automatic detection of models only works with messages with power on.
* set/getPower() now changes behaviour with different models.
* Report model type in the .toString() function.
* Update common a/c routines and unit tests.
* Unit test coverage for different power/model operation.

Fixes #814
commit a3663d4e (2019-07-15): Gree: Use less strict power mode detection.
Reported data in #814 indicates state[0] is the only thing that changes 
for power on/off.

Leave the existing bit setting/clearing for setPower() etc, but don't 
require the state[2] bit for getPower().

Fixes #814
commit 63d6059d (2019-07-22): Gree: Use less strict power mode detection.
Reported data in #814 indicates state[0] is the only thing that changes 
for power on/off.

Leave the existing bit setting/clearing for setPower() etc, but don't 
require the state[2] bit for getPower().

Fixes #814
commit ee0a288d (2019-07-22): Gree: Add differing model support.
* Add set/getModel(). e.g. YAW1F (1) [default] & YBOFB2 (2)
* Automatic detection of models only works with messages with power on.
* set/getPower() now changes behaviour with different models.
* Report model type in the .toString() function.
* Update common a/c routines and unit tests.
* Unit test coverage for different power/model operation.

Fixes #814
closed 2019-07-30 by crankyoldgit