crankyoldgit/IRremoteESP8266 #818: MQTT Example with MQTT not working

drschnalli, 2019-07-15:
Hello, i just flashed the example file onto my ESP8266. When i enable MQTT ( i dont find the credentials settings ) the webinterface is very very laggy and hangs. When i disable it all is fine. (nice work by the way) im using the newest libraries (of the ones needed) and the newest arduino IDE.

crankyoldgit, 2019-07-16:
See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L48

To set those, you need to go to the initial setup page which happens when the device is wiped.

It's under the "Admin" menu:

> **Wipe Settings** _Warning_: Resets the device back to original settings. ie. Goes back to AP/Setup mode.

Yes, without setting the correct MQTT details, it will be very very laggy.

drschnalli, 2019-07-16:
Hey thank you, that worked!

closed 2019-07-16 by drschnalli