crankyoldgit/IRremoteESP8266 #819: Transmitting Fahrenheit instead of Celsius (Midea)

drschnalli, 2019-07-16:
Hello, after i got it all running (MQTT IR Example - Aircon - Repo Cloned yesterday) i realized that my Midea Comfee AC is turning to Fahrenheit Mode (<- i cannot even set fahrenheit with the german original AC but when i change the temp it changes back to celsius).

I set my AC to Midea and i turned on Celsius Mode via Webinterface and later also again via the MQTT Command.

So when i now set the temperature to 24C° my AC shows something like 87°F but it doesnt change when i set another temperature. Do you know where my mistake is or has the code been created by an owner of the fahrenheit model.

crankyoldgit, 2019-07-17:
You are correct. The original person who gave the data & analysis for the Midea A/C was using a model that had/used Fahrenheit instead of Celsius. The library currently converts temps to degF to control the A/C because of that.
I'm willing to add native Celsius support if you can provide me with the codes for & descriptions of **all** the temp values from the lowest to the highest temp it supports.
e.g.
COOL + LOW FAN + 16 Degrees Celsius -> COOL + LOW FAN + 32 Degrees Celsius ... or what ever the temp ranges are. Please include half degrees etc if it supports it.

crankyoldgit, 2019-07-17:
Also, what is the brand & model numbers for the A/C unit and the remote please?

drschnalli, 2019-07-17:
Hello Cranky, very happy about that your reacting so fast! Thanks for your great work by the way. 

For Sure, i will let you know the Model of the Remote (and also the AC) and for sure also the Codes. 

I will do this when im back home again (approx 10. hours).

So at the moment the Celsius values are getting transmitted in the right Fahrenheit value to the device? I never had to do with Fahrenheit but i can check if its sending the right values.

My Remote also sends a temperature (it has an internal sensors) to the AC, i dont know if that is an useful information.

What i also found out that the swing command is not getting decoded right. When i press Swing it decodes it as Power: Off Mode: OFF. I will also send you the swing ir command.

Thank you

crankyoldgit, 2019-07-17:
It'd be best if you can record the messages from the remote with the `IRrecvDumpV2` program. It will provide me with the best info.

But yes, it should be doing the correct C to F conversion when it uses the web interface or the C++ class api. However, as I didn't have a Celsius message from a remote to decode with, I wouldn't trust the "text description" of the IRrecvDumpV2 program. The rest of the data should be fine. i.e. rawData[]/state[] etc.

drschnalli, 2019-07-17:
Heyho Cranky I Created a Google Spreadsheet: https://docs.google.com/spreadsheets/d/1pr6Ec_YVti8QtSIUDM3BKT_4aMrGJfaKMOKGK1G65CU/edit?usp=sharing

I dont exactly know, but somewhere in the code there is also the temperature of the internal sensor (within the remote) provided. I hope that does not confuse. I remember that some ir code i used also shown this temperature sent from the remote but IRrecvDumpV2 does not seem to recognize (even its using the same libraries i think)

The remote itself is a Keystone RG57H4(B)BGEF
The AC is a Comfee MPD1-12CRN7

I hope i got all information in this thread.

crankyoldgit, 2019-07-18:
Coolix and Midea are two different protocols. Midea often seems to use the Coolix protocol on some devices. If you spend the time analysing the bits of the protocol & work out exactly what does what, I can add it to the protocol support. Please open a new Issue for that (and for the swing etc too)

You also have to assume I know nothing about your particular A/C & remote. e.g. Does pressing "swing" or what ever toggle to swing mode just Vertically or does it set a bit that says "Keep doing vertical until I turn off that bit"? e.g. does the code change one subsequent presses of swing or not, and how does the A/C behave. Does the swing just move it a single position, and does it just change from swing up & down (or left & right) to no swing at all. Does it remember the non-swing position etc etc?

In the meantime, to close out this issue, can you please download and try the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue819 per PR #823 and confirm if it does what you want?

P.S. No, I don't need a photo of the remote, but you could always upload the A/C & remote's manual if you have them, or a link to them.

drschnalli, 2019-07-18:
Heyho David, sorry i edited my posting after posting it (maybe you saw only the text in the mail notification)  The Thing with coolix is just something i saw in another issue (i just mentioned it because i saw that someone with a Midea got Coolix Readings and also got shown the RC internal Temp sensor) <- Thats not important.

My Swing is working vertically and you can only stop and start the swing mode no positions are remembered or saved. The Command to enable and disable swing seems to be the same and does not include "mode/temp"fanspeed". When you stop the swing it just stays at its position and when you issue the swing command again it just stats moving up and down again. 

I really appreciate your work. Do you have a patreon or paypal donation account? 

I will test the new branch when im back home and give you a feedback to this.

crankyoldgit, 2019-07-18:
> I will test the new branch when im back home and give you a feedback to this.

Thanks. I look forward to it.

> Swing

Thanks for the info, that's helpful. I'll try to look into implementing it tomorrow.

> I really appreciate your work. Do you have a patreon or paypal donation account?

Thanks. No, I don't do patreon. I do have a paypal account, but not a "donation" account.
I'm not sure how I'd feel about putting something up like that to solicit donations anyway.
However, if you're in town/nearby etc, I won't say no to a beverage and a [chinwag](https://www.google.com/search?q=define%3A+chinwag). Just ask @sheppy99 ;-)

drschnalli, 2019-07-18:
Heyho David, your Branch Issue819 is sending the right celcius value to the ac and the ac displays the right value!! I checked it from 17 to 30°C (this is the range the device supports) and the stepsize is 1°C.

Shall we close this issue and continue in another one for the swing command? by the way im from germany =) where are you from?

crankyoldgit, 2019-07-19:
> Heyho David, your Branch Issue819 is sending the right celcius value to the ac and the ac displays the right value!! I checked it from 17 to 30°C (this is the range the device supports) and the stepsize is 1°C.

Excellent! Glad it's working. Thanks for the confirmation.

> Shall we close this issue and continue in another one for the swing command?

When PR #823 gets merged, it will automatically close this issue. I'll look at adding _swing_ soon.

> where are you from?

If you hover the mouse over my profile image on the Issue thread, it says "Australia". :-)

crankyoldgit, 2019-07-19:
> by the way im from germany =)

Bayern/München per chance? Either way, enjoy a Maß for me at during the next Oktoberfest!

crankyoldgit, 2019-07-19:
If you can please (re)download and try the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue819 per PR #823 and confirm if it does what you want? i.e. Vertical Swing

I think I've hacked in support for it, but it's getting quite complicated to emulate _toggle only_ style messages and maintain the desired state. i.e. See #821 / #825 So extensive testing and detailed descriptions of failure cases are probably needed.

drschnalli, 2019-07-19:
Yes its very stupid that they implemented is as toggle... so you can never be sure to what mode its set without having a look at it. I downloaded the branch again (the one i also downloaded yesterday but the updated one i hope) when i issue the swing command via the original remote the esp sets mode to off, power to off, temperature to -17,5. When i issue the swing v command via webinterface.

crankyoldgit, 2019-07-20:
I can't seem to reproduce the circumstances and results you are reporting. Can you please go into boring levels of detail about the state of the A/C emulation in the ESP before you start, and what exact steps you took to get it to do the " mode to off, power to off, temperature to -17,5" result?

Looking at that result btw, it is what I would kind-of expect **if** you hadn't successfully updated to the latest version of the branch/patch, compiled, and uploaded to the ESP.

If you are unsure if you have updated correctly. I suggest hand-checking the code in the directory for these changes: https://github.com/crankyoldgit/IRremoteESP8266/commit/b01602dd7e155fd5eab3a04b9c08d19f6c7bdf1d

and perhaps update the `examples/IRMQTTServer/IRMQTTServer.h` file in that directory and change the line:
`#define _MY_VERSION_ "v1.3.2-testing"`
and the `src/IRremoteESP8266.h` file and change the line:
`#define _IRREMOTEESP8266_VERSION_ "2.6.3"`
to something different. Those values will show up on the "Info" page. That way you're sure you're using the new code etc. 

Either that, or maybe the remote is producing a different code for the "toggle swing" command than what you reported earlier. e.g. `0xA201FFFFFF7C`

drschnalli, 2019-07-21:
Heyho, i saw that i forgot to update examples/IRMQTTServer/IRMQTTServer.h and ino i now updated the files, too and reuploaded the new image. I will give you more information tomorrow as im a little bit in hurry today. By the way how should i issue the swing command then? Should i set SwingV to Auto or Low High mid to see the result?

crankyoldgit, 2019-07-22:
For that A/C, it's `Off` -> `Off`, and _everything else_ -> `Auto`/swinging etc.

crankyoldgit, 2019-07-23:
@drschnalli Any results?

crankyoldgit, 2019-07-25:
@drschnalli Any feedback? If I don't hear anything soon. I'm going to assume all is now good and I'll merge the changes as is.

drschnalli, 2019-07-25:
Hello David, im back from a business trip tomorrow then i will put more information onto my post. Sorry for the Late feedback.

crankyoldgit, 2019-07-26:
@drschnalli I've merged the changes thus far, and that has closed this issue.
However, if you find it doesn't fix it when you try it. Ping me on this issue and I can reopen it, or create a new issue. Either way, confirmation that it's working (or not) would be great!

closed 2019-07-26 by crankyoldgit

crankyoldgit, 2019-07-26:
FYI, the code changes thus far have been included in the v2.6.4 release of the library.

commit 624862f6 (2019-07-18): Add more supported models to Midea

Fixes #819
commit ca5767a0 (2019-07-19): Midea: Add support for SwingV.
Swing is a special toggle-only command on Midea. Thus we need to try to 
code around it if we want to preserve state etc.
Try to do this as best as we can.
Handle toggle operation in IRMQTTServer.

Add/update unit tests accordingly.

Fixes #819
commit b01602dd (2019-07-19): Midea: Add support for SwingV.
Swing is a special toggle-only command on Midea. Thus we need to try to 
code around it if we want to preserve state etc.
Try to do this as best as we can.
Handle toggle operation in IRMQTTServer.

Add/update unit tests accordingly.

Fixes #819
commit b99b48c6 (2019-07-19): Add more supported models to Midea

Fixes #819