crankyoldgit/IRremoteESP8266 #821: please add support for Midea(RN08A/BG)

hassbian-ABC, 2019-07-17:
[Midea -RN08ABG.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3402750/Midea.-RN08ABG.txt)

hassbian-ABC, 2019-07-17:
When I press the remote control off, temp becomes 255.
When I press the remote control to swing, it also becomes 255
Swing (V) and swing (H) are auto.
But my remote control is swing (H)

1043717432, 2019-07-18:
```mode：auto, power: on, temp: 25, fanspeed: auto,

Timestamp : 000089.318
Encoding  : COOLIX
Code      : B21FC8 (24 bits)
Mesg Desc.: Power: On, Mode: 2 (AUTO), Fan: 0 (AUTO0), Temp: 25C, Zone Follow: Off, Sensor Temp: Ignored
Library   : v2.6.2

Raw Timing[199]:

+  4374, -  4462,    +   480, -  1704,    +   478, -   602,   +   476, -  1704,
+   478, -  1702,    +   480, -   604,    +   474, -   600,   +   478, -  1704,
+   478, -   602,    +   476, -   602,    +   476, -  1702,    +   480, -   600,
+   476, -   604,    +   474, -  1704,    +   480, -  1704,    +   480, -   602,
+   476, -  1702,    +   480, -   602,    +   476, -   600,    +   476, -   602,
+   476, -  1702,    +   482, -  1700,    +   480, -  1704,    +   480, -  1704,
+   478, -  1704,    +   478, -  1704,    +   480, -  1704,    +   478, -  1704,
+   480, -   602,    +   476, -   602,    +   476, -   602,    +   476, -   602,
+   476, -   600,    +   476, -  1704,    +   478, -  1702,    +   478, -   602,
+   478, -   626,    +   450, -  1702,    +   480, -   602,    +   478, -   600,
+   474, -   602,    +   476, -   600,    +   476, -   602,    +   476, -  1704,
+   478, -  1704,    +   480, -   626,    +   452, -  1704,    +   478, -  1702,
+   480, -  1704,    +   478, -  5280,    +  4376, -  4464,    +   478, -  1704,
+   478, -   600,    +   480, -  1700,    +   478, -  1704,    +   480, -   600,
+   476, -   602,    +   478, -  1700,    +   480, -   602,    +   478, -   600,
+   476, -  1704,    +   478, -   602,    +   476, -   602,    +   476, -  1704,
+   478, -  1704,    +   478, -   604,    +   474, -  1702,    +   480, -   600,
+   478, -   600,    +   476, -   602,    +   476, -  1702,    +   480, -  1704,
+   480, -  1702,    +   480, -  1704,    +   480, -  1702,    +   478, -  1704,
+   482, -  1700,    +   480, -  1704,    +   478, -   602,    +   476, -   602,
+   482, -   596,    +   476, -   626,    +   452, -   602,    +   476, -  1702,
+   480, -  1702,    +   480, -   600,    +   476, -   602,    +   476, -  1702,
+   480, -   602,    +   476, -   600,    +   478, -   602,    +   476, -   600,
+   476, -   600,    +   478, -  1704,    +   478, -  1704,    +   478, -   602,
+   476, -  1704,    +   478, -  1702,    +   480, -  1702,    +   480

uint16_t rawData[199] = {4374, 4462,  480, 1704,  478, 602,  476, 1704,  478, 1702,  
480, 604,  474, 600,  478, 1704,  478, 602,  476, 602,  476, 1702,  480, 600,  476, 
604,  474, 1704,  480, 1704,  480, 602,  476, 1702,  480, 602,  476, 600,  476, 602,  
476, 1702,  482, 1700,  480, 1704,  480, 1704,  478, 1704,  478, 1704,  480, 1704,  
478, 1704,  480, 602,  476, 602,  476, 602,  476, 602,  476, 600,  476, 1704,  478, 
1702,  478, 602,  478, 626,  450, 1702,  480, 602,  478, 600,  474, 602,  476, 600,  
476, 602,  476, 1704,  478, 1704,  480, 626,  452, 1704,  478, 1702,  480, 1704,  478, 
5280,  4376, 4464,  478, 1704,  478, 600,  480, 1700,  478, 1704,  480, 600,  476, 602,  
478, 1700,  480, 602,  478, 600,  476, 1704,  478, 602,  476, 602,  476, 1704,  478, 1704,  
478, 604,  474, 1702,  480, 600,  478, 600,  476, 602,  476, 1702,  480, 1704,  480, 1702,  
480, 1704,  480, 1702,  478, 1704,  482, 1700,  480, 1704,  478, 602,  476, 602,  482, 596,  
476, 626,  452, 602,  476, 1702,  480, 1702,  480, 600,  476, 602,  476, 1702,  480, 602,  
476, 600,  478, 602,  476, 600,  476, 600,  478, 1704,  478, 1704,  478, 602,  476, 1704,  
478, 1702,  480, 1702,  480};  // COOLIX B21FC8

uint64_t data = 0xB21FC8;

..................................................................................................................

{mode：auto, power: on, temp: 25, fanspeed: auto, led: on} swingh: on

Timestamp : 001183.168
Encoding  : COOLIX
Code      : B26BE0 (24 bits)
Mesg Desc.: Power: On, Swing: Toggle
Library   : v2.6.2

Raw Timing[199]:
+  4406, -  4434,    +   506, -  1676,    +   506, -   574,    +   504, -  1676,
+   506, -  1676,    +   506, -   572,    +   506, -   572,    +   506, -  1676,
+   506, -   572,    +   506, -   572,    +   506, -  1676,    +   506, -   572,
+   506, -   572,    +   534, -  1646,    +   508, -  1676,    +   508, -   572,
+   506, -  1676,    +   506, -   572,    +   518, -  1664,    +   506, -  1676,
+   504, -   574,    +   478, -  1704,    +   480, -   600,    +   476, -  1704,
[... 586 lines cut ...]
+   504, -  1676,    +   506, -  1676,    +   506, -  1676,    +   508, -  1676,
+   508, -   572,    +   506, -  1676,    +   508, -  1674,    +   508, -  1676,
+   508, -   572,    +   504, -   572,    +   506, -   572,    +   506, -   572,
+   506, -  1676,    +   506, -   572,    +   506, -   572,    +   506, -  1676,
+   508, -  1676,    +   506, -  1676,    +   506, -   574,    +   504, -   574,
+   504, -   572,    +   504, -   574,    +   506, -   572,    +   504, -   574,
+   504, -   574,    +   504, -   572,    +   506, -  1676,    +   508, -  1676,
+   508, -  1674,    +   508, -  1676,    +   508, -  1674,    +   506

uint16_t rawData[199] = {4378, 4462,  480, 1702,  480, 598,  480, 1702,  480, 1704,  
480, 600,  476, 600,  478, 1704,  480, 602,  476, 600,  478, 1704,  480, 600,  478,
 600,  478, 1702,  480, 1702,  480, 600,  478, 1702,  480, 600,  478, 1702,  480, 
 1702,  480, 1702,  480, 1702,  480, 602,  478, 1702,  482, 1702,  480, 1702,  480, 
 600,  478, 600,  478, 600,  478, 600,  478, 1702,  482, 600,  478, 600,  478, 1702, 
[... cut ...]
 508, 1676,  506, 574,  506, 1676,  506, 572,  504, 1676,  506, 1676,  506, 1676,  508,
 1676,  508, 572,  506, 1676,  508, 1674,  508, 1676,  508, 572,  504, 572,  506, 572, 
 506, 572,  506, 1676,  506, 572,  506, 572,  506, 1676,  508, 1676,  506, 1676,  506, 574, 
 504, 574,  504, 572,  504, 574,  506, 572,  504, 574,  504, 574,  504, 572,  506, 1676, 
 508, 1676,  508, 1674,  508, 1676,  508, 1674,  506};  // COOLIX B27BE0
                                                                      
uint64_t data = 0xB27BE0;

crankyoldgit, 2019-07-18:
@hassbian-ABC I'm assuming you are talking about when using `IRMQTTServer`? Is that correct?

hassbian-ABC, 2019-07-18:
@
yes ，
Protocol: COOLIX
Model: Default

hassbian-ABC, 2019-07-18:
@

> @ hassbian-ABC我假设你在谈论使用时`IRMQTTServer`？那是对的吗？

yes ，
Protocol: COOLIX
Model: Default

hassbian-ABC, 2019-07-18:
@crankyoldgit

yes
Protocol: COOLIX
model: default

hassbian-ABC, 2019-07-18:
@ crankyoldgit  When I use COOLIX to control temp cool and heat, it's right.

hassbian-ABC, 2019-07-18:
@crankyoldgit
 I use COOLIX to control temp cool and heat, it's right.

crankyoldgit, 2019-07-18:
@hassbian-ABC I think this is because COOLIX has a special command for off and swing that doesn't contain the rest of the state information. I'll see what I can do to hack in support to work around that problem.

hassbian-ABC, 2019-07-18:
@crankyoldgit 
If you switch from cool or heat to auto, the air conditioner swing (H) is on, but on the web, swing (H) is off.

hassbian-ABC, 2019-07-18:
@crankyoldgit 
And Led Toggle doesn't work

crankyoldgit, 2019-07-18:
@hassbian-ABC Can you please supply more context with your feedback. I can't read your mind or see what you are doing. For instance:

Are you saying the Led Toggle doesn't work via 1) the remote, 2) via the web interface, 3) via mqtt, 4) some other method? And how "doesn't (it) work"? e.g. It's out of phase/sync, nothing at all happens when you click it. The A/C doesn't respond at all? It beeps and does nothing?!?

e.g. See/read the initial Issue template that you deleted when you reported the issue, and the Troubleshooting guide on the wiki. Especially the "more information" & "be verbose" parts.

hassbian-ABC, 2019-07-18:
@crankyoldgit
via the web interface，when toggle led，The A/C doesn't respond at all，  It beeps and does nothing

crankyoldgit, 2019-07-18:
@hassbian-ABC Can you please download and try out the code for PR #825 and let me know how it goes for your initially reported issues? e.g. https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821

crankyoldgit, 2019-07-18:
> @crankyoldgit
> via the web interface，when toggle led，The A/C doesn't respond at all， It beeps and does nothing

FYI, Beeping is responding. ;-)

So, what happens when you go from "Coolix, Light: off" (send) -> "Coolix, Light: on" (send) -> "Coolix, Light: off" (send) etc

And can you please capture and report the same sequence using `IRrecvDumpV2` when using **just** your original IR remote?

hassbian-ABC, 2019-07-18:
@crankyoldgit
Thank you. I'll test it as soon as possible.

hassbian-ABC, 2019-07-18:
@crankyoldgit
Now, besides led, there are other perfections

hassbian-ABC, 2019-07-18:
@crankyoldgit
This is the turn-on and turn-off of the led.

hassbian-ABC, 2019-07-19:
@crankyoldgit
There is another problem, via the web interface, turn power to on，and turn swing（v）to on，Air conditioning is correct，but when swing (v) is on，via the web interface，turn power to off，turn power to on again，the Air conditioning swing(v) is off, On the Web Interface，swing(v) is on

hassbian-ABC, 2019-07-19:
@crankyoldgit
About LED，When I turn the LED to on/off through the web interface，Air conditioning beeps and does nothing，when via the remote turn the LED to on/off ，In the web interface, LIGHT hasn't changed at all.

crankyoldgit, 2019-07-21:
> Library : v2.6.2

Hey, I noticed this in your replies.
If you are using the code from https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821
it should be showing a different version. This makes me think you are not using the new updated code.

Can you please check/show me the version info section from the "Info" menu/tab on IRMQTTServer?

hassbian-ABC, 2019-07-21:
[image]
The problem still exists. @crankyoldgit

When I turn the LED to on/off through the web interface，Air conditioning beeps and does nothing，when via the remote turn the LED to on/off ，In the web interface,the LIGHT state is correct

via the web interface, turn power to on，and turn swing（v）to on，A/C is correct，but when swing (v) is on，via the web interface，turn power to off，turn power to on again，A/C swing(v) is off, in the Web Interface，swing(v) is on

hassbian-ABC, 2019-07-22:
I can control the LED through mosquitto_pub -h 10.0.0.4 -t ir_server/send -m ‘15, B5F5A5,24’
But in the web interface, it's always off
@crankyoldgit

crankyoldgit, 2019-07-22:
@hassbian-ABC I think I've found the problem. I really stuffed up the calling parameters for coolix in the common a/c stuff. (see https://github.com/crankyoldgit/IRremoteESP8266/commit/b86813581760c2c5168c5e5d1b04d16520380f45)

Please download and try the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821 and let me know if that works any better?

hassbian-ABC, 2019-07-22:
I downloaded and used https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821，
Unfortunately, the problem remains.

hassbian-ABC, 2019-07-22:
I downloaded and used https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821，
Unfortunately, the problem remains.
@crankyoldgit

crankyoldgit, 2019-07-22:
Can you please give me the state of the "AirCon" page you are using? I'll use that to write up a test case to see what is going wrong now.

hassbian-ABC, 2019-07-22:
[image]
@crankyoldgit

crankyoldgit, 2019-07-22:
Thanks for the data.
I've added some tests to confirm it is sending the Led/Light toggle message as it should be.
However, it may have been confusing the A/C unit by not having a big enough gap between the messages IRMQTTServer was producing. I've taken a **guess** at a big enough gap between the messages (i.e. 100ms, or 1/10th of a second). So it may work, it may not.
i.e. we may need to increase it.

**IF** it doesn't work, can you please change the line here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/Issue821/src/ir_Coolix.cpp#L90

from: `space(kDefaultMessageGap);` to `space(250000);  // 1/4 second` or  `space(500000);  // 1/2 second` or  `space(750000);  // 3/4 second` or  `space(1000000);  // 1 second`

I want to see what the required gap between messages is, and the smaller we can have the better.

So, please download and try the code in branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue821 again and let me know if that works any better? or if you need to change that value? or if it all is still not working.

Fingers crossed.

hassbian-ABC, 2019-07-22:
Thank you for your work, but I still can't control the turn-on and turn-off of LED/LIGHT through the web interface. i try to change the  ir_Coolix.cpp#L90 to space(250000); / space(500000);  /space(750000); and space(1000000); 
But I'm sorry.

hassbian-ABC, 2019-07-22:
Thank you for your work, but I still can't control the turn-on and turn-off of LED/LIGHT through the web interface. i try to change the ir_Coolix.cpp#L90 to space(250000); / space(500000); /space(750000); and space(1000000);
But I'm sorry.
@crankyoldgit

crankyoldgit, 2019-07-22:
@hassbian-ABC Okay. I'm really confused. I am at a loss as to why it is not working for you.

My testing indicates it should be working.
I've uploaded a binary of `IRMQTTServer` from the latest update to the "Issue821" branch to an ESP8266, and loaded `IRrecvDumpV2` on to another ESP8266 to capture the signals generated.

This is what happens when after I load up the "AirCon" state per your screenshot in https://github.com/crankyoldgit/IRremoteESP8266/issues/821#issuecomment-513708970, and **then** change "Light" to "On" and click "update & send".
i.e. this is just the turning on the light bit the web interface.

```
Timestamp : 000034.855
Library   : v2.6.3

Encoding  : COOLIX
Code      : B5F5A5 (24 bits)
Mesg Desc.: Power: On, Led: Toggle
Raw Timing[199]:
   +  4480, -  4506,    +   596, -  1658,    +   598, -   554,    +   574, -  1646,
   +   598, -  1648,    +   596, -   530,    +   598, -  1648,    +   596, -   530,
   +   596, -  1662,    +   594, -   534,    +   594, -  1652,    +   624, -   502,
   +   594, -   532,    +   626, -  1620,    +   592, -   532,    +   626, -  1620,
   +   592, -   536,    +   592, -  1654,    +   592, -  1680,    +   564, -  1656,
   +   600, -  1648,    +   598, -   554,    +   574, -  1646,    +   598, -   530,
   +   598, -  1648,    +   596, -   530,    +   596, -   530,    +   598, -   528,
   +   600, -   528,    +   590, -  1656,    +   600, -   528,    +   600, -  1646,
   +   598, -   530,    +   596, -  1650,    +   626, -   526,    +   570, -  1650,
   +   596, -   530,    +   596, -   530,    +   596, -  1650,    +   596, -   530,
   +   596, -  1650,    +   626, -   500,    +   596, -  1650,    +   594, -   532,
   +   594, -  1650,    +   596, -  1678,    +   566, -   532,    +   626, -  1620,
   +   594, -   532,    +   594, -  5012,    +  4508, -  4462,    +   598, -  1674,
   +   570, -   530,    +   620, -  1654,    +   570, -  1676,    +   568, -   532,
   +   596, -  1650,    +   594, -   532,    +   596, -  1650,    +   594, -   534,
   +   592, -  1654,    +   592, -   560,    +   568, -   558,    +   568, -  1652,
   +   594, -   534,    +   624, -  1622,    +   594, -   560,    +   598, -  1622,
   +   622, -  1624,    +   590, -  1656,    +   598, -  1674,    +   572, -   528,
   +   600, -  1648,    +   596, -   530,    +   598, -  1650,    +   596, -   530,
   +   598, -   530,    +   598, -   556,    +   572, -   554,    +   572, -  1646,
   +   598, -   528,    +   598, -  1648,    +   598, -   530,    +   596, -  1650,
   +   626, -   500,    +   596, -  1650,    +   594, -   532,    +   596, -   530,
   +   596, -  1650,    +   594, -   532,    +   596, -  1652,    +   594, -   532,
   +   594, -  1652,    +   624, -   502,    +   596, -  1650,    +   594, -  1654,
   +   592, -   536,    +   592, -  1654,    +   624, -   504,    +   622
uint16_t rawData[199] = {4480, 4506,  596, 1658,  598, 554,  574, 1646,  598, 1648,  596, 530,  598, 1
648,  596, 530,  596, 1662,  594, 534,  594, 1652,  624, 502,  594, 532,  626, 1620,  592, 532,  626,
1620,  592, 536,  592, 1654,  592, 1680,  564, 1656,  600, 1648,  598, 554,  574, 1646,  598, 530,  59
8, 1648,  596, 530,  596, 530,  598, 528,  600, 528,  590, 1656,  600, 528,  600, 1646,  598, 530,  59
6, 1650,  626, 526,  570, 1650,  596, 530,  596, 530,  596, 1650,  596, 530,  596, 1650,  626, 500,  5
96, 1650,  594, 532,  594, 1650,  596, 1678,  566, 532,  626, 1620,  594, 532,  594, 5012,  4508, 4462
,  598, 1674,  570, 530,  620, 1654,  570, 1676,  568, 532,  596, 1650,  594, 532,  596, 1650,  594, 5
34,  592, 1654,  592, 560,  568, 558,  568, 1652,  594, 534,  624, 1622,  594, 560,  598, 1622,  622,
1624,  590, 1656,  598, 1674,  572, 528,  600, 1648,  596, 530,  598, 1650,  596, 530,  598, 530,  598
, 556,  572, 554,  572, 1646,  598, 528,  598, 1648,  598, 530,  596, 1650,  626, 500,  596, 1650,  59
4, 532,  596, 530,  596, 1650,  594, 532,  596, 1652,  594, 532,  594, 1652,  624, 502,  596, 1650,  5
94, 1654,  592, 536,  592, 1654,  624, 504,  622};  // COOLIX B5F5A5
uint64_t data = 0xB5F5A5;

Timestamp : 000035.179
Library   : v2.6.3

Encoding  : COOLIX
Code      : B21F48 (24 bits)
Mesg Desc.: Power: On, Mode: 2 (AUTO), Fan: 0 (AUTO0), Temp: 24C, Zone Follow: Off, Sensor Temp: Ignored
Raw Timing[199]:
   +  4494, -  4470,    +   590, -  1656,    +   600, -   554,    +   564, -  1656,
   +   598, -  1648,    +   598, -   528,    +   598, -   554,    +   574, -  1646,
   +   620, -   508,    +   598, -   528,    +   598, -  1674,    +   572, -   528,
   +   598, -   528,    +   600, -  1646,    +   600, -  1646,    +   598, -   528,
   +   598, -  1650,    +   596, -   530,    +   596, -   530,    +   618, -   510,
   +   598, -  1648,    +   598, -  1676,    +   568, -  1650,    +   626, -  1622,
   +   592, -  1654,    +   590, -  1656,    +   598, -  1646,    +   598, -  1646,
   +   598, -   530,    +   598, -   528,    +   600, -   526,    +   600, -   526,
[... 22 lines cut ...]
0, 536,  592, 1654,  590, 536,  622, 504,  624, 1624,  622, 530,  566, 534,  594, 560,  566, 1680,  56
4, 536,  592, 1654,  590, 1656,  600, 526,  590, 1656,  600, 1672,  572, 1676,  570, 5010,  4500, 4466
,  594, 1652,  594, 532,  594, 1652,  624, 1622,  590, 536,  592, 534,  594, 1654,  590, 536,  590, 53
6,  592, 1654,  592, 536,  592, 560,  566, 1654,  590, 1656,  600, 526,  590, 1656,  598, 528,  598, 5
[... cut ...]
e.g. `space(kDefaultMessageGap);` -> `space(THIS_SHOULD_ERROR);`
If the firmware still builds fine, then I think you've not been compiling from the `Issue821` branch, but are using another copy of the library somewhere.

If it fails, then you are probably using the correct copy of the library.

Can you please try physically powering off the board etc to fully reset it. Does the board/program still control the A/C normally apart from the Light/led setting?
Can you try running at 80Mhz (the normal default speed) just in case that is the problem?

hassbian-ABC, 2019-07-23:
@crankyoldgit 
Thank you. Now I can control the LED/LIGHT through the web interface, but whenever I turn on or off the LED/LIGHT, A/C seems to receive information twice.

crankyoldgit, 2019-07-23:
> Thank you. Now I can control the LED/LIGHT through the web interface

So, what did you do that finally fixed it? So I can know if someone else encounters a similar issue?

> whenever I turn on or off the LED/LIGHT, A/C seems to receive information twice.

Yes. That's unfortunately correct and expected behaviour.

COOLIX protocol has a bunch of "special" codes it uses for some of the functions. e.g. LED/Light toggle, Turbo toggle, Sleep toggle etc. These special codes do not include the "normal" Temperature, operating mode, or fan speed settings etc. In order for `IRMQTTServer` to ensure all the settings are correct, it has to send these "special" codes first, then follow it up with a "normal" message to make sure the A/C is in the desired state as best it can.

e.g. Imagine you changed Swing from Off to On, AND changed mode from Heat to Cool, AND Protocol from "GREE" to "COOLIX" then pressed "Update & Send". The program has to make the sure it sends everything it knows to the a/c unit.

hassbian-ABC, 2019-07-23:
I haven't changed anything to fix it. I found that the OTA upgrade was unsuccessful.
@crankyoldgit

hassbian-ABC, 2019-07-23:
Thank you again for your work. It's a great project. @crankyoldgit

closed 2019-07-23 by crankyoldgit

crankyoldgit, 2019-07-26:
FYI, the code changes thus far have been included in the v2.6.4 release of the library.

1043717432, 2019-07-26:
> FYI, the code changes thus far have been included in the v2.6.4 release of the library.

Thank you again for your work

commit b8681358 (2019-07-22): Fix parameters for coolix in IRac::sendAc()
Fixes #821
commit e16b340a (2019-07-22): Fix parameters for coolix in IRac::sendAc()
Fixes #821
commit d6c858c6 (2019-07-22): Enforce a post-message gap for Coolix messages.
* Add unit tests to confirm LED/light toggle message is being sent when 
it should be.

Hopefully fixes #821
commit 2c18132d (2019-07-23): [BUG] Fix parameters for coolix in IRac::sendAc() (#829)

* Fix parameters for coolix in IRac::sendAc()
* Enforce a post-message gap for Coolix messages.
* Add unit tests to confirm LED/light toggle message is being sent when it should be.

Fixes #821